openwrt/mt76 #188: 2.4g issue remains

dimatto, 2018-07-06:
2.4g remains very laggy. I use a client which behind a wall, its maxspeed is 72m, but usually below 58m as displayed in luci. I try every option in txpower selection of luci, no use.

dimatto, 2018-07-07:
The reason is signal. When this client is behind wall, it’s very unstable and laggy, even signal displayed at luci is -55. When move straight sight, signal displayed -24 at luci, stable and no laggy, tx/rx speed all remain 72M. 
But it should be stable  at -55, there’s something wrong. Moreover, I try tx power from 17db to 30db, there’s no difference.

easyteacher, 2018-07-11:
2.4G WiFi will disappear after some time. Restart the WiFi in luci will cause luci dead. Only reboot can solve the problem.

psyborg55, 2018-07-22:
>> change frequency channel seems resolve this issue

that's a mitigation not a solution.

brzz, 2018-08-03:
same too

barhom, 2018-08-12:
Please rewrite the topic name to include which chipset so that we know which 2.4ghz chip you are referring to.

nbd168, 2018-08-24:
Please try the latest version

dimatto, 2018-08-25:
still need frequently change channel to get acceptable speed. and it seems even harder with latest commits.

julianocss, 2018-08-27:
Hello @nbd168

I installed latest version now.
I will test for next days.
I let you know if i had any problem.

Thanks for yor work.
Juliano

barhom, 2018-09-18:
@julianocss update?

julianocss, 2018-09-20:
Hello @barhom 

The problem persists, especially when i use multiple SSID´s (Master and Client).
Sometimes i have to reboot the router to bring connection back.
Another thing that i noticed is that behind only one wall, the connection drops the quality a lot.

I don´t test the latest commit yet.

Juliano

P.S.: My router uses MT7628 chip.

milankragujevic, 2018-10-02:
I tested latest everything, but the problem still remains. MT7628 / WR840N V5.

yujincheng08, 2018-10-16:
same issue on MT7260

yujincheng08, 2018-10-18:
@julianocss Is your issue the same as #169?

julianocss, 2018-10-18:
Hello @yujincheng08.

The problem is similar to #169.
I still can not reproduce what causes the problem.
I'm using wireless and suddenly clients disconnect and the AP disappears from the air and only rebooting recover the AP.

I checked the log now, i don´t have this message in the latest 17h.
"STA xx:xx:xx:xx:xx:xx IEEE 802.11: disconnected due to excessive missing ACKs"

Bellow is mine wireless file.

config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'platform/10300000.wmac'
	option country '00'
	option legacy_rates '1'
	option noscan '1'
	option htmode 'HT40'
	option txpower '20'
	option channel '3'

config wifi-iface
	option network 'wwan'
	option ssid 'Network302_2g'
	option encryption 'psk2'
	option device 'radio0'
	option mode 'sta'
	option bssid '70:4F:57:EC:CC:01'
	option key 'secret'
	option disassoc_low_ack '0'

config wifi-iface
	option device 'radio0'
	option mode 'ap'
	option ssid 'Network_402'
	option network 'lan'
	option disassoc_low_ack '0'
	option encryption 'psk2'
	option key 'secret'
	option wpa_disable_eapol_key_retries '1'

Juliano.

yujincheng08, 2018-10-18:
@julianocss I think we encounter the same problem even you miss the very message which, in fact, was eliminated after a certain commit while the problem remained.

nbd168, 2018-10-25:
Please try the latest version (master only, not 18.06).

julianocss, 2018-10-27:
Hello @nbd168.

Yesterday i noticed that snapshot builds for Mt76x8 routers are not update since 22/10. I don't know what is the problem. 

There is another way to test this last driver version?

Juliano

yujincheng08, 2018-10-27:
@julianocss Build this repo with sdk, and you will get several ipk files. Install them and then test the wifi.

julianocss, 2018-10-27:
Can you help me how to do that? Sorry for off topic, you send an email to me if you can.

Do you ir another user have an ideia what could be the problem with snapshot builds

yujincheng08, 2018-10-27:
@julianocss I don't think it helps because the latest commit seems to be related only with 76x0 rather than your chip 7628.

julianocss, 2018-10-27:
@yujincheng08 , hum, ok!

I thought that some improvement had been made for mt7628, because @nbd168 said to test the latest driver version.

nbd168, 2018-10-27:
The changes that I made affect MT7603 (and thus also MT7628) as well

julianocss, 2018-10-30:
Hello @nbd168.

Today i installed Linux on my computer and compiled the latest openwrt master branch to my router.

Now i can test the latest modifications that you made for the driver.

I give a feedback in the next days.

Thanks.
Juliano

ffrediani, 2018-11-07:
Hello
Wanted to mention that I made my report about instability issues on issue #169 for chip MT7628AN and add that as reported above in this thread that I also have multiple SSID in this case.

ffrediani, 2018-11-07:
Hello
I wanted to mention that I made my report about instability issues on issue #169 for chip MT7628AN and add that as reported above in this thread that I also have multiple SSID in this case.

sombree, 2018-11-08:
Hello
With latest commit ([417cf49](https://github.com/openwrt/mt76/commit/417cf49ce811287b980767a1b7b44118b0ccc8e4)) there are still some issues with 2.4GHz network - after some time wifi speed drops drasticaly. Only solution so far is to disconnect and then connect to the network or restart the router.
Btw. is this normal? phy0 is mt7603e; current uptime - ~19h.
```
root@OpenWrt:/# cat /sys/kernel/debug/ieee80211/phy0/mt76/reset
             TX hang: 0
   TX DMA busy stuck: 41
   RX DMA busy stuck: 0
        Beacon stuck: 5
   RX PSE busy stuck: 5
            MCU hang: 0
```

julianocss, 2018-11-09:
Hello @nbd168 

The speed improved a lot with latest commit (08/11) on my WR-840 v4 (MT7628AN)
I will keep testing, specially if will keep stable.

Thank you very much.
Juliano

sombree, 2018-11-12:
@nbd168 connection to 2,4GHz network (mt703e in YouHua WR1200JS) seems to be much more stable after changing "time interval for rekeying GTK" from default 600 seconds to 86400 seconds. I came up with this after reading [this](https://forum.turris.cz/t/solution-found-for-wifi-clients-disconnects/6065).

9000000, 2018-11-14:
latest OpenWrt master version not good,

The 2.4 GHz band is still faulty when using multiple devices at the same time for download. Downloaded after a few minutes will automatically be disconnected completely and no network

nbd168, 2018-11-14:
@9000000 Which device are you using?

milankragujevic, 2018-11-14:
I can confirm it is faulty. On TL-MR3420v5, it breaks after a few hours of usage, the WiFi disappears while the router still works. I haven't tried getting a log, I will attach it later today, I just rebooted the router and it fixed it temporarily. Latest custom build from Git, with LuCi and nothing else other than defaults. Clean install, no previous config. WiFi speed is good with WMM enabled, but it crashes so that's a real roadblock for me.

nbd168, 2018-11-14:
Did the latest commit make anything worse for you, or was it that way with older versions as well?

9000000, 2018-11-14:
@nbd168 I use the wifi Xiaomi 3G

ffrediani, 2018-11-14:
@9000000 @milankragujevic 
Have you tested with multiple SSID or not ? Your symptoms look similar to mine just when using multiple SSIDs.

9000000, 2018-11-14:
@ffrediani I only use 2 main ssid, 2.4ghz is a name, 5ghz is another name

5ghz is too ok, but 2.4 is too bad, the main need only 2.4ghz

sombree, 2018-11-14:
@nbd168 I can confirm - with latest master 2,4GHz network is much less stable than before. Is [this](https://pastebin.com/jCmbWdSC) normal? [Here](https://pastebin.com/GCwq5gX7) is my wireless config.

julianocss, 2018-11-15:
@nbd168.

Last commit was not good for me. 
Speed drops a lot. Router became very unstable. 

There is something that i can do to help? 

Juliano.

lukasz1992, 2018-11-15:
I can not reproduce speed drops, but somehow problem happened to me on https://github.com/openwrt/mt76/commit/88f68838d91e7b00e2abb5638cf40a514837b7da on Xiaomi 3G Router (MT7603E).

After expensive iperf tests and disconnecting from 2.4GHz WiFi, I could not connect to it again:
- I was the only connected client to the AP
- network was OPEN (without encryption)

I can not provide a reproduce steps.
1. I managed to get some values from /sys/kernel/debug/ieee80211/phy0/mt76:

```
reset:
             TX hang: 0
   TX DMA busy stuck: 0
   RX DMA busy stuck: 0
        Beacon stuck: 0
   RX PSE busy stuck: 0
            MCU hang: 0

reset_test:
0
```
2. I have another USB WiFi (ath9k_htc) adapter, and dumped the traffic - in attachment
[dump-2g.tar.zip](https://github.com/openwrt/mt76/files/2586368/dump-2g.tar.zip)
, but I am afraid it is incomplete - I am not sure how to exactly set monitor mode with minimum filtering.
Anyway 320s after start - I tried to connect, about 370s after start I reloaded driver, and about 420s after start I was able to connect.

Restarting WiFi (wifi up) on router did not help, I needed to unload mt7603e module, load it again and start AP, and it started to work immediately

lukasz1992, 2018-11-15:
Update - my previous comment https://github.com/openwrt/mt76/issues/188#issuecomment-439123869 is about tests on Ubuntu with 4.18 kernel. After that, I did tests on Windows 8.1 on the same device, and can confirm that what @julianocss affects me too. Reproduce is simple - just try to connect to WiFi and transfer data.

Screenshoot:

[image]

Traffic dump (USB ath9k_htc):

[DUMP-2G-01.zip](https://github.com/openwrt/mt76/files/2586738/DUMP-2G-01.zip)

Even number of retransmissions shows something is really wrong.

ffrediani, 2018-11-16:
Hi @nbd168 as I believe some of us that are reporting here use different radio chips (in my case 7628) the reports may differ on some ways.
Let us know exactly what details are important to take note in order to make a more detailed report. 
In my case I wasn't able to confirm the speed issues yet but it does happen very similar to when I have multiple SSIDs to the point that disable/enable the radio doesn't resolve. Only a reboot in the router.

Maybe if you find useful to provide an specific image with extra debug options enabled I am free to use it and provide further feedback.

nbd168, 2018-11-16:
I found and fixed some more bugs and pushed an update to OpenWrt master. Please test the latest version.

julianocss, 2018-11-16:
Thanks @nbd168.

I will compile now. 

Juliano

9000000, 2018-11-16:
Thanks @nbd168

Thank you very much, this is the best update ever so far for xiaomi r3g that I have ever used, I will keep this version in case of future updates if bug

julianocss, 2018-11-16:
Hello @nbd168 .

Running latest commit in the last hour.
Things seens better. The speed returned to the same level as the compilation on 08/11.
About stability, i will test in the nex days. I keep you updated.

Thank you very much, again!
Juliano

9000000, 2018-11-17:
Ping parameters are high and unstable. The video is good but if the game is too bad

julianocss, 2018-11-17:
Hello @nbd168 and @LorenzoBianconi .

I am running the last update for 17 hours. No crashes.

But it seems like there's still something wrong.
This morning, when I woke up, the connection was alive, but extremely slow.
 I could not even access the graphical interface of the router.
Now, about 3 hours later, without me doing anything, the connection returned to normal.

I looked at the log and did not see any relevant information.
I can send it to you, if you want.

Juliano

9000000, 2018-11-17:
I also get errors like @juliano when disconnecting too long and when connected back it is very slow, until restarting ruoter it works normal

julianocss, 2018-11-17:
Hello @nbd168.

I do not know if this information can be useful, but my setup are: Router_1 (Access Point) - Router_2 and Router_3 clients.
I had to go back to the previous version because network became unusable.

Thanks for your work.
I'm willing to help with whatever you need.

Juliano

nbd168, 2018-11-20:
Please try the latest version from OpenWrt master, I pushed some more fixes.

julianocss, 2018-11-20:
Hello @nbd168 .

Thanks. Compiling now.
There is a way to compile only the driver and integrate to current firmware of my router?

For every update that you release, i´m having to compile a clean buid for my router on my computer (buildbot of Openwrt is still with problem), and the proccess take ~4 hours. Maybe there is one more faster way to do that that i don´t know. :)

Regards,
Juliano

lukasz1992, 2018-11-20:
You don't have to do a clean build.
git pull ; make   should be sufficient.

julianocss, 2018-11-20:
@lukasz1992 .
 
" ./scripts/feeds update && ./scripts/feeds install -a " is necessary?

I will try in the next releases.
Thanks.
Juliano

julianocss, 2018-11-21:
Hello @nbd168 .

I´m running latest version (Nov 20) for 10 hours, no crashes.
But the behavior is the same as the previous versions.
When i turn on the router, everything works normal. But after some period of  inactivity (at night, in my case), and i come back to use again, the connection is still alive, but extremely slow. This began to happen since the Nov 13 version.

Today I noticed something different. I have an older computer that has a "bg" wireless adapter. At that moment, the connection to it is ok. But on my cellphones, which have "bgn" adapters, the connection is frozen at these speeds (attached image).

Do you want that i do some test?

Thanks.

[image]

lukasz1992, 2018-11-21:
@9000000 
Driver can not get current txpower (as needed in luci), but can set and sets txpower (in mt7603_mcu_set_tx_power)

julianocss, 2018-11-21:
@lukasz1992.

This information is new for me. 
Can you explain to me how to do it?

Thanks. 
Juliano

nbd168, 2018-11-21:
MT7603 always uses the maximum power by default at the moment.
By the way, I pushed another mt76 update with mt7603 fixes.
I was finally able to make the watchdog reset code reliable, so things should work better now.

julianocss, 2018-11-21:
@nbd168 

Flashed latest version now. Speed is OK!
Tomorrow i tell you if became stable.

Thanks.
Juliano

differentblue, 2018-11-21:
I would love to try testing the latest @nbd168 changes.  

When do the OpenWrt snapshots get rebuilt with these changes?   It looks like the MT76x8 targets have not been rebuilt since last week.  Any idea why that is?
[image]

julianocss, 2018-11-21:
Hello @differentblue.

There is almost 2 weeks that this problem exits. I reported in Openwrt forum, but no success. To test the latest versions, i installed Linux on my computer to compile my own builds.

What is your router? If you want i can add then on my list, so it will be included in my next compilation..

xback, 2018-11-21:
Not all host systems contain the "zip" tool which is required for a device
within this target.

A build will only succeed if it's get creates by 1 of the hosts who does
have it.

[72 more replies]

closed 2019-02-25 by dimatto