openwrt/mt76 #192: mt7603e: client disconnects

barhom, 2018-08-13:
Chip: mt7603e
Target: ZBT-WE3526

Problem:

Internet suddenly dissapears. Wi-Fi symbol on macbook pro, android phones etc are still showing as connected.
Ping to router-ip does not work.
Renewing DHCP on macbook pro does not work.
Only solution is turn wifi off and back on or switch network and then go back to the 2g network.

r0ck3r, 2018-08-24:
Try to disable next option: "Disassociate On Low Acknowledgement"

nbd168, 2018-08-24:
Please try the latest version

barhom, 2018-08-28:
@nbd168 
Trying latest now, will close ticket if problem does not happen again.

ein-shved, 2018-08-31:
@nbd168 still reproducing on Xiaomi Mi Router 3G. Android works perfectly as always, but 2 our iPhone SE can suddenly lost internet connection with internet (no gw ping). disassoc_low_ack=0

pataha, 2018-08-31:
I have same problem on iphone6s, 7
Xiaomi 3G with latest openwrt snapshot

Rising-Sun, 2018-09-01:
Have the same problem too with trunk r7986-dc9388ac55 with my iPhone 7.
I have disassoc_low_ack=1 set.

nbd168, 2018-09-04:
I don't think logs help at this point. What I need is a really simple way to reproduce the issue as quickly as possible. I will look into this some more next week. If you can figure out more about what triggers it, it would help.
Thanks!

barhom, 2018-09-14:
@nbd168 
Has it ever happened to you on your ZBT devices? All it takes is to use it  for a while

barhom, 2018-09-18:
I am trying hard to reproduce this issue reliably. I still cant. I'm not giving up.
Anyone know more in detail on when this happens exactly?

barhom, 2018-09-18:
I have been debugging more and more using a Samsung S7 that takes about 15 minutes to start showing the issue where the wifi is connected but no internet is available. 

Inside wifi-settings on the S7 it says "Connected, no internet".

This is my observation so far;

When the S7 is in this "weird state", I can tcpdump on the router,

192.168.20.224 = IP of Samsung S7

```tcpdump -i wlan0 src 192.168.20.224 or dst 192.168.20.224 -en ```

Every second I see the following,

```
22:42:21.946352 8c:f5:a3:a4:8a:e8 > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 42: Request who-has 192.168.20.1 tell 192.168.20.224, length 28
22:42:21.946449 8c:f5:a3:a4:8a:e8 > ff:ff:ff:ff:ff:ff, ethertype ARP (0x0806), length 42: Request who-has 192.168.20.1 tell 192.168.20.224, length 28
22:42:21.946675 78:a3:51:30:1a:c2 > 8c:f5:a3:a4:8a:e8, ethertype ARP (0x0806), length 42: Reply 192.168.20.1 is-at 78:a3:51:30:1a:c2, length 28
```

It seems that the S7 keeps sending ARP to ask about who has 192.168.20.1 (router IP).
I do not kno why it sends the ARP twice.
The router replies but since the phone keeps sending the ARP request again and again I must conclude that the phone is not receiving the packets.

My conclusion is the following so far,

Samsung S7 -> ROUTER connection is FINE
ROUTER -> Samsung S7 is NOT FINE.

I'll keep digging, hope this is of some help.

nbd168, 2018-09-29:
Please test the latest version from OpenWrt master or the 18.06 branch.

swg0101, 2018-09-30:
Just tried the latest snapshot, now the connection is almost unusable. My client that wasn't disconnecting before is disconnecting now whenever I try doing something on the connection (won't if I leave it idle). Gateway pings jump all over the place, with ~40% packet loss and almost nothing is loading on the connection. Same problem whether on Windows or Android.

Note that I am running STA+AP on channel 36 - the problem only happens on the AP side, but STA seems fine.

pataha, 2018-09-30:
I have the same thing with latest snapshot, client is now unable to connect. "Incorrect password" message appears most of time

swg0101, 2018-09-30:
Here's a snippet of how the traffic looks like:

[image]

The client machine (Intel) does not receive any beacons from the AP, keeps sending probe requests out to locate the AP, while the AP keeps retransmitting frames that are on the current session. Since the client has left the channel to send the probe requests, no acknowledgments are being sent in response to the packets sent to the client (the bluish red) pieces. Eventually, the client locates the AP via active scanning (probe response), and reassociates with the AP. However, since no beacons are actively sent, the client doesn't stay on the channel very long until this process starts again with the client sending out probe requests to locate the AP repeatedly.

[image]

As seen by an airodump-ng capture, there are no beacons sent from the Xiaomi with the 7A:11:DC MAC address, whereas my Linksys router 62:38:E0 router is sending beacons normally (in the one-minute timeframe).

nbd168, 2018-10-01:
Does the issue only happen in a STA+AP configuration, or also with only a single AP interface?

swg0101, 2018-10-01:
Haven't tried yet - I will play again with this tomorrow on my spare router.

nbd168, 2018-10-01:
I was able to reproduce the issue with AP+STA (but not AP-only)

nbd168, 2018-10-01:
It happens for me even with the old mt76 version and before the recent mac80211 upgrade

nbd168, 2018-10-01:
Found the bug and pushed a fix to master and 18.06

araujorm, 2018-10-01:
Hello. I noticed that the latest fix seems to be only for mt76x2 code. Or should it improve reliability on 2.4g chips (such as the 7603e that's on the subject of this thread) as well?

nbd168, 2018-10-01:
The latest fix is for mt76x2 code only, for the issue that @swg0101 reported. In the last round of commits there were fixes for mt7603 as well though.

barhom, 2018-10-02:
@nbd168 Why is it you are requesting testing on 18.06 and not on master branch?
I dont usually compile 18.06

Thanks for the info and y our work

nbd168, 2018-10-02:
I asked for tests on master or 18.06. Both have the fixes

tkso1997, 2018-10-25:
Is it confirmed to work now? 
Last time I tried, I had the same issue, mt7612 was working quite good, but mt7603 stopped working after a few minutes (while still showing that the device is connected). The only thing that helped was a reboot.

barhom, 2018-10-30:
I'm closing this issue because I have moved on to another chipset.
Its too bad that mt7621 based routers was never stable enough for me.

When I try MT routers again and see the same problem I'll open a new ticket.

closed 2018-10-30 by barhom

pataha, 2018-10-30:
I have same problem on IPhone 7, device still connected but can not connect to internet nor router
Pls keep it open

barhom, 2018-11-02:
@pataha @ein-shved 
Please open a new ticket with as much details as possible with your issues.
You can always reference this ticket #192.

Rising-Sun, 2018-11-07:
Happening with the latest mt76 drivers from 6.Nov using mir3g with iPhone 7 too.
But throughput increased by 20%

Rising-Sun, 2018-11-10:
Still happening with OpenWrt SNAPSHOT r8444-7dd42aa592  on my mir3g.

BlackCutpoint, 2020-04-18:
I can help you to debug.
I suppose Wi-Fi is failing on a machine with NetworkManager and systemd.
In /etc/systemd/journald.conf set
```ForwardToKMsg=yes```
Run something like
```
while true; do
if ! nm-online -t 1 > /dev/null; then
  nmcli con up id $ssid
fi
done
```
Then provide both the output of that script AND output of dmesg.
That will be all info necessary to debug.

pjgerardo, 2020-07-31:
I have a the exact same issue with several diverent versions on openWRT, current one 19.07.2 r10947-65030d81f3 and the issue pecists. Should I open a new thread ?

system type             : Atheros AR9130 rev 2
machine                 : Linksys WRT160NL

evelineraine, 2021-10-31:
I am seeing this (or similar) issue on my *Xiaomi Mi Router 3G*'s MT7603E 2.4GHz radio. In contrast, the second 5GHz MT76x2E radio works just fine.

Under heavy load (e.g. torrenting) one of the clients would suddenly loose all network connectivity, but remain *connected* as reported both on the client and on the router. Some of the times, reconnecting restores connectivity temporarily while all other clients remain unaffected. Other times, all clients lose connectivity and no client can reconnect to the network (stuck in *connecting* state forever) until the radio on the router is restarted.

In the second case, I would see strange log messages on my desktop like this when trying to force reconnect:
```
wpa_supplicant[836]: wlan: Trying to associate with SSID 'WIFI'
wpa_supplicant[836]: wlan: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
wpa_supplicant[836]: wlan: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="WIFI" auth_failures=2 duration=20 reason=CONN_FAILED
...
wpa_supplicant[836]: wlan: CTRL-EVENT-SSID-REENABLED id=0 ssid="WIFI"
<repeat>
```
(notice the **`bssid=00:00:00:00:00:00`**).

When these disconnects happen there are no log messages printed on the router, nor the client - neither seems to *notice* a change.

The issue happens pretty randomly - from minutes to hours of uninterrupted connectivity in between. I am not sure heavy traffic is a prerequisite.

escape0707, 2022-02-12:
> I am seeing this (or similar) issue on my _Xiaomi Mi Router 3G_'s MT7603E 2.4GHz radio. In contrast, the second 5GHz MT76x2E radio works just fine.
> 
> Under heavy load (e.g. torrenting) one of the clients would suddenly loose all network connectivity, but remain _connected_ as reported both on the client and on the router. Some of the times, reconnecting restores connectivity temporarily while all other clients remain unaffected. Other times, all clients lose connectivity and no client can reconnect to the network (stuck in _connecting_ state forever) until the radio on the router is restarted.
> 
> In the second case, I would see strange log messages on my desktop like this when trying to force reconnect:
> 
> ```
> wpa_supplicant[836]: wlan: Trying to associate with SSID 'WIFI'
> wpa_supplicant[836]: wlan: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
> wpa_supplicant[836]: wlan: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="WIFI" auth_failures=2 duration=20 reason=CONN_FAILED
> ...
> wpa_supplicant[836]: wlan: CTRL-EVENT-SSID-REENABLED id=0 ssid="WIFI"
> <repeat>
> ```
> 
> (notice the **`bssid=00:00:00:00:00:00`**).
> 
> When these disconnects happen there are no log messages printed on the router, nor the client - neither seems to _notice_ a change.
> 
> The issue happens pretty randomly - from minutes to hours of uninterrupted connectivity in between. I am not sure heavy traffic is a prerequisite.
> 
> OpenWrt release: `21.02.1 r16325-88151b8303`.

Still having the same problem on Redmi AC2100.

Many of my devices can't even connect to MT7603E 2.4G. Although MT7615E 5G always connect.

As for the traffic interruption after heavy load issue, I haven't confirmed that.