openwrt/mt76 #195: Package kmod-mt76x2 is missing dependencies for the following libraries

nasbdh9, 2018-09-08:
I got the latest commit but I got an error in compiling ( use GCC8

`make[3]: Entering directory '/home/nasbdh9/openwrt/package/kernel/mt76'
mkdir -p /home/nasbdh9/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/nasbdh9/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-09-06-fe650dab/.pkgdir/kmod-mt76-core/. /home/nasbdh9/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/nasbdh9/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.kmod-mt76-core_installed
mkdir -p /home/nasbdh9/openwrt/bin/targets/ramips/mt7621/packages /home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-09-06-fe650dab/ipkg-mipsel_24kc/kmod-mt76x2/CONTROL /home/nasbdh9/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo
mkdir -p /home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-09-06-fe650dab/ipkg-mipsel_24kc/kmod-mt76x2/etc/modules.d; ( echo "mt76x2e"; ) > /home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-09-06-fe650dab/ipkg-mipsel_24kc/kmod-mt76x2/etc/modules.d/mt76x2; 
install -d -m0755 /home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-09-06-fe650dab/ipkg-mipsel_24kc/kmod-mt76x2/lib/firmware
cp /home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-09-06-fe650dab/firmware/mt7662_rom_patch.bin /home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-09-06-fe650dab/firmware/mt7662.bin /home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-09-06-fe650dab/ipkg-mipsel_24kc/kmod-mt76x2/lib/firmware
find /home/nasbdh9/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-09-06-fe650dab/ipkg-mipsel_24kc/kmod-mt76x2 -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package kmod-mt76x2 is missing dependencies for the following libraries:
mt76x2-common.ko
Makefile:108: recipe for target '/home/nasbdh9/openwrt/bin/targets/ramips/mt7621/packages/kmod-mt76x2_4.14.68+2018-09-06-fe650dab-1_mipsel_24kc.ipk' failed
make[3]: *** [/home/nasbdh9/openwrt/bin/targets/ramips/mt7621/packages/kmod-mt76x2_4.14.68+2018-09-06-fe650dab-1_mipsel_24kc.ipk] Error 1
make[3]: Leaving directory '/home/nasbdh9/openwrt/package/kernel/mt76'
time: package/kernel/mt76/compile#0.71#0.24#0.96
package/Makefile:107: recipe for target 'package/kernel/mt76/compile' failed
make[2]: *** [package/kernel/mt76/compile] Error 2
make[2]: Leaving directory '/home/nasbdh9/openwrt'
package/Makefile:103: recipe for target '/home/nasbdh9/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/nasbdh9/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/nasbdh9/openwrt'
/home/nasbdh9/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
`

pparent76, 2018-09-10:
I get the same:

> Package kmod-mt76x2 is missing dependencies for the following libraries:
> mt76x2-common.ko

nasbdh9, 2018-09-10:
It seems that need to update the makefile

LorenzoBianconi, 2018-09-10:
You need to add kmod-mt76x2-common to DEPENDS of mt76x2 section (mt76 openwrt Makefile):

DEPENDS+=+kmod-mt76x2-common

closed 2018-09-10 by nasbdh9