openwrt/mt76 #197: mt7612e low  download  speed

leksmax, 2018-09-29:
hello  i  have  router  witch  mt7612e   use  openwrt  and  ihave  low  download  speed
[image]

root@OpenWrt:/# iwinfo
wlan0     ESSID: "OpenWrt_5g"
          Access Point: AC:84:C6:65:02:CD
          Mode: Master  Channel: 36 (5.180 GHz)
          Tx-Power: 15 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7662 14C3:7662 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

Wiphy phy0
        max # scan SSIDs: 4
        max scan IEs length: 2247 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 2:
                Capabilities: 0x1ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-15
                VHT Capabilities (0x018001b0):
                        Max MPDU length: 3895
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                        short GI (80 MHz)
                        TX STBC
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
                Frequencies:
                        * 5180 MHz [36] (16.0 dBm)
                        * 5200 MHz [40] (16.0 dBm)
                        * 5220 MHz [44] (16.0 dBm)
                        * 5240 MHz [48] (16.0 dBm)
                        * 5260 MHz [52] (15.0 dBm) (no IR, radar detection)
                        * 5280 MHz [56] (15.0 dBm) (no IR, radar detection)
                        * 5300 MHz [60] (15.0 dBm) (no IR, radar detection)
                        * 5320 MHz [64] (15.0 dBm) (no IR, radar detection)
                        * 5500 MHz [100] (15.0 dBm) (no IR, radar detection)
                        * 5520 MHz [104] (15.0 dBm) (no IR, radar detection)
                        * 5540 MHz [108] (15.0 dBm) (no IR, radar detection)
                        * 5560 MHz [112] (15.0 dBm) (no IR, radar detection)
                        * 5580 MHz [116] (15.0 dBm) (no IR, radar detection)
                        * 5600 MHz [120] (15.0 dBm) (no IR, radar detection)
                        * 5620 MHz [124] (15.0 dBm) (no IR, radar detection)
                        * 5640 MHz [128] (15.0 dBm) (no IR, radar detection)
                        * 5660 MHz [132] (15.0 dBm) (no IR, radar detection)
                        * 5680 MHz [136] (15.0 dBm) (no IR, radar detection)
                        * 5700 MHz [140] (15.0 dBm) (no IR, radar detection)
                        * 5745 MHz [149] (15.0 dBm) (no IR)
                        * 5765 MHz [153] (15.0 dBm) (no IR)
                        * 5785 MHz [157] (15.0 dBm) (no IR)
                        * 5805 MHz [161] (15.0 dBm) (no IR)
                        * 5825 MHz [165] (15.0 dBm) (no IR)
        valid interface combinations:
                 * #{ IBSS } <= 1, #{ managed, AP, mesh point } <= 8,
                   total <= 8, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Device supports VHT-IBSS.

mweinelt, 2018-10-27:
Can you reconfirm this issue with `iperf` or `iperf3`? This way we might get to know if it's really the WiFi or just your internet downlink.

lukasz1992, 2018-10-27:
Many months ago I tested it with iperf3 in https://github.com/openwrt/mt76/issues/151.
I had similar issue as @leksmax - problem was both in mt7603 and mt7612 drivers.

But I have to retest it now with latest commits.

lukasz1992, 2018-11-15:
I have closed https://github.com/openwrt/mt76/issues/151, meanthetime I think that situation is not perfect with MT7612E.

I have made some tests today on https://github.com/openwrt/mt76/commit/88f68838d91e7b00e2abb5638cf40a514837b7da

MT7612, VHT80

```
sta -> ap		0.00-30.04  sec   743 MBytes   207 Mbits/sec                  receiver
sta -> ap		0.00-30.04  sec   739 MBytes   206 Mbits/sec                  receiver
sta -> ap		0.00-30.04  sec   736 MBytes   206 Mbits/sec                  receiver
sta -> ap		0.00-10.04  sec   254 MBytes   213 Mbits/sec                  receiver

ap -> sta		0.00-30.09  sec   733 MBytes   204 Mbits/sec    0             sender
ap -> sta		0.00-30.06  sec  1016 MBytes   283 Mbits/sec    0             sender
ap -> sta		0.00-30.04  sec  1003 MBytes   280 Mbits/sec    0             sender
ap -> sta		0.00-30.08  sec   961 MBytes   268 Mbits/sec    1             sender
ap -> sta		0.00-30.09  sec   693 MBytes   193 Mbits/sec    0             sender
```

As I have no 5GHz networks here, it seems the upload speed is not as stable as here:
https://github.com/openwrt/mt76/issues/151#issuecomment-439114941 .

I consider 300-310 Mbit/s as a maximum for AC, on my Intel 1x1 WiFi card.

nbd168, 2019-01-25:
Should be better now, please test

sombree, 2019-01-31:
There is significant speed drop on 5GHz network (for both N and AC stations) on build with latest driver ([f665fb0](https://github.com/openwrt/openwrt/commit/f665fb058f5c040129ab8e134ea0c0ea38d98797)) - for me it dropped from ~50mbps to not more than 10mbps (wlan<->wan). 
Issue is observed with "802.11w Management Frame Protection" set to "Optional". Disabling 802.11w solves problem.
Btw. Tx-power reported by iwinfo is lower than one in /etc/config/wireless - it should be 20 instead of 16:
```
wlan1     ESSID: "xxx"
          Access Point: D4:5F:25:xx:xx:xx
          Mode: Master  Channel: 120 (5.600 GHz)
          Tx-Power: 16 dBm  Link Quality: 31/70
          Signal: -79 dBm  Noise: unknown
          Bit Rate: 265.5 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT7612E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

sombree, 2019-01-31:
I don't see any warnings with 802.11w enabled: [kernel log](https://pastebin.com/J1ybMVth) and [system log](https://pastebin.com/LjgwsaAk)

nbd168, 2019-01-31:
I've pushed a fix for this to mt76.git, will update openwrt master/18.06 soon. Please test.

sombree, 2019-01-31:
802.11w low speed problem is solved, thanks :)

nbd168, 2019-01-31:
Thanks for testing

closed 2019-01-31 by nbd168

Openwrtfunboy, 2020-02-24:
Ку, @leksmax Все еще есть эта проблема на последней стабилке?
/
Hi, @leksmax You have it problem in latest openwrt release?