openwrt/mt76 #199: Clients won't associate to 5GHz radio.

xavier83, 2018-10-06:
I'm running openwrt trunk (snapshot firmware on 5th October 2015)
device is (zbt-we1326), Thankfully 2.4 GHz is working fine.
```
root@OpenWrt:~# iw list
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2247 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 1 (up to 450m)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 2:
                Capabilities: 0x1ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-15
                VHT Capabilities (0x018001b0):
                        Max MPDU length: 3895
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                        short GI (80 MHz)
                        TX STBC
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
                Frequencies:
                        * 5180 MHz [36] (23.0 dBm)
                        * 5200 MHz [40] (23.0 dBm)
                        * 5220 MHz [44] (23.0 dBm)
                        * 5240 MHz [48] (23.0 dBm)
                        * 5260 MHz [52] (23.0 dBm) (radar detection)
                        * 5280 MHz [56] (23.0 dBm) (radar detection)
                        * 5300 MHz [60] (23.0 dBm) (radar detection)
                        * 5320 MHz [64] (23.0 dBm) (radar detection)
                        * 5500 MHz [100] (23.0 dBm) (radar detection)
                        * 5520 MHz [104] (23.0 dBm) (radar detection)
                        * 5540 MHz [108] (23.0 dBm) (radar detection)
                        * 5560 MHz [112] (23.0 dBm) (radar detection)
                        * 5580 MHz [116] (23.0 dBm) (radar detection)
                        * 5600 MHz [120] (23.0 dBm) (radar detection)
                        * 5620 MHz [124] (23.0 dBm) (radar detection)
                        * 5640 MHz [128] (23.0 dBm) (radar detection)
                        * 5660 MHz [132] (23.0 dBm) (radar detection)
                        * 5680 MHz [136] (23.0 dBm) (radar detection)
                        * 5700 MHz [140] (23.0 dBm) (radar detection)
                        * 5745 MHz [149] (30.0 dBm)
                        * 5765 MHz [153] (30.0 dBm)
                        * 5785 MHz [157] (30.0 dBm)
                        * 5805 MHz [161] (30.0 dBm)
                        * 5825 MHz [165] (30.0 dBm)
        valid interface combinations:
                 * #{ IBSS } <= 1, #{ managed, AP, mesh point } <= 8,
                   total <= 8, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Device supports VHT-IBSS.
Wiphy phy0
        max # scan SSIDs: 4
        max scan IEs length: 2257 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 1 (up to 450m)
        Available Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 1:
                Capabilities: 0x1fe
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-15
                Frequencies:
                        * 2412 MHz [1] (30.0 dBm)
                        * 2417 MHz [2] (30.0 dBm)
                        * 2422 MHz [3] (30.0 dBm)
                        * 2427 MHz [4] (30.0 dBm)
                        * 2432 MHz [5] (30.0 dBm)
                        * 2437 MHz [6] (30.0 dBm)
                        * 2442 MHz [7] (30.0 dBm)
                        * 2447 MHz [8] (30.0 dBm)
                        * 2452 MHz [9] (30.0 dBm)
                        * 2457 MHz [10] (30.0 dBm)
                        * 2462 MHz [11] (30.0 dBm)
                        * 2467 MHz [12] (disabled)
                        * 2472 MHz [13] (disabled)
                        * 2484 MHz [14] (disabled)
        valid interface combinations:
                 * #{ IBSS } <= 1, #{ managed, AP, mesh point } <= 4,
                   total <= 4, #channels <= 1, STA/AP BI must match
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
root@OpenWrt:~# iw reg get
global
country US: DFS-FCC
        (2402 - 2472 @ 40), (N/A, 30), (N/A)
        (5170 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
        (5490 - 5730 @ 160), (N/A, 23), (0 ms), DFS
        (5735 - 5835 @ 80), (N/A, 30), (N/A)
        (57240 - 63720 @ 2160), (N/A, 40), (N/A)
root@OpenWrt:~# cat /var/run/hostapd-phy1.conf 
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
channel=48

ieee80211n=1
ht_coex=0
ht_capab=[LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1]
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3]

interface=wlan1
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
wpa_passphrase=**********
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=WIFISSID
bridge=br-speed
wpa_disable_eapol_key_retries=1
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1
bssid=78:a3:51:32:47:3d

```

nbd168, 2018-10-06:
Please show me /etc/config/wireless

xavier83, 2018-10-07:
Hi @nbd168, Thanks for responding so quick... 

```

config wifi-device 'radio0'
        option type 'mac80211'
        option hwmode '11g'
        option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
        option country 'US'
        option legacy_rates '0'
        option distance '15'
        option channel '6'
        option htmode 'HT40'
        option txpower '15'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option mode 'ap'
        option ssid 'ANOTHERWIFISSID'
        option network 'lat'
        option encryption 'psk2'
        option key '**********'
        option wpa_disable_eapol_key_retries '1'
        option multicast_to_unicast '0'

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option country 'US'
        option legacy_rates '0'
        option distance '15'
        option txpower '15'
        option channel '48'
        option htmode 'VHT80'

config wifi-iface 'default_radio1'
        option device 'radio1'
        option mode 'ap'
        option ssid 'WIFISSID'
        option encryption 'psk2'
        option key '**********'
        option wpa_disable_eapol_key_retries '1'
        option multicast_to_unicast '0'
        option network 'speed'

config wifi-iface
        option device 'radio0'
        option mode 'ap'
        option ssid 'WIFISSID'
        option encryption 'psk2'
        option key '**********'
        option wpa_disable_eapol_key_retries '1'
        option multicast_to_unicast '0'
        option network 'speed'

```

nbd168, 2018-10-08:
Is this a new issue? What was the last working version?

xavier83, 2018-10-08:
Yes. It is a new issue. I've tested 5ghz clients with openwrt 18.06 stable release build on the same device and it was working fine with minor issues (ACS noise floor sth log spam when frequency set to auto)

xavier83, 2018-10-12:
@nbd168  any thoughts why it worked with the same firmware(18.06) earlier and not after reverting back from snapshot? did my hardware fail?

xavier83, 2018-10-13:
Tried again with snapshot(13th October 2018) still the issue persists.
attached the dmesg.
```
[    0.000000] Linux version 4.14.75 (buildbot@slashdirt-02) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r8313-7186520)) #0 SMP Fri Oct 12 07:36:05 2018
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is ZBT-WE1326
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 1c000000 @ 00000000 (usable)
[    0.000000]  memory: 04000000 @ 20000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] On node 0 totalpages: 131072
[    0.000000] free_area_init_node: node 0, pgdat 805ce300, node_mem_map 81003000
[    0.000000]   Normal zone: 512 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] random: get_random_bytes called from start_kernel+0x90/0x4ac with crng_init=0
[    0.000000] percpu: Embedded 14 pages/cpu @8148f000 s26256 r8192 d22896 u57344
[    0.000000] pcpu-alloc: s26256 r8192 d22896 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130560
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Writing ErrCtl register=0000cd50
[    0.000000] Readback ErrCtl register=0000cd50
[    0.000000] Memory: 512692K/524288K available (4670K kernel code, 241K rwdata, 1032K rodata, 252K init, 254K bss, 11596K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000010] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.007829] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.073977] pid_max: default: 32768 minimum: 301
[    0.078755] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.085262] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.094171] Hierarchical SRCU implementation.
[    0.099350] smp: Bringing up secondary CPUs ...
[    0.105419] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.105427] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.105439] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.105585] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.164097] Synchronize counters for CPU 1: done.
[    0.205437] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.205446] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.205456] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.205532] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.255212] Synchronize counters for CPU 2: done.
[    0.286547] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.286554] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.286564] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.286646] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.340392] Synchronize counters for CPU 3: done.
[    0.370266] smp: Brought up 1 node, 4 CPUs
[... 215 lines cut ...]
[  241.032969] kmodloader: done loading kernel modules from /etc/modules.d/*
[  241.254517] kmodloader: loading kernel modules from /etc/modules.d/*
[  241.264531] kmodloader: done loading kernel modules from /etc/modules.d/*
[  241.364939] kmodloader: loading kernel modules from /etc/modules.d/*
[  241.375049] kmodloader: done loading kernel modules from /etc/modules.d/*
[  241.591576] kmodloader: loading kernel modules from /etc/modules.d/*
[  241.601485] kmodloader: done loading kernel modules from /etc/modules.d/*
[  241.676877] kmodloader: loading kernel modules from /etc/modules.d/*
[  241.686157] kmodloader: done loading kernel modules from /etc/modules.d/*
[  241.807416] kmodloader: loading kernel modules from /etc/modules.d/*
[  241.816624] kmodloader: done loading kernel modules from /etc/modules.d/*
[... cut ...]
[  248.267130] kmodloader: done loading kernel modules from /etc/modules.d/*
[  324.962760] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[  324.972311] br-lan: port 2(wlan1) entered blocking state
[  324.977619] br-lan: port 2(wlan1) entered disabled state
[  324.983448] device wlan1 entered promiscuous mode
[  325.632672] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  325.639581] br-lan: port 2(wlan1) entered blocking state
[  325.645013] br-lan: port 2(wlan1) entered forwarding state
```

xavier83, 2018-10-18:
Still no luck after updating to snapshot (18th October 2018).

xavier83, 2018-10-18:
I hope this isn't a hardware issue that can't be fixed, no point breaking my head over this.. my 5ghz clients can see the network being advertised but just can't connect to it -> doesn't feel like a hw issue though. is there any other related issue that is causing this?.

xavier83, 2018-10-24:
retried with the snapshot firmware again(23 October 2018) still no luck.even see

@keulu I'm not getting any such Hostapd  log. Yes the antennas are plugged. 
FWIW, i tried changing the country code to CN, my 5ghz capable devices can't even detect the SSID broadcast. If I switch it back to US and the client can at least see the SSID broadcast on 5ghz. I've been testing while 2.4ghz radio is disabled as well.

WBINVD, 2018-11-30:
I have the same issue if I change the mac address of the 5ghz radio. 
I'm using a youhua wr1200js on the latest snapshot.

I changed the 5ghz mac address by adding something like "option macaddr '12:34:56:78:90:ab'" in the wifi-iface section of the 5ghz radio in /etc/config/wireless. The SSID is visible with the altered mac, but it's impossible to connect to it. There appears to be no traffic except for the beacon.

The 2.4ghz radio is unaffected by this and functions normally with or without the macaddr option. Disabling the 2.4ghz radio has no effect on the 5ghz radio. The 5ghz radio only works without the macaddr option.

I don't have any other dual-band hardware to test this on.

xavier83, 2018-12-08:
Actually the issue isn't limited to clients unable to associate with the router,I tried to scan for 5ghz networks using the router, it wouldn't list any networks. seems the 5ghz radio is fully defunct except for AP beacons..

xavier83, 2019-02-11:
Updated to Openwrt 18.02 still issue persists.

ShapeShifter499, 2019-04-20:
@xavier83 I stumbled across this after noticing issues with one of my routers, thing is I have more than one and the other routers were fine with 5GHz on the same OpenWrt 18.06.2 image.

I got it working by copying the config over from one of my other routers, I noticed the ordering of the entries in /etc/config/wireless

My router is a WD My Net n750 

While I don't have the same router as you can you try this config for me? You can edit the options to match your settings but make sure the order of the options are the same as what I have below. If it works then there is a bug, otherwise my hunch is incorrect.

 ```
config wifi-iface 'default_radio1'                                              
        option device 'radio1'                                                  
        option network 'lan'                                                    
        option mode 'ap'                                                        
        option encryption 'psk2+ccmp'                                           
        option key 'SomePassword'
        option wpa_disable_eapol_key_retries '1'
        option ssid Wifi 5GHz
```

xavier83, 2019-04-21:
> @xavier83 I stumbled across this after noticing issues with one of my routers, thing is I have more than one and the other routers were fine with 5GHz on the same OpenWrt 18.06.2 image.
>
> I got it [...]

Thanks for the suggestion, I tried these settings/options in the same order but no luck for my case. The weird thing is the network is broadcast as soon as the router is booted for a few seconds then it disappears, it is visible on macbook but won't associate to it. I suspect my error is something to do with the eeprom settings. I don't know how to fix it though.

xavier83, 2019-05-03:
Also noticed that the tx power always remains 10dbm even if set it to the max power listed in luci(30dbm) or anything inbetween.

xavier83, 2019-07-03:
Still no luck as of 18.06.4
/var/run/hostapd-phy1.conf is set to
```
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=US
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
channel=36

ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3]

interface=wlan1
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
auth_algs=1
wpa=0
ssid=UBISPLAT
bridge=br-speed
bssid=78:a3:51:32:47:3d
```

iscilyas, 2019-08-04:
FWIW, i can confirm that on the MIR3P 5ghz is working fine (can associate, etc). not as fast or robust-feeling as the closed-source driver, but usable.

xavier83, 2019-08-08:
Thanks for the info @iscilyas , BTW did you face the issues described here https://forum.openwrt.org/t/mt7621-pcie-dts-issues-mt76-stopped-working-with-latest-updates/14623/1 before it started working?

iscilyas, 2019-08-08:
@xavier83 no. no issues at all. of course, the MIR3P  is a dual mt7615e system... i don't know why i was under the impression that the ZBT-WE1326 also had mt7615e... sorry for any confusion.

btw, in that link you referenced... why on earth don't they have the `compatible` keyword in the .dts?

here's the PR for getting mt76/mt7615e to work on the MIR3P (for some reason no-one has even touched it in a week)....

https://github.com/openwrt/openwrt/pull/2293

xavier83, 2019-09-14:
I had been thinking 2.4 GHz was working fine. turns-out it hadn't been utilizing 40MHz bandwidth only just 20 MHz even after setting the channel width to 40 MHz on luci. Noticed it with a WifiAnalyzer App.

xavier83, 2019-12-02:
Not yet working on OpenWrt 19.07.0-rc1 realease, Tested on 02 Dec 2019.

lukasz1992, 2019-12-19:
@xavier83 Because OpenWRT 19.07 currently stays on commit from August.

leonardcoutinho, 2020-01-02:
openwrt 19.07 rc1 and rc2 works for me, but snapshots don't, after 10/10/19, builds stopped working with 5ghz radio, my devices can't connect to my router.

xavier83, 2020-01-15:
Have tested on Openwrt 19.07.0-rc2 and 19.07 stable release still no luck, Have tried with the latest snapshot dated 15 Jan 2020, attached the syslog for the same with no luck either, even the beacon broadcast doesn't seem to be working with the latest snapshot. 
[we1326-15jan-syslog.txt](https://github.com/openwrt/mt76/files/4067489/we1326-15jan-syslog.txt)

BTW did anybody have any luck with the proprietary driver, if so can you please share the steps to use the kernel module/build process using the proprietary driver? Thanks.

dangowrt, 2020-01-15:
@xavier83 afaik the proprietary driver requires a vendor-specific and outdated version of the  Linux kernel. Hence it can only be supported in outdated vendor specific OpenWrt forks...

xavier83, 2020-03-10:
Tested Openwrt 19.07.2 and still 5 Ghz is dead attaching the syslog and kernel log.
[openwrt-19.17.2-kern.log](https://github.com/openwrt/mt76/files/4312689/openwrt-19.17.2-kern.log)
[openwrt-19.17.2-syslog.log](https://github.com/openwrt/mt76/files/4312690/openwrt-19.17.2-syslog.log)

xavier83, 2020-05-26:
Tested Openwrt 19.07.3 no improvement still. attaching the kernel log.
[openwrt-19.17.3-kern.log](https://github.com/openwrt/mt76/files/4680449/openwrt-19.17.3-kern.log)

xavier83, 2020-06-07:
Tested latest snapshot 7 June 2020, and the device gets bricked.

rozhuk-im, 2020-06-15:
Try to set ```Beacon Interval``` to 1000.
This helps with DIR-510L (MT7610EN).

xavier83, 2020-06-16:
> Try to set `Beacon Interval` to 1000.
> This helps with DIR-510L (MT7610EN).

No luck with this device.

xavier83, 2020-06-16:
@nbd168 I wonder if the eeprom files listed at https://github.com/i80s/mtk-sources/tree/master/mt76x2e/src/eeprom will be of any use here. 
will it help my device's 5Ghz radio to work? is there any way i can use them to build them and test?

xavier83, 2020-07-15:
More logs.
[system.log](https://github.com/openwrt/mt76/files/4927146/system.log)

ShirokovPN, 2020-08-05:
Hi, I have partially the same problem. I have 2 devices Lenovo Newifi 3 D2.
First device works fine with VHT80. Second device is available by wireless only with VHT40 settings, with VHT80 no one client can find them. OpenWrt 19.07.3 r11063-85e04e9f46.
I can start some tests or something else, that would help to close this problem. Just say it :)

xavier83, 2020-10-04:
Tested again on OpenWRT 19.07.4 , 5GHz signal is completely dead.

rdslw, 2020-10-04:
there are few commits to mt76 on openwrt/master in August and September.

Could you please @xavier83 test recent SNAPSHOT build?

xavier83, 2020-10-16:
@rdslw I tried the latest snapshot dated 16 October and device continues to be in a boot loop.
noticed this message when trying to flash though, doesn't make sense though :stuck_out_tongue: 
```
Device zbt-we1326 not supported by this image Supported devices: zbtlink,zbt-we1326 zbt-we1326 - Image version mismatch: image 1.1, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: Config cannot be migrated from swconfig to DSA Image check failed.

The uploaded image file does not contain a supported format. Make sure that you choose the generic image format for your platform.
```

rdslw, 2020-10-17:
Thanks for test, looks like we're doomed :( Probably in two months it would be worthwhile to test again.

Message is ok, as upgrading from 1.0 to 1.1 is expected to show that. Just to be sure: you DID force reset of configuration ? /it is required, without it it will not boot ok/

xavier83, 2020-10-17:
I did force reset the configuration.

xavier83, 2020-10-19:
Is the antenna configuration something that i should look into. does anyone here know if we can bug anyone from Mediatek to provide a fix for this hardware? not even 5Ghz scanning is working.
```
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2247 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 1 (up to 450m)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 2:
                Capabilities: 0x1ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-15
                VHT Capabilities (0x318001b0):
                        Max MPDU length: 3895
                        Supported Channel Width: neither 160 nor 80+80
                        RX LDPC
                        short GI (80 MHz)
                        TX STBC
                        RX antenna pattern consistency
                        TX antenna pattern consistency
                VHT RX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                        1 streams: MCS 0-9
                        2 streams: MCS 0-9
                        3 streams: not supported
                        4 streams: not supported
                        5 streams: not supported
                        6 streams: not supported
                        7 streams: not supported
                        8 streams: not supported
                VHT TX highest supported: 0 Mbps
                Frequencies:
                        * 5180 MHz [36] (21.0 dBm)
                        * 5200 MHz [40] (21.0 dBm)
                        * 5220 MHz [44] (21.0 dBm)
                        * 5240 MHz [48] (21.0 dBm)
[... 78 lines cut ...]
                        * 2442 MHz [7] (21.0 dBm)
                        * 2447 MHz [8] (21.0 dBm)
                        * 2452 MHz [9] (21.0 dBm)
                        * 2457 MHz [10] (21.0 dBm)
                        * 2462 MHz [11] (21.0 dBm)
                        * 2467 MHz [12] (disabled)
                        * 2472 MHz [13] (disabled)
                        * 2484 MHz [14] (disabled)
        valid interface combinations:
                 * #{ IBSS } <= 1, #{ managed, AP, mesh point } <= 4,
                   total <= 4, #channels <= 1, STA/AP BI must match
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Supported extended features:
                * [ RRM ]: RRM
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs

```

gregd72002, 2020-11-29:
@xavier83 

For my ZBT-WE1326 (256MB):
```
root@OpenWrt:~# cat /proc/mtd 
dev:    size   erasesize  name
mtd0: 00030000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00010000 00010000 "factory"
mtd3: 00fb0000 00010000 "firmware"
mtd4: 001ca1a1 00010000 "kernel"
mtd5: 00de5e5f 00010000 "rootfs"
mtd6: 00b50000 00010000 "rootfs_data"
root@OpenWrt:~# 
root@OpenWrt:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 2.8M      2.8M         0 100% /rom
tmpfs                   123.9M    116.0K    123.8M   0% /tmp
/dev/mtdblock6           11.3M      1.0M     10.3M   9% /overlay
overlayfs:/overlay       11.3M      1.0M     10.3M   9% /
tmpfs                   512.0K         0    512.0K   0% /dev
root@OpenWrt:~# 
```

xavier83, 2020-12-01:
Thanks @gregd72002 . Maybe the newer zbt-we1326 with 256MB ram doesn't have this issue. 

BTW i learnt some interesting things about this hardware. I was under the impression that the 5ghz radio in zbt-we1326 was mt7612e (https://wikidevi.wi-cat.ru/ZBT_WE1326) but it turns out it is reported as mt7662e. not sure if the mt76x2e supports the mt7662e as well because according to https://www.mediatek.com/products/broadbandWifi/mt7662e it support 802.11ac draft standard where as mt7612e supports the full 802.11ac standard. mt7662e apparently support bluetooth 2.1 + EDR,  3.0,  4.0 (Low Energy) :exploding_head:  

```
root@OpenWrt:~# lspci
00:00.0 PCI bridge: Device 0e8d:0801 (rev 01)
00:01.0 PCI bridge: Device 0e8d:0801 (rev 01)
01:00.0 Network controller: MEDIATEK Corp. MT7662E 802.11ac PCI Express Wireless Network Adapter
02:00.0 Network controller: MEDIATEK Corp. Device 7603
```
Als the only other issue that mentions this radio is https://github.com/openwrt/mt76/issues/222
more discussion on the forum about mt7662 https://forum.openwrt.org/t/mt7662e-txpower-problem/20842

```
root@OpenWrt:~# cat /proc/interrupts 
           CPU0       CPU1       CPU2       CPU3       
  8:    9796410    9796387    9796384    9796377  MIPS GIC Local   1  timer
  9:    5783040          0          0          0  MIPS GIC  63  IPI call
 10:          0    4717968          0          0  MIPS GIC  64  IPI call
 11:          0          0    7491602          0  MIPS GIC  65  IPI call
 12:          0          0          0    4138782  MIPS GIC  66  IPI call
 13:    1206674          0          0          0  MIPS GIC  67  IPI resched
 14:          0    1150113          0          0  MIPS GIC  68  IPI resched
 15:          0          0    2656526          0  MIPS GIC  69  IPI resched
 16:          0          0          0    1219207  MIPS GIC  70  IPI resched
 19:         12          0          0          0  MIPS GIC  33  ttyS0
 20:          0          0          0          0  MIPS GIC  27  1e130000.sdhci
 21:          0          0          0          0  MIPS GIC  29  xhci-hcd:usb1
 22:    4711322          0          0          0  MIPS GIC  10  1e100000.ethernet
 23:         10          0          0          0  MIPS GIC  30  gsw
 25:          2          0     193284          0  MIPS GIC  31  mt76x2e
 26:         23          0          0   13159170  MIPS GIC  32  mt7603e
ERR:         39
```
@nbd168 does this explain why 5ghz might not be working on this device. if so what can i try next? Thanks.

rozhuk-im, 2020-12-08:
Can you test https://github.com/openwrt/mt76/pull/481 ?

dfateyev, 2022-12-05:
I know this is an old thread, but anyway. Related issues were investigated in the openwrt/openwrt/pull/11220 thread, and seems no more reproduced — or are fixed — already.
BTW, anybody is encouraged to test master snapshots with "kernel-5.15", and compare with 22.03: ramips is heading towards switching kernel to 5.15 in master, and more feedback is appreciated.