openwrt/mt76 #200: XBox One Wireless Adapter v2 doesn't work

diizzyy, 2018-10-08:
In order to avoid any confusion, it's this adapter:
https://mspoweruser.com/new-xbox-wireless-adapter-significantly-slimmer/

Hardware:
Orange Pi PC (Allwinner H3)

Software:
OpenWrt master (commit 16c712501cdcb10f333f4abc027a0dcdb8f8101b)
(with some hacking of the mt76 package(s))

I tried adding the device id however that doesn't seem enough:
```{ USB_DEVICE(0x045e, 0x02fe) }, /* XBox One Wireless Adapter v2 */```

**dmesg**
```
[    7.482412] mt76x2u 3-1:1.0: ASIC revision: 76320044
[    8.770685] mt76x2u 3-1:1.0: could not get hardware semaphore for ROM PATCH
[    9.906948] mt76x2u: probe of 3-1:1.0 failed with error -110
[    9.912812] usbcore: registered new interface driver mt76x2u
```

**lsusb** (serial zeroed)
```
Bus 003 Device 002: ID 045e:02fe Microsoft Corp.
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x045e Microsoft Corp.
  idProduct          0x02fe
  bcdDevice            1.00
  iManufacturer           1 Microsoft Inc.
  iProduct                2 XBOX ACC
  iSerial                 3 000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           74
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          2 XBOX ACC
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           8
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 XBOX ACC
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x07  EP 7 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x09  EP 9 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength           12
  bNumDeviceCaps          1
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000f41e
      Link Power Management (LPM) Supported
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```

Windows drivers can be obtained here (includes firmware):
http://www.catalog.update.microsoft.com/Search.aspx?q=xbox%20wireless%20adapter

18.X - 21.X seems to be for the older adapter
1.0.X is for the new one

Having a quick look at the drivers they're using different firmwares but that's as far as I can tell.
If there's anything else I can provide just ask and I'll try my best.

@LorenzoBianconi

LorenzoBianconi, 2018-10-24:
>
> Software: OpenWrt master (commit e1db49310e9912f4076789e704180ab3b9ea0184)
>
> Same error message as before
>
> *dmesg*
>
> [    7.772409] mt76x2u 3-1:1.0: ASIC revision: 76320044
> [    7.830443] mt76x2u 3-1:1.0: ROM patch build: 20141115060606a
> [    7.969521] mt76x2u 3-1:1.0: Firmware Version: 0.0.00
> [    7.974646] mt76x2u 3-1:1.0: Build: 1
> [    7.978315] mt76x2u 3-1:1.0: Build Time: 201507311614____
> [    8.187039] mt76x2u 3-1:1.0: error: MCU resp evt:9 seq:1-0
> [    8.192562] mt76x2u 3-1:1.0: error: MCU resp evt:9 seq:1-0
> [    8.198058] mt76x2u 3-1:1.0: error: MCU resp evt:9 seq:1-0
>
> <https://github.com/openwrt/mt76/issues/200#issuecomment-432126219>, or mute
> the thread

Are you sure that chipset is compatible with mt76 driver? can you point out
some reference code for it?

Regards,
Lorenzo

diizzyy, 2018-10-25:
Hi,
I did the (blantant) assumption as the Windows driver does refer it as "mt7612US" which is also true for the supported v1 adapter. If you extract the .cab files from the Windows site you'll see what I mean. Of course, it can be a variant that's crippled. Unfortunately I do not have the knowledge to compare firmwares (if possible?) and figure out even if they're even remotely similar or some other way of determine if it's possible to support.

Best regards,
Daniel

LorenzoBianconi, 2019-01-16:
@diizzyy : are you using an AMD motherboard/CPU? if so, could you please double check if IOMMU is enabled?

diizzyy, 2019-01-16:
@LorenzoBianconi 
No, ARM - Allwinner H3 SoC (sunxi), OrangePi PC

ddragic, 2019-03-30:
It works for me on Fedora 29 x86-64 with kernel v5.0.3, I just added the new device id to mt76x2/usb.c as @diizzyy did and built it as an out of tree module:

Kernel log:

```
[14210.387490] usb 9-1.2: new high-speed USB device number 3 using xhci_hcd
[14210.464722] usb 9-1.2: New USB device found, idVendor=045e, idProduct=02fe, bcdDevice= 1.00
[14210.464725] usb 9-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[14210.464727] usb 9-1.2: Product: XBOX ACC
[14210.464729] usb 9-1.2: Manufacturer: Microsoft Inc.
[14210.464731] usb 9-1.2: SerialNumber: 291586
[14210.629576] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[14210.631343] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[14210.706863] mt76: loading out-of-tree module taints kernel.
[14210.706954] mt76: module verification failed: signature and/or required key missing - tainting kernel
[14210.784848] usb 9-1.2: reset high-speed USB device number 3 using xhci_hcd
[14210.865300] mt76x2u 9-1.2:1.0: ASIC revision: 76320044
[14210.894545] mt76x2u 9-1.2:1.0: ROM patch build: 20141115060606a
[14211.059284] mt76x2u 9-1.2:1.0: Firmware Version: 0.0.00
[14211.059286] mt76x2u 9-1.2:1.0: Build: 1
[14211.059288] mt76x2u 9-1.2:1.0: Build Time: 201507311614____
[14211.317795] mt76x2u 9-1.2:1.0: error: MCU resp evt:9 seq:1-0
[14211.317830] mt76x2u 9-1.2:1.0: error: MCU resp evt:9 seq:1-0
[14211.317870] mt76x2u 9-1.2:1.0: error: MCU resp evt:9 seq:1-0
[14211.862129] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[14211.863085] usbcore: registered new interface driver mt76x2u
[14211.896214] mt76x2u 9-1.2:1.0 wlp3s0u1u2: renamed from wlan0
```
I haven't tested it extensively, but everything seems to work ok.

I wasn't able to build the latest code from this repo, the build was complaining about an implicit declaration of ieee80211_schedule_txq (I think I need a newer kernel for the recent changes).

perillamint, 2019-11-16:
AFAIK, it uses MT7600U rather then MT7612U. Wikidevi has information about dongles but now Wikidevi is dead. :(

ddragic, 2019-11-17:
The older, bigger, adapter uses MT7600U for sure, but it looks like the newer one (model 1790) is based on MT7632U as can be seen from the dmesg output above: **ASIC revision: 76320044**

Also on TechInfoDepot it's listed as MT7632TUN:
http://en.techinfodepot.shoutwiki.com/wiki/Microsoft_1790