openwrt/mt76 #201: mt7612e: unusually high inactive timeout for stations

mweinelt, 2018-10-16:
On ath9k/10k the most I see for inactive time is around 300s. On a D-Link DIR860L I see inactivity times up to half an hour on the 5 GHz radio (MT7612E) only.

```
# iw client0 station dump | grep inactive
	inactive time:	15157640 ms
	inactive time:	2845020 ms
	inactive time:	2614110 ms
	inactive time:	2273520 ms
	inactive time:	0 ms
	inactive time:	1790190 ms
	inactive time:	1109130 ms
	inactive time:	989860 ms
	inactive time:	987530 ms
	inactive time:	916600 ms
	inactive time:	927660 ms
	inactive time:	744370 ms
	inactive time:	580820 ms
	inactive time:	565460 ms
	inactive time:	80290 ms
	inactive time:	100280 ms
```

nbd168, 2018-10-25:
Please try the latest version (master only, not 18.06).

mweinelt, 2018-10-26:
Inactive stations now disappear when higher than 300s. Thanks!

nbd168, 2018-10-26:
Thanks for testing, I will backport the change to 18.06

closed 2018-10-26 by nbd168