openwrt/mt76 #202: mt76x0e doesn't work

LGA1150, 2018-10-16:
Platform is MT7620A+MT7610E

Can confirm mt76x0e module is loaded, but phy0 isn't up
```
[   12.474663] mt76x0e 0000:01:00.0: card - bus=0x1, slot = 0x0 irq=4
[   12.481175] mt76x0e 0000:01:00.0: ASIC revision: 76100002
[   12.489275] mt76x0e 0000:01:00.0: Firmware Version: 0.1.00
[   12.578946] mt76x0e 0000:01:00.0: EEPROM ver:01 fae:00
[   12.587562] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   12.700344] rt2800_wmac 10180000.wmac: loaded eeprom from mtd device "factory"
[   12.707739] ieee80211 phy1: rt2x00_set_rt: Info - RT chipset 6352, rev 0500 detected
[   12.715719] ieee80211 phy1: rt2x00_set_rf: Info - RF chipset 7620 detected
[   12.723445] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   12.732955] kmodloader: done loading kernel modules from /etc/modules.d/*
[   14.010429] urandom_read: 5 callbacks suppressed
[   14.010439] random: jshn: uninitialized urandom read (4 bytes read)
[   29.765565] 8021q: adding VLAN 0 to HW filter on device eth0
[   29.783742] device eth0 entered promiscuous mode
[   29.803159] br-lan: port 1(eth0.1) entered blocking state
[   29.808686] br-lan: port 1(eth0.1) entered disabled state
[   29.814674] device eth0.1 entered promiscuous mode
[   29.897983] br-lan: port 1(eth0.1) entered blocking state
[   29.903576] br-lan: port 1(eth0.1) entered forwarding state
[   29.909485] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   30.791963] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   38.124220] ieee80211 phy1: rt2800_rf_self_txdc_cal: Info - RF Tx self calibration start
[   38.133042] ieee80211 phy1: rt2800_rf_self_txdc_cal: Info - RF Tx self calibration end
[   44.717469] ieee80211 phy1: rt2800_loft_iq_calibration: Info - LOFT Calibration Done!
[   45.230479] ieee80211 phy1: rt2800_iq_search: Info - IQCalibration Start!
[   45.251819] ieee80211 phy1: rt2800_iq_search: Info - IQCalibration Done! CH = 0, (gain= 0, phase= 2)
[   45.262069] ieee80211 phy1: rt2800_iq_search: Info - IQCalibration Start!
[   45.283383] ieee80211 phy1: rt2800_iq_search: Info - IQCalibration Done! CH = 1, (gain= 1, phase= 1)
[   45.292740] ieee80211 phy1: rt2800_loft_iq_calibration: Info - TX IQ Calibration Done!
[   45.802525] ieee80211 phy1: rt2800_rxiq_calibration: Warning - Wait MAC Status to MAX !!!
[   49.170036] ieee80211 phy1: rt2800_rxiq_calibration: Warning - RXIQ calibration FAIL
[   49.185884] ieee80211 phy1: rt2800_rxiq_calibration: Warning - RXIQ calibration FAIL
[   49.279513] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   49.339164] br-lan: port 2(wlan1) entered blocking state
[   49.344701] br-lan: port 2(wlan1) entered disabled state
[   49.350836] device wlan1 entered promiscuous mode
[   49.458144] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   49.465319] br-lan: port 2(wlan1) entered blocking state
[   49.470754] br-lan: port 2(wlan1) entered forwarding state
[   56.881858] random: crng init done
```

LorenzoBianconi, 2018-10-16:
>
> Platform is MT7620A+MT7610E
>
> Can confirm mt76x0e module is loaded, but phy0 isn't up
>
>
Could you please provide your /etc/config/wireless? Please note just sta is
currently supported

LGA1150, 2018-10-16:
Only STA mode? I see, so do you have ETA on AP mode support?

LorenzoBianconi, 2018-10-16:
>
> Only STA mode? I see, so do you have ETA on AP mode support?

I am currently working on it

-- 
UNIX is Sexy: who | grep -i blonde | talk; cd ~; wine; talk; touch;
unzip; touch; strip; gasp; finger; gasp; mount; fsck; more; yes; gasp;
umount; make clean; sleep

LGA1150, 2018-10-24:
Working with latest commit, great job!

closed 2018-10-24 by LGA1150