openwrt/mt76 #204: 请教如何用最新的源码支持1701版本的openwrt喃？

hzjnet, 2018-10-21:
如题 ：1701版本如何使用现在的源码？

再请教：何时能增加对7615芯片的支持喃？  

谢谢

LGA1150, 2018-10-21:
English please

hzjnet, 2018-10-25:
看来改成鸟语还是然并卵啊

yousong, 2018-10-26:
Be professional please.

> 1 How can we use your code of the newest driver in lede-17.01 ?

Version bumps in release code are usually only for bug fixes or security issues, etc.  You are on your own to try and test newest version on lede-17.01 if this is not happening.

> 2 I wonder why I often find myself offline when using 2.4G?

Provide info on your environment, versions of critical components, detailed description of symptoms that can help diagnose the issue and find the cause.

> 3 When will the support to chip mt7615 be added ?

When it comes to open source projects overall, I'd like to it comes when it comes ;)   I will leave this question open to others ;)

closed 2018-10-26 by yousong