openwrt/mt76 #206: MT76x0e does not stay associated to my router

jefflancaster, 2018-10-22:
Using a simple wpa_supplicant, the device appears to associate to my router, but after several seconds the driver will disconnect and retry. On some routers it does not associate at all.

wpa_supplicant -i wlan1 -c wpa.conf -dd

root@OpenWrt:~# wpa_supplicant -i wlan1 -c wpa.conf -dd
Successfully initialized wpa_supplicant
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:76 (SSID='KILN_Member' freq=5220 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:76 (SSID='KILN_Member' freq=5220 MHz)
wlan1: Associated with e6:cb:ac:88:7a:76
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:76 [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:76 completed [id=0 id_str=]
wlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:76 reason=3 locally_generated=1
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:a3 (SSID='KILN_Member' freq=5200 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:a3 (SSID='KILN_Member' freq=5200 MHz)
wlan1: Associated with e6:cb:ac:88:7a:a3
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:a3 [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:a3 completed [id=0 id_str=]
wlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:a3 reason=3 locally_generated=1
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:76 (SSID='KILN_Member' freq=5220 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:76 (SSID='KILN_Member' freq=5220 MHz)
wlan1: Associated with e6:cb:ac:88:7a:76
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:76 [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:76 completed [id=0 id_str=]
wlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:76 reason=3 locally_generated=1
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:a3 (SSID='KILN_Member' freq=5200 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:a3 (SSID='KILN_Member' freq=5200 MHz)
wlan1: Associated with e6:cb:ac:88:7a:a3
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:a3 [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:a3 completed [id=0 id_str=]
wlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:a3 reason=3 locally_generated=1
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:76 (SSID='KILN_Member' freq=5220 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:76 (SSID='KILN_Member' freq=5220 MHz)
wlan1: Associated with e6:cb:ac:88:7a:76
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:76 [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:76 completed [id=0 id_str=]
wlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:76 reason=3 locally_generated=1
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:a3 (SSID='KILN_Member' freq=5200 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:a3 (SSID='KILN_Member' freq=5200 MHz)
wlan1: Associated with e6:cb:ac:88:7a:a3
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:a3 [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:a3 completed [id=0 id_str=]
wlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:a3 reason=3 locally_generated=1
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:9d (SSID='KILN_Member' freq=5240 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:9d (SSID='KILN_Member' freq=5240 MHz)
wlan1: Associated with e6:cb:ac:88:7a:9d
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:9d [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:9d completed [id=0 id_str=]
wlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:9d reason=3 locally_generated=1
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:76 (SSID='KILN_Member' freq=5220 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:76 (SSID='KILN_Member' freq=5220 MHz)
wlan1: Associated with e6:cb:ac:88:7a:76
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:76 [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:76 completed [id=0 id_str=]
wlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:76 reason=3 locally_generated=1
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:a3 (SSID='KILN_Member' freq=5200 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:a3 (SSID='KILN_Member' freq=5200 MHz)
wlan1: Associated with e6:cb:ac:88:7a:a3
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:a3 [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:a3 completed [id=0 id_str=]
wlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:a3 reason=3 locally_generated=1
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:9d (SSID='KILN_Member' freq=5240 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:9d (SSID='KILN_Member' freq=5240 MHz)
wlan1: Associated with e6:cb:ac:88:7a:9d
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:9d [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:9d completed [id=0 id_str=]
wlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:9d reason=3 locally_generated=1
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:76 (SSID='KILN_Member' freq=5220 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:76 (SSID='KILN_Member' freq=5220 MHz)
wlan1: Associated with e6:cb:ac:88:7a:76
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:76 [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:76 completed [id=0 id_str=]
wlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:76 reason=3 locally_generated=1
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:a3 (SSID='KILN_Member' freq=5200 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:a3 (SSID='KILN_Member' freq=5200 MHz)
wlan1: Associated with e6:cb:ac:88:7a:a3
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:a3 [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:a3 completed [id=0 id_str=]
wlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:a3 reason=3 locally_generated=1
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:9d (SSID='KILN_Member' freq=5240 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:9d (SSID='KILN_Member' freq=5240 MHz)
wlan1: Associated with e6:cb:ac:88:7a:9d
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:9d [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:9d completed [id=0 id_str=]

nbd168, 2018-10-23:
Please try the latest version, it should work now.

jefflancaster, 2018-10-24:
The latest changes look good and our device seem to work fine.

Let me know if the following events are issues on the 2.4ghz radio.

root@OpenWrt:~# wpa_supplicant -i wlan0 -c wpa.conf

Successfully initialized wpa_supplicant
wlan0: SME: Trying to authenticate with e6:cb:bc:88:7a:76 (SSID='KILN_Member' freq=2462 MHz)
wlan0: Trying to associate with e6:cb:bc:88:7a:76 (SSID='KILN_Member' freq=2462 MHz)
wlan0: Associated with e6:cb:bc:88:7a:76
wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan0: WPA: Key negotiation completed with e6:cb:bc:88:7a:76 [PTK=CCMP GTK=CCMP]
wlan0: CTRL-EVENT-CONNECTED - Connection to e6:cb:bc:88:7a:76 completed [id=0 id_str=]
wlan0: CTRL-EVENT-BEACON-LOSS
wlan0: CTRL-EVENT-BEACON-LOSS
wlan0: CTRL-EVENT-BEACON-LOSS
wlan0: CTRL-EVENT-BEACON-LOSS
wlan0: CTRL-EVENT-BEACON-LOSS
wlan0: CTRL-EVENT-BEACON-LOSS
wlan0: CTRL-EVENT-BEACON-LOSS
wlan0: CTRL-EVENT-BEACON-LOSS
wlan0: CTRL-EVENT-BEACON-LOSS
wlan0: CTRL-EVENT-BEACON-LOSS
wlan0: CTRL-EVENT-BEACON-LOSS
wlan0: CTRL-EVENT-BEACON-LOSS
wlan0: CTRL-EVENT-BEACON-LOSS
^Cwlan0: CTRL-EVENT-DISCONNECTED bssid=e6:cb:bc:88:7a:76 reason=3 locally_generated=1
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
wlan0: CTRL-EVENT-TERMINATING

root@OpenWrt:~# wpa_supplicant -i wlan1 -c wpa.conf
Successfully initialized wpa_supplicant
wlan1: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan1: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan1: CTRL-EVENT-REGDOM-CHANGE init=BEACON_HINT type=UNKNOWN
wlan1: SME: Trying to authenticate with e6:cb:ac:88:7a:76 (SSID='KILN_Member' freq=5220 MHz)
wlan1: Trying to associate with e6:cb:ac:88:7a:76 (SSID='KILN_Member' freq=5220 MHz)
wlan1: Associated with e6:cb:ac:88:7a:76
wlan1: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan1: WPA: Key negotiation completed with e6:cb:ac:88:7a:76 [PTK=CCMP GTK=CCMP]
wlan1: CTRL-EVENT-CONNECTED - Connection to e6:cb:ac:88:7a:76 completed [id=0 id_str=]
^Cwlan1: CTRL-EVENT-DISCONNECTED bssid=e6:cb:ac:88:7a:76 reason=3 locally_generated=1
nl80211: deinit ifname=wlan1 disabled_11b_rates=0
wlan1: CTRL-EVENT-TERMINATING

5ghz radio does not have beacon loss events.

Jeff Lancaster
Principal Software Engineer
Komodo Systems, LLC
2000 Ashton Blvd.
Lehi, Utah 84045

closed 2018-10-25 by nbd168