openwrt/mt76 #211: NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out

swg0101, 2018-11-04:
It looks like on my Xiaomi 3G device the device crashes every other day or so and would require a hard reboot to get traffic to flow again. This problem has been there for a while now, and the problem exists as of OpenWrt SNAPSHOT r8393-dd9da51 (10/30/18). Any ideas why this is happening? I have seen there were some attempts earlier in fixing this but it's still occurring on this device. Thanks!

```
[58659.651455] ------------[ cut here ]------------
[58659.656103] WARNING: CPU: 3 PID: 0 at net/sched/sch_generic.c:320 dev_watchdog+0x1ac/0x324
[58659.664352] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[58659.671277] Modules linked in: rt2800usb rt2800lib pppoe ppp_async rt2x00usb rt2x00lib rndis_host pppox ppp_generic mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_MASQUERADE ebtable_nat ebtable_filter ebtable_broute cp210x ch341 cfg80211 cdc_ether xt_u32 xt_time xt_tcpmss xt_string xt_statistic xt_state xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_bpf xt_addrtype xt_TRACE xt_TEE xt_TCPMSS xt_REDIRECT xt_NFQUEUE xt_NFLOG xt_NETMAP xt_LOG xt_LED xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_DELUDE xt_TARPIT ipt_REJECT xt_tcpudp xt_CHAOS wireguard usbserial usbnet ums_usbat ums_sddr55 ums_sddr09
[58659.742286]  ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm slhc rtl8150 r8152 nfnetlink_queue nfnetlink_log nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_dup_ipv6 nf_dup_ipv4 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack_netlink iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ip6table_raw ip_tables ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_nflog ebt_mark_m ebt_mark ebt_log ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 crc_itu_t crc_ccitt compat_xtables compat br_netfilter arptable_filter arpt_mangle arp_tables act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb
[58659.812907]  sch_hfsc sch_ingress ledtrig_usbport cryptodev xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 nf_nat nf_conntrack ip6t_rt ip6t_frag ip6t_hbh ip6t_eui64 ip6t_mh ip6t_ah ip6t_ipv6header ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables bonding ip6_udp_tunnel udp_tunnel ipcomp6 xfrm6_tunnel xfrm6_mode_tunnel xfrm6_mode_transport xfrm6_mode_beet esp6 ah6 ipcomp xfrm4_tunnel
[58659.883257]  xfrm4_mode_tunnel xfrm4_mode_transport xfrm4_mode_beet esp4 ah4 tunnel6 tunnel4 tun af_key xfrm_user xfrm_ipcomp xfrm_algo ntfs dm_crypt dm_mirror dm_region_hash dm_log dm_mod dax twofish_generic serpent_generic wp512 twofish_common tgr192 tea khazad cast6_generic cast5_generic cast_common camellia_generic anubis xts echainiv des_generic usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd uhci_hcd ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 jbd2 mbcache exfat usbcore nls_base usb_common mii crc32c_generic crc32_generic
[58659.935453] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.78 #0
[58659.941432] Stack : 00000000 8fe45840 805c0000 8006f654 805e0000 805842ec 00000000 00000000
[58659.949772]         8054fbf8 8fc15dc4 8fc44e7c 805be927 8054ace8 00000001 8fc15d68 1cc282ae
[58659.958104]         00000000 00000000 80630000 00007d78 00000000 00000181 00000008 00000000
[58659.966438]         00000000 805c0000 000e461d 00000000 00000000 00000000 805e0000 80396a38
[58659.974772]         00000009 00000140 00000003 8fe45840 00000003 802bf4d0 0000000c 8063000c
[58659.983107]         ...
[58659.985544] Call Trace:
[58659.988006] [<800106c0>] show_stack+0x58/0x100
[58659.992450] [<80487d2c>] dump_stack+0x9c/0xe0
[58659.996793] [<8002e408>] __warn+0xe0/0x114
[58660.000869] [<8002e46c>] warn_slowpath_fmt+0x30/0x3c
[58660.005820] [<80396a38>] dev_watchdog+0x1ac/0x324
[58660.010521] [<800865c4>] call_timer_fn.isra.3+0x24/0x84
[58660.015726] [<800867e0>] run_timer_softirq+0x1bc/0x248
[58660.020859] [<804a5200>] __do_softirq+0x128/0x2ec
[58660.025551] [<80032b34>] irq_exit+0xac/0xc8
[58660.029734] [<8027323c>] plat_irq_dispatch+0xfc/0x138
[58660.034766] [<8000b5e8>] except_vec_vi_end+0xb8/0xc4
[58660.039709] [<8000cfb0>] r4k_wait_irqoff+0x1c/0x24
[58660.044505] [<800666cc>] do_idle+0xe4/0x168
[58660.048671] [<80066948>] cpu_startup_entry+0x24/0x2c
[58660.053667] ---[ end trace e91c8f91e07b1e43 ]---
[58660.058272] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[58660.064436] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[58660.070419] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0ebc0000, max=0, ctx=2323, dtx=2323, fdx=2322, next=2323
[58660.081251] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0d750000, max=0, calc=2946, drx=2947
[58660.096895] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[58660.109472] mtk_soc_eth 1e100000.ethernet: PPE started
```

swg0101, 2018-11-04:
Also to note that this router is constantly transferring around 12Mbps of data between Wi-Fi and LAN 24/7 and that both router/NAT/software offloading is turned off (since that was causing crashes if I tried it earlier)...

kristrev, 2018-11-08:
This issue is not connected with the wifi driver, but with the switch. I have experienced the same issue and started a [discussion](https://www.mail-archive.com/lede-dev@lists.infradead.org/msg08327.html) some time ago on the OpenWRT-mailing list. We have not been able to find a solution, but if you skip to the last email you can read a description of my current work-around. The short version is that in my case, the timeout problem was caused by pause frames. Disabling flow control on the switch made the issue go away in my case.

If you (or other) are interested, I can brush up my patch and submit it to the OpenWRT mailing list for review.

swg0101, 2018-11-08:
Count me in... Have you experienced any issues with flow control turned off?

swg0101, 2018-11-09:
Great - is there somewhere I can monitor this so I can get updates on it after closing this issue? Thanks!

swg0101, 2018-11-21:
Well looks like in the latest commits I am not seeing this anymore... I do see there were some changes made to mt76 - maybe they are related - will close this for now.

closed 2018-11-21 by swg0101

camelcamro, 2019-07-21:
this issues still exists .... tried newest TRUNK version ... (MT7620 MT 7621)

how is it possible to disable the FLOW CONTROL ? is that possible via settings on /etc/cfg dir ?
cu camel

problems ...
eg:
```
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.590371] ------------[ cut here ]------------
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.595142] WARNING: CPU: 3 PID: 538 at net/sched/sch_generic.c:320 dev_watchdog+0x1ac/0x324
Sun Jul 21 00:10:15 2019 kern.info kernel: [100743.603727] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.610841] Modules linked in: pppoe ppp_async option usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usbserial ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.681662]  nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conntrack macvlan iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat br_netfilter sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.753181]  ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 nfs ip_gre gre ifb ip_tunnel tun vfat fat lockd sunrpc grace dns_resolver dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb usb_storage ext4 mbcache jbd2 crc32c_generic mmc_block mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.799794] CPU: 3 PID: 538 Comm: jffs2_gcd_mtd6 Not tainted 4.14.131 #0
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.806570] Stack : 00000000 8fd3cd40 80600000 8007cecc 80630000 805cdb88 00000000 00000000
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.815005]         80596cf8 8fc0fdc4 8fd7580c 8060aa47 80591b38 00000001 8fc0fd68 ac07f59a
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.823428]         00000000 00000000 80680000 000087e0 00000000 00000194 00000008 00000000
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.831851]         00000000 80610000 000c3432 666a203a 00000000 00000000 80630000 805c79e8
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.840278]         803cafa8 00000140 00000003 8fd3cd40 00000000 802eab28 0000000c 8067000c
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.848704]         ...
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.851229] Call Trace:
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.853783] [<800108d0>] show_stack+0x58/0x100
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.858322] [<804c95ec>] dump_stack+0x9c/0xe0
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.862760] [<800339b8>] __warn+0xe0/0x138
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.866927] [<80033a40>] warn_slowpath_fmt+0x30/0x3c
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.871976] [<803cafa8>] dev_watchdog+0x1ac/0x324
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.876752] [<80093f4c>] call_timer_fn.isra.25+0x24/0x84
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.882133] [<80094208>] run_timer_softirq+0x1bc/0x248
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.887355] [<804e6828>] __do_softirq+0x128/0x2ec
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.892139] [<80038664>] irq_exit+0xac/0xc8
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.896429] [<80298620>] plat_irq_dispatch+0xfc/0x138
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.901558] [<8000b588>] except_vec_vi_end+0xb8/0xc4
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.906586] [<804e01f8>] __udelay+0x48/0x58
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.910861] [<8034d738>] mt7621_spi_read_half_duplex+0xd4/0x17c
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.916840] [<8034d948>] mt7621_spi_transfer_one_message+0x168/0x370
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.923256] [<8034b108>] __spi_pump_messages+0x4a0/0x534
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.928631] [<8034b370>] __spi_sync+0x1c8/0x208
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.933230] [<8034b3e0>] spi_sync+0x30/0x58
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.937502] [<8032077c>] m25p80_read+0x310/0x3f0
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.942193] [<8033134c>] spi_nor_read+0xd0/0x200
Sun Jul 21 00:10:15 2019 kern.warn kernel: [100743.946892] [<80311e18>] part_read+0x60/0xb8
```

dchard, 2019-10-04:
Latest snapshot as of today (on a Dlink 860L, mtk7621):

[88155.836241] ------------[ cut here ]------------
[88155.845483] WARNING: CPU: 3 PID: 0 at net/sched/sch_generic.c:320 dev_watchdog+0x1ac/0x324
[88155.861946] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[88155.875801] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[88156.015588] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 4.14.146 #0
[88156.027704] Stack : 00000000 87fb6240 80600000 80078fcc 80630000 805cb458 00000000 00000000
[88156.044336]         80594470 87c0fdc4 87c3518c 80608a47 8058f270 00000001 87c0fd68 53261673
[88156.060964]         00000000 00000000 80780000 00005d20 00000000 00000121 00000008 00000000
[88156.077593]         00000000 80610000 00003ce4 00000000 00000000 00000000 80630000 805c52a4
[88156.094223]         803c8040 00000140 00000003 87fb6240 00000000 802e7498 0000000c 8077000c
[88156.110855]         ...
[88156.115715] Call Trace:
[88156.120596] [<8000c8d0>] show_stack+0x58/0x100
[88156.129456] [<804c646c>] dump_stack+0x9c/0xe0
[88156.138127] [<8002fa08>] __warn+0xe0/0x138
[88156.146268] [<8002fa90>] warn_slowpath_fmt+0x30/0x3c
[88156.156148] [<803c8040>] dev_watchdog+0x1ac/0x324
[88156.165508] [<8009004c>] call_timer_fn.isra.25+0x24/0x84
[88156.176076] [<80090308>] run_timer_softirq+0x1bc/0x248
[88156.186312] [<804e3708>] __do_softirq+0x128/0x2ec
[88156.195670] [<800346b4>] irq_exit+0xac/0xc8
[88156.204002] [<80294f00>] plat_irq_dispatch+0xfc/0x138
[88156.214059] [<80007588>] except_vec_vi_end+0xb8/0xc4
[88156.223935] [<80008f50>] r4k_wait_irqoff+0x1c/0x24
[88156.233571] ---[ end trace e2d29df5b5ae6d1f ]---
[88156.242767] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[88156.255073] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[88156.267046] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=06a90000, max=0, ctx=2008, dtx=2008, fdx=2007, next=2008
[88156.288692] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=06220000, max=0, calc=1278, drx=1279
[88156.309926] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[88156.329680] mtk_soc_eth 1e100000.ethernet: PPE started

Assader, 2019-12-30:
This repo contains MTK wireless drivers, while the issue is related to MT7530 commutation chip, but also this page is one of the first, when you are trying to google error message, so i will dare to post here.
Initially, the community have connected the issue with hardware flow control bug. They also decided, that the bug will be fixed in future by MTK, so the solution was to disable flow control for one specific chip revision.
Later, chips with higher chip revision was released, but the flow control bug was not fixed and users (above) started reporting the issue again.
So the maintainers probably need to replace "==" [here](https://github.com/openwrt/openwrt/blob/master/target/linux/ramips/files-4.14/drivers/net/ethernet/mediatek/gsw_mt7621.c#L101) to ">=".
I have MT7621 with chip revision 0x0103 and turning flow controll off seems to stabilize situation.

dchard, 2019-12-31:
> I have MT7621 with chip revision 0x0103 and turning flow controll off seems to stabilize situation.

Can you explain how to do that on the router side? I also have mt7621, but I was not able to turn it off via normal methods. Thanks!

Assader, 2019-12-31:
@dchard as soon as i remember, using standard interfaces like debugfs, there is no way to check chip revision (i did it, adding pr_info("Chip rev: %u", rt_sysc_r32(SYSC_REG_CHIP_REV_ID)) to the code), nor a way to disable flow control.
To disable flow control you may compile OpenWrt with patch:
```
diff -u a/drivers/net/ethernet/mediatek/gsw_mt7621.c b/drivers/net/ethernet/mediatek/gsw_mt7621.c
--- a/drivers/net/ethernet/mediatek/gsw_mt7621.c
2019-11-25 14:14:15.152253091 +0500
+++ b/drivers/net/ethernet/mediatek/gsw_mt7621.c
2019-12-30 15:31:52.119516791 +0500
@@ -98,8 +98,8 @@
  mt7530_mdio_w32(gsw, 0x7000, 0x3);
  usleep_range(10, 20);
 
- if ((rt_sysc_r32(SYSC_REG_CHIP_REV_ID) & 0xFFFF) == 0x0101) {
- /* (GE1, Force 1000M/FD, FC ON, MAX_RX_LENGTH 1536) */
+ if ((rt_sysc_r32(SYSC_REG_CHIP_REV_ID) & 0xFFFF) >= 0x0101) {
+ /* (GE1, Force 1000M/FD, FC OFF, MAX_RX_LENGTH 1536) */
  mtk_switch_w32(gsw, 0x2305e30b, GSW_REG_MAC_P0_MCR);
  mt7530_mdio_w32(gsw, 0x3600, 0x5e30b);
  } else {
```
, placed into the file: target/linux/ramips/patches-4.14/220-mt7621-disable-flow-control.patch

Mushoz, 2020-01-22:
@Assader Do you have any plans for making a pull request for this patch to Openwrt, so it can be mainlined?

@kristrev did you ever make a pull request for your patch since you mentioned you were gonna do that? :)

denisandroid, 2020-09-16:
MT7620, I faced this problem, 19.07.03, 4.14kernel.

Perhaps I missed the patch above, you will have to rebuild. :(

hellion86, 2020-10-30:
Same problem
Model	Xiaomi Mi Router 3G
Architecture	MediaTek MT7621 ver:1 eco:3
Firmware Version	OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.136.49537-fb2f363
```

[11093.040136] ------------[ cut here ]------------
[11093.044786] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:320 0x8038ba10
[11093.051827] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[11093.058754] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY wireguard slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netportnet
[11093.130344]  ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip_gre gre ip6_udp_tunnel udp_tunnel ipip tunnel4 ip_tunnel leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[11093.177365] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.14.180 #0
[11093.183431] Stack : 00000000 00000000 00000000 8fe9f540 00000000 00000000 00000000 00000000
[11093.191767]         00000000 00000000 00000000 00000000 00000000 00000001 8fc0bd60 1cc28254
[11093.200102]         8fc0bdf8 00000000 00000000 00006480 00000038 8049c0d8 00000008 00000000
[11093.208454]         00000000 80550000 0002b4d5 00000000 8fc0bd40 00000000 00000000 8050ae44
[11093.216791]         8038ba10 00000140 00000001 8fe9f540 00000000 802accb0 00000004 806b0004
[11093.225127]         ...
[11093.227562] Call Trace:
[11093.227577] [<8049c0d8>] 0x8049c0d8
[11093.233476] [<8038ba10>] 0x8038ba10
[11093.236946] [<802accb0>] 0x802accb0
[11093.240418] [<8000c1a0>] 0x8000c1a0
[11093.243889] [<8000c1a8>] 0x8000c1a8
[11093.247357] [<80484f34>] 0x80484f34
[11093.250830] [<80071a90>] 0x80071a90
[11093.254301] [<8002e608>] 0x8002e608
[11093.257770] [<8038ba10>] 0x8038ba10
[11093.261244] [<8002e690>] 0x8002e690
[11093.264714] [<800550e8>] 0x800550e8
[11093.268183] [<8038ba10>] 0x8038ba10
[11093.271656] [<80099940>] 0x80099940
[11093.275127] [<8038b864>] 0x8038b864
[11093.278596] [<8008850c>] 0x8008850c
[11093.282067] [<8005f1fc>] 0x8005f1fc
[11093.285536] [<800887c8>] 0x800887c8
[11093.289004] [<800790f8>] 0x800790f8
[11093.292481] [<804a2ed8>] 0x804a2ed8
[11093.295955] [<80032fb4>] 0x80032fb4
[11093.299422] [<8025a2f0>] 0x8025a2f0
[11093.302894] [<80007488>] 0x80007488
[11093.306362] 
[11093.307906] ---[ end trace 1eeab0d01990d637 ]---
[11093.312528] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[11093.318683] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[11093.324685] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0ebc0000, max=0, ctx=639, dtx=639, fdx=638, next=639
[11093.335184] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0df60000, max=0, calc=1798, drx=1799
[11093.347900] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[11093.360558] mtk_soc_eth 1e100000.ethernet: PPE started
[36804.131736] conntrack: generic helper won't handle protocol 47. Please consider loading the specific helper module.
[51119.838502] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[51119.844705] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[51119.850717] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0de70000, max=0, ctx=2403, dtx=2403, fdx=2402, next=2403
[51119.861557] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0bc40000, max=0, calc=2683, drx=2684
[51119.874059] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[51119.886726] mtk_soc_eth 1e100000.ethernet: PPE started
[89229.695179] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[89229.701380] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[89229.707395] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0ed30000, max=0, ctx=836, dtx=836, fdx=835, next=836
[89229.717886] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0bf30000, max=0, calc=3466, drx=3467
[89229.730663] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[89229.743335] mtk_soc_eth 1e100000.ethernet: PPE started
[113529.605540] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[113529.611808] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000065
[113529.617908] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0bf30000, max=0, ctx=2724, dtx=2724, fdx=2723, next=2724
[113529.628843] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0bde0000, max=0, calc=3154, drx=3155
[113529.641841] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
[113529.654635] mtk_soc_eth 1e100000.ethernet: PPE started

```

vortigont, 2020-10-30:
It might (or might not) be related, I have the same router that drew my blood for months with same switch issues until I found [this ](https://4pda.ru/forum/index.php?showtopic=837667&st=5500#entry75122115) volt-mod scheme. Looks like this router is undervolted by default compared to datasheet spec. Once I've modded it to 1.1 volts it became rock-stable. I've never saw such crashes for about 2 month already.

gregd72002, 2020-11-02:
Same problem here on ZBT-WE1326

When i was on 19.07.03 i had over 100days uptime. Now on 19.07.04 the issue occurs daily. Today the router lasted on 30min
`[   25.397303] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   91.345155] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   91.352000] br-lan: port 2(wlan1) entered blocking state
[   91.357330] br-lan: port 2(wlan1) entered forwarding state
[ 1148.038731] ------------[ cut here ]------------
[ 1148.043368] WARNING: CPU: 1 PID: 0 at net/sched/sch_generic.c:320 0x8038c0d0
[ 1148.050418] NETDEV WATCHDOG: eth0 (mtk_soc_eth): transmit queue 0 timed out
[ 1148.057372] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun mmc_block mtk_sd mmc_core leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd gpio_button_hotplug usbcore nls_base
[ 1148.128378]  usb_common
[ 1148.130861] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 4.14.195 #0
[ 1148.136927] Stack : 00000000 00000000 00000000 8fe20240 00000000 00000000 00000000 00000000
[ 1148.145262]         00000000 00000000 00000000 00000000 00000000 00000001 8fc0bd60 ac07f5da
[ 1148.153599]         8fc0bdf8 00000000 00000000 000058f0 00000038 8049c858 00000007 00000000
[ 1148.161935]         00000000 80550000 0001ff2d 00000000 8fc0bd40 00000000 00000000 8050aed8
[ 1148.170273]         8038c0d0 00000140 00000001 8fe20240 00000000 802ad210 00000004 806b0004
[ 1148.178605]         ...
[ 1148.181041] Call Trace:
[ 1148.181093] [<8049c858>] 0x8049c858
[ 1148.186991] [<8038c0d0>] 0x8038c0d0
[ 1148.190468] [<802ad210>] 0x802ad210
[ 1148.193945] [<8000c1a0>] 0x8000c1a0
[ 1148.197413] [<8000c1a8>] 0x8000c1a8
[ 1148.200886] [<804856b4>] 0x804856b4
[ 1148.204356] [<80071ab0>] 0x80071ab0
[ 1148.207841] [<8002e608>] 0x8002e608
[ 1148.211320] [<8038c0d0>] 0x8038c0d0
[ 1148.214799] [<8002e690>] 0x8002e690
[ 1148.218274] [<800550e8>] 0x800550e8
[ 1148.221752] [<8038c0d0>] 0x8038c0d0
[ 1148.225223] [<80099a00>] 0x80099a00
[ 1148.228707] [<8038bf24>] 0x8038bf24
[ 1148.232182] [<80088568>] 0x80088568
[ 1148.235652] [<8005f214>] 0x8005f214
[ 1148.239130] [<80088824>] 0x80088824
[ 1148.242601] [<80079158>] 0x80079158
[ 1148.246075] [<804a3658>] 0x804a3658
[ 1148.249553] [<80032fb4>] 0x80032fb4
[ 1148.253028] [<8025a5f0>] 0x8025a5f0
[ 1148.256502] [<80007488>] 0x80007488
[ 1148.259974] 
[ 1148.261550] ---[ end trace 84f2d55f19c351b0 ]---
[ 1148.266172] mtk_soc_eth 1e100000.ethernet eth0: transmit timed out
[ 1148.272360] mtk_soc_eth 1e100000.ethernet eth0: dma_cfg:80000067
[ 1148.278361] mtk_soc_eth 1e100000.ethernet eth0: tx_ring=0, base=0ec00000, max=0, ctx=2486, dtx=2090, fdx=2090, next=2486
[ 1148.289227] mtk_soc_eth 1e100000.ethernet eth0: rx_ring=0, base=0e1a0000, max=0, calc=3886, drx=3887
[ 1148.702904] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x5f60000c, 0x10c = 0x80818
[ 1148.716724] mtk_soc_eth 1e100000.ethernet: PPE started`

Str1ker17, 2020-11-05:
Confirming the same issue on TP-Link Archer C20 v1 (mt7620) and OpenWRT 19.07.4.
I installed 19.07.3 preserving configuration and the bug has disappeared. So downgrading to 19.07.3 may be a solution for end users.

araujorm, 2020-11-05:
Hey guys please stop spamming this issue, which originally had nothing to do with mt76 driver, is already closed for almost 2 years, and besides it has nothing to do with the problem you are facing now. Not only it does not make sense, but also the devs won't read or reply to it. And bothers everyone else who is following the driver's evolution and receiving the true issues' messages by email (which is my case).

I'm not affiliated with the team developing the driver or openwrt, but please search or seek help in the appropriate channels, in this case the openwrt forum and the openwrt bug tracker.

Anyway your problem is only on the 19.07 branch and was introduced in someone's attempt to fix what didn't need fixing in here: https://github.com/openwrt/openwrt/commit/7ac454014a11347887323a131415ac7032d53546

That was reverted here, after the 19.07.4 release: https://github.com/openwrt/openwrt/commit/34a96529041d4e9502c490c66f8af0154187c6d2

So your options are (assuming you want to remain on the stable branch):
- checkout the commit with the fix, rebase as needed if you want to keep the stable repo and recompile (see my GitHub openwrt fork tags and you'll find some examples)
- go back to 19.07.3
- or wait for next stable release

But it makes zero sense to complain of a ethernet driver problem on a closed issue of the the WiFi driver. As I said, next time please search the forum.

joluxer, 2022-08-05:
For reference for everyone like me, who stumbles across this thread while searching for this symptom, here a little hint for a possible cause beside kernel issues.

If you are using the image builder or do somehow else mess around with the network configuration:

If you miss to configure the internal ethernet switch of the mt76, i.e. not include the `swconfig` package (i.e. just by accident added `-swconfig` to the iamge builder package list) into the image build or do somehow else not configure the internal switch, you run into this symptom right away from device start.