openwrt/mt76 #213: Low 2.4GHz speeds and high packet loss (mir3g)

jpolnetpl, 2018-11-18:
I've recently started to notice very high packet loss (after recent OpenWrt snapshot upgrades of my mir3g).

The problem exists at least for 2,4GHz WLAN client. 'iwinfo wlan0 info' invoked on mir3g shows low link speeds (6.5Mbps) and I have exceptionally (>70%) packet loss (even when I ping from client to the mir3g itself, so there is no packet forwarding and Internet transmission involved). There is almost no other traffic in my network during these tests.

iwinfo output on problematic snapshot:
```
wlan0     ESSID: "XYZ"
          Access Point: 50:64:2B:12:34:56
          Mode: Master  Channel: 6 (2.437 GHz)
          Tx-Power: 0 dBm  Link Quality: 45/70
          Signal: -65 dBm  Noise: unknown
          Bit Rate: 6.5 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 14C3:7603 14C3:7603 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

```
I'm certain that the issue occurs with mt76 commits from 2018-11-16 (OpenWrt snapshot r8466-251c350) and almost certain that it occurred also on snapshot r8459-3a6bddd (mt76 commits from 2018-11-13). Probably also on r8452-e95e9fc (mt76 commits up to ffccb48), but I'm not entirely sure if I remember correctly when I first started to notice the issue. Of course I'm not certain that the problem is caused by mt76, but I strongly suspect so.

To stabilize my router quickly I've reverted back to the OpenWrt snapshot that I've used without any problems for the longest period of time recently - to r8210-031c31c (dated 2018-10-02). After the downgrade all of the described problems disappear (link speed is back to normal ~50-58Mbps, I see no packet loss and reasonable download/upload speeds).

iwinfo wlan0 info on the reverted snapshot without problems:
```
wlan0     ESSID: "XYZ"
          Access Point: 50:64:2B:12:34:56
          Mode: Master  Channel: 6 (2.437 GHz)
          Tx-Power: 0 dBm  Link Quality: 42/70
          Signal: -68 dBm  Noise: unknown
          Bit Rate: 50.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 14C3:7603 14C3:7603 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
```

OpenWrt wireless config for this device is:
```
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '6'
        option hwmode '11g'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option htmode 'HT20'
        option distance '16'
        option txpower '17'
        option legacy_rates '0'
        option country 'PL'
        option disabled '0'
```
Option "distance" was added during my diagnostics of the problem, but it haven't changed anything. I've also tried different "txpower" values (20, 18 and driver default - by removing it completely). Before diagnostics I've usually used txpower '16'. But txpower basically has no effect on this issue.

I could try to further narrow down the list of suspected mt76 commits, however I've never built OpenWrt image myself, so I'd have to setup build environment etc. So this might take some (substantial) time.

nbd168, 2018-11-20:
Please try the latest version from OpenWrt master, I pushed some more fixes.

swg0101, 2018-11-21:
Tried the latest snapshots - 2.4Ghz not usable on Mi3g - looking at an air trace, it looks like acknowledgments are messed up on the 2.4Ghz side so the router keeps retransmitting information which then the client seems to discard. If I put the adapter into monitor mode and do
tcpdump -i wlan[x] wlan[0] == 0xd4 - I get no packets on the 2.4Ghz radio and plenty on the 5Ghz side regardless of the channel I put it on.

jpolnetpl, 2018-11-21:
Today I've tried latest official snapshot (r8473-8184bdd), which includes mt76 up to c09e8a4 (so without today's updates).                                                                     

For about 40 minutes none of my previously reported problems manifested. So I've powered off my computer (I've seen that some of your fixes are related to station removal).

After I've powered it back on (after half an hour or so) the reported problem was instantly visible. At first I couldn't get any packets (ping, DNS, not even ARP), despite that /proc/net/wireless on my station indicated usual - for working interface - link levels. Bringing the computer's wifi interface down and back up (with 'ifdown' and 'ifup' from Debian's "ifupdown" package) brought back the connectivity. However I'm experiencing visible packet loss and 'iwinfo' on mir3g again shows at best 6.5 MBit/s Bit Rate (however real internet download speeds are around 0.33 MBit/s, and simultaneously a 5 GHz client works flawlessly).

Logs on OpenWrt that mention station showing the symptoms don't seem to indicate anything really suspicious:
```
root@OpenWrt:~# logread | grep 11:22:33
Wed Nov 21 17:02:03 2018 daemon.info hostapd: wlan0: STA 60:e3:27:11:22:33 IEEE 802.11: authenticated
Wed Nov 21 17:02:03 2018 daemon.info hostapd: wlan0: STA 60:e3:27:11:22:33 IEEE 802.11: associated (aid 1)
Wed Nov 21 17:02:03 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 60:e3:27:11:22:33
Wed Nov 21 17:02:03 2018 daemon.info hostapd: wlan0: STA 60:e3:27:11:22:33 WPA: pairwise key handshake completed (RSN)
Wed Nov 21 17:45:32 2018 daemon.info dnsmasq-dhcp[1956]: DHCPRELEASE(br-lan) 192.168.33.132 60:e3:27:11:22:33 unknown lease
Wed Nov 21 17:45:32 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 60:e3:27:11:22:33
Wed Nov 21 18:26:31 2018 daemon.info hostapd: wlan0: STA 60:e3:27:11:22:33 IEEE 802.11: authenticated
Wed Nov 21 18:26:31 2018 daemon.info hostapd: wlan0: STA 60:e3:27:11:22:33 IEEE 802.11: associated (aid 2)
Wed Nov 21 18:26:31 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 60:e3:27:11:22:33
Wed Nov 21 18:26:31 2018 daemon.info hostapd: wlan0: STA 60:e3:27:11:22:33 WPA: pairwise key handshake completed (RSN)
Wed Nov 21 18:26:37 2018 daemon.info dnsmasq-dhcp[1956]: DHCPDISCOVER(br-lan) 192.168.33.132 60:e3:27:11:22:33
Wed Nov 21 18:26:37 2018 daemon.info dnsmasq-dhcp[1956]: DHCPOFFER(br-lan) 192.168.33.132 60:e3:27:11:22:33
Wed Nov 21 18:26:37 2018 daemon.info dnsmasq-dhcp[1956]: DHCPREQUEST(br-lan) 192.168.33.132 60:e3:27:11:22:33
Wed Nov 21 18:26:37 2018 daemon.info dnsmasq-dhcp[1956]: DHCPACK(br-lan) 192.168.33.132 60:e3:27:11:22:33 mars
Wed Nov 21 18:29:28 2018 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 60:e3:27:11:22:33
Wed Nov 21 18:29:35 2018 daemon.info hostapd: wlan0: STA 60:e3:27:11:22:33 IEEE 802.11: authenticated
Wed Nov 21 18:29:35 2018 daemon.info hostapd: wlan0: STA 60:e3:27:11:22:33 IEEE 802.11: associated (aid 2)
Wed Nov 21 18:29:35 2018 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 60:e3:27:11:22:33
Wed Nov 21 18:29:35 2018 daemon.info hostapd: wlan0: STA 60:e3:27:11:22:33 WPA: pairwise key handshake completed (RSN)
Wed Nov 21 18:29:42 2018 daemon.info dnsmasq-dhcp[1956]: DHCPDISCOVER(br-lan) 192.168.33.132 60:e3:27:11:22:33                                                                                 
Wed Nov 21 18:29:42 2018 daemon.info dnsmasq-dhcp[1956]: DHCPOFFER(br-lan) 192.168.33.132 60:e3:27:11:22:33                                                                                    
Wed Nov 21 18:29:42 2018 daemon.info dnsmasq-dhcp[1956]: DHCPREQUEST(br-lan) 192.168.33.132 60:e3:27:11:22:33                                                                                  
Wed Nov 21 18:29:42 2018 daemon.info dnsmasq-dhcp[1956]: DHCPACK(br-lan) 192.168.33.132 60:e3:27:11:22:33 mars                                                                                 
```

I'll wait till tomorrow for the next official snapshot to test today's changes because my build system on VirtualBox seems extremely slow. Maybe next time I'll try to use Docker instead.

jpolnetpl, 2018-11-22:
It seems my issue is fixed with your yesterday's commits. Thank you!

Similar to described above test scenario revealed no unexpected behaviour on official build of OpenWrt snapshot r8474-0191a06 today.

Should I try to find specific change that introduced this issue by building images bisecting between commits 94d4445 and da666a7 of mt76 driver? Or maybe a shorter route, find out which of yesterday's changes fixed it?

swg0101, 2018-11-23:
Still stuck here... reverting to last week's build did fix it though. Will try to see if I can narrow it down when I have some time...

swg0101, 2018-11-23:
Here's the problem I was referring to: wlan1 = 5.8Ghz, wlan0 = 2.4Ghz - they are captured in a busy environment so there are tons of WLAN traffic around:

```
root@Greenfield:~# tcpdump -nni wlan1 "wlan[0] == 0xd4" -c 10
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan1, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
07:32:50.518211 24.0 Mb/s 5180 MHz 11a -69dBm signal 0dBm signal antenna 0 -98dBm signal antenna 1 Acknowledgment RA:34:f3:9a:d4:9f:7c
07:32:50.597971 24.0 Mb/s 5180 MHz 11a -69dBm signal 0dBm signal antenna 0 -95dBm signal antenna 1 Acknowledgment RA:34:f3:9a:d4:9f:7c
07:32:50.698570 24.0 Mb/s 5180 MHz 11a -68dBm signal 0dBm signal antenna 0 -97dBm signal antenna 1 Acknowledgment RA:34:f3:9a:d4:9f:7c
07:32:50.737123 24.0 Mb/s 5180 MHz 11a -68dBm signal 0dBm signal antenna 0 -96dBm signal antenna 1 Acknowledgment RA:34:f3:9a:d4:9f:7c
07:32:50.877043 24.0 Mb/s 5180 MHz 11a -68dBm signal 0dBm signal antenna 0 -96dBm signal antenna 1 Acknowledgment RA:34:f3:9a:d4:9f:7c
07:32:50.961868 24.0 Mb/s 5180 MHz 11a -68dBm signal 0dBm signal antenna 0 -96dBm signal antenna 1 Acknowledgment RA:34:f3:9a:d4:9f:7c
07:32:51.041068 24.0 Mb/s 5180 MHz 11a -68dBm signal 0dBm signal antenna 0 -95dBm signal antenna 1 Acknowledgment RA:34:f3:9a:d4:9f:7c
07:32:51.124243 24.0 Mb/s 5180 MHz 11a -68dBm signal 0dBm signal antenna 0 -96dBm signal antenna 1 Acknowledgment RA:34:f3:9a:d4:9f:7c
07:32:51.223558 24.0 Mb/s 5180 MHz 11a -69dBm signal 0dBm signal antenna 0 -97dBm signal antenna 1 Acknowledgment RA:34:f3:9a:d4:9f:7c
07:32:51.339085 24.0 Mb/s 5180 MHz 11a -69dBm signal 0dBm signal antenna 0 -98dBm signal antenna 1 Acknowledgment RA:34:f3:9a:d4:9f:7c
10 packets captured
12 packets received by filter
0 packets dropped by kernel
```

```
root@Greenfield:~# tcpdump -nni wlan0 "wlan[0] == 0xd4" -c 10
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlan0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
^C
0 packets captured
4 packets received by filter
0 packets dropped by kernel
root@Greenfield:~#
```

```
root@Greenfield:~# opkg list *mt76*
kmod-mt76-core - 4.14.82+2018-11-21-da666a74-1
kmod-mt7603 - 4.14.82+2018-11-21-da666a74-1
kmod-mt76x02-common - 4.14.82+2018-11-21-da666a74-1
kmod-mt76x2 - 4.14.82+2018-11-21-da666a74-1
kmod-mt76x2-common - 4.14.82+2018-11-21-da666a74-1
```

```
root@Greenfield:~# uname -a
Linux Greenfield 4.14.82 #0 SMP Thu Nov 22 15:30:38 2018 mips GNU/Linux
```

Rascal6, 2018-11-23:
ZBT-WG3526 
OpenWrt SNAPSHOT r8483-952b117

wifi 2.4 not work

Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.329065] ------------[ cut here ]------------
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.333754] WARNING: CPU: 1 PID: 1984 at /var/lib/buildbot/slaves/slashdirt-03/MAIN/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2018-11-21-da666a74/mt7603/mac.c:196 mt7603_filter_tx+0x100/0x164 [mt7603e]
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.353608] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables mmc_block mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.424238]  sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.431307] CPU: 1 PID: 1984 Comm: hostapd Not tainted 4.14.82 #0
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.437377] Stack : 00000000 00000001 80390840 8006f714 805d0000 8056dea8 00000000 00000000
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.445720]         80538818 8f625a54 8ef841fc 805aa947 80533908 00000001 8f6259f8 ac07f54b
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.454063]         00000000 00000000 80610000 000073f8 00000000 0000015c 00000008 00000000
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.462403]         00000000 805b0000 000694cb 00000000 80000000 00000000 805d0000 8f1143a0
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.470743]         00000009 000000c4 8f110000 00000001 00000000 802a2d48 00000004 80610004
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.479088]         ...
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.481536] Call Trace:
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.484015] [<800106c0>] show_stack+0x58/0x100
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.488482] [<8047272c>] dump_stack+0x9c/0xe0
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.492834] [<8002e368>] __warn+0xe0/0x114
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.496926] [<8002e42c>] warn_slowpath_null+0x1c/0x30
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.501988] [<8f1143a0>] mt7603_filter_tx+0x100/0x164 [mt7603e]
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.507905] [<8f111578>] mt7603_sta_remove+0x84/0xb0 [mt7603e]
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.513748] [<8e8b2090>] mt76_sta_state+0x110/0x300 [mt76]
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.519364] [<8ef07c60>] sta_set_sinfo+0xbf8/0xcd0 [mac80211]
Fri Nov 23 05:24:32 2018 kern.warn kernel: [  510.525278] ---[ end trace 46e5bc60e211c3bf ]---

camelcamro, 2018-11-23:
not sure, if it is also related to this crash what i had (newest trunk)

<1>[162146.511610] CPU 1 Unable to handle kernel paging request at virtual address 00000628, epc == 8010c560, ra == 8010c444
<4>[162146.522422] Oops[#1]:
<4>[162146.524889] CPU: 1 PID: 18638 Comm: tc Not tainted 4.14.81 #0
<4>[162146.530762] task: 8fe18000 task.stack: 8e584000
<4>[162146.535375] $ 0   : 00000000 00000001 00000000 814a5700
<4>[162146.540686] $ 4   : 805a61b8 00000001 00000001 00000628
<4>[162146.545994] $ 8   : 03ac679b 03ac679a 00000000 80000000
<4>[162146.551302] $12   : 8e584000 00000000 00000000 77fd32c0
<4>[162146.556613] $16   : 8fc02900 014102c0 8e506000 8033f818
<4>[162146.561923] $20   : 00001000 8da44300 00000b98 00000000
<4>[162146.567230] $24   : 00000000 8038e438
<4>[162146.572538] $28   : 8e584000 8e585cb0 00000000 8010c444
<4>[162146.577849] Hi    : 08418c00
<4>[162146.580803] Lo    : 07395a80
<4>[162146.583798] epc   : 8010c560 __kmalloc_track_caller+0x1d4/0x228
<4>[162146.589789] ra    : 8010c444 __kmalloc_track_caller+0xb8/0x228
<4>[162146.595681] Status: 11007c03     KERNEL EXL IE
<4>[162146.599948] Cause : 40800008 (ExcCode 02)
<4>[162146.604022] BadVA : 00000628
<4>[162146.606977] PrId  : 0001992f (MIPS 1004Kc)
<4>[162146.611139] Modules linked in: pppoe ppp_async option usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usbserial usb_storage ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table
<4>[162146.682805]  nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda macvlan libcrc32c iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat br_netfilter fuse sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip
<4>[162146.754220]  ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables nfsv4 nfsv3 nfs ip_gre gre ifb ip_tunnel tun vfat fat lockd sunrpc grace ext4 jbd2 mbcache dns_resolver dm_crypt dm_mirror dm_region_hash dm_log dm_mod dax nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb crc32c_generic mmc_block mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
<4>[162146.799627] Process tc (pid: 18638, threadinfo=8e584000, task=8fe18000, tls=77fd4d98)
<4>[162146.807507] Stack : 00000000 80338f38 00000000 00000000 8e815000 8e585d0f 014000c0 8033f718
<4>[162146.815946]         00000000 8e585d50 8e585d54 fffffff5 8e815000 00000f5f 014000c0 8fc20600
<4>[162146.824385]         00000000 8033f818 00000000 00000b98 00000000 00000000 8e585ee8 8da44300
<4>[162146.832821]         8e815b40 8e3e0800 8e815b40 00000f40 00000f24 8e585ee8 8da44300 8038e1ec
<4>[162146.841257]         00000000 803477f8 8e3e0800 8e815b40 8e3e0800 8034657c 8e585ee8 8e3e0800
<4>[162146.849695]         ...
<4>[162146.852225] Call Trace:
<4>[162146.854761] [<8010c560>] __kmalloc_track_caller+0x1d4/0x228
<4>[162146.860427] [<8033f718>] __kmalloc_reserve.isra.7+0x40/0xa4
<4>[162146.866075] [<8033f818>] __alloc_skb+0x9c/0x180
<4>[162146.870692] [<8038e1ec>] netlink_dump+0x84/0x2d0
<4>[162146.875387] [<8038e608>] netlink_recvmsg+0x1d0/0x344
<4>[162146.880449] [<80335818>] ___sys_recvmsg+0xc0/0x178
<4>[162146.885320] [<803368dc>] __sys_recvmsg+0x50/0x80
<4>[162146.890029] [<80019578>] syscall_common+0x34/0x58
<4>[162146.894807] Code: 00000000  8e020014  00e23821 <8ce20000> 10000009  cc400000  1040ffbd  00000000  8e060010
<4>[162146.904644]
<4>[162146.907236] ---[ end trace 294fcbb12d1cc3ec ]---

nbd168, 2019-01-20:
Please try the latest version

nbd168, 2019-03-03:
I just pushed more fixes to OpenWrt master, things should work much better now.

closed 2019-03-03 by nbd168