openwrt/mt76 #214: kernel crash - Unable to handle kernel paging request at virtual address

camelcamro, 2018-11-23:
Model	ZBT-WG3526 (16M)
Architecture	MediaTek MT7621 ver:1 eco:3
Firmware Version	OpenWrt SNAPSHOT r8473-8184bdd / LuCI Master (git-18.325.74447-562ae5f)
Kernel Version	4.14.81

and again crashes ...

<1>[162146.511610] CPU 1 Unable to handle kernel paging request at virtual address 00000628, epc == 8010c560, ra == 8010c444
<4>[162146.522422] Oops[#1]:
<4>[162146.524889] CPU: 1 PID: 18638 Comm: tc Not tainted 4.14.81 #0
<4>[162146.530762] task: 8fe18000 task.stack: 8e584000
<4>[162146.535375] $ 0   : 00000000 00000001 00000000 814a5700
<4>[162146.540686] $ 4   : 805a61b8 00000001 00000001 00000628
<4>[162146.545994] $ 8   : 03ac679b 03ac679a 00000000 80000000
<4>[162146.551302] $12   : 8e584000 00000000 00000000 77fd32c0
<4>[162146.556613] $16   : 8fc02900 014102c0 8e506000 8033f818
<4>[162146.561923] $20   : 00001000 8da44300 00000b98 00000000
<4>[162146.567230] $24   : 00000000 8038e438
<4>[162146.572538] $28   : 8e584000 8e585cb0 00000000 8010c444
<4>[162146.577849] Hi    : 08418c00
<4>[162146.580803] Lo    : 07395a80
<4>[162146.583798] epc   : 8010c560 __kmalloc_track_caller+0x1d4/0x228
<4>[162146.589789] ra    : 8010c444 __kmalloc_track_caller+0xb8/0x228
<4>[162146.595681] Status: 11007c03     KERNEL EXL IE
<4>[162146.599948] Cause : 40800008 (ExcCode 02)
<4>[162146.604022] BadVA : 00000628
<4>[162146.606977] PrId  : 0001992f (MIPS 1004Kc)
<4>[162146.611139] Modules linked in: pppoe ppp_async option usb_wwan pptp pppox ppp_mppe ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_NETMAP xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usbserial usb_storage ts_fsm ts_bm slhc nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table
<4>[162146.682805]  nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_broadcast ts_kmp nf_conntrack_amanda macvlan libcrc32c iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat br_netfilter fuse sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip
<4>[162146.754220]  ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables nfsv4 nfsv3 nfs ip_gre gre ifb ip_tunnel tun vfat fat lockd sunrpc grace ext4 jbd2 mbcache dns_resolver dm_crypt dm_mirror dm_region_hash dm_log dm_mod dax nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic ecb crc32c_generic mmc_block mtk_sd mmc_core leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd ahci libahci libata sd_mod scsi_mod gpio_button_hotplug usbcore nls_base usb_common
<4>[162146.799627] Process tc (pid: 18638, threadinfo=8e584000, task=8fe18000, tls=77fd4d98)
<4>[162146.807507] Stack : 00000000 80338f38 00000000 00000000 8e815000 8e585d0f 014000c0 8033f718
<4>[162146.815946]         00000000 8e585d50 8e585d54 fffffff5 8e815000 00000f5f 014000c0 8fc20600
<4>[162146.824385]         00000000 8033f818 00000000 00000b98 00000000 00000000 8e585ee8 8da44300
<4>[162146.832821]         8e815b40 8e3e0800 8e815b40 00000f40 00000f24 8e585ee8 8da44300 8038e1ec
<4>[162146.841257]         00000000 803477f8 8e3e0800 8e815b40 8e3e0800 8034657c 8e585ee8 8e3e0800
<4>[162146.849695]         ...
<4>[162146.852225] Call Trace:
<4>[162146.854761] [<8010c560>] __kmalloc_track_caller+0x1d4/0x228
<4>[162146.860427] [<8033f718>] __kmalloc_reserve.isra.7+0x40/0xa4
<4>[162146.866075] [<8033f818>] __alloc_skb+0x9c/0x180
<4>[162146.870692] [<8038e1ec>] netlink_dump+0x84/0x2d0
<4>[162146.875387] [<8038e608>] netlink_recvmsg+0x1d0/0x344
<4>[162146.880449] [<80335818>] ___sys_recvmsg+0xc0/0x178
<4>[162146.885320] [<803368dc>] __sys_recvmsg+0x50/0x80
<4>[162146.890029] [<80019578>] syscall_common+0x34/0x58
<4>[162146.894807] Code: 00000000  8e020014  00e23821 <8ce20000> 10000009  cc400000  1040ffbd  00000000  8e060010
<4>[162146.904644]
<4>[162146.907236] ---[ end trace 294fcbb12d1cc3ec ]---

nbd168, 2018-11-28:
Please try the latest version

closed 2019-01-21 by nbd168

YearMonthDay, 2020-08-01:
Hello, can you share me with the solution about this issue?