openwrt/mt76 #216: low tx power m7610e

leksmax, 2018-12-04:
root@OpenWrt:~# iw phy
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2257 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short long limit: 2
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0 RX 0
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 1:
                Capabilities: 0x2fe
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 2-streams
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 2 usec (0x04)
                HT TX/RX MCS rate indexes supported: 0-15, 32
                Frequencies:
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm) (no IR)
                        * 2472 MHz [13] (20.0 dBm) (no IR)
                        * 2484 MHz [14] (20.0 dBm) (no IR)
        valid interface combinations:
                 * #{ managed, AP, mesh point } <= 8,
                   total <= 8, #channels <= 1
        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
Wiphy phy0
        max # scan SSIDs: 4
        max scan IEs length: 2243 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Available Antennas: TX 0x1 RX 0x1
[... 115 lines cut ...]
          Tx-Power: 11 dBm  Link Quality: 21/70
          Signal: -89 dBm  Noise: unknown
          Bit Rate: 58.5 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgnac
          Hardware: 14C3:7650 14C3:7650 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: "OpenWrt"
          Access Point: 00:0C:43:76:20:20
          Mode: Master  Channel: 11 (2.462 GHz)
          Tx-Power: 20 dBm  Link Quality: 48/70
          Signal: -62 dBm  Noise: unknown
          Bit Rate: 57.8 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1

root@OpenWrt:~# iwinfo 
 in  stock  i  have  20dBm

leksmax, 2018-12-04:
root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/mt76/txpower
Target power: 0
     Delta:  0  0
and  router  have  FEM (SKY11)

LorenzoBianconi, 2018-12-04:
@leksmax : what device are you running? Could you please provide device eeprom partition?

leksmax, 2018-12-04:
i running  on  this  device https://github.com/openwrt/openwrt/pull/1550/commits
eeprom https://drive.google.com/file/d/1UdSt4R7XWXZX1VqrEOyTsNFaLSwApN5c/view?usp=sharing

LorenzoBianconi, 2018-12-06:
@leksmax are you referring to 2GHz or 5GHz band?

leksmax, 2018-12-06:
@LorenzoBianconi  i am  referring  stock 5g  and  mt76 driver 5g

leksmax, 2018-12-07:
maybe  sombady  know  how  use  fem sky11  in this  driver ?

leksmax, 2018-12-09:
with  this  driver  wifi  on  mt7610  work  well 
https://github.com/sunsky13122156/LEDE-Archer-C2/tree/master/package/mediatek/drivers/mt7610e

LorenzoBianconi, 2018-12-09:
@leksmax ok, I will take a look, thx

leksmax, 2018-12-11:
maybe  i  need  turn on  gpio to FEM working ?

leksmax, 2018-12-11:
@LorenzoBianconi What is the probability that in the near future Fem support will appear for this chip?

LorenzoBianconi, 2018-12-20:
> with this driver wifi on mt7610 work well
> https://github.com/sunsky13122156/LEDE-Archer-C2/tree/master/package/mediatek/drivers/mt7610e

Could you please try to make a diff between MT7610E-V10-FEM.bin and device eeprom partition? (just wifi part)

leksmax, 2018-12-24:
@LorenzoBianconi 
cmp -b -l lavaeeprom.bin  MT7610E-V10-FEM.bin 
  4   1 ^A     2 ^B
 10 140 `    100 @
 36  37 ^_    67 7
 54 375 M-}    5 ^E
 56   0 ^@    20 ^P
 57   6 ^F   377 M-^?
 59 154 l    144 d
 67 377 M-^?  21 ^Q
 68   0 ^@     6 ^F
 70  12 ^J    11 ^I
 74  11 ^I    14 ^L
 78   5 ^E    14 ^L
121  20 ^P    17 ^O
122  20 ^P    16 ^N
123  17 ^O    15 ^M
124  17 ^O    14 ^L
125  17 ^O    13 ^K
126  17 ^O    12 ^J
127  16 ^N    11 ^I
128  16 ^N    11 ^I
129  16 ^N    11 ^I
130  16 ^N    11 ^I
131  15 ^M    11 ^I
132  15 ^M    11 ^I
133  11 ^I    14 ^L
134  11 ^I    14 ^L
135  11 ^I    14 ^L
136  11 ^I    14 ^L
137  12 ^J    14 ^L
138  12 ^J    14 ^L
139  12 ^J    14 ^L
140  12 ^J    14 ^L
141  12 ^J    14 ^L
142  12 ^J    14 ^L
143  12 ^J    14 ^L
144  13 ^K    14 ^L
145  13 ^K    14 ^L
146  13 ^K    14 ^L
148  16 ^N    13 ^K
149  16 ^N    13 ^K
150  16 ^N    13 ^K
151  16 ^N    12 ^J
152  17 ^O    12 ^J
153  17 ^O    12 ^J
154  17 ^O    12 ^J
155  17 ^O    13 ^K
210 377 M-^? 374 M-|
211  32 ^Z    34 ^\
241 377 M-^? 200 M-^@
242 377 M-^? 200 M-^@
243 377 M-^? 200 M-^@
244 377 M-^? 200 M-^@
245 377 M-^? 362 M-r
246 377 M-^? 370 M-x
247 377 M-^? 374 M-|
248 377 M-^?   3 ^C
249 377 M-^?  16 ^N
250 377 M-^?  20 ^P
251 377 M-^? 177 ^?
252 377 M-^? 177 ^?
253 377 M-^? 177 ^?
254 377 M-^? 177 ^?
255 377 M-^? 200 M-^@
256 377 M-^? 200 M-^@
257 377 M-^? 200 M-^@
258 377 M-^? 200 M-^@
259 377 M-^? 356 M-n
260 377 M-^? 365 M-u
261 377 M-^? 372 M-z
262 377 M-^?   4 ^D
263 377 M-^?  17 ^O
264 377 M-^?  24 ^T
265 377 M-^? 177 ^?
266 377 M-^? 177 ^?
267 377 M-^? 177 ^?
268 377 M-^? 177 ^?
269 377 M-^? 200 M-^@
270 377 M-^? 374 M-|
lavaeeprom.bin
https://drive.google.com/open?id=17qhD2S3J5kbeZqXzJw25eb7S_g-GEPX3

orangepizza, 2019-01-11:
I have same problem with Tplink Archer C2, txpower for mt7610e limited to 11dbm.

ZerBea, 2019-01-26:
Doesn't work (kernel 4.20.3 / 5.0-rc3 - INTEL core system, AMD Ryzen not tested because of iommu issue as described here https://bugzilla.kernel.org/show_bug.cgi?id=202241):
phy#1
	Interface wlp0s20f0u3
		ifindex 4
		wdev 0x100000001
		addr da:20:99:ff:c0:34
		type managed
		txpower 14.00 dBm
shown: 14.00 dBm, but real tx power is much less.

Eprom not read correctly:
			* 2484 MHz [14] (disabled)

Also, driver crashed under heavy load.
Described from here:
https://bugzilla.kernel.org/show_bug.cgi?id=202243#c89
up to here:
https://bugzilla.kernel.org/show_bug.cgi?id=202243#c102

Wiphy phy2
	max # scan SSIDs: 4
	max scan IEs length: 2243 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports RSN-IBSS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x1 RX 0x1
	Supported interface modes:
		 * managed
		 * monitor
	Band 1:
		Capabilities: 0x17e
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			RX STBC 1-stream
			Max AMSDU length: 3839 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 4 usec (0x05)
		HT TX/RX MCS rate indexes supported: 0-7
		Bitrates (non-HT):
			* 1.0 Mbps (short preamble supported)
			* 2.0 Mbps (short preamble supported)
			* 5.5 Mbps (short preamble supported)
			* 11.0 Mbps (short preamble supported)
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
[... 133 lines cut ...]
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	HT Capability overrides:
		 * MCS: ff ff ff ff ff ff ff ff ff ff
		 * maximum A-MSDU length
		 * supported channel width
		 * short GI for 40 MHz
		 * max A-MPDU length exponent
		 * min MPDU start spacing
	Device supports TX status socket option.
	Device supports HT-IBSS.
	Device supports SAE with AUTHENTICATE command
	Device supports low priority scan.
	Device supports scan flush.
	Device supports AP scan.
	Device supports per-vif TX power setting
	Driver supports full state transitions for AP/GO clients
	Driver supports a userspace MPM
	Device supports active monitor (which will ACK incoming frames)
	Device supports configuring vdev MAC-addr on create.

LorenzoBianconi, 2019-01-26:
>
>
> Doesn't work (kernel 4.20.3 / 5.0-rc3 - INTEL core system, AMD Ryzen not
> tested because of iommu issue as described here
> https://bugzilla.kernel.org/show_bug.cgi?id=202241):
> phy#1
> Interface wlp0s20f0u3
> ifindex 4
> wdev 0x100000001
> addr da:20:99:ff:c0:34
> type managed
> txpower 14.00 dBm
> shown: 14.00 dBm, but real tx power is much less.
>
> Eprom not read correctly:
> * 2484 MHz [14] (disabled)
>
> Also, driver crashed under heavy load.
> Described from here:
> https://bugzilla.kernel.org/show_bug.cgi?id=202243#c89
> up to here:
> https://bugzilla.kernel.org/show_bug.cgi?id=202243#c102
>
> Wiphy phy2
> max # scan SSIDs: 4
> max scan IEs length: 2243 bytes
> max # sched scan SSIDs: 0
> max # match sets: 0
> max # scan plans: 1
> max scan plan interval: -1
> max scan plan iterations: 0
> Retry short limit: 7
> Retry long limit: 4
> Coverage class: 0 (up to 0m)
> Device supports RSN-IBSS.
> Supported Ciphers:
> * WEP40 (00-0f-ac:1)
> * WEP104 (00-0f-ac:5)
> * TKIP (00-0f-ac:2)
> * CCMP-128 (00-0f-ac:4)
> * CCMP-256 (00-0f-ac:10)
> * GCMP-128 (00-0f-ac:8)
> * GCMP-256 (00-0f-ac:9)
> * CMAC (00-0f-ac:6)
> * CMAC-256 (00-0f-ac:13)
> * GMAC-128 (00-0f-ac:11)
> * GMAC-256 (00-0f-ac:12)
> Available Antennas: TX 0x1 RX 0x1
> Supported interface modes:
> * managed
> * monitor
> Band 1:
> Capabilities: 0x17e
> HT20/HT40
> SM Power Save disabled
> RX Greenfield
> RX HT20 SGI
> RX HT40 SGI
> RX STBC 1-stream
> Max AMSDU length: 3839 bytes
> No DSSS/CCK HT40
> Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
> Minimum RX AMPDU time spacing: 4 usec (0x05)
> HT TX/RX MCS rate indexes supported: 0-7
> Bitrates (non-HT):
> * 1.0 Mbps (short preamble supported)
> * 2.0 Mbps (short preamble supported)
> * 5.5 Mbps (short preamble supported)
> * 11.0 Mbps (short preamble supported)
> * 6.0 Mbps
[... 156 lines cut ...]
> * short GI for 40 MHz
> * max A-MPDU length exponent
> * min MPDU start spacing
> Device supports TX status socket option.
> Device supports HT-IBSS.
> Device supports SAE with AUTHENTICATE command
> Device supports low priority scan.
> Device supports scan flush.
> Device supports AP scan.
> Device supports per-vif TX power setting
> Driver supports full state transitions for AP/GO clients
> Driver supports a userspace MPM
> Device supports active monitor (which will ACK incoming frames)
> Device supports configuring vdev MAC-addr on create.
>
> <https://github.com/openwrt/mt76/issues/216#issuecomment-457847192>, or mute
> the thread

Are you using usb device or PCIe one?
Could you please provide the output of:
- cat /sys/kernel/debug/ieee80211/phy0/mt76/txpower
- cat /sys/kernel/debug/ieee80211/phy0/mt76/rate_txpower

Regards,
Lorenzo

ZerBea, 2019-01-26:
USB device:
148f:761a Ralink Technology, Corp. MT7610U ("Archer T2U" 2.4G+5G WLAN)

ZerBea, 2019-01-26:
and the other values:
$  cat /sys/kernel/debug/ieee80211/phy1/mt76/txpower
[Target power: 32
     Delta:  0  0

cat /sys/kernel/debug/ieee80211/phy1/mt76/rate_txpower
       CCK: -4 -4 -4 -4
      OFDM: -4 -4 -4 -4 -4 -4 -4 -4
      STBC: -4 -4 -4 -4 -4 -4 -4 -4 -4 -4
        HT: -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4
       VHT: -4 -4 -4 -4 -4 -4 -4 -4 -4 -4

orangepizza, 2019-01-26:
> 
> 
> could you please try this repo? https://github.com/LorenzoBianconi/wireless-drivers-next/tree/mt76x0_init_target_power
> I added some fixes in eeprom tx power parsing (please not I have not been able to test it, just compiled)

I have device with mt7610e (tplink c2), but have no idea how to use this for openwrt? makefiile of mt76 is just a link to git (openwrt/mt76), so should I fork that and apply last few commits form there to it myself?

LorenzoBianconi, 2019-01-27:
> and the other values:
> /sys/kernel/debug/ieee80211/phy1/mt76/txpower
> [Target power: 32
> Delta: 0 0
> 
> /sys/kernel/debug/ieee80211/phy1/mt76/rate_txpower
> CCK: -4 -4 -4 -4
> OFDM: -4 -4 -4 -4 -4 -4 -4 -4
> STBC: -4 -4 -4 -4 -4 -4 -4 -4 -4 -4
> HT: -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4
> VHT: -4 -4 -4 -4 -4 -4 -4 -4 -4 -4
> 
> looking not good, so far.
> 
> running the first tests using iw and hcxdumptool

Could you please provide me your eeprom data (/sys/kernel/debug/ieee80211/phy0/mt76/eeprom)?

ZerBea, 2019-01-29:
Hi Lorenzo.
Got the latest driver from linux-next and applied patch from here:
https://patchwork.kernel.org/patch/10782801/
Unfortunately, we run into some issues again:
$ iw dev
Interface wlp0s20f0u1
	ifindex 4
	wdev 0x100000001
	addr 76:22:24:c8:2d:95
	type monitor
	wiphy 1
	channel 11 (2462 MHz), width: 20 MHz (no HT), center1: 2462 MHz
	txpower 16.00 dBm

$ iw phy1 info
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm)
			* 2472 MHz [13] (20.0 dBm)
			* 2484 MHz [14] (disabled)

$ iw dev
Interface wlp0s20f0u1
	ifindex 4
	wdev 0x100000001
	addr 76:22:24:c8:2d:95
	type monitor
	wiphy 1
	channel 11 (2462 MHz), width: 20 MHz (no HT), center1: 2462 MHz
	txpower 16.00 dBm

the real tx output is still much lower than 20.0 dBm

ZerBea, 2019-01-29:
Also channel 14 isn't disabled, as from the driver reported:
* 2484 MHz [14] (disabled)

phy#1
	Interface wlp0s20f0u1
		ifindex 5
		wdev 0x200000001
		addr 50:3e:aa:92:e3:26
		type monitor
		channel 14 (2484 MHz), width: 20 MHz (no HT), center1: 2484 MHz
		txpower 16.00 dBm

$ hcxdumptool -i wlp0s20f0u1 -C 
initialization...
available channels:
1,2,3,4,5,6,7,8,9,10,11,12,13,14,36,40,44,48,52,56,60,64,100,104,108,112,116,120,124,128,132,136,140,149,153,157,161,165 
terminated...

tx output power is the same as on ch 1...13

LorenzoBianconi, 2019-01-30:
@ZerBea using this [repo](https://github.com/LorenzoBianconi/wireless-drivers-next/tree/mt76x0_init_target_power), could you please provide me the output of:
- /sys/kernel/debug/ieee80211/phy1/mt76/txpower
-  /sys/kernel/debug/ieee80211/phy1/mt76/rate_txpower
(at 2412MHz and 5180MHz)

ZerBea, 2019-01-30:
@LorenzoBianconi here we go:
repo: https://github.com/LorenzoBianconi/wireless-drivers-next/tree/mt76x0_init_target_power
	ifindex 4
	wdev 0x100000001
	addr 5e:8d:d9:1a:e0:88
	type monitor
	wiphy 1
	channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
	txpower 14.00 dBm

/sys/kernel/debug/ieee80211/phy1/mt76/txpower
Target power: 32
     Delta:  0  0

/sys/kernel/debug/ieee80211/phy1/mt76/rate_txpower
       CCK: -4 -4 -4 -4
      OFDM: -4 -4 -4 -4 -4 -4 -4 -4
      STBC: -4 -4 -4 -4 -4 -4 -4 -4 -4 -4
        HT: -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4
       VHT: -4 -4 -4 -4 -4 -4 -4 -4 -4 -4

Interface wlp0s20f0u3
	ifindex 4
	wdev 0x100000001
	addr 5e:8d:d9:1a:e0:88
	type monitor
	wiphy 1
	channel 36 (5180 MHz), width: 20 MHz (no HT), center1: 5180 MHz
	txpower 17.00 dBm

/sys/kernel/debug/ieee80211/phy1/mt76/txpower
Target power: 28
     Delta:  0  0

/sys/kernel/debug/ieee80211/phy1/mt76/rate_txpower
       CCK:  0  0  0  0
      OFDM:  6  6  6  6  4  4  0  0
      STBC: -3 -3 -3 -3  0  0 -4 -4  0  0
        HT:  6  6  6  6  4  4  0  0  0  0  0  0  0  0  0  0
       VHT:  6  6  6  6  4  4  0 -1 -1  0

Please do not wonder about "type monitor". I'm using hcxdumptool to set channel.
BTW:
Compared to 4.19, driver is extremely slow. hcxdumptool takes twice more time to set a channel.

LorenzoBianconi, 2019-01-30:
@ZerBea I think the results you posted are not directly from the eeprom you posted since I parsed it and I got different results for rate_power (especially @2GHz) . Anyway the txpower is limited to 17dbm (@5180MHz) by the eeprom calibration value

ZerBea, 2019-01-30:
That could be. All values  on that driver are weird. Attached a pcapng from my test range.
TP-LINK router -> 1 meter -> TP-LINK Archer T2UH
reported RSSI -81 dBm (normal = -23... -28 dBm)
distant stations are not received at all!
[dirvernew.pcapng.zip](https://github.com/openwrt/mt76/files/2814026/dirvernew.pcapng.zip)

ZerBea, 2019-02-01:
Deletetd RSSI comment (measurement error here), because RSSI values are correct:
Archer T2UH -23 dBm (old wrong measurement: -81 dBm because WireShark wasn't able to switch channel to AP channel)
ASUS AC51 -33 dBm

ZerBea, 2019-02-03:
Just received 4.20.6.arch1-1:
This are the values for T2UH running 4.20.6.arch1-1:
phy#1
	Interface wlp0s20f0u3
		ifindex 4
		wdev 0x100000001
		addr 2e:86:ab:40:e7:82
		type monitor
		channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
		txpower 20.00 dBm

rate_txpower
       CCK:  0  0  0  0
      OFDM: -3 -3 -3 -3  0  0 -4 -4
      STBC: -3 -3 -3 -3  0  0 -4 -4  0  0
        HT: -3 -3 -3 -3  0  0 -4  0  0  0  0  0  0  0  0  0
       VHT: -3 -3 -3 -3  0  0 -4 -1 -1  0

phy#1
	Interface wlp0s20f0u3
		ifindex 4
		wdev 0x100000001
		addr 2e:86:ab:40:e7:82
		type monitor
		channel 36 (5180 MHz), width: 20 MHz (no HT), center1: 5180 MHz
		txpower 23.00 dBm

rate_txpower
       CCK:  0  0  0  0
      OFDM:  7  7  7  7  4  4  0  0
      STBC: -3 -3 -3 -3  0  0 -4 -4  0  0
        HT:  7  7  7  7  4  4  0  0  0  0  0  0  0  0  0  0
       VHT:  7  7  7  7  4  4  0 -1 -1  0

Cheers
Mike

orangepizza, 2019-02-03:
so we need to backport the patches in https://bugzilla.kernel.org/show_bug.cgi?id=202243   ?

ZerBea, 2019-02-03:
No, I don't think so. The latest code from wireless-drivers-next is much better (except of some init issues). Many functions are simplified and/or merged together (mt76x0 <-> Mt76x2).
Also the mt7601 code is working fine (eeprom 0d was applied by latest patch). So it's a good way to look forward.

LorenzoBianconi, 2019-02-04:
@ZerBea : the fix I posted upstream is not in wireless-drivers-next yet, it is in wireless-drivers. Could you please try it or backport the fix on wireless-drivers-next? However I guess the max tx power is limited by eeprom calibration data

ZerBea, 2019-02-04:
@LorenzoBianconi ok, but before I start - I'm here: * mt76x0_init_target_power
Is that the branch?

LorenzoBianconi, 2019-02-04:
>
>
> @LorenzoBianconi <https://github.com/LorenzoBianconi> ok, but before I
> start - I'm here: * mt76x0_init_target_power
> Is that the branch?
>
> <https://github.com/openwrt/mt76/issues/216#issuecomment-460340718>, or mute
> the thread
What git repo are you using?

ZerBea, 2019-02-04:
https://github.com/LorenzoBianconi/wireless-drivers-next/tree/mt76x0_init_target_power

LorenzoBianconi, 2019-02-04:
Yes, it is fine. Commit bba7187b1bcf4df8e07be20f7b1dc3edf77451fb?

ZerBea, 2019-02-04:
commit bba7187b1bcf4df8e07be20f7b1dc3edf77451fb (HEAD -> mt76x0_init_target_power, origin/mt76x0_init_target_power)

LorenzoBianconi, 2019-02-04:
> running this commit I retrieve the weird eeprom data.

Why weird data? it depends on the eeprom values

ZerBea, 2019-02-04:
So it's ok if this values changed from driver version to driver version. I thought they should be in ever case the same

ZerBea, 2019-02-04:
cat ASUS eeprom data running driver from 4.20.6 and compared them to bba7187b1bcf4df8e07be20f7b1dc3edf77451fb - same values

cat T2UH eeprom data running driver from 4.20.6 and compared them to bba7187b1bcf4df8e07be20f7b1dc3edf77451fb - different values

LorenzoBianconi, 2019-02-04:
>
>
> So it's ok if this values changed from driver version to driver version. I
> thought they should be in ever case the same
What do you mean? Before this commit? This commit fixes a bug in eeprom
parsing. Anyway the tx power values depend on eeprom calibration data, so
they can change from one device to another

> <https://github.com/openwrt/mt76/issues/216#issuecomment-460344766>, or mute
> the thread

ZerBea, 2019-02-04:
That is ok, but they should not change from the same device to the same device

ZerBea, 2019-02-04:
Ok, Ill try it again.

LorenzoBianconi, 2019-02-04:
>
> That is ok, but they should not change from the same device to the same device

From the same device over time: no
From two devices on the same type (e.g T2UH): yes

ZerBea, 2019-02-04:
Ok, I'll try to explain it in a better way:
we are on kernel 4.20.6 and use the included driver.
I cat the eeprom data from the ASUS and the T2UH

Then I compiled your repo, unloaded the old driver and loaded the new driver.
After this I did a cat eeprom data from the ASUS and the T2UH, again

No I compared the values:
ASUS 4.20.6 and ASUS your repo = same value
T2UH 4.20.6 and T2UH your repo = different values

LorenzoBianconi, 2019-02-04:
>
>
> Ok, I'll try to explain it in a better way:
> we are on kernel 4.20.6 and use the included driver.
> I cat the eeprom data from the ASUS and the T2UH
>
> Then I compiled your repo, unloaded the old driver and loaded the new
> driver.
> After this I did a cat eeprom data from the ASUS and the T2UH, again
>
> No I compared the values:
> ASUS 4.20.6 and ASUS your repo = same value
> T2UH 4.20.6 and T2UH your repo = different values
>
> <https://github.com/openwrt/mt76/issues/216#issuecomment-460348569>, or mute
> the thread
It depends on eeprom values. Could you provide me the eeprom data and the
working channel?

ZerBea, 2019-02-04:
This are the T2UH values running 4.20.6
https://github.com/openwrt/mt76/files/2825528/mt76eeprom420.zip

This are the T2UH values from your repo:
https://github.com/openwrt/mt76/files/2814107/mt76eeprom.zip

ZerBea, 2019-02-04:
Ok, give me a second to retrieve the values

ZerBea, 2019-02-04:
phy#1
	Interface wlp0s20f0u3
		ifindex 4
		wdev 0x100000001
		addr 8e:0c:19:fd:b2:24
		type monitor
		channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
		txpower 14.00 dBm

[mt76eepromt2uhtxpwrrepo.zip](https://github.com/openwrt/mt76/files/2829263/mt76eepromt2uhtxpwrrepo.zip)

LorenzoBianconi, 2019-02-04:
>
>
> phy#1
> Interface wlp0s20f0u3
> ifindex 4
> wdev 0x100000001
> addr 8e:0c:19:fd:b2:24
> type monitor
> channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
> txpower 14.00 dBm
>
> mt76eepromt2uhtxpwrrepo.zip
>
> <https://github.com/openwrt/mt76/issues/216#issuecomment-460352937>, or mute
> the thread

could you please paste output of txpower and rate_power?

ZerBea, 2019-02-04:
Target power: 32
     Delta:  0  0

      CCK: -4 -4 -4 -4
      OFDM: -4 -4 -4 -4 -4 -4 -4 -4
      STBC: -4 -4 -4 -4 -4 -4 -4 -4 -4 -4
        HT: -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4 -4
       VHT: -4 -4 -4 -4 -4 -4 -4 -4 -4 -4

LorenzoBianconi, 2019-02-04:
@ZerBea Could you please describe how you get these values?
I overwrite my eeprom data with the one you provided and I got different rate_power values. I double checked them and they are fine according to the vendor driver

[root@localhost mt76]# iw dev wlp0s4u1 info
Interface wlp0s4u1
        ifindex 13
        wdev 0x1
        addr 18:31:bf:57:22:3a
        type monitor
        wiphy 0
        channel 1 (2412 MHz), width: 20 MHz (no HT), center1: 2412 MHz
        txpower 16.00 dBm
[root@localhost mt76]# cat rate_txpower 
       CCK:  0  0  0  0
      OFDM: -3 -3 -3 -3  0  0 -4 -4
      STBC: -3 -3 -3 -3  0  0 -4 -4  0  0
        HT: -3 -3 -3 -3  0  0 -4  0  0  0  0  0  0  0  0  0
       VHT: -3 -3 -3 -3  0  0 -4 -1 -1  0
[root@localhost mt76]# cat txpower
Target power: 32
     Delta:  0  0

ZerBea, 2019-02-05:
1) git clone of the repo:
$ git clone -b mt76x0_init_target_power --single-branch https://github.com/LorenzoBianconi/wireless-drivers-next
2) cd to drivers/../mediatek
3) compile driver
make -C /lib/modules/`uname -r`/build M=$PWD clean
make -C /lib/modules/`uname -r`/build M=$PWD
4) compress modules running xz
xz mt76/*.ko
xz mt76/mt76x0/*.ko
xz mt76/mt76x2/*.ko
5) moving modules to /lib/....
cp mt76/*.xz  /lib/modules/4.20.6-arch1-1-ARCH/kernel/drivers/net/wireless/mediatek/mt76
cp mt76/mt76x0/*.xz /lib/modules/4.20.6-arch1-1-ARCH/kernel/drivers/net/wireless/mediatek/mt76/mt76x0/
cp mt76/mt76x2/*.xz /lib/modules/4.20.6-arch1-1-ARCH/kernel/drivers/net/wireless/mediatek/mt76/mt76x2/
6) activate monitor mode and stop remaining processes like NetworkManager and wpa_supplicant
iw dev wlp0s20f0u3 set type monitor
iw dev wlp0s20f0u3 set channel 1
or variant 2 (my prefered)
hcxdumptool -i wlp0s20f0u3 -c 1
We can do this running iw or hcxdumptool (I added iw functionality to hcxdumpttol). Tested both variants.
7) get interface info
$ iw dev
phy#1
	Interface wlp0s20f0u3
		ifindex 5
		wdev 0x200000001
		addr 50:3e:aa:92:e3:26
		type monitor
		txpower 14.00 dBm
8) copy requested data 
cat /sys/kernel/debug/ieee80211/phy1/mt76/txpower > txp
cat /sys/kernel/debug/ieee80211/phy1/mt76/rate_txpower > rtxp
cat /sys/kernel/debug/ieee80211/phy1/mt76/eeprom > eeprom
9) zip eeprom

remarks:
If I'm doing this on ASUS USB AC51, everything is fine.
If 'm doing this on both(!) T2UH, values are weird.

Am I' doing something wrong?

orangepizza, 2019-02-05:
from main openwrt trunk on TPlink-C2 v1 (it uses mt7610e for 5GHz)
lt looks like changes made on https://github.com/LorenzoBianconi/wireless-drivers-next/tree/mt76x0_init_target_power was pulled to openwrt/mt76 at Jan 30. but the bug didn't fixed sadly. It's still limited 12dBm. 

root@OpenWrt:/sys/kernel/debug/ieee80211/phy0/mt76# cat txpower
Target power: 17
     Delta:  0  0
root@OpenWrt:/sys/kernel/debug/ieee80211/phy0/mt76# cat rate_txpower
       CCK:  0  0  0  0
      OFDM:  7  7  7  7  4  4  0  0
      STBC:  0  0  0  0  0  0  0  0  0  0
        HT:  7  7  7  7  4  4  0  0  0  0  0  0  0  0  0  0
       VHT:  7  7  7  7  4  4  0 -1 -1  0
[tplink c2.zip](https://github.com/openwrt/mt76/files/2831744/tplink.c2.zip)

LorenzoBianconi, 2019-02-05:
>
>
> from main openwrt trunk on TPlink-C2 v1 (it uses mt7610e for 5GHz)
> lt looks like changes made on
> https://github.com/LorenzoBianconi/wireless-drivers-next/tree/mt76x0_init_target_power
> was pulled to openwrt/mt76 at Jan 30. but the bug didn't fixed sadly. It's
> still limited 12dBm.
>
> root@OpenWrt:/sys/kernel/debug/ieee80211/phy0/mt76# cat txpower
> Target power: 17
> Delta: 0 0
> root@OpenWrt:/sys/kernel/debug/ieee80211/phy0/mt76# cat rate_txpower
> CCK: 0 0 0 0
> OFDM: 7 7 7 7 4 4 0 0
> STBC: 0 0 0 0 0 0 0 0 0 0
> HT: 7 7 7 7 4 4 0 0 0 0 0 0 0 0 0 0
> VHT: 7 7 7 7 4 4 0 -1 -1 0
> tplink c2.zip
>
> <https://github.com/openwrt/mt76/issues/216#issuecomment-460590876>, or mute
> the thread
@2GHz or @5GHz?

orangepizza, 2019-02-05:
> 
 .
> @2GHz or @5ghz?
> […](#)
> -- UNIX is Sexy: who | grep -i blonde | talk; cd ~; wine; talk; touch; unzip; touch; strip; gasp; finger; gasp; mount; fsck; more; yes; gasp; umount; make clean; sleep

it's used for  5GHz.

Wiphy phy0
        max # scan SSIDs: 4
        max scan IEs length: 2243 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports RSN-IBSS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x1 RX 0x1
        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
                 * mesh point
        Band 1:
                Capabilities: 0x17e
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        RX STBC 1-stream
                        Max AMSDU length: 3839 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: 4 usec (0x05)
                HT TX/RX MCS rate indexes supported: 0-7
                Bitrates (non-HT):
                        * 1.0 Mbps (short preamble supported)
                        * 2.0 Mbps (short preamble supported)
                        * 5.5 Mbps (short preamble supported)
                        * 11.0 Mbps (short preamble supported)
                        * 6.0 Mbps
                        * 9.0 Mbps
                        * 12.0 Mbps
                        * 18.0 Mbps
                        * 24.0 Mbps
                        * 36.0 Mbps
                        * 48.0 Mbps
                        * 54.0 Mbps
                Frequencies:
                        * 2412 MHz [1] (13.0 dBm)
                        * 2417 MHz [2] (13.0 dBm)
                        * 2422 MHz [3] (13.0 dBm)
[... 133 lines cut ...]
                 * AP/VLAN
                 * monitor
        valid interface combinations:
                 * #{ IBSS } <= 1, #{ managed, AP, mesh point } <= 8,
                   total <= 8, #channels <= 1, STA/AP BI must match, radar detect widths: { 20 MHz (no HT), 20 MHz, 40 MHz, 80 MHz }

        HT Capability overrides:
                 * MCS: ff ff ff ff ff ff ff ff ff ff
                 * maximum A-MSDU length
                 * supported channel width
                 * short GI for 40 MHz
                 * max A-MPDU length exponent
                 * min MPDU start spacing
        Device supports TX status socket option.
        Device supports HT-IBSS.
        Device supports SAE with AUTHENTICATE command
        Device supports low priority scan.
        Device supports scan flush.
        Device supports AP scan.
        Device supports per-vif TX power setting
        Driver supports full state transitions for AP/GO clients
        Driver supports a userspace MPM
        Device supports active monitor (which will ACK incoming frames)
        Device supports configuring vdev MAC-addr on create.
        Device supports VHT-IBSS.

LorenzoBianconi, 2019-02-05:
Sorry I need operating channel and channel width

Regards,
Lorenzo

orangepizza, 2019-02-05:
Channel: 36 (5.180 GHz) and 40MHz?

Interface wlan0
        ifindex 9
        wdev 0x2
        addr 00:0c:43:26:60:90
        ssid OpenWrt_5G
        type AP
        wiphy 0
        channel 36 (5180 MHz), width: 40 MHz, center1: 5190 MHz
        txpower 12.00 dBm

LorenzoBianconi, 2019-02-05:
@orangepizza : I double-checked the values you posted and they are corrects. Even if the channel allows to tx @30dbm, the eeprom limits the tx power to 12dbm (this value is due to calibration data)

orangepizza, 2019-02-05:
problem is 12dBm is nowhere near what stock firmware allowed.
maybe TPlink don't used normal calibration date space and invented its own?

sgruszka, 2019-02-05:
Is possible that stock firmware override device EEPROM from external file.

orangepizza, 2019-02-05:
> 
> 
> Is possible that stock firmware override device EEPROM from external file.

I noticed there is 64kb 'radio' partiton on main flash, Maybe somewhere in this partition is where real calibration data sits for this router.
[backup-OpenWrt-radio-2019-02-05.zip](https://github.com/openwrt/mt76/files/2832271/backup-OpenWrt-radio-2019-02-05.zip)

[97 more replies]

closed 2019-07-10 by nbd168