openwrt/mt76 #219: eeprom file support for mt76

musashino205, 2018-12-06:
Are there plans to support "eeprom file" in mt76 driver?

I opened the PR (https://github.com/openwrt/openwrt/pull/1448) to add support for ELECOM WRC-733GHBK.

This device has a MT7610E for 5GHz wifi, and eeprom data is stored in Factory partition.
But eeprom does not have device specific MAC address and it needs to be patched.

Therefore, eeprom file needs to be used instead of mtd-eeprom on this device.

LGA1150, 2018-12-10:
I suggest creating a hotplug script like those in ath79 and other platform
https://github.com/openwrt/openwrt/blob/master/target/linux/ath79/base-files/etc/hotplug.d/ieee80211/10_fix_wifi_mac

musashino205, 2018-12-11:
Thanks for your suggestion.
I tried it before, but the correct MAC address was not applied to the device.

commit:
https://github.com/musashino-build/openwrt/commit/e541a519bcd0e42083cb594adca8332b65cd7639

```
root@OpenWrt:/# cat /sys/class/ieee80211/phy0/macaddress
00:90:fe:c9:cb:e7
root@OpenWrt:/# iw dev
phy#1
        Interface wlan1
                ifindex 5
                wdev 0x100000001
                addr 00:90:fe:c9:cb:e6
                type managed
                txpower 0.00 dBm
phy#0
        Interface wlan0
                ifindex 4
                wdev 0x1
                addr 00:aa:bb:cc:dd:e8
                type managed
                txpower 0.00 dBm
```

LGA1150, 2018-12-11:
Maybe it's a bug related to mt76. Overriding MAC works with rt2x00 (MT7620)

LGA1150, 2018-12-11:
Have you tried PR #209 ?

musashino205, 2018-12-11:
No.
OK, I'll try it later.

musashino205, 2018-12-18:
I checked it, but it does not work...
```
root@OpenWrt:/# cat /sys/class/ieee80211/phy0/macaddress
00:90:fe:c9:cb:e7
root@OpenWrt:/# iw dev
phy#1
        Interface wlan1
                ifindex 5
                wdev 0x100000001
                addr 00:90:fe:c9:cb:e6
                type managed
                txpower 0.00 dBm
phy#0
        Interface wlan0
                ifindex 4
                wdev 0x1
                addr 00:aa:bb:cc:dd:e8
                type managed
                txpower 0.00 dBm
```

lukasz1992, 2018-12-18:
> iw dev
does shows driver MAC address.

@musashino205 Please do it in this way: set this macaddr in interface section in /etc/config/wireless, and restart WiFi. If you think that AP BSSID is still not changed, please dump the traffic with sniffer.

musashino205, 2018-12-18:
@lukasz1992 Thanks for your opinion.
However, since this issue is related to device support, setting the MAC address directly to *"/etc/config/wireless"* will change the point of issue.

ptpt52, 2019-02-22:
The same problem On the HC5861, I hope mt76 can support it, just like rt2x00 (MT7620)

ptpt52, 2019-02-23:
@LGA1150 
I dive into the code of rt2x00, but don't see/understand how the mac(/sys/class/ieee80211/phy0/macaddress) apply to phy

ptpt52, 2019-02-28:
@nbd168 any sugguestion on implement of this function?

I want to set mac in this way: 
https://github.com/openwrt/openwrt/blob/master/target/linux/ath79/base-files/etc/hotplug.d/ieee80211/10_fix_wifi_mac

ptpt52, 2019-04-16:
I can do this trick by
https://github.com/openwrt/openwrt/pull/1906

ynezz, 2019-05-03:
@musashino205 FYI https://github.com/openwrt/openwrt/pull/1906#issuecomment-489093161

musashino205, 2019-05-03:
@ynezz Thank you for your work.

mans0n, 2019-09-03:
I wish for this feature too, not for overriding MAC address but loading eeprom itself.
I'm working with a device which has a eMMC, and mt76 eeprom data is stored somewhere in it.
But AFAIK we can't create a mtd partition on mmc.
Is there another way to load eeprom data from MMC?