openwrt/mt76 #222: Doubled phys on 18.06

nickberry17, 2018-12-10:
The wifi is running pretty well, better than before, great improvements in the recent build.  I'm not really sure how to describe this issue except that after using the commit below, and 18.06.1 I now have 4 phy's instead of 2.  I just want 2 phy's.  Not really experienced enough to know what all is useful for this problem, so I'll just put this.

How can I find out the problem and possibly fix it? 

**Kernel:**
4.14.82

**mt76 commit:**
ffccb48c5082b2c6f43a0235b77b62ca388810f0

**lspci**
```
00:00.0 PCI bridge: Device 0e8d:0801 (rev 01)
00:01.0 PCI bridge: Device 0e8d:0801 (rev 01)
01:00.0 Network controller: MEDIATEK Corp. Device 7603
02:00.0 Network controller: MEDIATEK Corp. MT7662E 802.11ac PCI Express Wireless Network Adapter
```

**iwinfo**
```
wlan0     ESSID: "OpenWrt 2.4GHz"
          Access Point: 02:00:00:00:00:00
          Mode: Master  Channel: 36 (5.180 GHz)
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgnac
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

wlan1     ESSID: "OpenWrt 5GHz"
          Access Point: 02:00:00:00:01:00
          Mode: Master  Channel: 44 (5.220 GHz)
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgnac
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy1

wlan2     ESSID: unknown
          Access Point: 00:0C:43:26:76:03
          Mode: Client  Channel: unknown (unknown)
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 14C3:7603 14C3:7603 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy2

wlan3     ESSID: unknown
          Access Point: F8:62:AA:50:00:01
          Mode: Client  Channel: unknown (unknown)
          Tx-Power: 3 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7662 14C3:7662 [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
[... 508 lines cut ...]
[    2.469049] pci 0000:02:00.0: BAR 6: assigned [mem 0x60200000-0x6020ffff pref]
[    2.483377] pci 0000:00:01.0: PCI bridge to [bus 02]
[    2.493212] pci 0000:00:01.0:   bridge window [mem 0x60100000-0x601fffff]
[    2.506704] pci 0000:00:01.0:   bridge window [mem 0x60200000-0x602fffff pref]
...
[   15.770959] mac80211_hwsim: initializing netlink
[   15.776899] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[   15.780532] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   15.795358] usbcore: registered new interface driver mt7601u
[   15.804458] bus=0x1, slot = 0x0, irq=0xff
[   15.808543] PCI: Enabling device 0000:00:00.0 (0004 -> 0006)
[   15.814478] mt7603e 0000:01:00.0: ASIC revision: 76030010
[   15.823880] mt7603e 0000:01:00.0: Firmware Version: ap_pcie
[   15.829455] mt7603e 0000:01:00.0: Build Time: 20160107100755
[   15.872615] mt7603e 0000:01:00.0: firmware init done
[   16.039991] ieee80211 phy2: Selected rate control algorithm 'minstrel_ht'
[   16.060565] bus=0x2, slot = 0x1, irq=0xff
[   16.064680] PCI: Enabling device 0000:00:01.0 (0004 -> 0006)
[   16.070544] mt76x2e 0000:02:00.0: ASIC revision: 76120044
[   16.735842] mt76x2e 0000:02:00.0: ROM patch build: 20141115060606a
[   16.745826] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[   16.751351] mt76x2e 0000:02:00.0: Build: 1
[   16.755592] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[   16.782688] mt76x2e 0000:02:00.0: Firmware running!
```

lukasz1992, 2018-12-10:
You have installed mac80211_hwsim module, and mt7601u.

phy2 - mt7603
phy3 - mt76x2
phy0, phy1 - hwsim

Everything is correct.

nickberry17, 2018-12-10:
Oh, so that's what that was :)

closed 2018-12-10 by nickberry17