openwrt/mt76 #223: Kernel oops with latest commit

kristrev, 2018-12-14:
After compiling the latest OpenWRT-nightly and installing it on my router (a ZBT3526), I sometimes see the following oops:

[   36.519177] CPU 1 Unable to handle kernel paging request at virtual address 00000018, epc == 8ed80668, ra == 8ed03454
[   36.529874] Oops[#1]:
[   36.532203] CPU: 1 PID: 3306 Comm: hostapd Not tainted 4.14.87 #0
[   36.538305] task: 8ff7ea40 task.stack: 8e0ca000
[   36.542824] $ 0   : 00000000 00000001 ffffffff fffffffe
[   36.548064] $ 4   : 8f2d5560 8f1a2c40 00000000 8f1a2d70
[   36.553312] $ 8   : 00000001 00007c00 811e3960 00002408
[   36.558534] $12   : 00000000 00000901 ffffffff 40000000
[   36.563755] $16   : 8f1a2480 8f2d4bc0 8f1a2000 00000001
[   36.568977] $20   : 8f1a275c ffffffff 00000000 8f1a2480
[   36.574197] $24   : 0f046c00 8ec2808c                  
[   36.579419] $28   : 8e0ca000 8e0cbd20 00000000 8ed03454
[   36.584641] Hi    : 51eb644c
[   36.587506] Lo    : c28f3f71
[   36.590423] epc   : 8ed80668 mt76x02_add_interface+0x6c/0x88 [mt76x02_lib]
[   36.597419] ra    : 8ed03454 drv_add_interface+0x7c/0x94 [mac80211]
[   36.603656] Status: 11007c03 KERNEL EXL IE 
[   36.607840] Cause : 4080000c (ExcCode 03)
[   36.611833] BadVA : 00000018
[   36.614701] PrId  : 0001992f (MIPS 1004Kc)
[   36.618776] Modules linked in: rt2800pci rt2800mmio rt2800lib qcserial ppp_async option usb_wwan rt2x00pci rt2x00mmio rt2x00lib rndis_host qmi_wwan ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6e
[   36.689939]  nfnetlink_acct nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_s
[   36.761201]  ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipporti
[   36.832214]  xhci_hcd uhci_hcd ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug usbcore nls_base usb_common mii
[   36.845181] Process hostapd (pid: 3306, threadinfo=8e0ca000, task=8ff7ea40, tls=77fb9e90)
[   36.853321] Stack : 8f733000 8ed49af8 8f2d4bc0 8f1a2488 8f2d4bc0 8ed03454 00000001 8f1a275c
[   36.861674]         00000000 00000000 8f1a2488 8ed17dd4 8f1a2000 00001003 8ed5e668 00000000
[   36.870025]         8f1a2000 8f1a2030 8ed5e668 00000000 7f834e68 00000000 8f1a2000 8f3d3e00
[   36.878376]         00000000 80342ce4 00000000 00000000 00000000 00000000 8f1a2000 8e0cbee8
[   36.886726]         8f1a2000 00001003 00000001 8034302c 8e0cbee0 7f834d38 00000000 8faac180
[   36.895080]         ...
[   36.897529] Call Trace:
[   36.899986] [<8ed80668>] mt76x02_add_interface+0x6c/0x88 [mt76x02_lib]
[   36.906572] [<8ed03454>] drv_add_interface+0x7c/0x94 [mac80211]
[   36.912555] [<8ed17dd4>] ieee80211_do_open+0x3b4/0x9dc [mac80211]
[   36.918685] [<80342ce4>] __dev_open+0xdc/0x14c
[   36.923117] [<8034302c>] __dev_change_flags+0xe8/0x1ac
[   36.928239] [<80343118>] dev_change_flags+0x28/0x70
[   36.933114] [<803c11c8>] devinet_ioctl+0x2ec/0x830
[   36.937917] [<8031ed54>] sock_ioctl+0x300/0x35c
[   36.942457] [<80127a54>] do_vfs_ioctl+0x778/0x818
[   36.947148] [<80127b44>] SyS_ioctl+0x50/0x9c
[   36.951423] [<80019578>] syscall_common+0x34/0x58
[   36.956109] Code: 24a70130  a0a20185  a0a30184 <acc70018> 0fb0ac59  8ca50120  8fbf0014  00001025  03e00008 
[   36.965857] 
[   36.967761] ---[ end trace e291e07e09000323 ]---
[   36.976033] Kernel panic - not syncing: Fatal exception
[   36.983245] Rebooting in 3 seconds..

I have only seen the oops are upgrading the software on the router, and the wifi configuration before the upgrade does not seem to matter. The oops is triggered regardless if the wifi-interfaces were used as client or station. I do not see the oops after every upgrade, but once it happens then the router is stuck in a boot loop as the oops is triggered on every noot.

LorenzoBianconi, 2018-12-14:
>
> After compiling the latest OpenWRT-nightly and installing it on my router
> (a ZBT3526), I sometimes see the following oops:
>
> Are you running the latest commit?
(da00af00c503250ef8e7d1da74a123191def2893)

> [ 36.519177] CPU 1 Unable to handle kernel paging request at virtual
> address 00000018, epc == 8ed80668, ra == 8ed03454
> [ 36.529874] Oops[#1 <https://github.com/openwrt/mt76/pull/1>]:
> [ 36.532203] CPU: 1 PID: 3306 Comm: hostapd Not tainted 4.14.87 #0
> [ 36.538305] task: 8ff7ea40 task.stack: 8e0ca000
> [ 36.542824] $ 0 : 00000000 00000001 ffffffff fffffffe
> [ 36.548064] $ 4 : 8f2d5560 8f1a2c40 00000000 8f1a2d70
> [ 36.553312] $ 8 : 00000001 00007c00 811e3960 00002408
> [ 36.558534] $12 : 00000000 00000901 ffffffff 40000000
> [ 36.563755] $16 : 8f1a2480 8f2d4bc0 8f1a2000 00000001
> [ 36.568977] $20 : 8f1a275c ffffffff 00000000 8f1a2480
> [ 36.574197] $24 : 0f046c00 8ec2808c
> [ 36.579419] $28 : 8e0ca000 8e0cbd20 00000000 8ed03454
> [ 36.584641] Hi : 51eb644c
> [ 36.587506] Lo : c28f3f71
> [ 36.590423] epc : 8ed80668 mt76x02_add_interface+0x6c/0x88 [mt76x02_lib]
> [ 36.597419] ra : 8ed03454 drv_add_interface+0x7c/0x94 [mac80211]
> [ 36.603656] Status: 11007c03 KERNEL EXL IE
> [ 36.607840] Cause : 4080000c (ExcCode 03)
> [ 36.611833] BadVA : 00000018
> [ 36.614701] PrId : 0001992f (MIPS 1004Kc)
> [ 36.618776] Modules linked in: rt2800pci rt2800mmio rt2800lib qcserial
> ppp_async option usb_wwan rt2x00pci rt2x00mmio rt2x00lib rndis_host
> qmi_wwan ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6e
> [ 36.689939] nfnetlink_acct nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic
> nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4
> nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_s
> [ 36.761201] ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface
> ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net
> ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet
> ip_set_hash_ipporti
> [ 36.832214] xhci_hcd uhci_hcd ohci_platform ohci_hcd ehci_platform sd_mod
> scsi_mod ehci_hcd gpio_button_hotplug usbcore nls_base usb_common mii
> [ 36.845181] Process hostapd (pid: 3306, threadinfo=8e0ca000,
> task=8ff7ea40, tls=77fb9e90)
> [ 36.853321] Stack : 8f733000 8ed49af8 8f2d4bc0 8f1a2488 8f2d4bc0 8ed03454
> 00000001 8f1a275c
> [ 36.861674] 00000000 00000000 8f1a2488 8ed17dd4 8f1a2000 00001003
> 8ed5e668 00000000
> [ 36.870025] 8f1a2000 8f1a2030 8ed5e668 00000000 7f834e68 00000000
> 8f1a2000 8f3d3e00
> [ 36.878376] 00000000 80342ce4 00000000 00000000 00000000 00000000
> 8f1a2000 8e0cbee8
> [ 36.886726] 8f1a2000 00001003 00000001 8034302c 8e0cbee0 7f834d38
> 00000000 8faac180
> [ 36.895080] ...
> [ 36.897529] Call Trace:
> [ 36.899986] [<8ed80668>] mt76x02_add_interface+0x6c/0x88 [mt76x02_lib]
> [ 36.906572] [<8ed03454>] drv_add_interface+0x7c/0x94 [mac80211]
> [ 36.912555] [<8ed17dd4>] ieee80211_do_open+0x3b4/0x9dc [mac80211]
> [ 36.918685] [<80342ce4>] __dev_open+0xdc/0x14c
> [ 36.923117] [<8034302c>] __dev_change_flags+0xe8/0x1ac
> [ 36.928239] [<80343118>] dev_change_flags+0x28/0x70
> [ 36.933114] [<803c11c8>] devinet_ioctl+0x2ec/0x830
> [ 36.937917] [<8031ed54>] sock_ioctl+0x300/0x35c
> [ 36.942457] [<80127a54>] do_vfs_ioctl+0x778/0x818
> [ 36.947148] [<80127b44>] SyS_ioctl+0x50/0x9c
> [ 36.951423] [<80019578>] syscall_common+0x34/0x58
> [ 36.956109] Code: 24a70130 a0a20185 a0a30184 0fb0ac59 8ca50120 8fbf0014
> 00001025 03e00008
> [ 36.965857]
> [ 36.967761] ---[ end trace e291e07e09000323 ]---
> [ 36.976033] Kernel panic - not syncing: Fatal exception
> [ 36.983245] Rebooting in 3 seconds..
>
> I have only seen the oops are upgrading the software on the router, and
> the wifi configuration before the upgrade does not seem to matter. The oops
> is triggered regardless if the wifi-interfaces were used as client or
> station. I do not see the oops after every upgrade, but once it happens
> then the router is stuck in a boot loop as the oops is triggered on every
> noot.

Could you please run gdb on mt76x02_lib module?

Regards,
Lorenzo

kristrev, 2018-12-14:
> Are you running the latest commit?

Yes, my image includes OpenWRT-commit 4fd788282274e132b974e074a35cf8729c72819e, which includes mt76-commit you mention. I installed the same OpenWRT-image on a couple of more routers, and do not see the oops.

> Could you please run gdb on mt76x02_lib module?

I will try. I do not know how, but I will try to figure it out. However, the oops only happens right after boot, which I guess complicates the situation a bit?

Thanks the reply!

kristrev, 2018-12-20:
I will close this issue and open it again if I can see the issue. I have been testing quite extensively over the last week, but the oops has not occurred. I suspect something dirty in my build directory.

kristrev, 2019-01-05:
I have found a device where I am able to reliably trigger the issue. I have not been able to run gdb, but I instrumented the driver to write something before each line in the path starting with the call to `mt76x02_add_interface()`. The last output I see, is the output I added before this line `mtxq->wcid = &mvif->group_wcid;` in `mt76x02_vif_init()`.

I am running nightly OpenWRT, so all the most recent commits to mt76 are part of my firmware.

kristrev, 2019-01-05:
After some more debugging, I think I found the problem. Turns out that while `mtxq` is not NULL, `vif->txq` is NULL. Looking through the git history for `mt76x02_util.c`, I see that there was a function `mt76x02_txq_init()` that was removed in 7b1373e7e2b4181cf676b0ee4a7b74ec1b297c21. This function contained a check for if txq wa NULL, and if that was the case then we just returned.

When removing the function, the NULL-check seem to have been lost. As a test, I added the check for NULL to `mt76x02_add_interface()`, and now the router boots just fine and wifi works fine. I am not very familiar with `mac80211`, so I do not know if the NULL-check was removed for a reason or if restoring it is correct. However, the problem seems to be history on my devices now.

kristrev, 2019-01-05:
I was a bit quick to declare victory. The following error appeared when connecting a client to my AP:

```
[  118.920461] CPU 0 Unable to handle kernel paging request at virtual address 0000002f, epc == 8f3f35fc, ra == 8f3f35cc
[  118.931112] Oops[#1]:
[  118.933379] CPU: 0 PID: 4015 Comm: hostapd Not tainted 4.14.90 #0
[  118.939444] task: 8fe33200 task.stack: 8e11c000
[  118.943948] $ 0   : 00000000 00000001 00000088 00000000
[  118.949159] $ 4   : 00000000 00000008 8ed98610 00000001
[  118.954368] $ 8   : 00000000 00000015 8ed98610 00000008
[  118.959576] $12   : 000000ff 000000ff 00000001 8efecce0
[  118.964785] $16   : 00000000 000005d4 8ed81560 8efeccc0
[  118.969994] $20   : 000005d0 00000000 8f3e25d8 8f3e26a8
[  118.975203] $24   : 8efecce0 00000000                  
[  118.980411] $28   : 8e11c000 8e11da28 8ef21c2e 8f3f35cc
[  118.985620] Hi    : 0000008b
[  118.988493] Lo    : 00000006
[  118.991398] epc   : 8f3f35fc mt76_tx+0xe0/0x2ac [mt76]
[  118.996521] ra    : 8f3f35cc mt76_tx+0xb0/0x2ac [mt76]
[  119.001635] Status: 11008403 KERNEL EXL IE 
[  119.005816] Cause : 40800008 (ExcCode 02)
[  119.009806] BadVA : 0000002f
[  119.012674] PrId  : 0001992f (MIPS 1004Kc)
[  119.016749] Modules linked in: rt2800pci rt2800mmio rt2800lib qcserial ppp_async option usb_wwan rt2x00pci rt2x00mmio rt2x00lib rndis_host qmi_wwan ppp_generic nf_nat_pptp nf_conntrack_pptp nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether ax88179_178a asix xt_time xt_tcpe
[  119.087909]  nfnetlink_acct nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_conntrack_ipv4 nf_nat_ipv4 nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_cs
[  119.159164]  ledtrig_usbport xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 ni
[  119.230172]  xhci_hcd uhci_hcd ohci_platform ohci_hcd ehci_platform sd_mod scsi_mod ehci_hcd gpio_button_hotplug usbcore nls_base usb_common mii
[  119.243132] Process hostapd (pid: 4015, threadinfo=8e11c000, task=8fe33200, tls=77fc6e90)
[  119.251269] Stack : 8ed81500 00000000 00000000 80470b78 00000001 8e11daf0 8e11daf4 8efeccc0
[  119.259620]         8ed80bc0 00000000 8ed80f68 8e11daf4 00000000 8f3e25d8 8fd59c40 8ecaa654
[  119.267969]         8efeccc0 8ef21c00 8ed80bc0 8ed821b4 8f3e25d8 8efeccc0 8e11daf4 00000288
[  119.276317]         8ed80bc0 00000085 00000000 805a0000 00000000 ffffffff 8fd59054 8ecaa798
[  119.284666]         8ecf0000 00000002 8efeccc0 8fd59480 00000000 8ed80bc0 8efeccc0 8fd59480
[  119.293017]         ...
[  119.295465] Call Trace:
[  119.297914] [<8f3f35fc>] mt76_tx+0xe0/0x2ac [mt76]
[  119.302850] [<8ecaa654>] ieee80211_csa_set_counter+0x2e8/0xa20 [mac80211]
[  119.309704] [<8ecaa798>] ieee80211_csa_set_counter+0x42c/0xa20 [mac80211]
[  119.316506] Code: 00031880  02c31821  8c630000 <9064002f> 1080000c  00000000  8e6500ac  94a40000  3086008c 
[  119.326253] 
[  119.328113] ---[ end trace 79f93aa339f9f0c5 ]---
[  119.336787] Kernel panic - not syncing: Fatal exception in interrupt
[  119.345555] Rebooting in 3 seconds..
```

kristrev, 2019-01-06:
I spent some more time looking into this issue and I am starting to suspect that the problem might lie outside of the mt76-driver. I have some devices (also 3526) running OpenWRT 18.06.1, but with mt76 bumped to the latest version and they do not have the problem of the crash loop. I will instrument my 18.06.1 build and see if I can for example see the missing TXQ wuth that OpenWRT-version as well.

kristrev, 2019-01-06:
I think I have found the source of the problem. It seems that mac80211-package was not rebuilt or not rebuilt properly after the last bump in OpenWRT. When adding some debug output to package, it was completely rebuilt and since then the problems have gone away. I am not able to reproduce the crash any more. I still don't quite see what has caused the issue, but then again, I am not so familiar with the code so the problem is probably over my head :)

closed 2019-01-06 by kristrev

nbd168, 2019-01-06:
Thanks for tracking this down. I've pushed a change to OpenWrt master to prevent this issue from happening again on future upgrades.

kristrev, 2019-01-06:
Thanks for a quick fix!

camelcamro, 2019-01-07:
does this means, that it should be solved in the next trunk build ?