openwrt/mt76 #226: mt7603 on mir3g totally unusable

dimatto, 2019-01-06:
I usually use driver commit 0853d79 on AUG 25,2018. Sometimes it’s slowly. The wifi client is a set top box and there’s a wall beteewn it and mir3g. Today I upgrade driver to latest, then it totally unusable.below is the luci capture
[image]

nbd168, 2019-01-11:
Please try the latest version

tkso1997, 2019-01-13:
With the newest version its seems to be much much better, 2.4g radio is usable, I had no disconnects in the last 5 hours. Lets hope it stays!!
5g is a bit slower than it was when I tried last time in December but still absolutely usable.

iperf3 on the 2.4g network:
`Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.166 port 57719 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  6.99 MBytes  58.5 Mbits/sec                  
[  5]   1.00-2.00   sec  8.41 MBytes  70.7 Mbits/sec                  
[  5]   2.00-3.00   sec  9.44 MBytes  79.0 Mbits/sec                  
[  5]   3.00-4.00   sec  10.1 MBytes  84.8 Mbits/sec                  
[  5]   4.00-5.00   sec  10.1 MBytes  84.8 Mbits/sec                  
[  5]   5.00-6.00   sec  8.59 MBytes  72.1 Mbits/sec                  
[  5]   6.00-7.00   sec  9.79 MBytes  82.3 Mbits/sec                  
[  5]   7.00-8.00   sec  8.71 MBytes  73.1 Mbits/sec                  
[  5]   8.00-9.00   sec  7.85 MBytes  65.6 Mbits/sec                  
[  5]   9.00-10.00  sec  6.47 MBytes  54.5 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  86.5 MBytes  72.5 Mbits/sec                  sender
[  5]   0.00-10.02  sec  86.2 MBytes  72.2 Mbits/sec                  receiver

iperf Done.`

iperf3 on 5g network:

`
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.166 port 57776 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  19.0 MBytes   159 Mbits/sec                  
[  5]   1.00-2.00   sec  21.7 MBytes   182 Mbits/sec                  
[  5]   2.00-3.00   sec  24.2 MBytes   203 Mbits/sec                  
[  5]   3.00-4.00   sec  23.4 MBytes   196 Mbits/sec                  
[  5]   4.00-5.00   sec  22.3 MBytes   187 Mbits/sec                  
[  5]   5.00-6.00   sec  17.9 MBytes   150 Mbits/sec                  
[  5]   6.00-7.00   sec  20.6 MBytes   172 Mbits/sec                  
[  5]   7.00-8.00   sec  20.9 MBytes   175 Mbits/sec                  
[  5]   8.00-9.00   sec  20.2 MBytes   169 Mbits/sec                  
[  5]   9.00-10.00  sec  17.2 MBytes   144 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec   207 MBytes   174 Mbits/sec                  sender
[  5]   0.00-10.01  sec   207 MBytes   173 Mbits/sec                  receiver

iperf Done.
`

most important for me no dramatic latency issues!!

dimatto, 2019-01-14:
@nbd168 yes it’s better, but still laggy. This issue is brought by some commits during May to July, 2018, because there’s no this kind of problem before May.

tkso1997, 2019-01-14:
I registered some lags but it is def. better. Is there an easy way to get the mt76 version from before May working with a recent openwrt? Do I have to compile the whole firmware or is there some work around?

nbd168, 2019-01-14:
You can copy the mt76 makefile from the old firmware state into a new build. With a bit of luck, it's still compatible. It would be nice if you could find out which commit broke it.

nbd168, 2019-01-20:
Please try the latest version, I've added some more fixes

popurlti, 2019-01-25:
same problem here, might be related to this old issue #145 ?

pataha, 2019-01-27:
Problem still persisted when connected overnight. It connected but cant do anything

Not sure the log helps but here is warning
‘’’
[ 8515.268748] ------------[ cut here ]------------
[ 8515.273410] WARNING: CPU: 0 PID: 0 at /builder/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2019-01-25-0b939dc7/mt7603/beacon.c:52 mt7603_eeprom_init+0x4dc/0x518 [mt7603e]
[ 8515.291620] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
[ 8515.362537] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.95 #0
[ 8515.368518] Stack : 00000000 00000001 8f154bc0 800703d8 805d0000 805755ec 00000000 00000000
[ 8515.376856]         8053fde0 8fc09d84 805b0dbc 805b0947 8053acb8 00000001 8fc09d28 1cc282bf
[ 8515.385187]         00000000 00000000 80610000 00005db0 00000000 00000127 00000008 00000000
[ 8515.393517]         00000000 805b0000 00058829 00000000 00000000 00000000 805d0000 8f1779ac
[ 8515.401849]         00000009 00000034 8f155560 00000001 00000003 802a41d8 00000000 80610000
[ 8515.410180]         ...
[ 8515.412616] Call Trace:
[ 8515.415076] [<800106c0>] show_stack+0x58/0x100
[ 8515.419508] [<804788dc>] dump_stack+0x9c/0xe0
[ 8515.423851] [<8002ee98>] __warn+0xe0/0x114
[ 8515.427929] [<8002ef5c>] warn_slowpath_null+0x1c/0x30
[ 8515.432966] [<8f1779ac>] mt7603_eeprom_init+0x4dc/0x518 [mt7603e]
[ 8515.439167] ---[ end trace ae8902d1b6dfd982 ]---
‘’’

tkso1997, 2019-01-28:
I compile a version of last march.
I can't try it at the moment, but if anybody is willing to fell free to try the modules. 
I compiled it with the most recent OpenWRT, so I guess you should be able to just  unload and uninstall the current ipks and install and load the modules of the ones in my link.
@nbd168 Correct me if I am wrong.

tkso1997, 2019-01-31:
Just testing the newest version of mt76 2.4GHz. Seems to run quite stable and very responsive!

redchenjs, 2019-02-06:
> Just testing the newest version of mt76 2.4GHz. Seems to run quite stable and very responsive!
> 
> and just 5 minutes later it sopped working ... same problem as before, you need to disconnect and reconnect again ...

On my Newifi D2 (MT7603E for 2.4GHz), only "N" mode has this problem, "Legacy" mode is quite stable (iperf3 ~25Mbps).

tahamsaglam, 2019-02-11:
+1 with mir3g

tahamsaglam, 2019-02-12:
+1 With Mir3G.
But i can only see this problem with my two different intel 3165 clients. No problem on my qcom phone.
Can anyone else confirm if client type matters?

Tom-Brouwer, 2019-02-12:
For me also client type seems to matter. Nokia 5 with qualcomm chip is always working, but having troubles with multiple intel adapters:

Intel 6105 on a ubuntu laptop
Intel 6135 on a windows 10 pc

With previous versions internet would stop within 1 or two minutes on both devices. I now have the latest running, and the Ubuntu seems to be stable, but the windows one still has issues sometimes, although it works for a much longer time on end now.

I saw a recent commit about ccmp (fc31457), of which I thought it would fix the issue, but after installing that one, I still had disconnects very often. As said, on the latest commit, problems seem much less (I can stream live tv for longer periods on the Ubuntu one), but the windows one still sometimes has no internet. Overall it has become much better however...

pataha, 2019-02-23:
it's unstable, new crash log
```
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.519696] ------------[ cut here ]------------
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.524442] WARNING: CPU: 0 PID: 0 at /build/openwrt-18.06/slaves/phase1/ramips_mt7621/build/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2019-01-25-0b939dc7/mt7603/beacon.c:52 0x8f0f79ac [mt7603e@8f0f0000+0x8d40]
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.544624] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables leds_gpio xhci_mtk xhci_plat_hcd xhci_pci xhci_hcd gpio_button_hotplug usbcore nls_base usb_common
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.615477] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.14.95 #0
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.621539] Stack : 00000000 00000000 00000000 00000001 00000000 00000000 00000000 00000000
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.629959]         00000000 00000000 00000000 00000000 00000000 00000001 8fc09d20 1cc282bf
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.638381]         8fc09db8 00000000 00000000 00005dd8 00000038 80480738 00000008 00000000
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.646799]         00000000 80530000 00096435 70617773 8fc09d00 00000000 80550000 8f0f79ac
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.655218]         00000009 00000034 8e849560 00000001 00000000 802994b8 00000000 80590000
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.663636]         ...
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.666158] Call Trace:
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.666171] [<80480738>] 0x80480738
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.672243] [<8f0f79ac>] 0x8f0f79ac [mt7603e@8f0f0000+0x8d40]
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.678065] [<802994b8>] 0x802994b8
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.681624] [<80010090>] 0x80010090
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.685180] [<80010098>] 0x80010098
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.688736] [<8046975c>] 0x8046975c
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.692290] [<80070fd4>] 0x80070fd4
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.695847] [<8002e948>] 0x8002e948
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.699406] [<8f0f79ac>] 0x8f0f79ac [mt7603e@8f0f0000+0x8d40]
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.705218] [<8f0f78bc>] 0x8f0f78bc [mt7603e@8f0f0000+0x8d40]
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.711019] [<8002ea0c>] 0x8002ea0c
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.714584] [<8f0f79ac>] 0x8f0f79ac [mt7603e@8f0f0000+0x8d40]
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.720385] [<80486b2c>] 0x80486b2c
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.723946] [<8f0f3464>] 0x8f0f3464 [mt7603e@8f0f0000+0x8d40]
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.729754] [<8f233c30>] 0x8f233c30 [mac80211@8f200000+0x67e00]
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.735738] [<8f0f7a6c>] 0x8f0f7a6c [mt7603e@8f0f0000+0x8d40]
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.741548] [<80032c94>] 0x80032c94
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.745105] [<80486c68>] 0x80486c68
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.748664] [<80033084>] 0x80033084
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.752219] [<8024ffec>] 0x8024ffec
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.755776] [<8000b4e8>] 0x8000b4e8
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.759329]
Sat Feb 16 17:00:25 2019 kern.warn kernel: [313927.760996] ---[ end trace ff07e132223115da ]---
```

dimatto, 2019-02-25:
it’s usable now although very laggy, I’ll start another issue to find out why.

closed 2019-02-25 by dimatto