openwrt/mt76 #227: Various issues with 5Ghz MT7610E

ffrediani, 2019-01-07:
Using a TP-Link Archer C20 v4 with MT7610E I have installed kmod-mt76x0e which doesn't come by default and it detects the 5Ghz card. Configuring it it works to connect but a few minutes after booting the router the 5 Ghz SSID disappear and doesn't come back soon.

Sometimes the SSID may appear again but for short periods and then disappear again, very intermittently, therefore it is unusable.
Tested with Wifi Analyser and the SSID isn't even advertised.

I tried various different configurations as such:
- Changing the mode from AC to N
- Using 40 Mhz channel
- Changing the channel to use the groups 36-64, 100-140 and 149-165 but made no difference
- Changing the Transmit Power from Auto to different levels, but although it shows it on LuCI it doesn't seem to make any difference on the power signal.

If I am not wrong the Maximum Transmit Power for this card is 13dBm but it is possible to configure it with more and it shows up more under "iwinfo wlan1 info"

Also the Scan option doesn't seem to work either.

@nbd168 did you mentioned recently some work has been done on this driver asking us to test or am I mistaken and confused with other chip ? Is there anything else other than related here that is worth to give attention ?

ffrediani, 2019-01-29:
Hello @nbd168 after upgrading to aa470d8 the same issue happens using various different configurations for this radio.
The SSID barely advertises and appears as available for connecting. It stays invisible or not working most of the time (not even Wifi Analyser detects it). If it appears again it doesn't last much and therefore it is still unusable.

Let us know of anything significant about the improvements for it or any specific feedback you may require we can provide accurately.

Alpha2121, 2019-03-06:
I have a c20 v1 with same 5ghz chipset and kmod-mt76x0e driver isn't doing anything for me.  I installed latest snapshot with kmod-mt76x0e pre installed, yet the 5ghz radio isn't showing up under wireless settings.. What am i doing wrong?

ffrediani, 2019-03-26:
Hello @nbd168 
Not sure if you have been doing any work on this recently but wanted to feedback that I have done successive upgrades to snapshot version to this TP-Link C20 v4 with kmod-mt76x0e driver now coming by default in the image and had zero improvements since them. Currently using 6fc50ea9 and Kernel 4.14.107 but all symptoms I have reported in the first message seem to be still present.

Can't make a simple scan, SSID disappear, doesn't connect at all regardless the channel used, etc.

So although the 5Ghz radio is detected and appears in the system it remains unusable.
Let us know you output so we can provide more fine feedback in order to help advance with this driver.

ffrediani, 2019-06-25:
This issue should never beem closed. It remains the same and MT7610E keeps not working properly at all. It is detected by default snapshot but doesn't work and has the same issues reported.

@nbd168 have you or anyone done any work about this driver ?

I believe the results will be the same if the radio hardware is the same regardless the router, but to mention in more detail the router in case is a TP-Link C20 v4

ffrediani, 2019-06-26:
Just to update the thread I have updated to Snapshot r10348-577174cf60 with a recent mt76 fix and now it seems to be able to control the  Transmit Power correctly. Also the Scan button that has never worked  correctly seems to work now.

xavier83, 2019-09-14:
Seem like I'm experiencing the same issue

https://github.com/openwrt/mt76/issues/199

davidzmn, 2019-09-25:
> I have a c20 v1 with same 5ghz chipset and kmod-mt76x0e driver isn't doing anything for me. I installed latest snapshot with kmod-mt76x0e pre installed, yet the 5ghz radio isn't showing up under wireless settings.. What am i doing wrong?

Hi there I have the same problem 
Already installed Kmod-mt76x0e and 5GHz radio isn't showing  
Any Suggestion?
Thank you

AndyX90, 2019-11-09:
> Hi there I have the same problem
> Already installed Kmod-mt76x0e and 5GHz radio isn't showing
> Any Suggestion?
> Thank you

Same for me. kmod-mt76x0e installed and no wifi showing.
DMESG-Output:

> [   14.531320] mt76x0e 0000:02:00.0: enabling device (0000 -> 0002)
> [   14.537510] mt76x0e 0000:02:00.0: ASIC revision: 76100002
> [   14.570734] mt76x0e 0000:02:00.0: Firmware Version: 0.1.00
> [   14.845282] mt76x0e 0000:02:00.0: driver does not support default EEPROM
> [   14.850755] mt76x0e: probe of 0000:02:00.0 failed with error -22

Lankaster, 2019-12-18:
Same problem on Asus RT-AC51U, Openwrt 19.07 (commit d2d12346e8e6dd5fbb6befd4f9e3e1d5304a2e31)
```
[  112.353442] mt76x0e 0000:01:00.0: card - bus=0x1, slot = 0x0 irq=4
[  112.366014] mt76x0e 0000:01:00.0: ASIC revision: 76100002
[  112.379113] mt76x0e 0000:01:00.0: Firmware Version: 0.1.00
[  112.476377] mt76x0e 0000:01:00.0: EEPROM data check failed: ffff
[  112.500587] mt76x0e 0000:01:00.0: driver does not support default EEPROM
[  112.514050] mt76x0e: probe of 0000:01:00.0 failed with error -22
[  112.542572] rt2800_wmac 10180000.wmac: loaded eeprom from mtd device "factory"
[  112.557020] ieee80211 phy1: rt2x00_set_rt: Info - RT chipset 6352, rev 0500 detected
[  112.572438] ieee80211 phy1: rt2800_init_eeprom: Error - Invalid RF chipset 0xffff detected
[  112.588875] ieee80211 phy1: rt2x00lib_probe_dev: Error - Failed to allocate device
```
wifi (both 2.4 & 5 ) doesn't work.

Lankaster, 2020-01-17:
> Same problem on Asus RT-AC51U, Openwrt 19.07 (commit d2d12346e8e6dd5fbb6befd4f9e3e1d5304a2e31)
>
> ```
> [  112.353442] mt76x0e 0000:01:00.0: card - bus=0x1, slot = 0x0 irq=4
> [  112.366014] mt76x0e [...]

In my case calibration data was erased. Stock firmware shows wrong MAC adresses for LAN and WAN (00:11:11:11:11:11, 00:22:22:22:22:22)
hier ist bootlog with errors from stock FW
```
Aug  1 00:00:37 kernel: MT7610ap:[FullCalibration] Reload from flash! current channel : 42 
Aug  1 00:00:37 kernel: MT7610ap:<==== ReloadLowCalResult() 
Aug  1 00:00:37 kernel: MT7610ap:====> ReloadLowCalResult() 
Aug  1 00:00:37 kernel: MT7610ap:==== 2C70 [ffffffff] ,RXBackup [ffffffff]
Aug  1 00:00:37 kernel: MT7610ap:2C70 [ffffffff] not 0 , backup [ffffffff] 
Aug  1 00:00:37 kernel: MT7610ap:SYNC - BBP R4 to 20MHz.l
Aug  1 00:00:37 kernel: MT7610ap:FullCalibration():  docal = [ffff] valid bit[10]
Aug  1 00:00:37 kernel: MT7610ap:[FullCalibration] Reload from flash! current channel : 36 
Aug  1 00:00:37 kernel: MT7610ap:<==== ReloadLowCalResult() 
Aug  1 00:00:37 kernel: MT7610ap:====> ReloadLowCalResult() 
Aug  1 00:00:37 kernel: MT7610ap:==== 2C70 [ffffffff] ,RXBackup [ffffffff]
Aug  1 00:00:37 kernel: MT7610ap:2C70 [ffffffff] not 0 , backup [ffffffff] 
Aug  1 00:00:37 kernel: MT7610ap:==== 2C70 [ffffffff] ,RXBackup [ffffffff]
Aug  1 00:00:37 kernel: MT7610ap:2C70 [ffffffff] not 0 , backup [ffffffff] 
Aug  1 00:00:37 kernel: MT7610ap:FullCalibration():  docal = [ffff] valid bit[10]
Aug  1 00:00:37 kernel: MT7610ap:[FullCalibration] Reload from flash! current channel : 40 
Aug  1 00:00:37 kernel: MT7610ap:<==== ReloadLowCalResult() 
Aug  1 00:00:37 kernel: MT7610ap:====> ReloadLowCalResult() 
Aug  1 00:00:37 kernel: MT7610ap:==== 2C70 [ffffffff] ,RXBackup [ffffffff]
Aug  1 00:00:37 kernel: MT7610ap:2C70 [ffffffff] not 0 , backup [ffffffff] 
```

ffrediani, 2020-04-04:
Hello @nbd168 
I have a newer report about remaining issues related to MT7610E.
Just for reference the device used here is a TP-Link C20 v4
This was tested with OpenWrt 18.06.8 (installing the extra kmod-7610e package driver) and with 19.07.0 and 19.07.2

When trying to set the channel for 5Ghz radio. It simply doesn't let you do that. You must use it in Auto otherwise when the device boots up the 5Ghz radius simply doesn't broadcast SSID and no clients can connect.
I tried set several different channels from all three groups: 36-64, 100-140 and 149-165 and it didn't work at all. Only if I leave it configured on auto. When set in Auto it always default to channel 36.
When the device boots it may take a little while for the radio to start working and to let clients connect.

Below is my /etc/config/wireless relevant part for the 5Ghz radio:

config wifi-device 'radio1'
        option type 'mac80211'
        option hwmode '11a'
        option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
        option country 'US'
        option distance '150'
        option beacon_int '70'
        option txpower '18'
        option channel 'auto'
        option legacy_rates '0'
        option htmode 'VHT40'
        option __toggle 'Enable'

rimil0uz, 2020-05-04:
> Hello @nbd168
> I have a newer report about remaining issues related to MT7610E.
> Just for reference the device used here is a TP-Link C20 v4
> This was tested with OpenWrt 18.06.8 (installing the extra [...]

I have exactly the same issue with WAVLINK-WN570HA1 using OpenWRT 19.07.3 Master. If i change channels on 5ghz network the radio can't be started. What's more with the radio set on auto. It allows me to connect for a short time and then in like 5 minutes it starts failing, my devices start needing more than one attemp to connect and late are not able to connect at all. With the radio not showing anymore.

ffrediani, 2020-05-04:
It was announced today in the OpenWrt-Devel list that there is a plan to release 19.07.3 and 18.06.9 soon. It would be interesting if these issues could be fixed before that happens so they are included in these newer builds.

LorenzoBianconi, 2020-05-05:
> Same problem on Asus RT-AC51U, Openwrt 19.07 (commit d2d12346e8e6dd5fbb6befd4f9e3e1d5304a2e31)
>
> ```
> [  112.353442] mt76x0e 0000:01:00.0: card - bus=0x1, slot = 0x0 irq=4
> [  112.366014] mt76x0e [...]

@Lankaster have you been able to restore it? what is the tx power @ 5GHz?

Lankaster, 2020-05-05:
@LorenzoBianconi I bought RT-AC51U with erased calibration data partition. Unfortunately it doesn't possible to do something (see my comment from 17 Jan)

LorenzoBianconi, 2020-05-05:
> @LorenzoBianconi I bought RT-AC51U with erased calibration data partition. Unfortunately it doesn't possible to do something (see my comment from 17 Jan)

My current issue is the tx power on 5GHz band is very weak. IIUC your device has been shipped with corrupted calibration data or has the factory partition been wiped? In the latter case, do you remember what was the signal strength using the stock firmware?

Lankaster, 2020-05-05:
@LorenzoBianconi factory partition was wiped. Default APs on stock firmware wasn't visible. It didn't make sense for me check something more.

LorenzoBianconi, 2020-05-05:
@Lankaster I can share my factory partition if you want to try to restore it

senolcalisan, 2020-05-11:
Hi there,

I just bricked my C20i and desoldered Eeprom than flashed with usb-spi writer.
I was facing the disappearing 5Ghz AP in stock firmware so i wanted to try openwrt.
After flashing 19.07 Lede, 5ghz seams to working fine , no dissapering any more however, 5ghz working laggy. I am getting too many lost packets when i trying to ping the AP.
Hope it will be fixed soon

rimil0uz, 2020-05-14:
In my WAVLINK-wl-wn570ha1 which according to wikidev uses this MT7610E wchip for the 5ghx band i get the next log when my wlan fails:

`Thu Feb 27 21:12:21 2020 user.notice mac80211: Failed command: iw phy phy1 set antenna 0xffffffff 0xffffffff
Thu Feb 27 21:12:22 2020 daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
Thu Feb 27 21:12:22 2020 kern.info kernel: [  439.902903] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
Thu Feb 27 21:12:22 2020 kern.info kernel: [  439.919586] br-lan: port 2(wlan1) entered blocking state
Thu Feb 27 21:12:22 2020 kern.info kernel: [  439.930215] br-lan: port 2(wlan1) entered disabled state
Thu Feb 27 21:12:22 2020 kern.info kernel: [  439.941134] device wlan1 entered promiscuous mode
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: Channel 100 (primary) not allowed for AP mode, flags: 0x10095b NO-IR RADAR
Thu Feb 27 21:12:22 2020 daemon.warn hostapd: wlan1: IEEE 802.11 Configured channel (100) not found from the channel list of current mode (2) IEEE 802.11a
Thu Feb 27 21:12:22 2020 daemon.warn hostapd: wlan1: IEEE 802.11 Hardware does not support configured channel
Thu Feb 27 21:12:22 2020 daemon.err hostapd: Could not select hw_mode and channel. (-3)
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->DISABLED
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: wlan1: AP-DISABLED
Thu Feb 27 21:12:22 2020 daemon.err hostapd: wlan1: Unable to setup interface.
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: wlan1: interface state DISABLED->DISABLED
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: wlan1: AP-DISABLED
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Thu Feb 27 21:12:22 2020 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1 wasn't started
Thu Feb 27 21:12:22 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Thu Feb 27 21:12:22 2020 kern.info kernel: [  439.957085] device wlan1 left promiscuous mode
Thu Feb 27 21:12:22 2020 kern.info kernel: [  439.966103] br-lan: port 2(wlan1) entered disabled state
Thu Feb 27 21:12:22 2020 daemon.notice netifd: radio1 (2059): cat: can't open '/var/run/wifi-phy1.pid': No such file or directory
Thu Feb 27 21:12:22 2020 daemon.notice netifd: radio1 (2059): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process  path (/proc/exe)
Thu Feb 27 21:12:22 2020 daemon.notice netifd: radio1 (2059): Command failed: Invalid argument
Thu Feb 27 21:12:22 2020 daemon.notice netifd: radio1 (2059): Device setup failed: HOSTAPD_START_FAILED
Thu Feb 27 21:12:35 2020 kern.info kernel: [  452.919482] rt3050-esw 10110000.esw: link changed 0x00
Thu Feb 27 21:12:39 2020 kern.info kernel: [  457.703818] rt3050-esw 10110000.esw: link changed 0x01
Thu Feb 27 21:12:40 2020 daemon.info dnsmasq-dhcp[1436]: DHCPREQUEST(br-lan) 192.168.1.210 40:8d:5c:b6:97:78
Thu Feb 27 21:12:40 2020 daemon.info dnsmasq-dhcp[1436]: DHCPACK(br-lan) 192.168.1.210 40:8d:5c:b6:97:78 Xeon-1220VE3
Thu Feb 27 21:12:40 2020 daemon.info dnsmasq[1436]: read /etc/hosts - 4 addresses
Thu Feb 27 21:12:40 2020 daemon.info dnsmasq[1436]: read /tmp/hosts/odhcpd - 0 addresses
Thu Feb 27 21:12:40 2020 daemon.info dnsmasq[1436]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Feb 27 21:12:40 2020 daemon.info dnsmasq-dhcp[1436]: read /etc/ethers - 0 addresses
Thu Feb 27 21:12:41 2020 daemon.info dnsmasq[1436]: read /etc/hosts - 4 addresses
Thu Feb 27 21:12:41 2020 daemon.info dnsmasq[1436]: read /tmp/hosts/odhcpd - 1 addresses
Thu Feb 27 21:12:41 2020 daemon.info dnsmasq[1436]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Thu Feb 27 21:12:41 2020 daemon.info dnsmasq-dhcp[1436]: read /etc/ethers - 0 addresses`

rimil0uz, 2020-05-14:
I went back to "original" firmware. And on my Wavlink ac600. I said "" cos could not find the firmware in wavlink website and just downloaded one I found... link: [https://www.youtube.com/watch?v=YkqO_J3EQaE](url) The original firmware is really buggy, however the 5GHZ wifi unlike than on Openwrt 19.07.2 works great...

ffrediani, 2020-05-14:
@nbd168 are you following this ? It would be great to have you input on it as author of mt76
Let me know.

amk316, 2020-05-18:
i noticed than i can connect to device after a reboot no problem, but if i reconnect there is no traffic go throw even the wifi is connected.
i also noticed my Samsung S10 do not have any problem, it connect all the time!!

rimil0uz, 2020-05-20:
Upgraded to Openwrt 19.07.3 still same problem. If i change channer radio fails to boot, if I dont (auto 36) its too weak and authentifications fails.

juhench, 2020-09-08:
The issue is also present on the Zyxel NBG6515.
openwrt 19.07.03:
```
[   12.748223] mt76x0e 0000:01:00.0: card - bus=0x1, slot = 0x0 irq=4
[   12.760867] mt76x0e 0000:01:00.0: ASIC revision: 76100002
[   12.797555] mt76x0e 0000:01:00.0: Firmware Version: 0.1.00
[   12.869106] urngd: jent-rng init failed, err: 2
[   12.946854] mt76x0e 0000:01:00.0: driver does not support default EEPROM
[   12.960368] mt76x0e: probe of 0000:01:00.0 failed with error -22
[   12.977888] kmodloader: done loading kernel modules from /etc/modules.d/*
[   21.470171] random: crng init done
```
Hardware on the Zyxel NBG6515:
MT7620A
MT7530WU (Switch)
MT7610EN (presumably handling all WiFi, 2.4 GHz, and 5 GHz?)

It is rather unfortunate, everything else seems to work.

juhench, 2020-09-17:
Zyxel: NBG6515
The problem persists on 19.07.4. Having tried all driver combinations, WLAN still does not come up. Also, copying the stock "factory" contents to the flash does not change anything. Stock MTD was saved before flashing from TFTP-booted openWRT.

rozhuk-im, 2020-12-08:
Can you test https://github.com/openwrt/mt76/pull/481 ?

g0053, 2021-01-01:
Hello everyone, facing with the same 5Ghz wifi on this device:

https://openwrt.org/toh/tp-link/tl-wr902ac_v3

wifi is working unstable, SSID is disheartening...

ffrediani, 2021-01-04:
Is mt76 still being developed ? Is MediaTek still sponsoring it ?
Anyone else besides @nbd168 looks after it ?

suppenkasper0815, 2021-01-17:
Any news to send more than 9dBm TX on MT7610e in Openwrt?

ffrediani, 2021-01-17:
Until now I also have trouble in setting up Tx Power for MT7610e as well.

desaiamogh, 2021-03-05:
Whats the status on 5Ghz for TP-Link Archer C20 v4 ??  Are the issues fixed and does it work stably on latest snapshot ?

ffrediani, 2021-03-05:
@nbd168 are you still developing it ?

sanitariu, 2021-03-05:
Still not working. Just tested latest git on a TP-Link Archer C20 version 5.
5Ghz -14 dbm which is 25 mw too low ! 5Ghz does not work / pass through walls so good as 2.4Ghz and 
distributing the signal with even lower TX-Power makes it really bad after 4 meters away.
5Ghz on mediatek (7628) is not really usable at the moment.

derkostka, 2021-04-24:
Any updates here ?

I got a 7610E in a TP-Link WR902A v3. With 19.07.7 5GHz was very unstable, as soon as you put load on the channel, it breaks.

Updated to Snapshot, now much better, but back to old problems (data loss) as soon as more then one device is connected.

Mediatek 76x0 is unusable, shall be marked somewhere....

ffrediani, 2021-04-24:
I guess nbd is not being sponsored by MediaTek anymore to develop mt76. If it was the case it wouldn't take the time is taking to fix some bugs, so he is probably doing on best effort which we know always takes quiet a while for things to move. Unless other people are maintaining mt76 holder of these hardwares are just on the best of luck really.
A possibility is to poke MediaTek to return with sponsoring for this type of development in a more continuous way.

sanitariu, 2021-06-01:
Tested today git and still the same. Archer C20 version 5 with 5G chip MT7610E not working good. TX-power is 25mw only which is strange. Doing just a little traffic make 5G stuck and clients need reconnect. Do not know what is wrong with that driver but seems like mediatek is so so low quality chip......

ffrediani, 2021-06-01:
@nbd168 are you still following this thread ? Could you point us in some direction if you have any information to share ?

sanitariu, 2021-06-07:
There is a patch for Archer C20 version 4 - https://github.com/openwrt/mt76/pull/287/commits/625619849a2b994235f345787420c3ed8a03f646
Maybe that patch broke version 5 ?
As far as i understand MT7610E needs some additional flags to be initialized so you can have normal TX power.
Also downloading full dump from Archer C20 version 4 and uploading via programmer to Archer C20 version 5 make the latest router work 5ghz normally and with 23db tx-power exactly like the original.

desaiamogh, 2021-06-08:
@sanitariu Great, So who will do a new release with these fixes ?

suppenkasper0815, 2021-06-10:
@sanitariu Can also C20 V1 and other devices with MT7610E and MT7610 chipsets contribute on this great fix? So is there a major openwrt package like MT76 to bring in that patch and let every device with this chipset get big improvement? Maybe also directly to kernel-packages in Linux?

sanitariu, 2021-06-14:
Ok i touched this file: build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2021-06-06-22b69033/mt76x0/eeprom.c

And commented / changed val value.
        /* vht mcs 8, 9 5GHz */
//      val = mt76x02_eeprom_get(dev, 0x132);
        val = 20;

txpower seems okay now or at least it is according to ./build_dir/target-mipsel_24kc_musl/wireless-regdb-2021.04.21/db.txt. If reading from wrong eeprom places gives an error lets just use regulatory database !
For my country now it is okay.
I do not know if this is legal or not for you but for my country it is 20db.
Will test how stable is 5G driver.... Last time it crashes after 20 minutes of heavy usage for example torrents+video+downloads....

sanitariu, 2021-06-14:
Txpower seems good but the device is disconnecting and doing packet losses at 5G band. Tested auto and fixed channel like 36. Encryption is sae-mixed.  Here is the crash log:

[  234.456253] ------------[ cut here ]------------
[  234.461049] WARNING: CPU: 0 PID: 639 at backports-5.10.42-1/net/mac80211/airtime.c:457 0x83655df0 [mac80211@d20ed623+0x6c600]
[  234.472528] Modules linked in: pppoe ppp_async mt76x0e mt76x0_common iptable_nat xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mt76x02_lib mt7603e mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG ts_kmp ts_fsm ts_bm slhc nfnetlink nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_raw iptable_mangle iptable_filter ip_tables crc_ccitt compat asn1_decoder nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 tun leds_gpio gpio_button_hotplug
[  234.557993] CPU: 0 PID: 639 Comm: mt76-tx phy1 Not tainted 5.4.124 #0
[  234.565215] Stack : 00000000 800572c4 804c0000 8048dae8 00000000 00000000 00000000 00000000
[  234.573773]         00000000 00000000 00000000 00000000 00000000 00000001 837fdc60 07499da9
[  234.582256]         837fdcf8 00000000 00000000 00000000 00000038 80427f64 65746e69 2e352064
[  234.590729]         00000000 0000000b 00000000 000883a9 00000000 837fdc40 00000000 83655df0
[  234.599211]         00000009 00000014 00000004 00000002 00000000 80255c24 00800317 00800357
[  234.607694]         ...
[  234.610168] Call Trace:
[  234.610183] [<800572c4>] 0x800572c4
[  234.616933] [<80427f64>] 0x80427f64
[  234.620501] [<83655df0>] 0x83655df0 [mac80211@d20ed623+0x6c600]
[  234.626564] [<80255c24>] 0x80255c24
[  234.630108] [<8000969c>] 0x8000969c
[  234.633651] [<800096a4>] 0x800096a4
[  234.637183] [<8002305c>] 0x8002305c
[  234.640728] [<83655df0>] 0x83655df0 [mac80211@d20ed623+0x6c600]
[  234.647167] [<80023104>] 0x80023104
[  234.650757] [<83655df0>] 0x83655df0 [mac80211@d20ed623+0x6c600]
[  234.656831] [<83603658>] 0x83603658 [mac80211@d20ed623+0x6c600]
[  234.662848] [<83655fd8>] 0x83655fd8 [mac80211@d20ed623+0x6c600]
[  234.668853] [<83638c4c>] 0x83638c4c [mac80211@d20ed623+0x6c600]
[  234.674870] [<836560b8>] 0x836560b8 [mac80211@d20ed623+0x6c600]
[  234.680880] [<802b7c60>] 0x802b7c60
[  234.684455] [<83552a24>] 0x83552a24 [mt76x02_lib@88149594+0x9440]
[  234.690635] [<834e048c>] 0x834e048c [mt76@fc72f307+0x8fe0]
[  234.696211] [<834e6264>] 0x834e6264 [mt76@fc72f307+0x8fe0]
[  234.701821] [<834e048c>] 0x834e048c [mt76@fc72f307+0x8fe0]
[  234.707386] [<83554fe4>] 0x83554fe4 [mt76x02_lib@88149594+0x9440]
[  234.713578] [<834e048c>] 0x834e048c [mt76@fc72f307+0x8fe0]
[  234.719227] [<834e051c>] 0x834e051c [mt76@fc72f307+0x8fe0]
[  234.724823] [<80429510>] 0x80429510
[  234.728374] [<8003e620>] 0x8003e620
[  234.731917] [<8003e4e8>] 0x8003e4e8
[  234.735460] [<8003e4e8>] 0x8003e4e8
[  234.738999] [<80004fd8>] 0x80004fd8
[  234.742546]
[  234.744054] ---[ end trace 14ae1ae2d558bf62 ]---

allhavebrainimplantsandmore, 2021-06-21:
5Ghz radio not working. The device can scan but no SSID seem to be advertised. 
Back a couple of years when I was setting up OpenWRT there was something strange in the firewall log. I think I saw there were packets generated from within the router circuitry, hitting on one of the internal interfaces coming from a subnet that I did not have set up whatsoever (like 192.168.30.something).

ffrediani, 2021-06-21:
Sorry, didn´t mean to close this. It was a mistake.

allhavebrainimplantsandmore, 2021-06-22:
> @allhavebrainimplantsandmore have you tested with OpenWrt 21.02.0-rc3 or Snapshot versions ? It seems mt76 has received a few improvments more recentlly.

Just the mt75x0E driver?

ffrediani, 2021-06-22:
No, the mt76 driver which is the one used for MT7610E

derkostka, 2021-06-22:
Hi everyone. I am running latest snapshot on a TP-Link TL-WR902AC v3 with MT7610E.

Compared to 19.07 and snapshots some weeks ago, stability is greatly improved - in fact the device is usable not for me the first time on 5 GHz.

My problem was a lot of data loss (ping timeouts, high latencies) and is is really minimized. 

- if multiple clients are connected and produce traffic, I still see data loss and increased ping times,  but is is usable now as my android devices do not disconnect anymore (which they did before due to that high lags)
- some data loss can be seen if the net is full (many clients on 5 GHz) but also here it feels more stable

So thanks, hope there is room for improvement still, let me know if you need some tests to be done.

alpha7504, 2021-06-29:
> @sanitariu Can also C20 V1 and other devices with MT7610E and MT7610 chipsets contribute on this great fix? So is there a major openwrt package like MT76 to bring in that patch and let every device with this chipset get big improvement? Maybe also directly to kernel-packages in Linux?

Hello, 

I am having the same device c20 v1, were you able to fix the issue with mt7610e?

thiagokokada, 2021-07-14:
> There is a patch for Archer C20 version 4 - [6256198](https://github.com/openwrt/mt76/commit/625619849a2b994235f345787420c3ed8a03f646)

I ported this patch to current `master` (working with commit 99a22d48f2003910f0968369b61b9131a8f53d80 from OpenWrt) and build an image for my TP-Link RE200v4. It seems to increase the stability of 5Ghz WiFi, at least with this patch applied I didn't had any disconnects yet (while without it I can see my Chromecast TV disconnect after some minutes).

If someone wants to try I think you can copy the contents of this file to `package/kernel/mt76/patches/999-mt76x0_init_hack.patch` and it should work (maybe it needs to commit also to avoid `make` resetting the branch state).

```patch
Index: mt76-2021-06-06-22b69033/mt76x0/phy.c
===================================================================
--- mt76-2021-06-06-22b69033.orig/mt76x0/phy.c
+++ mt76-2021-06-06-22b69033/mt76x0/phy.c
@@ -428,12 +428,16 @@ static void mt76x0_phy_ant_select(struct
 	u16 ee_ant = mt76x02_eeprom_get(dev, MT_EE_ANTENNA);
 	u16 ee_cfg1 = mt76x02_eeprom_get(dev, MT_EE_CFG1_INIT);
 	u16 nic_conf2 = mt76x02_eeprom_get(dev, MT_EE_NIC_CONF_2);
-	u32 wlan, coex3;
+	u32 wlan, coex3, xo1;
 	bool ant_div;
 
 	wlan = mt76_rr(dev, MT_WLAN_FUN_CTRL);
 	coex3 = mt76_rr(dev, MT_COEXCFG3);
 
+	xo1 = mt76_rr(dev, MT_XO_CTRL1);
+	xo1 &= 0xffff0000;
+	xo1 |= (u32)ee_cfg1;
+
 	ee_ant &= ~(BIT(14) | BIT(12));
 	wlan  &= ~(BIT(6) | BIT(5));
 	coex3 &= ~GENMASK(5, 2);
@@ -467,6 +471,7 @@ static void mt76x0_phy_ant_select(struct
 	mt76_rmw(dev, MT_CSR_EE_CFG1, GENMASK(15, 0), ee_cfg1);
 	mt76_clear(dev, MT_COEXCFG0, BIT(2));
 	mt76_wr(dev, MT_COEXCFG3, coex3);
+	mt76_wr(dev, MT_XO_CTRL1, xo1);
 }
 
 static void
```

**Disclaimer**: I don't have a clue what the patch above does, and this is why I am not qualified to open a PR with it.

thiagokokada, 2021-07-15:
> Ok i touched this file: build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2021-06-06-22b69033/mt76x0/eeprom.c

I also decided to patch `mt76x0/eeprom.c`, even more aggressively (if you want to test, add the contents to `package/kernel/mt76/patches/999-mt76x0_tx_val_hack.patch`):

```patch
Index: mt76-2021-06-06-22b69033/mt76x0/eeprom.c
===================================================================
--- mt76-2021-06-06-22b69033.orig/mt76x0/eeprom.c
+++ mt76-2021-06-06-22b69033/mt76x0/eeprom.c
@@ -201,7 +201,8 @@ void mt76x0_get_tx_power_per_rate(struct
 	t->stbc[6] = t->stbc[7] = s6_to_s8(val >> 8);
 
 	/* vht mcs 8, 9 5GHz */
-	val = mt76x02_eeprom_get(dev, 0x132);
+	/* val = mt76x02_eeprom_get(dev, 0x132); */
+	val = 30;
 	t->vht[8] = s6_to_s8(val);
 	t->vht[9] = s6_to_s8(val >> 8);
 
```

My `iw phy1 info` now list:

```
Frequencies:
	* 5180 MHz [36] (21.0 dBm)
	* 5200 MHz [40] (20.0 dBm)
	* 5220 MHz [44] (20.0 dBm)
	* 5240 MHz [48] (19.0 dBm)
	* 5260 MHz [52] (19.0 dBm) (radar detection)
	* 5280 MHz [56] (20.0 dBm) (radar detection)
	* 5300 MHz [60] (20.0 dBm) (radar detection)
	* 5320 MHz [64] (20.0 dBm) (radar detection)
	* 5500 MHz [100] (20.0 dBm) (radar detection)
	* 5520 MHz [104] (21.0 dBm) (radar detection)
	* 5540 MHz [108] (21.0 dBm) (radar detection)
	* 5560 MHz [112] (21.0 dBm) (radar detection)
	* 5580 MHz [116] (22.0 dBm) (radar detection)
	* 5600 MHz [120] (22.0 dBm) (radar detection)
	* 5620 MHz [124] (22.0 dBm) (radar detection)
	* 5640 MHz [128] (22.0 dBm) (radar detection)
	* 5660 MHz [132] (22.0 dBm) (radar detection)
	* 5680 MHz [136] (22.0 dBm) (radar detection)
	* 5700 MHz [140] (22.0 dBm) (radar detection)
	* 5720 MHz [144] (23.0 dBm) (radar detection)
	* 5745 MHz [149] (23.0 dBm)
	* 5765 MHz [153] (23.0 dBm)
	* 5785 MHz [157] (23.0 dBm)
	* 5805 MHz [161] (23.0 dBm)
	* 5825 MHz [165] (23.0 dBm)
	* 5845 MHz [169] (disabled)
	* 5865 MHz [173] (disabled)
```

This seems to be compliant with my current region:

```
$ iw reg get
global
country US: DFS-FCC
	(2400 - 2472 @ 40), (N/A, 30), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
	(5250 - 5350 @ 80), (N/A, 23), (0 ms), DFS, AUTO-BW
	(5470 - 5730 @ 160), (N/A, 23), (0 ms), DFS
	(5730 - 5850 @ 80), (N/A, 30), (N/A)
	(57240 - 71000 @ 2160), (N/A, 40), (N/A)
```

So it doesn't blindly trust this value, it just simply seems to be the limit? Also, the values are kindly inconsistent (channel 36~48 should have been 23 dBm for example).

cgm999, 2021-07-16:
@thiagokokada , your patch with MT_XO_CTRL1 should do the same as (MT_XO_CTRL1 and MT_CSR_EE_CFG1 are 0x0104) :
 	mt76_rmw(dev, MT_CSR_EE_CFG1, GENMASK(15, 0), ee_cfg1);

newmem12, 2021-07-25:
> @thiagokokada , your patch with MT_XO_CTRL1 should do the same as (MT_XO_CTRL1 and MT_CSR_EE_CFG1 are 0x0104) :
> mt76_rmw(dev, MT_CSR_EE_CFG1, GENMASK(15, 0), ee_cfg1);

and how to do that? can you give  full code for mt76_rmw(dev, MT_CSR_EE_CFG1, GENMASK(15, 0), ee_cfg1);  ?
Thanks! Sorry for my bad English...

thiagokokada, 2021-07-27:
> I also decided to patch `mt76x0/eeprom.c`, even more aggressively (if you want to test, add the contents to `package/kernel/mt76/patches/999-mt76x0_tx_val_hack.patch`):
> 
> ```diff
> Index: mt76-2021-06-06-22b69033/mt76x0/eeprom.c
> ===================================================================
> --- mt76-2021-06-06-22b69033.orig/mt76x0/eeprom.c
> +++ mt76-2021-06-06-22b69033/mt76x0/eeprom.c
> @@ -201,7 +201,8 @@ void mt76x0_get_tx_power_per_rate(struct
>  	t->stbc[6] = t->stbc[7] = s6_to_s8(val >> 8);
>  
>  	/* vht mcs 8, 9 5GHz */
> -	val = mt76x02_eeprom_get(dev, 0x132);
> +	/* val = mt76x02_eeprom_get(dev, 0x132); */
> +	val = 30;
>  	t->vht[8] = s6_to_s8(val);
>  	t->vht[9] = s6_to_s8(val >> 8);

Just another update, using this patch doesn't make any difference. It seems that the change are only cosmetic (I can set it to 23dBm at Luci interface, but I still get the same signal as before when the maximum was 12dBm). This issue seem similar to this one: https://github.com/openwrt/mt76/issues/216

Using OpenWrt SNAPSHOT r17217-39f81b0bf6 this is what I get:

```
VHT TX highest supported: 0 Mbps
Frequencies:
	* 5180 MHz [36] (10.0 dBm)
	* 5200 MHz [40] (9.0 dBm)
	* 5220 MHz [44] (9.0 dBm)
	* 5240 MHz [48] (8.0 dBm)
	* 5260 MHz [52] (8.0 dBm) (radar detection)
	* 5280 MHz [56] (9.0 dBm) (radar detection)
	* 5300 MHz [60] (9.0 dBm) (radar detection)
	* 5320 MHz [64] (9.0 dBm) (radar detection)
	* 5500 MHz [100] (9.0 dBm) (radar detection)
	* 5520 MHz [104] (10.0 dBm) (radar detection)
	* 5540 MHz [108] (10.0 dBm) (radar detection)
	* 5560 MHz [112] (10.0 dBm) (radar detection)
	* 5580 MHz [116] (11.0 dBm) (radar detection)
	* 5600 MHz [120] (11.0 dBm) (radar detection)
	* 5620 MHz [124] (11.0 dBm) (radar detection)
	* 5640 MHz [128] (11.0 dBm) (radar detection)
	* 5660 MHz [132] (11.0 dBm) (radar detection)
	* 5680 MHz [136] (11.0 dBm) (radar detection)
	* 5700 MHz [140] (11.0 dBm) (radar detection)
	* 5720 MHz [144] (12.0 dBm) (radar detection)
	* 5745 MHz [149] (12.0 dBm)
	* 5765 MHz [153] (12.0 dBm)
	* 5785 MHz [157] (12.0 dBm)
	* 5805 MHz [161] (12.0 dBm)
	* 5825 MHz [165] (12.0 dBm)
	* 5845 MHz [169] (disabled)
	* 5865 MHz [173] (disabled)
```

So yeah, I think the `tx_power` should be set at the EEPROM in some way, otherwise this will not work.

sanitariu, 2021-09-02:
I am using latest snapshot and with val = 23; 
I have the following:

		Frequencies:
			* 5180 MHz [36] (22.0 dBm)
			* 5200 MHz [40] (21.0 dBm)
			* 5220 MHz [44] (21.0 dBm)
			* 5240 MHz [48] (20.0 dBm)
			* 5260 MHz [52] (19.0 dBm) (radar detection)
			* 5280 MHz [56] (19.0 dBm) (radar detection)
			* 5300 MHz [60] (19.0 dBm) (radar detection)
			* 5320 MHz [64] (19.0 dBm) (radar detection)
			* 5500 MHz [100] (disabled)
			* 5520 MHz [104] (disabled)
			* 5540 MHz [108] (disabled)
			* 5560 MHz [112] (disabled)
			* 5580 MHz [116] (disabled)
			* 5600 MHz [120] (disabled)
			* 5620 MHz [124] (disabled)
			* 5640 MHz [128] (disabled)
			* 5660 MHz [132] (disabled)
			* 5680 MHz [136] (disabled)
			* 5700 MHz [140] (disabled)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (24.0 dBm)
			* 5765 MHz [153] (24.0 dBm)
			* 5785 MHz [157] (24.0 dBm)
			* 5805 MHz [161] (24.0 dBm)
			* 5825 MHz [165] (24.0 dBm)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)

Do not how to test exactly/correct if it is really okay because lowering and raising the power seems the same.
Will test stability first.

sanitariu, 2021-09-03:
Little update:
Using val = 22 gives the following:

cat /sys/kernel/debug/ieee80211/phy1/mt76/rate_txpower
       CCK:  1  1  1  1
      OFDM:  9  9  7  7  3  3  1  1
      STBC:  1  1  1  1  1  1  1  1  1  1
        HT:  7  7  5  5  3  3  1  1  1  1  1  1  1  1  1  1
       VHT:  7  7  5  5  3  3  1  1 21  1

sanitariu, 2021-09-07:
Ok seems like there is a fix for this:
https://github.com/openwrt/mt76/pull/577

lynxthecat, 2021-09-08:
Is the RE200 

> > There is a patch for Archer C20 version 4 - [6256198](https://github.com/openwrt/mt76/commit/625619849a2b994235f345787420c3ed8a03f646)
> 
> I ported this patch to current `master` (working with commit [openwrt/openwrt@99a22d4](https://github.com/openwrt/openwrt/commit/99a22d48f2003910f0968369b61b9131a8f53d80)) and build an image for my TP-Link RE200v4. It seems to increase the stability of 5Ghz WiFi, at least with this patch applied I didn't had any disconnects yet (while without it I can see my Chromecast TV disconnect after some minutes).
> 
> If someone wants to try I think you can copy the contents of this file to `package/kernel/mt76/patches/999-mt76x0_init_hack.patch` and it should work (maybe it needs to commit also to avoid `make` resetting the branch state).
> 
> ```diff
> Index: mt76-2021-06-06-22b69033/mt76x0/phy.c
> ===================================================================
> --- mt76-2021-06-06-22b69033.orig/mt76x0/phy.c
> +++ mt76-2021-06-06-22b69033/mt76x0/phy.c
> @@ -428,12 +428,16 @@ static void mt76x0_phy_ant_select(struct
>  	u16 ee_ant = mt76x02_eeprom_get(dev, MT_EE_ANTENNA);
>  	u16 ee_cfg1 = mt76x02_eeprom_get(dev, MT_EE_CFG1_INIT);
>  	u16 nic_conf2 = mt76x02_eeprom_get(dev, MT_EE_NIC_CONF_2);
> -	u32 wlan, coex3;
> +	u32 wlan, coex3, xo1;
>  	bool ant_div;
>  
>  	wlan = mt76_rr(dev, MT_WLAN_FUN_CTRL);
>  	coex3 = mt76_rr(dev, MT_COEXCFG3);
>  
> +	xo1 = mt76_rr(dev, MT_XO_CTRL1);
> +	xo1 &= 0xffff0000;
> +	xo1 |= (u32)ee_cfg1;
> +
>  	ee_ant &= ~(BIT(14) | BIT(12));
>  	wlan  &= ~(BIT(6) | BIT(5));
>  	coex3 &= ~GENMASK(5, 2);
> @@ -467,6 +471,7 @@ static void mt76x0_phy_ant_select(struct
>  	mt76_rmw(dev, MT_CSR_EE_CFG1, GENMASK(15, 0), ee_cfg1);
>  	mt76_clear(dev, MT_COEXCFG0, BIT(2));
>  	mt76_wr(dev, MT_COEXCFG3, coex3);
> +	mt76_wr(dev, MT_XO_CTRL1, xo1);
>  }
>  
>  static void
> ```
> 
> **Disclaimer**: I don't have a clue what the patch above does, and this is why I am not qualified to open a PR with it.

Is the RE200 v4 usable right now? There are a lot of people out there with this hardware that want OpenWrt to work properly on it. I have no idea why OpenWrt states this device is supported in 21.02.0 since the 5Ghz radio is unusable when I tried. See here: 

https://forum.openwrt.org/t/port-to-tp-link-re200/50540/277

sanitariu, 2021-09-08:
The fix does not fix tx-power to read correctly from eeprom. At least for Archer C20 version 5.

cgm999, 2021-10-14:
Hi, 
Is it really that other vendors have good wifi signal?

[38 more replies]