openwrt/mt76 #228: mt76x2e MCU message  timed out

ptpt52, 2019-01-09:
after hours running 5G wifi hang:

[53845.369152] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[53847.449130] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[53849.529111] mt76x2e 0000:02:00.0: MCU message 31 (seq 3) timed out
[53851.609113] mt76x2e 0000:02:00.0: MCU message 31 (seq 4) timed out
[53853.689102] mt76x2e 0000:02:00.0: MCU message 31 (seq 5) timed out
[53855.769142] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[53857.849088] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[53859.929123] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[53862.009125] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[53864.089055] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[53866.169046] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[53868.249050] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[53870.329051] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[53872.409030] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[53874.489040] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out

openwrt master HEAD code
commit: https://github.com/openwrt/openwrt/commit/634c733065256979835ca3d7218ae8203a693683

ptpt52, 2019-01-09:
device: mir3g

K900, 2019-01-10:
Same issue here on MiWiFi mini.

nbd168, 2019-01-11:
Please try the latest version

nbd168, 2019-01-12:
Thanks for testing

closed 2019-01-12 by nbd168

ptpt52, 2019-01-23:
issue again?
[ 7438.186009] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[ 7440.265766] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[ 7442.345639] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out

ptpt52, 2019-01-23:
happened on MIR3G

openwrt master HEAD code

@nbd168

K900, 2019-02-03:
Still happening for me on miwifi too :(

K900, 2019-02-04:
Been running it for 20-ish hours, seems stable so far. Will have to see if it breaks with a few days of uptime.

K900, 2019-02-07:
Seems stable still, no timeouts to be seen. Thanks :)

nbd168, 2019-02-07:
Thanks for testing

K900, 2019-02-23:
So, about 10 days of stability, and then it goes down within about 12 hours of a reboot. Might just be bad luck, but it's still the same error. Any more info I can provide to help troubleshoot?

ptpt52, 2019-02-23:
same error message on my friend's MIR3G, with the latest code build.

nbd168, 2019-02-24:
Please try the latest version of mt76.git (not pushed to OpenWrt yet)

mgurgev, 2019-02-24:
> Please try the latest version of mt76.git (not pushed to OpenWrt yet)

I will test.
I have problems with my DLink DIR-860L B1

`[16758.771223] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[16760.851224] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out`

After this error my router need manually power off/on. No ping, no internet, no ssh... Totally dead before power off/on.
5GHz is disabled, only 2,4GHz is enabled.

I have two DIR-860L B1 with same firmware(the firmware is compiled from me and updated every 1-2 weeks). Router 1 has no problems with 5GHz and 2,4GHz enabled, but router 2 have only 2,4GHz and this router have the problem. Now both 5GHz and 2,4GHz is enabled on router 2... I will test for 1-2 days.

P.S.: Sorry for my bad english.

K900, 2019-02-25:
Trying 18af219 right now, we'll see how it goes.

nbd168, 2019-02-27:
Pushed some more fixes that should hopefully make it possible to recover from these errors now.

K900, 2019-02-27:
Updated to the latest commit, no issues so far on miwifi or r6120. Will keep it running and see.