openwrt/mt76 #230: VHT-NSS 2 reported for device with 1x1:1 antenna

michaeltandy, 2019-01-09:
I have a [ASUS USB-AC51](https://wikidevi.com/wiki/ASUS_USB-AC51) using the MediaTek MT7610U, which (AFAIK) is a abgn+ac device, with a 1x1:1 antenna configuration. The advertised maximum speed is 433 Mbps, consistent with supporting a single spatial stream (at 80MHz, VHT-MCS 9 VHT-NSS 1, short guard interval)

However, when I run `wavemon` it reports `rx rate: 130.0 Mbit/s VHT-MCS 7 VHT-NSS 2 (exp: 30.5 MB/s)` i.e. more than one spatial stream.

Is the report of NSS=2 correct, despite the 1x1:1 design? Or is the report than NSS=2 incorrect?

Thanks for taking the time to read my bug report, and apologies if this is the wrong forum for me to ask this!

LorenzoBianconi, 2019-01-09:
> I have a ASUS USB-AC51 <https://wikidevi.com/wiki/ASUS_USB-AC51> using
> the MediaTek MT7610U, which (AFAIK) is a abgn+ac device, with a 1x1:1
> antenna configuration. The advertised maximum speed is [...]
Can you please provide a traffic trace using a monitor interface?

> Thanks for taking the time to read my bug report, and apologies if this is
> the wrong forum for me to ask this!

michaeltandy, 2019-01-09:
I can try :)

Do you mean you want the type of capture I'd get by running `tshark -i wlx107b44e70d1b -I -y IEEE802_11_RADIO -a duration:10 -w linklayer-a.pcapng` on the problem device? Or do you mean I should try to capture on a different device while the problem device starts up? Or are you looking for a trace of USB traffic and I've misunderstood? :)

LorenzoBianconi, 2019-01-09:
Please add a monitor interface and sniff traffic while using the sta one:
$iw phy phy0 interface add mon0 type monitor
$ifconfig mon0 up
$tcpdump -i mon0 -s0 -w /tmp/trace.pcap (or wireshark)

michaeltandy, 2019-01-10:
Sure thing, here's a capture with some traffic generated by iperf: [monitor-interface-trace-during-iperf.pcapng.gz](https://github.com/openwrt/mt76/files/2744699/monitor-interface-trace-during-iperf.pcapng.gz)

I assume you'll be looking at this header:-
```
Radiotap Header v0, Length 36
    Header revision: 0
    Header pad: 0
    Header length: 36
    Present flags
    Flags: 0x00
    Channel frequency: 5280 [A 56]
    Channel flags: 0x0140, Orthogonal Frequency-Division Multiplexing (OFDM), 5 GHz spectrum
    Antenna signal: -62dBm
    RX flags: 0x0000
    VHT information
        Known VHT information: 0x400
            .... .... .... ...0 = STBC: Absent
            .... .... .... ..0. = TXOP_PS_NOT_ALLOWED: Absent
            .... .... .... .1.. = Guard interval: Present
            .... .... .... 0... = SGI Nsym disambiguation: Absent
            .... .... ...0 .... = LDPC extra OFDM symbol: Absent
            .... .... ..0. .... = Beamformed: Absent
            .... .... .1.. .... = Bandwidth: Present
            .... .... 0... .... = Group ID: Absent
            .... ...0 .... .... = Partial AID: Absent
        .... .1.. = Guard interval: short (1)
        Bandwidth: 20 MHz (0)
        User 0: MCS 7
            0111 .... = MCS index 0: 7 (64-QAM 5/6)
            .... 0010 = Spatial streams 0: 2
            .... ...0 = Coding 0: BCC (0)
            [Data Rate: 144.4 Mb/s]
    Antenna signal: 0dBm
    Antenna: 0
```

LorenzoBianconi, 2019-01-10:
>
>
> Sure thing, here's a capture with some traffic generated by iperf:
> monitor-interface-trace-during-iperf.pcapng.gz
>
> I assume you'll be looking at this header:-
>
> Radiotap Header v0, Length 36
>     Header revision: 0
>     Header pad: 0
>     Header length: 36
>     Present flags
>     Flags: 0x00
>     Channel frequency: 5280 [A 56]
>     Channel flags: 0x0140, Orthogonal Frequency-Division Multiplexing (OFDM), 5 GHz spectrum
>     Antenna signal: -62dBm
>     RX flags: 0x0000
>     VHT information
>         Known VHT information: 0x400
>             .... .... .... ...0 = STBC: Absent
>             .... .... .... ..0. = TXOP_PS_NOT_ALLOWED: Absent
>             .... .... .... .1.. = Guard interval: Present
>             .... .... .... 0... = SGI Nsym disambiguation: Absent
>             .... .... ...0 .... = LDPC extra OFDM symbol: Absent
>             .... .... ..0. .... = Beamformed: Absent
>             .... .... .1.. .... = Bandwidth: Present
>             .... .... 0... .... = Group ID: Absent
>             .... ...0 .... .... = Partial AID: Absent
>         .... .1.. = Guard interval: short (1)
>         Bandwidth: 20 MHz (0)
>         User 0: MCS 7
>             0111 .... = MCS index 0: 7 (64-QAM 5/6)
>             .... 0010 = Spatial streams 0: 2
>             .... ...0 = Coding 0: BCC (0)
>             [Data Rate: 144.4 Mb/s]
>     Antenna signal: 0dBm
>     Antenna: 0
>
> <https://github.com/openwrt/mt76/issues/230#issuecomment-453037858>, or mute
> the thread

Yes, thx . I will look into it later today.

Regards,
Lorenzo

LorenzoBianconi, 2019-01-11:
@michaeltandy : could you please try following commit?
https://github.com/LorenzoBianconi/mt76/commit/140835499b39fed46d5597125dcc07830d040ee7

michaeltandy, 2019-01-16:
I applied this diff atop kernel v4.20, but `wavemon` still seems to report `rx rate: 117.0 Mbit/s VHT-MCS 6 VHT-NSS 2 (exp: 30.5 MB/s)`

LorenzoBianconi, 2019-01-16:
First of all thx for testing :)
Could you please try this repo
https://github.com/LorenzoBianconi/wireless-drivers-next/tree/mt76_nss_limit?files=1?
Felix did an important tx status fix. Moreover could you please provide a
traffic trace (as you did before)?

michaeltandy, 2019-01-17:
I tested with `70d181330b75` from `wireless-drivers-next` and wavemon reports `rx rate: 117.0 Mbit/s VHT-MCS 6 VHT-NSS 2 (exp: 30.5 MB/s)` - here's a sample of that data: [monitor-interface-70d181330b75.pcapng.gz](https://github.com/openwrt/mt76/files/2770103/monitor-interface-70d181330b75.pcapng.gz)

```
VHT information
    Known VHT information: 0x400
    .... .0.. = Guard interval: long (0)
    Bandwidth: 20 MHz (0)
    User 0: MCS 5
        0101 .... = MCS index 0: 5 (64-QAM 2/3)
        .... 0010 = Spatial streams 0: 2
        .... ...0 = Coding 0: BCC (0)
        [Data Rate: 104.0 Mb/s]
```

I accidentally also tested with the master branch (209b43759d65) which makes wavemon report `rx rate: 52.0 Mbit/s MCS 5 (exp: 26.1 MB/s)` i.e. removing the NSS but also the VHT.

michaeltandy, 2019-01-17:
Incidentally, when shutting down I get this on the console:
```
usb usb1: USB disconnect, device number 1
usb 1-1: USB disconnect, device number 2
usb 1-1.4: USB disconnect, device number 3
mt76x0u 1-1.4:1.0: TX DMA did not stop
mt76x0u 1-1.4:1.0: Warning: MAC TX did not stop!
mt76x0u 1-1.4:1.0: Warning: MAC RX did not stop!
mt76x0u 1-1.4:1.0: RX DMA did not stop
ci_hdrc ci_hdrc.1: USB bus 1 deregistered
```
Don't know if it's relevant? It's at shutdown, so probably not :)

LorenzoBianconi, 2019-01-17:
>
>
> I tested with 70d181330b75 from wireless-drivers-next and wavemon reports rx
> rate: 117.0 Mbit/s VHT-MCS 6 VHT-NSS 2 (exp: 30.5 MB/s) - here's a sample
> of that data: monitor-interface-70d181330b75.pcapng.gz
>
> VHT information
>     Known VHT information: 0x400
>     .... .0.. = Guard interval: long (0)
>     Bandwidth: 20 MHz (0)
>     User 0: MCS 5
>         0101 .... = MCS index 0: 5 (64-QAM 2/3)
>         .... 0010 = Spatial streams 0: 2
>         .... ...0 = Coding 0: BCC (0)
>         [Data Rate: 104.0 Mb/s]
>
> I accidentally also tested with the master branch (209b43759d65) which
> makes wavemon report rx rate: 52.0 Mbit/s MCS 5 (exp: 26.1 MB/s) i.e.
> removing the NSS but also the VHT.
>
> <https://github.com/openwrt/mt76/issues/230#issuecomment-455280201>, or mute
> the thread

Now I figured out you are talking about rx side, I was just looking at tx
one. I will work on it, thx

Regards,
Lorenzo

LorenzoBianconi, 2019-01-20:
@michaeltandy : could you please try this commit? https://github.com/LorenzoBianconi/wireless-drivers-next/commit/611062117f1037517e22c589e100abb767328cc7

michaeltandy, 2019-01-24:
With  LorenzoBianconi/wireless-drivers-next@611062117f10

`rx rate: 58.5 Mbit/s VHT-MCS 6 VHT-NSS 1 (exp: 30.5 MB/s),`

Looks good :grinning:

LorenzoBianconi, 2019-01-24:
Cool :) Could you please close the bug?

Regards,
Lorenzo

michaeltandy, 2019-01-24:
Sure thing! And if there's any way I can buy you a beer, I'd be glad to :)

closed 2019-01-24 by michaeltandy

LorenzoBianconi, 2019-01-24:
Oh, thx :)
-- 
UNIX is Sexy: who | grep -i blonde | talk; cd ~; wine; talk; touch; unzip;
touch; strip; gasp; finger; gasp; mount; fsck; more; yes; gasp; umount;
make clean; sleep