openwrt/mt76 #233: mt7612e fails to receive packets if the bssid is changed

Nieuwejaar, 2019-01-13:
I've been working on tracking down a bug in the mt76 driver, and I think I've gotten about as far as I can without a datasheet/PRM.  There are enough registers with similar names that I can't guess at their intended behavior.

My big-picture goal is to support 3 SSIDs, with a mixture of WPA-PSK and WPA-EAP, but I'm blocked by a problem with only a single SSID.

The nic I'm testing with is an mt7612e with a mac address of `fa:62:aa:50:15:2d`.  If I run hostapd with a single SSID, I can successfully connect to this nic.

To support multiple SSIDs, I need to set the bssid to `fa:62:aa:50:15:2c`.  If I change my hostapd.conf to use that bssid with only a single SSID, I can no longer connect to that SSID.  The client is reporting timeouts waiting for auth responses.  Using a wireless monitor, I can see that the packets are being sent from the client correctly.  If I instrument the mt76 driver stack, I can see that the client's probe packets are being received, but not the auth packets.

If I modify `mt76x02_add_interface()` to call `mt76x02_mac_setaddr()` when the new address doesn't match the built-in address, then the driver will receive and process the auth packets, and the client will connect successfully.

If a try to start a 2-SSID hostap instance with that change in place, clients can connect only to the first SSID - the second SSID's packets don't show up in the receive queue.

My theory (with no documentation to look at) is that the driver needs to set a register with either the number of bssids  or a mask.

Nieuwejaar, 2019-01-16:
It's possible that this and https://github.com/openwrt/mt76/issues/164 have the root cause?

WBINVD, 2019-02-01:
It should be fixed now in the latest snapshots. The WiFi mac can be changed in /etc/config/wireless under wifi-iface of the 5ghz radio using option macaddr '12:34:56:78:90:ab' or any valid mac.

Djfe, 2023-02-26:
@Nieuwejaar Are you still experiencing this issue? If not, then please close this issue :)