openwrt/mt76 #235: mt7628  - tp-link 840 version 6

sanitariu, 2019-01-17:
You stay connected but no ping to router.
After sometime it starts running. Then after some minutes again stop/start.
Also ssh packets to remote station not passing all or you have disconnects.
There is nothing in the router log files.

sanitariu, 2019-01-23:
I just tested latest and this happened again. I was connected to router and everything works perfect. I was logged remotely to a ssh server. Strange thing was that i was able to type in the ssh server but no ping to router at all and also no internet and no pings to any websites. At the same time ssh was live !?
How this can happen ? Bug is still here. I turned my device wifi on/off. Also this happens on other devices and phones. This happened again while i was trying to comment here :)

nbd168, 2019-01-25:
Please update again and test the current version.

sanitariu, 2019-01-28:
I am testing latest one. So far until now at least working as expected. Also remote ssh lagging seems fixed too. I will need more time at least 2-3 days to be sure for everything.
Thanks nbd168 for your support !

sanitariu, 2019-01-29:
I have the same problems as others reported already. Using mediatek 7628 chip. Everything working for some hours then suddenly router stop passing traffic for some minutes without disconnecting you and after it goes on again. There is nothing in the log file ! No crashes nothing.

nbd168, 2019-01-30:
I've pushed some more fixes, please test again

sanitariu, 2019-02-01:
Hello,
Testing the latest commit. Problem still the same. I am staying connected to wifi but no ping to router and outside networks. After 1-2 minutes all is okay without doing nothing. This can happen from 2 to 10 or more times per day. I am 1 meter away from the router all the time. Channel 4, 40Mhz, WPA2,  Short preamble,  Disassociate On Low Acknowledgement, Force 40MHz mode.
If i can help more just let me know.

sanitariu, 2019-02-06:
Testing each new commit. Router still stopping the same way. 1-2 times per day stop for some minutes no ping to router or outside and at the same time you stay connected. If you disconnect/connect it is working again. No need to restart.

sanitariu, 2019-02-12:
Sometimes internet disconnects can happen 5-6 times per hour which is annoying. If someone have any patches or something to test i can do it.

sanitariu, 2019-02-18:
Disabling "Allow legacy 802.11b rates" - does not help. Still sometimes you stay connected without internet. You must disable/enable your device wifi card to be able to connect again.

nbd168, 2019-02-26:
Please try the latest version

tarasian666, 2019-03-05:
same problem, helps only legacy mode. The worst situation is on iPhone.

sanitariu, 2019-03-07:
Yes no improvement at all. iPhones are working very bad. I am using 40Mhz and legacy is disabled. On MacBook i think its working better but only 1 day passed. Maybe there is something specific with qualcomm chips.

lumixen, 2019-03-20:
I'm experiencing a pretty similar issue with Mac Book Pro 2018. A connection is active, but routing does not work and pings to router do not work as well. But for me it happens very frequently - 2-4 times per 10 minutes.

lukasz1992, 2019-03-24:
@sanitariu Please check the latest version, it should be much better now.

closed 2019-06-27 by nbd168