openwrt/mt76 #236: Memory leak on mt7603

sotux, 2019-01-21:
I tested the latest version of mt7603 fc31457cd99cb85c8cea9329eedc5edd80038f29, (openwrt master branch) and found there was a memory leak problem. 
My router has 128M ram. When it started up, the free memory is abount 78M, then my android mobile phone (HuaWei honor 9) connected to the 2.4g (mt7603) and test wifi speed by the app Speedtest . After serial times, the free memory will exhausted and the luci has no response because of out of memory.
The 5g (mt7612) is no problem.

My Wifi configure is below
~~~
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
	option htmode 'HT40'
	option legacy_rates '1'
	option noscan '1'
	option channel 'auto'
	option country 'CN'
	option disabled '0'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'psk2+ccmp'
	option key '1234567890'
	option wps_pushbutton '0'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'VHT80'
	option channel 'auto'
	option country 'CN'
	option legacy_rates '1'
	option disabled '0'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'OpenWrt-5G'
	option encryption 'psk2+ccmp'
	option key '1234567890'
	option wps_pushbutton '0'
~~~

@nbd168 Thank you for your effort on this project.

sotux, 2019-01-21:
Additional test:

| reverted version | hash | memory leak |
|----------|:-------------:|:------:|
| 2019-01-10 | a946b78acfda091cd6730a54fa9dfc87f125b17d | Yes |
| 2018-12-28 | 3598046d01f1d7e9490ebf11e269323634702b05 | No |

mweinelt, 2019-01-21:
My DIR860L (7602E/7612E) leaks memory since I updated OpenWrt 18.06 past
> f34ad1a8f0 mt76: update to the latest version

So that is likely related.

[image]

nbd168, 2019-01-21:
The only commit I see that could possibly be related to this in the revision range that @sotux pointed out is commit f84b00822268767806ec9cad123cd21cce54e3b4. Could you please test if reverting that commit makes the memory leak disappear?

mweinelt, 2019-01-21:
Yeah, that looks better from what I can see so far.

[image]

mweinelt, 2019-01-22:
Yup, that's the culprit.

[image]

nbd168, 2019-01-22:
For the version that's triggering the memory leak, is the memory freed again if you restart wifi without rebooting the router?

sotux, 2019-01-22:
Hi, @nbd168, I tried to stop the wifi even remove all mt76 mods but the memory is not freed.

nbd168, 2019-01-22:
Does applying this patch on top of the latest mt76.git version fix the issue?

nbd168, 2019-01-22:
Please test the latest version from mt76.git to see if it resolves the issue.
If not, please try adding http://nbd.name/mt76-pktid.patch on top.

mweinelt, 2019-01-22:
Issue still persists on aa470d8655c6b95c9930be768e9b85d95abc0314, will now try the patch on top.

mweinelt, 2019-01-23:
I noticed that the 2.4 GHz Mesh on the 7602E is acting up again - it's back with the problems of #168. I see neighbours with good signal strength [-30,-50] dBm but I just get a lot of packet loss and high latency answers again. This happens directly after boot.

Mesh had been working quite stable with versions from mid-november 2018 (around OpenWrt commit 24ca1cda38) to the end of mid january 2019 (until OpenWrt commit f34ad1a8f0).
https://stats.darmstadt.freifunk.net/d/000000021/router-meshviewer-export?var-node=908d7823af1c&refresh=15m&orgId=1&from=now-90d&to=now-1m&panelId=18&fullscreen

```
root@64289-rocket-cloud:~# iw dev mesh1 station dump
Station 7a:fc:d9:66:55:15 (on mesh1)
	inactive time:	60 ms
	rx bytes:	19605942
	rx packets:	136102
	tx bytes:	11944
	tx packets:	133
	tx retries:	3
	tx failed:	0
	rx drop misc:	416
	signal:  	-29 [0, -29] dBm
	signal avg:	-29 [0, -29] dBm
	Toffset:	672795661185 us
	tx bitrate:	13.0 MBit/s MCS 1
	rx bitrate:	130.0 MBit/s MCS 15
	expected throughput:	8.879Mbps
	mesh llid:	15068
	mesh plid:	46114
	mesh plink:	ESTAB
	mesh local PS mode:	ACTIVE
	mesh peer PS mode:	ACTIVE
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short slot time:yes
	connected time:	2321 seconds
Station 72:f2:91:1d:c9:f9 (on mesh1)
	inactive time:	200 ms
	rx bytes:	17381760
	rx packets:	119966
	tx bytes:	8822
	tx packets:	98
	tx retries:	2
	tx failed:	0
	rx drop misc:	323
	signal:  	-50 [0, -53] dBm
	signal avg:	-50 [0, -52] dBm
	tx bitrate:	19.5 MBit/s MCS 2
	rx bitrate:	72.2 MBit/s MCS 7 short GI
	expected throughput:	13.366Mbps
	mesh llid:	37265
	mesh plid:	52202
	mesh plink:	ESTAB
	mesh local PS mode:	ACTIVE
	mesh peer PS mode:	ACTIVE
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	short slot time:yes
	connected time:	2321 seconds
[... 243 lines cut ...]
64 bytes from fe80::78fc:d9ff:fe66:5515: seq=98 ttl=64 time=1.797 ms (DUP!)
64 bytes from fe80::40f7:31ff:fe6f:6cc9: seq=98 ttl=64 time=2.048 ms (DUP!)
64 bytes from fe80::70f2:91ff:fe1d:c9f9: seq=98 ttl=64 time=3.006 ms (DUP!)
64 bytes from fe80::948d:78ff:fe23:af1c: seq=99 ttl=64 time=0.464 ms
64 bytes from fe80::78fc:d9ff:fe66:5515: seq=99 ttl=64 time=1.337 ms (DUP!)
64 bytes from fe80::40f7:31ff:fe6f:6cc9: seq=99 ttl=64 time=1.468 ms (DUP!)
64 bytes from fe80::70f2:91ff:fe1d:c9f9: seq=99 ttl=64 time=2.160 ms (DUP!)
64 bytes from fe80::948d:78ff:fe23:af1c: seq=100 ttl=64 time=0.498 ms
64 bytes from fe80::78fc:d9ff:fe66:5515: seq=100 ttl=64 time=1.029 ms (DUP!)
64 bytes from fe80::70f2:91ff:fe1d:c9f9: seq=100 ttl=64 time=1.231 ms (DUP!)
64 bytes from fe80::40f7:31ff:fe6f:6cc9: seq=100 ttl=64 time=1.599 ms (DUP!)
64 bytes from fe80::948d:78ff:fe23:af1c: seq=101 ttl=64 time=0.471 ms
64 bytes from fe80::78fc:d9ff:fe66:5515: seq=101 ttl=64 time=3.012 ms (DUP!)
64 bytes from fe80::40f7:31ff:fe6f:6cc9: seq=101 ttl=64 time=3.160 ms (DUP!)
64 bytes from fe80::70f2:91ff:fe1d:c9f9: seq=101 ttl=64 time=3.505 ms (DUP!)
^C
--- ff02::1%mesh1 ping statistics ---
102 packets transmitted, 102 packets received, 89 duplicates, 0% packet loss
round-trip min/avg/max = 0.412/4877.649/31902.706 ms
```

http://[2001:67c:2ed8:1001:928d:78ff:fe23:af1c]/cgi-bin/status
The TQ value in the table is the packet loss metric of batman-adv (in percent).

Nothing interesting in dmesg yet.

mweinelt, 2019-01-24:
I was unable to find a trigger for the memory leak so far, it just accumulates memory by just being part of the mesh.

Link Quality, the router was booted with the new image at 23:00. And weirdly enough mesh is up after an hour of uptime. 
[image]

mweinelt, 2019-01-24:
Now dmesg is full of MCU message timeouts, approximately one every 2s.

```
[56240.594530] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[56242.674505] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[56244.754507] mt76x2e 0000:02:00.0: MCU message 31 (seq 3) timed out
[56246.834501] mt76x2e 0000:02:00.0: MCU message 31 (seq 4) timed out
[56248.914497] mt76x2e 0000:02:00.0: MCU message 31 (seq 5) timed out
[56250.994502] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[56253.074491] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[56255.154453] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[56257.234440] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[56259.314451] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[56261.394432] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[56263.474433] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[56265.554413] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[56267.634399] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[56269.714403] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[56271.794383] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[56273.874390] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[56275.954393] mt76x2e 0000:02:00.0: MCU message 31 (seq 3) timed out
[56278.034352] mt76x2e 0000:02:00.0: MCU message 31 (seq 4) timed out
[56280.114342] mt76x2e 0000:02:00.0: MCU message 31 (seq 5) timed out
[56282.194348] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[56284.274340] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[56286.354337] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[56288.434337] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[56290.514309] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[56292.594303] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[56294.674292] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[56296.754284] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[56298.834301] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[56300.914246] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[56302.994260] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[56305.074251] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[56307.154232] mt76x2e 0000:02:00.0: MCU message 31 (seq 3) timed out
[56309.234254] mt76x2e 0000:02:00.0: MCU message 31 (seq 4) timed out
[56311.314192] mt76x2e 0000:02:00.0: MCU message 31 (seq 5) timed out
[56313.394189] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[56315.474180] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[56317.554171] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[56319.634164] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[56321.714154] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[56323.794137] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[56325.874141] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[56327.954132] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[56330.034123] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[56332.114109] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[56334.194091] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[56336.274077] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[56338.354069] mt76x2e 0000:02:00.0: MCU message 31 (seq 3) timed out
[56340.434140] mt76x2e 0000:02:00.0: MCU message 31 (seq 4) timed out
[56342.514052] mt76x2e 0000:02:00.0: MCU message 31 (seq 5) timed out
[56344.594043] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[56346.674037] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[56348.754026] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[56350.834063] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[56352.914022] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[56354.993996] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[56357.073999] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[56359.154010] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[56361.234012] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[56363.313954] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[56365.393943] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[56367.473940] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[56369.553940] mt76x2e 0000:02:00.0: MCU message 31 (seq 3) timed out
[56371.633929] mt76x2e 0000:02:00.0: MCU message 31 (seq 4) timed out
[56373.713951] mt76x2e 0000:02:00.0: MCU message 31 (seq 5) timed out
[56375.793951] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[56377.873895] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[... 1078 lines cut ...]
[58622.184366] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[58624.264358] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[58626.344372] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[58628.424365] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[58630.504350] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[58632.584365] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[58634.664319] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[58636.744330] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[58638.824315] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[... cut ...]
[58655.464254] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[58657.544251] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[58659.624272] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[58661.704263] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[58663.784208] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[58665.864211] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[58667.944257] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[58670.024191] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
```

nbd168, 2019-01-24:
@mweinelt, are you using batman-adv with the BATMAN_V routing protocol?
If so, please try this batman-adv patch to see if it fixes your memory leak:
http://nbd.name/batman-adv-memleak.patch

mweinelt, 2019-01-24:
@nbd168 nope, it's batman-adv v15 with BATMAN_IV protocol.

mweinelt, 2019-01-25:
At least calling `wifi` seems to have reduced the CPU load.

[image]

ecsv, 2019-01-25:
@nbd168 I have send your patch to the B.A.T.M.A.N. mailing list  -https://patchwork.open-mesh.org/patch/17751/ . I hope this is ok for you. If not, please send an adjusted patch to b.a.t.m.a.n@lists.open-mesh.org.

nbd168, 2019-01-25:
@mweinelt please show me your wireless config, I can't reproduce the memleak with mesh and batman-adv

mweinelt, 2019-01-25:
@nbd168 

```
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
	option country 'DE'
	option htmode 'VHT20'
	option channel '48'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'HT20'
	option country 'DE'
	option channel '5'
	list supported_rates '6000'
	list supported_rates '9000'
	list supported_rates '12000'
	list supported_rates '18000'
	list supported_rates '24000'
	list supported_rates '36000'
	list supported_rates '48000'
	list supported_rates '54000'
	list basic_rate '6000'
	list basic_rate '9000'
	list basic_rate '18000'
	list basic_rate '36000'
	list basic_rate '54000'

config wifi-iface 'mesh_radio0'
	option ifname 'mesh0'
	option network 'mesh_radio0'
	option device 'radio0'
	option mesh_fwding '0'
	option mesh_id 'ffda-mesh-dom1'
	option mcast_rate '12000'
	option mode 'mesh'
	option disabled '0'
	option macaddr '92:8d:78:23:af:1e'

config wifi-iface 'mesh_radio1'
	option ifname 'mesh1'
	option network 'mesh_radio1'
	option device 'radio1'
	option mesh_fwding '0'
	option mesh_id 'ffda-mesh-dom1'
	option mcast_rate '12000'
	option mode 'mesh'
	option disabled '0'
	option macaddr '96:8d:78:23:af:1c'

config wifi-iface 'client_radio0'
	option ifname 'client0'
	option network 'client'
	option disabled '0'
	option device 'radio0'
	option mode 'ap'
	option macaddr '90:8d:78:23:af:1e'
	option ssid 'darmstadt.freifunk.net'

config wifi-iface 'client_radio1'
	option ifname 'client1'
	option network 'client'
	option disabled '0'
	option device 'radio1'
	option mode 'ap'
	option macaddr '92:8d:78:23:af:1c'
	option ssid 'darmstadt.freifunk.net'
```

nbd168, 2019-01-25:
Please try the latest version from mt76.git, I pushed a change that fixes some tx status handling issues

nbd168, 2019-01-25:
Great, thanks for testing and reporting back!

closed 2019-01-25 by nbd168