openwrt/mt76 #237: ibss mode not supported

lucabianchi91, 2019-01-22:
Hi, 
we have a mt7610u module (TPLINK AC600) and it is recognized as wireless interface, but "iw" says that ibss mode is not supported. In the past we were using a modified version of myria-de (https://github.com/Myria-de/mt7610u_wifi_sta_v3002_dpo_20130916) without any problem in ibss mode. 
Is the lack of support to ibss mode a bug or is it just a feature that is currently not supported? In the latter case, there are plans to integrate ibss mode in the future?
Many Thanks

kuba-moo, 2019-01-22:
FWIW mt7601u is a different driver ;)  I'm not aware of anyone planning to add IBSS mode to it, but I may simply not know.

lucabianchi91, 2019-01-22:
Hi kuba-moo, our wifi module is mt7610u, not mt7601u

kuba-moo, 2019-01-23:
Misread, my bad :speak_no_evil:

sgruszka, 2019-01-23:
I plan to add IBSS support for mt76x02u driver.

sgruszka, 2019-01-23:
Actually I think it should be sufficient to mark IBSS as supported to make things work, like this

-               wiphy->interface_modes = BIT(NL80211_IFTYPE_STATION);
+               wiphy->interface_modes = BIT(NL80211_IFTYPE_STATION) |
+                                        BIT(NL80211_IFTYPE_ADHOC);

Could you test this ?

LorenzoBianconi, 2019-01-23:
I think the issue here (as for AP support) is the lack of tbtt interrupt support to transmit beacons. We can switch to a sw timer for it but I am little bit concerned about performances since timers according to [this](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=500462a9de65) commit are not so precise. Maybe we could use a hr timer. We can experiment :)

nbd168, 2019-01-23:
I think using hrtimer is probably the way to go.

sgruszka, 2019-01-23:
Is possible to support beaconing without TBTT interrupt, but without PS buffered frames.

LorenzoBianconi, 2019-01-23:
>
>
> Is possible to support beaconing without TBTT interrupt, but without PS
> buffered frames.
I guess hr timer is the only way to send period frames (beacons) for usb
drivers. As soon as we have hr timer we can take care of PS buffered frames
as well.

Regards,
Lorenzo

> <https://github.com/openwrt/mt76/issues/237#issuecomment-456741170>, or mute
> the thread

sgruszka, 2019-01-23:
Hardware will send beacons periodically once BCN registers and data is programmed. It can be done on mt76x02_bss_info_changed() .  But indeed hrtimer will allow to support HOST_BROADCAST_PS_BUFFERING , not sure how that feature is impotent thought.

sgruszka, 2019-01-23:
/impotent/important :-)

LorenzoBianconi, 2019-01-23:
>
>
> /impotent/important :-)
>
> <https://github.com/openwrt/mt76/issues/237#issuecomment-456754751>, or mute
> the thread

If I understood correctly, you are saying it is not necessary to reprogram
the hw every tbtt interval in order to send the beacon but we can do it
once. Are we on the same page?

jsantala, 2019-01-23:
As a side note, when I last tried, 1-2 months ago, the IBSS on mt76 was quite unusable with VoCore2 and Omega2 at least. See ticket https://github.com/openwrt/mt76/issues/184

LorenzoBianconi, 2019-01-23:
>
>
> Yes, but it will require to disable HOST_BROADCAST_PS_BUFFERING .
>
> <https://github.com/openwrt/mt76/issues/237#issuecomment-456782971>, or mute
> the thread

oh, ok. Thx for sharing this info. Maybe if we go with hr timer we can
support ap as well, do you agree?

sgruszka, 2019-01-23:
AP can be supported without hrtimer too. The benefit of hrtimer is that we can share more code between USB and MMIO and have PS buffering. But it's not clear if it will work reliably (if we will get hrtimer interrupt before HW beacon send).  Assuming if it will work reliably on USB,  I'm for it.

sgruszka, 2019-01-24:
I just tested , beaconing is indeed needed to IBSS support. I'm going to implement it...

lucabianchi91, 2019-01-24:
Hi guys,
thanks for your suggestions and sorry for the delay. We applied the following:
> Actually I think it should be sufficient to mark IBSS as supported to make things work, like this
> [mt76x02u_ibss.txt](https://github.com/openwrt/mt76/files/2786564/mt76x02u_ibss.txt)
> Could you test this ?

and 

> Yes, but it will require to disable HOST_BROADCAST_PS_BUFFERING .

and we managed to make two devices communicate, even though the maximum rate (tested with iperf) was only 10Mbps.

sgruszka, 2019-01-25:
That surprising , since it didn't work for me without beacon support. I have some more complete patchset for test here:
https://lore.kernel.org/linux-wireless/1548344649-10404-1-git-send-email-sgruszka@redhat.com/

sgruszka, 2019-01-25:
> and we managed to make two devices communicate, even though the maximum rate (tested with iperf) was only 10Mbps.

How did you configure the connection ? If by iw , the HT rates will not be used. HT require RSN  encryption so should be configured via wpa_supplicant .

jsantala, 2019-01-25:
Afaik 11 Mbps is the maximum with adhoc.

lukasz1992, 2019-01-25:
iwpriv with aes encryption? AFAIK it is not possible in mac80211 stack, maybe with Atheros/Mediatek/Metalink closed-source one.

Try with wpa_supplicant as @sgruszka said.

sgruszka, 2019-01-25:
> We used iw, without specifying ht and without encryption:
> iw dev wlan0 set type ibss
> iw dev wlan0 ibss join MYNET 5220

You need to specify channel width for HT and VHT rates, for example: 
iw dev wlan0 ibss join MYNET 5220 HT40+

Giangio1986, 2019-01-25:
> You need to specify channel width for HT and VHT rates, for example:
iw dev wlan0 ibss join MYNET 5220 HT40+

Dear sgruszka, these days with lucabianchi91 we have been testing many drivers (also multiple version of rtl8812au, rtl8814au, rtl88xxau and other mt7610u/12u) so I may be doing some confusion. I believe that HT40 was not accepted by iw but now I'm going to apply all your patches, test again, and let you know in details what we get. Thanks a lot for the help. Also let us know if you may want us to test other features.

Just one quick question/curiosity. I read the iwpriv has been deprecated but with mediatek drivers it allowed us to configure the wifi interfaces very easily with all the features we desired. Will the support private ioctl be ever added also to this version of the driver? Thanks again

sgruszka, 2019-01-25:
It is supported:

iw dev <devname> ibss join <SSID> <freq in MHz> [HT20|HT40+|HT40-|NOHT] [fixed-freq] [<fixed bssid>] [beacon-interval <TU>] [basic-rates <rate in Mbps,rate2,...>] [mcast-rate <rate in Mbps>] [key d:0:abcde]

but maybe you have to update iw tool.

iwpriv is deprecated and will not be implemented. All needed functionality is achievable via nl80211 and tools that use it like iw and wpa_supplicant.