openwrt/mt76 #238: MT76 for CE?

jjliuhk, 2019-01-24:
I am working on a IOT project which is running on MT7688. It is running openwrt with MT76 driver. I wonder how can I set it to continuous transmission mode, and set different frequency, HT bandwidth, and mode for CE test on MT76 driver?

rettichschnidi, 2019-01-27:
We faced the same problem a few weeks ago.

Everything you are asking is supported by the driver, except for the continuous transmission which we substituted by using using iperf3. To simplify the usage for the testers, we wrote our own little tool [1]. It worked quite well for us and our mt7688 based device passed the EMC tests a few weeks ago.

[1] https://github.com/husqvarnagroup/smart-garden-gateway-yocto-meta-gardena/blob/master/recipes-devtools/wifi-certification-util/wifi-certification-util/config_wireless.py

rettichschnidi, 2019-01-27:
btw: We also passed the [Wi-Fi Alliance certification](http://certifications.prod.wi-fi.org/pdf/certificate/public/download?cid=WFA81035) using commit 6203d46fcc4577065209ea0ed9334d89df4f63f7

jjliuhk, 2019-02-14:
Thx a lot. It helps so much.

BTW, did you use jtag to debug uboot? I refered to https://github.com/Angelic47/MT7688-OpenOCD/blob/master/debug.cfg but it fails to do "reset halt". I tried different combination on setting for reset but they didn't work.

rettichschnidi, 2019-02-14:
@jjliuhk Nope, I never had had to debug U-Boot myself. We hired DENX (Stefan Roese) and he took care of everything we ever wished for. Was awesome, can highly recommend!

CodeFetch, 2019-03-16:
@rettichschnidi Where can I find your U-Boot sources?

rettichschnidi, 2019-03-16:
Everything we did is upstreamed by now. This should be a good starting point: http://git.denx.de/?p=u-boot.git;a=tree;f=board/gardena/smart-gateway-mt7688;h=25c42abc996573af9dc85251bc5f419072be5a8f;hb=HEAD

jjliuhk, 2019-03-18:
> @rettichschnidi Where can I find your U-Boot sources?

You may download it by cloning it from github.
 git clone https://github.com/MediaTek-Labs/linkit-smart-uboot.git

rettichschnidi, 2019-03-18:
@jjliuhk The link you posted is based on a *very* old version of U-Boot (more than 10 yeards old). I suggest strongly to use the mainlined version for the MT7688.

Foeg, 2019-04-03:
@rettichschnidi  just wanted to contact you wrt your comment on wifi alliance certification.
I note that the certification is for MT7688.

Just wondering did you certify at the chip level or did you happen to use the the 'linkit smart 7688' module?

jjliuhk, 2019-04-03:
> @jjliuhk The link you posted is based on a _very_ old version of U-Boot (more than 10 yeards old). I suggest strongly to use the mainlined version for the MT7688.

Thank you for your advise. I am going to check it out.

CodeFetch, 2019-04-03:
@Foeg You should really think twice whether you want/need your devices to be WiFi alliance certified. A certification for a wireless chipset is not possible. The whole system including the OS needs to act compliant to their standards. So far I have not seen a device that was really compliant and not using a reduced feature set. Since reading this article: https://www.smallnetbuilder.com/wireless/wireless-features/31743-bye-bye-40-mhz-mode-in-24-ghz-part-1
I don't care about certification anymore.

Foeg, 2019-04-03:
@CodeFetch - you can certify a module and then reuse that module in products without having to recertify each product.  Or specifically you can certify a module in a product and then use that certification in other products provided you use the same module etc.

Yes - its a pain, but we need homekit certification and a prerequisite for homekit cert is a wifi alliance cert.

CodeFetch, 2019-04-03:
@Foeg Whether a device is compliant is also a question of software. I doubt that you can get a certification for a module/chipset. If you want to reuse the certification neither the hardware nor the WiFi driver part must have been changed. Many devices only get certified, because they restrict the chipset's functions in software. Actually I think that WiFi certification is only a marketing gag, but if you need it, give them some money.

rettichschnidi, 2019-04-04:
@Foeg We have our own design, including own our own PCB antenna. And yes, we also did it in order to get the HomeKit certification.

rettichschnidi, 2019-04-04:
@CodeFetch In our case the certification showed some problems in mt76 running on the mt7688 which we would not have noticed withouth it. Our product got undoubtably improved because of this testing.

Foeg, 2019-04-05:
@rettichschnidi  - wondering if you can email me on geoff@homeboy.com

rettichschnidi, 2019-05-26:
For anyone else doing the Wi-Fi Alliance certification: Their current test suite will report a false false for the test Security Validation Test "SVD-5.2.5":

> Just to sum up the issue, the tool reports an IV reutilization, but that reutilization occurs AFTER a reassoc, so a new key is used and there is no security problem on using the same IV.

Our test house talked back to the Wi-Fi Alliance which accepted and gave us a pass.