openwrt/mt76 #242: Wrong 5 Ghz primary channel set

WBINVD, 2019-02-09:
Setting a channel within an 802.11ac 80mhz block gets shifted up to the next 20mhz channel, except for the channels at the end of a 80mhz block.

For example, the default of channel 36 and 80mhz channel width results in channel 40 as the primary channel being used. Setting channel 157 results in 161 being used. Setting channels 48 and 161 work correctly. 20mhz channel widths are unaffected. Setting a channel then rebooting has no effect.

I used the web interface to set the channels. The channel I set is shown correctly in the drop-down and /etc/config/wireless. The actual channel being used (which is the next 20mhz channel) is shown in the status screens, the output of "iw dev", and the wireless clients. I am using the latest snapshot on a youhua wr1200js.

nbd168, 2019-03-04:
What OpenWrt version are you using? This does not look like an mt76 bug to me - the operating channel is controlled by mac80211, not the driver.

WBINVD, 2019-03-04:
I was using the lastest snapshot at that time from https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/ and now retested with the latest one. The bug still exists, but the behavior isn't as consistent as I originally described.

I boot with channel 36 and 80 mhz width already set and use the luci web interface to change the channel sequentially to 40, 44, 48, 149, 153, 157, 161. The bug usually appears after the second channel change. Sometimes it boots with the wrong channel set or takes at least 4 channel changes. The wrong channel is an adjacent channel, usually the next one and sometimes the lower one. I repeated this by setting to channel 36 and rebooting after the bug appears.

I also randomly changed the channels from those 8 channels with the same result.

The bug will appear eventually with enough channel changes and only channels 48 and 161 work consistently.

And unlike #255, I have no connectivity issues.

nbd168, 2019-03-10:
Please show me the output of logread from a run where it switched the primary channel

WBINVD, 2019-03-11:
Thank you for pointing out the output of logread. I feel dumb for not thinking to look there. There is no bug in mt76.

Running logread -f while changing channels reveals: `daemon.notice hostapd: Switch own primary and secondary channel to get secondary channel with no Beacons from other BSSes`. The channel selection behavior from hostapd is in the code and is run by default. Weak neighbor APs (less than -85dbm) were responsible for the randomness.

Adding option noscan '1' under wifi-device in /etc/config/wireless overrides this behavior, and I have had no problems setting and changing to a specific channel with this option set.

Apparently, manual channel selection in the luci web interface can be misleading and there is no notice about this behavior in the interface. In my opinion, it doesn't make sense to do a channel scan within 40mhz only once on boot or manual channel change when neighboring AP channels can be dynamic.

---
logread output (cleaned up and mac modified) for reference:

without neighbor bss:
```
daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
daemon.notice hostapd: wlan1: AP-DISABLED
daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
daemon.notice netifd: Network device 'wlan1' link is down
user.notice mac80211: Failed command: iw phy phy1 set antenna all all
daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
daemon.err hostapd: Using interface wlan1 with hwaddr 12:34:56:78:90:ab and ssid "test_nomap"
daemon.notice hostapd: wlan1: interface state HT_SCAN->ENABLED
daemon.notice hostapd: wlan1: AP-ENABLED
daemon.notice netifd: Network device 'wlan1' link is up
```
with neighbor bss:
```
daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
daemon.notice hostapd: wlan1: AP-DISABLED
daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
daemon.notice netifd: Network device 'wlan1' link is down
user.notice mac80211: Failed command: iw phy phy1 set antenna all all
daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->HT_SCAN
daemon.notice hostapd: Switch own primary and secondary channel to get secondary channel with no Beacons from other BSSes
daemon.err hostapd: Using interface wlan1 with hwaddr 12:34:56:78:90:ab and ssid "test_nomap"
daemon.notice hostapd: wlan1: interface state HT_SCAN->ENABLED
daemon.notice hostapd: wlan1: AP-ENABLED
daemon.notice netifd: Network device 'wlan1' link is up
```
with noscan:
```
daemon.notice hostapd: wlan1: interface state ENABLED->DISABLED
daemon.notice hostapd: wlan1: AP-DISABLED
daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
daemon.notice netifd: Network device 'wlan1' link is down
user.notice mac80211: Failed command: iw phy phy1 set antenna all all
daemon.err hostapd: Configuration file: /var/run/hostapd-phy1.conf
daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
daemon.err hostapd: Using interface wlan1 with hwaddr 12:34:56:78:90:ab and ssid "test_nomap"
daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->ENABLED
daemon.notice hostapd: wlan1: AP-ENABLED
daemon.notice netifd: Network device 'wlan1' link is up
```

closed 2019-03-11 by WBINVD