openwrt/mt76 #245: Poor performance of mt7603e with Huawei honor 9

sotux, 2019-02-12:
Hi @nbd168 , I installed an iperf3 on my mobile phone (huawei honor 9) to test the latest driver (74075efe4432b562182cb6284a833f7a780c4654) of mt7603e. The performance of mt7603e is very poor when connects to my mobile phone. But my another intel 3160 wifi adapter gets good performance when connects to mt7603e.

#### Below is iperf3 test runs on a PC and sends data to huawei honor 9

| Device |  IP |
| ---- | --- |
| huawei honor9 | 192.168.1.237 |
| PC | 192.168.1.162 |

~~~
iperf3 -c 192.168.1.237 -t 60
Connecting to host 192.168.1.237, port 5201
[  4] local 192.168.1.162 port 45978 connected to 192.168.1.237 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  5.27 MBytes  44.2 Mbits/sec    0    223 KBytes       
[  4]   1.00-2.00   sec  4.35 MBytes  36.5 Mbits/sec   34    263 KBytes       
[  4]   2.00-3.00   sec  3.79 MBytes  31.8 Mbits/sec   78    143 KBytes       
[  4]   3.00-4.00   sec  4.10 MBytes  34.4 Mbits/sec   22    167 KBytes       
[  4]   4.00-5.00   sec  1.24 MBytes  10.4 Mbits/sec   10   1.41 KBytes       
[  4]   5.00-6.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]   8.00-9.00   sec  1.93 MBytes  16.2 Mbits/sec   43   99.0 KBytes       
[  4]   9.00-10.00  sec  4.35 MBytes  36.5 Mbits/sec    0    122 KBytes       
[  4]  10.00-11.00  sec  4.35 MBytes  36.5 Mbits/sec    1    109 KBytes       
[  4]  11.00-12.00  sec  4.16 MBytes  34.9 Mbits/sec    0    136 KBytes       
[  4]  12.00-13.00  sec  4.10 MBytes  34.4 Mbits/sec    2    115 KBytes       
[  4]  13.00-14.00  sec  3.73 MBytes  31.3 Mbits/sec    2   96.2 KBytes       
[  4]  14.00-15.00  sec  3.60 MBytes  30.2 Mbits/sec   19   74.9 KBytes       
[  4]  15.00-16.00  sec  3.67 MBytes  30.8 Mbits/sec    1    107 KBytes       
[  4]  16.00-17.00  sec  3.98 MBytes  33.4 Mbits/sec    5   94.7 KBytes       
[  4]  17.00-18.00  sec  3.60 MBytes  30.2 Mbits/sec    5   82.0 KBytes       
[  4]  18.00-19.00  sec  3.73 MBytes  31.3 Mbits/sec   11    110 KBytes       
[  4]  19.00-20.00  sec  2.98 MBytes  25.0 Mbits/sec   66   87.7 KBytes       
[  4]  20.00-21.00  sec  2.55 MBytes  21.4 Mbits/sec    0    107 KBytes       
[  4]  21.00-22.00  sec  3.79 MBytes  31.8 Mbits/sec    3    127 KBytes       
[  4]  22.00-23.00  sec  1.99 MBytes  16.7 Mbits/sec    8    102 KBytes       
[  4]  23.00-24.00  sec  3.60 MBytes  30.2 Mbits/sec    0    124 KBytes       
[  4]  24.00-25.00  sec   636 KBytes  5.21 Mbits/sec    1   1.41 KBytes       
[  4]  25.00-26.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  26.00-27.00  sec   573 KBytes  4.69 Mbits/sec   52   96.2 KBytes       
[  4]  27.00-28.00  sec  2.98 MBytes  25.0 Mbits/sec    2    119 KBytes       
[  4]  28.00-29.00  sec  3.60 MBytes  30.2 Mbits/sec   69    107 KBytes       
[  4]  29.00-30.00  sec  2.49 MBytes  20.8 Mbits/sec   10    116 KBytes       
[  4]  30.00-31.00  sec  2.17 MBytes  18.2 Mbits/sec   42   1.41 KBytes       
[  4]  31.00-32.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  32.00-33.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  33.00-34.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  34.00-35.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  35.00-36.00  sec  1.37 MBytes  11.5 Mbits/sec   19    102 KBytes       
[  4]  36.00-37.00  sec  2.92 MBytes  24.5 Mbits/sec    0    122 KBytes       
[  4]  37.00-38.00  sec  0.00 Bytes  0.00 bits/sec    2   1.41 KBytes       
[  4]  38.00-39.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  39.00-40.00  sec  2.42 MBytes  20.3 Mbits/sec   45    105 KBytes       
[  4]  40.00-41.00  sec  1.68 MBytes  14.1 Mbits/sec    2   1.41 KBytes       
[  4]  41.00-42.00  sec  2.24 MBytes  18.8 Mbits/sec   94   90.5 KBytes       
[  4]  42.00-43.00  sec   445 KBytes  3.65 Mbits/sec    2   1.41 KBytes       
[  4]  43.00-44.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  44.00-45.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  45.00-46.00  sec  0.00 Bytes  0.00 bits/sec    0   96.2 KBytes       
[  4]  46.00-47.00  sec  1.43 MBytes  12.0 Mbits/sec   47   55.1 KBytes       
[  4]  47.00-48.00  sec  1.93 MBytes  16.2 Mbits/sec    1    110 KBytes       
[  4]  48.00-49.00  sec  3.60 MBytes  30.2 Mbits/sec   16    103 KBytes       
[  4]  49.00-50.00  sec  4.35 MBytes  36.5 Mbits/sec    4   91.9 KBytes       
[  4]  50.00-51.00  sec  4.23 MBytes  35.4 Mbits/sec    0    123 KBytes       
[  4]  51.00-52.00  sec   318 KBytes  2.60 Mbits/sec    3   1.41 KBytes       
[  4]  52.00-53.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  53.00-54.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  54.00-55.00  sec   509 KBytes  4.17 Mbits/sec   50   59.4 KBytes       
[  4]  55.00-56.00  sec  1018 KBytes  8.34 Mbits/sec    1   1.41 KBytes       
[  4]  56.00-57.00  sec  0.00 Bytes  0.00 bits/sec    2   1.41 KBytes       
[  4]  57.00-58.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  58.00-59.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  59.00-60.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-60.00  sec   116 MBytes  16.2 Mbits/sec  785             sender
[  4]   0.00-60.00  sec   114 MBytes  16.0 Mbits/sec                  receiver

iperf Done.
~~~
During the test, I also use tcpdump to capture the wlan0.
[Here](https://mega.nz/#!8igGHAYa!ULItVfi0hjm4EmMba3Lz-wudB054VmTtPqqO0Qv0CR8) is the capture file. Hope will be helpful.
Another thing is I run iperf3 test several times and one time the router rebooted during the test. But I didn't catch the syslog. Sorry for the inconvenience.

Thank  you.

orangepizza, 2019-02-13:
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
I wonder why it's G mode?

sotux, 2019-02-13:
> config wifi-device 'radio0'
> option type 'mac80211'
> option hwmode '11g'
> I wonder why it's G mode?

Radio0 is mt7603e, it's a 2.4g radio chip. Isn't it a G mode? I configure it via luci.

jsantala, 2019-02-13:
Afaik 11g here just defines the channel range. 11a is the other option. 11n/11ac is done via htmode option.

lukasz1992, 2019-02-15:
Does changing
`option legacy_rates '1'`
to
`option legacy_rates '0'`
improve performance?

sotux, 2019-02-19:
> @sotux Does changing
> `option legacy_rates '1'`
> to
> `option legacy_rates '0'`
> improve performance?

no, the same result.

nbd168, 2019-02-26:
Please try the latest version

nbd168, 2019-03-03:
I just pushed more fixes to OpenWrt master, things should work much better now.

sotux, 2019-03-04:
> I just pushed more fixes to OpenWrt master, things should work much better now.

Thanks, @nbd168 
I've tested 461f3b0930b0fd9972a434289d7420519634294d on openwrt master, the performance of my honor 9 mobile phone seems still poor. Below is the test result.

Device | IP
-- | --
huawei honor9 | 192.168.1.237
PC | 192.168.1.139

~~~
iperf3 -c 192.168.1.237 -t 30
Connecting to host 192.168.1.237, port 5201
[  4] local 192.168.1.139 port 56862 connected to 192.168.1.237 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  6.76 MBytes  56.7 Mbits/sec   94    109 KBytes       
[  4]   1.00-2.00   sec  5.16 MBytes  43.3 Mbits/sec    0    136 KBytes       
[  4]   2.00-3.00   sec  5.84 MBytes  49.0 Mbits/sec    0    164 KBytes       
[  4]   3.00-4.00   sec   764 KBytes  6.25 Mbits/sec    2   1.41 KBytes       
[  4]   4.00-5.00   sec  2.49 MBytes  20.9 Mbits/sec    3    167 KBytes       
[  4]   5.00-6.00   sec  3.73 MBytes  31.3 Mbits/sec  119   1.41 KBytes       
[  4]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]   9.00-10.00  sec   318 KBytes  2.61 Mbits/sec   33    153 KBytes       
[  4]  10.00-11.00  sec  3.54 MBytes  29.7 Mbits/sec  191    123 KBytes       
[  4]  11.00-12.00  sec  3.42 MBytes  28.7 Mbits/sec   11   1.41 KBytes       
[  4]  12.00-13.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  13.00-14.00  sec   127 KBytes  1.04 Mbits/sec    1    146 KBytes       
[  4]  14.00-15.00  sec  2.49 MBytes  20.9 Mbits/sec    0    161 KBytes       
[  4]  15.00-16.00  sec  5.16 MBytes  43.3 Mbits/sec   15    129 KBytes       
[  4]  16.00-17.00  sec  4.78 MBytes  40.1 Mbits/sec  104    106 KBytes       
[  4]  17.00-18.00  sec  4.10 MBytes  34.4 Mbits/sec   64   94.7 KBytes       
[  4]  18.00-19.00  sec  4.66 MBytes  39.1 Mbits/sec    0    126 KBytes       
[  4]  19.00-20.00  sec  5.34 MBytes  44.8 Mbits/sec    5    140 KBytes       
[  4]  20.00-21.00  sec  5.90 MBytes  49.5 Mbits/sec    1    175 KBytes       
[  4]  21.00-22.00  sec   954 KBytes  7.81 Mbits/sec   65   1.41 KBytes       
[  4]  22.00-23.00  sec  0.00 Bytes  0.00 bits/sec    2   1.41 KBytes       
[  4]  23.00-24.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  24.00-25.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  25.00-26.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  26.00-27.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  27.00-28.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  28.00-29.00  sec  2.86 MBytes  24.0 Mbits/sec   16    175 KBytes       
[  4]  29.00-30.00  sec  4.10 MBytes  34.4 Mbits/sec   29    122 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-30.00  sec  72.4 MBytes  20.3 Mbits/sec  759             sender
[  4]   0.00-30.00  sec  70.8 MBytes  19.8 Mbits/sec                  receiver
~~~
[Here](https://mega.nz/#!cugSEAbQ!O_zCi5Tyoz0RoFG9Je6kvJdq1bi6Y54WegKFkuRKAbY) is the capture file. I can see many TCP Out-of-Order & Retransmission when the speed down to 0kb during the test.

nbd168, 2019-03-10:
Do you have another device that you could use to create a monitor mode capture between the phone and the AP?
Also, please run the next test with the latest version from mt76.git

sotux, 2019-03-11:
> Do you have another device that you could use to create a monitor mode capture between the phone and the AP?
> Also, please run the next test with the latest version from mt76.git

Thanks, @nbd168. I haven't another same model router. Could I use Netgear wndr3800 as a monitor router? If can't, I will test the latest version first.

sotux, 2019-03-11:
Hi @nbd168, I use an Atheros AR9485 on my notebook and turned it to the monitor mode to test my router. Below is iperf3 test result on the latest mt76 version 00ac79d1af995ac2c8975c63e57a97bd4ea207d8.
~~~
iperf3 -c 192.168.1.237 -t 30
Connecting to host 192.168.1.237, port 5201
[  4] local 192.168.1.101 port 35200 connected to 192.168.1.237 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  7.56 MBytes  63.4 Mbits/sec    0    362 KBytes       
[  4]   1.00-2.00   sec  4.97 MBytes  41.7 Mbits/sec  330    214 KBytes       
[  4]   2.00-3.00   sec  5.90 MBytes  49.5 Mbits/sec    0    233 KBytes       
[  4]   3.00-4.00   sec  4.23 MBytes  35.4 Mbits/sec    1    444 KBytes       
[  4]   4.00-5.00   sec  6.34 MBytes  53.2 Mbits/sec   76    363 KBytes       
[  4]   5.00-6.00   sec  5.59 MBytes  46.9 Mbits/sec  272    199 KBytes       
[  4]   6.00-7.00   sec  1.80 MBytes  15.1 Mbits/sec   66   1.41 KBytes       
[  4]   7.00-8.00   sec   636 KBytes  5.21 Mbits/sec    3    194 KBytes       
[  4]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    0    194 KBytes       
[  4]   9.00-10.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  10.00-11.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  11.00-12.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  12.00-13.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  13.00-14.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  14.00-15.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  15.00-16.00  sec  5.28 MBytes  44.4 Mbits/sec   45    126 KBytes       
[  4]  16.00-17.00  sec   191 KBytes  1.56 Mbits/sec    2   1.41 KBytes       
[  4]  17.00-18.00  sec  2.55 MBytes  21.4 Mbits/sec    2    141 KBytes       
[  4]  18.00-19.00  sec   573 KBytes  4.69 Mbits/sec    1   1.41 KBytes       
[  4]  19.00-20.00  sec  2.61 MBytes  21.9 Mbits/sec    3    156 KBytes       
[  4]  20.00-21.00  sec  5.72 MBytes  48.0 Mbits/sec    0    180 KBytes       
[  4]  21.00-22.00  sec  4.97 MBytes  41.7 Mbits/sec    0    198 KBytes       
[  4]  22.00-23.00  sec  0.00 Bytes  0.00 bits/sec    2   1.41 KBytes       
[  4]  23.00-24.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  24.00-25.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]  25.00-26.00  sec  2.86 MBytes  24.0 Mbits/sec    3    260 KBytes       
[  4]  26.00-27.00  sec  3.79 MBytes  31.8 Mbits/sec    9   1.41 KBytes       
[  4]  27.00-28.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  28.00-29.00  sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]  29.00-30.00  sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-30.00  sec  65.5 MBytes  18.3 Mbits/sec  820             sender
[  4]   0.00-30.00  sec  62.8 MBytes  17.6 Mbits/sec                  receiver

iperf Done.
~~~
Attach file is the captured data by wireshark.
My mobile phone's wifi mac is 78:62:56:64:8e:d7 and my router's mt7603e mac is d4:5f:25:eb:58:58.
Hope this file will be helpful. I use airmon-ng to turn my wifi adapter to the monitor mode. If it's not the correct way, please let me know. I'm not familiar with this command.
[mt7603e_test.pcapng.tar.gz](https://github.com/openwrt/mt76/files/2952442/mt7603e_test.pcapng.tar.gz)

sotux, 2019-03-26:
I've tested the latest version. This bug has been fixed. This issue should be the same as https://github.com/openwrt/mt76/issues/261
Thank you very much @nbd168 .

closed 2019-03-27 by nbd168