openwrt/mt76 #246: MT7602E: Radio becomes unusable after ~3 days

mweinelt, 2019-02-12:
The 2.4 GHz Radio (MT7602E) on my DIR-860L currently becomes unusuable again after ~3 days. It is configured to use two VAPs, one for AP, one for Mesh Point.

[image]
[image]
[image]

The kernel ringbuffer only contains these MCU message timeouts:
```
[...]
[Sun Feb 10 13:15:21 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[Sun Feb 10 13:15:23 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[Sun Feb 10 13:15:25 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[Sun Feb 10 13:15:27 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[Sun Feb 10 13:15:29 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[Sun Feb 10 13:15:31 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 3) timed out
[Sun Feb 10 13:15:33 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 4) timed out
[Sun Feb 10 13:15:36 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 5) timed out
[Sun Feb 10 13:15:38 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[Sun Feb 10 13:15:40 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[Sun Feb 10 13:15:42 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[Sun Feb 10 13:15:44 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[Sun Feb 10 13:15:46 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[Sun Feb 10 13:15:48 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[Sun Feb 10 13:15:50 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[Sun Feb 10 13:15:52 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[Sun Feb 10 13:15:54 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[Sun Feb 10 13:15:56 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[Sun Feb 10 13:15:58 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[Sun Feb 10 13:16:00 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[Sun Feb 10 13:16:03 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 3) timed out
[Sun Feb 10 13:16:05 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 4) timed out
[Sun Feb 10 13:16:07 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 5) timed out
[Sun Feb 10 13:16:09 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[Sun Feb 10 13:16:11 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[Sun Feb 10 13:16:13 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[Sun Feb 10 13:16:15 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[Sun Feb 10 13:16:17 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[Sun Feb 10 13:16:19 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[Sun Feb 10 13:16:21 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[Sun Feb 10 13:16:23 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[Sun Feb 10 13:16:25 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[Sun Feb 10 13:16:28 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[Sun Feb 10 13:16:30 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[Sun Feb 10 13:16:32 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[Sun Feb 10 13:16:34 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 3) timed out
[Sun Feb 10 13:16:36 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 4) timed out
[Sun Feb 10 13:16:38 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 5) timed out
[Sun Feb 10 13:16:40 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[Sun Feb 10 13:16:42 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[Sun Feb 10 13:16:44 2019] batman_adv: bat0: Interface deactivated: mesh0
[Sun Feb 10 13:16:44 2019] batman_adv: bat0: Removing interface: mesh0
[Sun Feb 10 13:16:44 2019] device client0 left promiscuous mode
[Sun Feb 10 13:16:44 2019] br-client: port 3(client0) entered disabled state
[Sun Feb 10 13:16:44 2019] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[Sun Feb 10 13:16:44 2019] IPv6: ADDRCONF(NETDEV_UP): mesh0: link is not ready
[Sun Feb 10 13:16:44 2019] device client1 left promiscuous mode
[Sun Feb 10 13:16:44 2019] br-client: port 4(client1) entered disabled state
[Sun Feb 10 13:16:46 2019] mt76x2e 0000:02:00.0: MCU message 30 (seq 9) timed out
[Sun Feb 10 13:16:47 2019] mt76x2e 0000:02:00.0: MCU message 30 (seq 10) timed out
[Sun Feb 10 13:16:47 2019] batman_adv: bat0: Interface deactivated: mesh1
[Sun Feb 10 13:16:47 2019] batman_adv: bat0: Removing interface: mesh1
[Sun Feb 10 13:16:49 2019] mt76x2e 0000:02:00.0: MCU message 2 (seq 13) timed out
[Sun Feb 10 13:16:50 2019] mt76x2e 0000:02:00.0: MCU message 30 (seq 14) timed out
[Sun Feb 10 13:16:51 2019] mt76x2e 0000:02:00.0: MCU message 30 (seq 15) timed out
[Sun Feb 10 13:16:51 2019] IPv6: ADDRCONF(NETDEV_UP): client1: link is not ready
[Sun Feb 10 13:16:51 2019] br-client: port 3(client1) entered blocking state
[Sun Feb 10 13:16:51 2019] br-client: port 3(client1) entered disabled state
[Sun Feb 10 13:16:52 2019] device client1 entered promiscuous mode
[Sun Feb 10 13:16:53 2019] mt76x2e 0000:02:00.0: MCU message 30 (seq 1) timed out
[Sun Feb 10 13:16:54 2019] mt76x2e 0000:02:00.0: MCU message 30 (seq 2) timed out
[... 121 lines cut ...]
    37b2ad3 mt76: dfs: run mt76x02_dfs_set_domain atomically
    51b6daf mt76x2: init: set default value for MT_TX_LINK_CFG
    9661da4 mt76: add energy detect CCA support to mt76x{0,2}e drivers
    876d0e9 mt76: mac: minor optimizations in mt76x02_mac_tx_rate_val
    c78e317 mt76: dma: do not build skb if reported len does not fit in buf_size
    3598046 mt76: mmio: introduce mt76x02_check_tx_hang watchdog
    58988a3 mt76: fix signedness of rx status signal field
    bce700d mt7603: fix signal strength reporting on single-stream devices
    148219d mt7603: fix checkpatch issues
    2a092e2 mt7603: fix per-rate retry accounting
    962152b mt7603: fix WMM TXOP limit configuration
    24ec040 mt7603: fix BSSID configuration in AP mode
    48fb011 mt7603: fix CF-End transmit rate when 11b stations are connected
    9daa5ff mt76: make const array 'data' static, shrinks object size
    7d4a95c mt76: dma: avoid indirect call in mt76_dma_tx_queue_skb
    f84b008 mt76: fix tx status reporting for non-probing frames
    8167074 Revert "mt7603: update firmware to 20161027164355"
    2ad54b2 mt76: move wcid rssi ewma init to mt76 core
    d77c861 mt76: fix rssi ewma tracking
    eca96cd mt76: use proper name for __MT76x02_H macro
    d1bc504 mt76: fix building without CONFIG_LEDS_CLASS
    a946b78 mt76: add led support to mt76x0e driver
    
    Signed-off-by: Felix Fietkau <nbd@nbd.name>
```

nbd168, 2019-02-13:
What's the latest version that you tested so far?

mweinelt, 2019-02-13:
OpenWrt v18.06.2-2-g13eeee7b2b
13eeee7b2b764794dd54f39c596f94a64f47d2ca

nbd168, 2019-02-13:
I just saw that I had not pushed my latest backport commit to the stable branch yet. Please try the latest version.

mweinelt, 2019-02-13:
Mesh died after 8 hours of uptime on https://github.com/openwrt/openwrt/commit/e5ace807593fd26b8bb0df0891cecb9a891ab8d4.

[image]
[image]

The load is unaffected at this time.
[image]

```
[    0.000000] Linux version 4.14.98 (hexa@gaia.lossy.network) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7691+15-e5ace80759)) #0 SMP Fri Feb 8 20:20:20 2019
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is D-Link DIR-860L B1
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000] free_area_init_node: node 0, pgdat 805b9f40, node_mem_map 81003000
[    0.000000]   Normal zone: 256 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] random: get_random_bytes called from start_kernel+0x90/0x4ac with crng_init=0
[    0.000000] percpu: Embedded 14 pages/cpu @8110f000 s26256 r8192 d22896 u57344
[    0.000000] pcpu-alloc: s26256 r8192 d22896 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=000102c0
[    0.000000] Readback ErrCtl register=000102c0
[    0.000000] Memory: 123220K/131072K available (4607K kernel code, 240K rwdata, 1016K rodata, 268K init, 254K bss, 7852K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.015490] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.087803] pid_max: default: 32768 minimum: 301
[    0.097109] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.110127] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.125789] Hierarchical SRCU implementation.
[    0.135175] smp: Bringing up secondary CPUs ...
[    6.200157] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    6.200166] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    6.200176] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    6.200303] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.204587] Synchronize counters for CPU 1: done.
[    6.328028] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    6.328036] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    6.328044] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    6.328112] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.325098] Synchronize counters for CPU 2: done.
[    6.440714] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    6.440723] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    6.440731] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[... 236 lines cut ...]
[   86.314519] br-client: port 3(client0) entered blocking state
[   86.326054] br-client: port 3(client0) entered forwarding state
[   86.437896] IPv6: ADDRCONF(NETDEV_UP): mesh0: link is not ready
[   86.622269] IPv6: ADDRCONF(NETDEV_CHANGE): mesh0: link becomes ready
[   87.019355] IPv6: ADDRCONF(NETDEV_UP): client1: link is not ready
[   87.038900] br-client: port 4(client1) entered blocking state
[   87.050466] br-client: port 4(client1) entered disabled state
[   87.062884] device client1 entered promiscuous mode
[   87.368173] IPv6: ADDRCONF(NETDEV_CHANGE): client1: link becomes ready
[   87.381679] br-client: port 4(client1) entered blocking state
[   87.393212] br-client: port 4(client1) entered forwarding state
[   87.480852] IPv6: ADDRCONF(NETDEV_UP): mesh1: link is not ready
[   87.624425] batman_adv: bat0: Adding interface: mesh0
[   87.634577] batman_adv: bat0: Interface activated: mesh0
[   87.753118] IPv6: ADDRCONF(NETDEV_CHANGE): mesh1: link becomes ready
[   88.036608] batman_adv: bat0: Changing gw mode from: off to: client
[   88.049732] batman_adv: bat0: hop_penalty: Changing from: 30 to: 15
[   88.063581] batman_adv: bat0: multicast_mode: Changing from: enabled to: disabled
[   88.079310] batman_adv: bat0: orig_interval: Changing from: 1000 to: 5000
[   88.360729] batman_adv: bat0: Adding interface: mesh1
[   88.370915] batman_adv: bat0: Interface activated: mesh1
[   92.592780] batman_adv: bat0: IGMP Querier appeared
[   92.602496] batman_adv: bat0: MLD Querier appeared
[ 1208.589142] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
```

mweinelt, 2019-02-15:
After another reboot it just behaves weird. Starts out meh, after 7 hours it starts working quite well, after 20h hours meh again, ok again after 22 hours uptime. And now it seems to go down again after 24 hours of uptime. I have neighbouring that are meshing just fine, so there are no rf issues.

[image]
[image]

Client associations with the AP seem to work though.

[image]

```
[   28.627832] IPv6: ADDRCONF(NETDEV_CHANGE): mesh1: link becomes ready
[   29.125495] batman_adv: bat0: Adding interface: mesh0
[   29.135698] batman_adv: bat0: Interface activated: mesh0
[   29.163872] batman_adv: bat0: Adding interface: mesh1
[   29.174058] batman_adv: bat0: Interface activated: mesh1
[   32.803097] batman_adv: bat0: IGMP Querier appeared
[   32.812816] batman_adv: bat0: MLD Querier appeared
[17373.446328] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[17375.526331] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[17377.606366] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[17379.686353] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[17381.766324] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[25843.180268] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[25845.260251] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[25847.340247] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
```

mweinelt, 2019-02-16:
After 1 day of uptime on e5ace807593fd26b8bb0df0891cecb9a891ab8d4 mesh and client connections are dead. The load has appeared again at the same time the 2.4 GHz radio died. The dmesg log is full of MCU timeouts, one roughly every 2.5 seconds.

[image]
[image]
[image]

kofec, 2019-02-22:
I see the same issue

nbd168, 2019-02-24:
Please try the latest version of mt76.git (not pushed to OpenWrt yet)

mweinelt, 2019-02-25:
After 8 hours cpu load went up, mesh connections died, client connections vanished.

8c8eb9835b4a794d6e1495d5661af3427bd7e451

nbd168, 2019-02-25:
When that happens and the CPU load is up, could you please show me the number of interrupts per second for wifi and then do `echo t > /proc/sysrq-trigger` and send me the full syslog or kernel log output.

mweinelt, 2019-02-25:
```
           CPU0       CPU1       CPU2       CPU3       
 22:    7330615          0          0          0  MIPS GIC  11  mt76x2e
 23:    8811073          0          0          0  MIPS GIC  31  mt76x2e
sleep 1s
 22:    7330884          0          0          0  MIPS GIC  11  mt76x2e
 23:    8811092          0          0          0  MIPS GIC  31  mt76x2e
sleep 1s
 22:    7331142          0          0          0  MIPS GIC  11  mt76x2e
 23:    8811112          0          0          0  MIPS GIC  31  mt76x2e
sleep 1s
 22:    7331385          0          0          0  MIPS GIC  11  mt76x2e
 23:    8811132          0          0          0  MIPS GIC  31  mt76x2e
sleep 1s
 22:    7331612          0          0          0  MIPS GIC  11  mt76x2e
 23:    8811151          0          0          0  MIPS GIC  31  mt76x2e
sleep 1s
 22:    7331867          0          0          0  MIPS GIC  11  mt76x2e
 23:    8811171          0          0          0  MIPS GIC  31  mt76x2e
sleep 1s
 22:    7332114          0          0          0  MIPS GIC  11  mt76x2e
 23:    8811192          0          0          0  MIPS GIC  31  mt76x2e
sleep 1s
 22:    7332339          0          0          0  MIPS GIC  11  mt76x2e
 23:    8811212          0          0          0  MIPS GIC  31  mt76x2e
sleep 1s
 22:    7332523          0          0          0  MIPS GIC  11  mt76x2e
 23:    8811232          0          0          0  MIPS GIC  31  mt76x2e
sleep 1s
 22:    7332726          0          0          0  MIPS GIC  11  mt76x2e
 23:    8811251          0          0          0  MIPS GIC  31  mt76x2e
```

But getting the process list failed, from dmesg:
```
[48882.129603] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[48884.209597] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[48886.289609] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[48888.369609] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[48890.449616] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[48892.529692] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[48894.609628] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[48896.689661] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[48898.769631] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[48900.849644] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[48902.929621] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[48905.009637] mt76x2e 0000:02:00.0: MCU message 31 (seq 3) timed out
[48907.089625] mt76x2e 0000:02:00.0: MCU message 31 (seq 4) timed out
[48909.169637] mt76x2e 0000:02:00.0: MCU message 31 (seq 5) timed out
[48911.249624] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[48913.329620] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[48915.409643] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[48917.489639] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[48919.569645] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[48921.649643] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[48923.729651] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[48925.809631] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[48927.889682] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[48929.969670] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[48932.049692] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[48934.129674] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[48936.209659] mt76x2e 0000:02:00.0: MCU message 31 (seq 3) timed out
[48938.289679] mt76x2e 0000:02:00.0: MCU message 31 (seq 4) timed out
[48940.369680] mt76x2e 0000:02:00.0: MCU message 31 (seq 5) timed out
[48942.449653] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[48944.529651] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[48946.609656] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[48948.689683] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[48950.769667] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[48952.849668] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[... 1531 lines cut ...]
[49940.850262] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[49942.930269] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[49945.010262] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[49947.090252] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[49949.170241] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[49951.250275] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[49953.330239] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[49955.410282] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[49957.490335] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
[49959.570235] mt76x2e 0000:02:00.0: MCU message 31 (seq 15) timed out
[49961.650250] mt76x2e 0000:02:00.0: MCU message 31 (seq 1) timed out
[49963.730276] mt76x2e 0000:02:00.0: MCU message 31 (seq 2) timed out
[49965.810250] mt76x2e 0000:02:00.0: MCU message 31 (seq 3) timed out
[49967.890253] mt76x2e 0000:02:00.0: MCU message 31 (seq 4) timed out
[49969.970265] mt76x2e 0000:02:00.0: MCU message 31 (seq 5) timed out
[49972.050289] mt76x2e 0000:02:00.0: MCU message 31 (seq 6) timed out
[49974.130273] mt76x2e 0000:02:00.0: MCU message 31 (seq 7) timed out
[49976.210288] mt76x2e 0000:02:00.0: MCU message 31 (seq 8) timed out
[49978.290271] mt76x2e 0000:02:00.0: MCU message 31 (seq 9) timed out
[49980.370254] mt76x2e 0000:02:00.0: MCU message 31 (seq 10) timed out
[49982.450314] mt76x2e 0000:02:00.0: MCU message 31 (seq 11) timed out
[49984.540275] mt76x2e 0000:02:00.0: MCU message 31 (seq 12) timed out
[49986.610259] mt76x2e 0000:02:00.0: MCU message 31 (seq 13) timed out
[49988.690252] mt76x2e 0000:02:00.0: MCU message 31 (seq 14) timed out
```

mweinelt, 2019-02-25:
I updated PKG_SOURCE_VERSION to the new commit id and dropped PKG_MIRROR_HASH.

Is this the correct way to bump the package version?

The resulting packages look like this:

```
~# opkg list-installed | grep mt76
kmod-mt76-core - 4.14.101+2019-02-03-8c8eb983-1
kmod-mt76x02-common - 4.14.101+2019-02-03-8c8eb983-1
kmod-mt76x2 - 4.14.101+2019-02-03-8c8eb983-1
kmod-mt76x2-common - 4.14.101+2019-02-03-8c8eb983-1
```

nbd168, 2019-02-25:
Yes, that works.

nbd168, 2019-02-25:
I found some more bugs related to ED/CCA, which could be causing this. Please try the latest version

mweinelt, 2019-02-26:
The device has seen its first bunch of MCU message timeouts but no visible impact after 7h uptime. I'm hopeful.

lukasz1992, 2019-02-26:
> 
> 
> The device has seen its first bunch of MCU message timeouts but no visible impact after 7h uptime. I'm hopeful.
The driver now resets device after MCU message timeout, so it is now able to recover itself.

mweinelt, 2019-02-26:
> > The device has seen its first bunch of MCU message timeouts but no visible impact after 7h uptime. I'm hopeful.
> 
> The driver now resets device after MCU message timeout, so it is now able to recover itself.

Yes, I understand what the code is supposed to do. Except after ~15 hours of uptime the mesh connections died, cpu load is up, the ap still seems to accept station connections though.

nbd168, 2019-02-27:
Pushed some more fixes to mt76.git that should hopefully make it possible to recover from these errors now.

mweinelt, 2019-02-28:
No 2.4 GHz meshing ever occured with 

The load is fine, but alot more connectivity was lost.

[image]

This was what I could read from dmesg now, I certainly can get a more complete log if necessary.

```
[  440.490096]  ebtable_broute
[  440.501170]  sch_dsmark sch_teql em_cmp em_text
[  440.506734]  dm9601
[  440.521789]  sch_pie sch_sfq sch_fq sch_multiq sch_red act_skbedit act_mirred
[  440.527359]  cfg80211
[  440.536356]  em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw
[  440.540550]  cdc_subset
[  440.554736]  sch_htb sch_hfsc ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6
[  440.559276]  cdc_ether
[  440.571387]  nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables dummy ip_tunnel
[  440.576266]  cdc_eem
[  440.591665]  veth tun vfat fat ntfs nls_utf8 nls_iso8859_2
[  440.596370]  ax88179_178a
[  440.613497]  nls_iso8859_15 nls_iso8859_13 nls_iso8859_1 nls_cp852 nls_cp850 nls_cp437
[  440.617853]  asix
[  440.628764]  nls_cp1250 vxlan udp_tunnel
[  440.633977]  xt_time
[  440.649723]  ip6_udp_tunnel uas usb_storage leds_gpio xhci_mtk xhci_plat_hcd xhci_pci
[  440.653566]  xt_tcpudp
[  440.661353]  xhci_hcd sd_mod scsi_mod act_police cls_basic sch_tbf
[  440.665710]  xt_state
[  440.681279]  sch_ingress gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base
[  440.685986]  xt_quota
[  440.698271]  usb_common mii crc32c_generic
[  440.698303] CPU: 3 PID: 66 Comm: kworker/3:1 Tainted: G        W       4.14.102 #0
[  440.702801]  xt_pkttype
[  440.717429] Workqueue: events_freezable ieee80211_alloc_hw_nm [mac80211]
[  440.721953]  xt_owner
[  440.730073] Stack : 805217e4 87cf59fc 80590000 87d25100 80521878 8730351c 00000009 00000011
[  440.745160]  xt_nat
[  440.749994]         86ba1014 800702ac 87d4357c 80594947 80590000 00000001 87cf59c0 53261687
[  440.763358]  xt_multiport
[  440.767839]         00000000 00000000 00000000 00000124 00000063 6d6d6f63 6d206e6f 63206969
[  440.784480]  xt_mark
[  440.788623]         87cf57e4 805a0000 000aa79f 00000000 00000000 00000000 805c0000 8730351c
[  440.805264]  xt_mac
[  440.810443]         00000009 00000011 86ba1014 00000000 00000001 8029b7e8 0000000c 8060000c
[  440.827084]  xt_limit
[  440.831401]         ...
[  440.831410] Call Trace:
[  440.831444] [<800106c0>] show_stack+0x58/0x100
[  440.831486] [<8046290c>] dump_stack+0x9c/0xe0
[  440.848090]  xt_conntrack
[  440.852271] [<8002e948>] __warn+0xe0/0x114
[  440.868886]  xt_comment
[  440.873353] [<8002e9ac>] warn_slowpath_fmt+0x30/0x3c
[  440.878203]  xt_addrtype
[  440.883164] [<8730351c>] drv_sta_state+0xd0/0x1c4 [mac80211]
[  440.891872]  xt_TCPMSS
[  440.900579] [<87306dd8>] sta_info_move_state+0x298/0x35c [mac80211]
[  440.905727]  xt_REDIRECT
[  440.913912] [<873079a4>] sta_set_sinfo+0xa74/0xbb0 [mac80211]
[  440.918720]  xt_LOG
[  440.933609]  xt_FLOWOFFLOAD
[  440.944986] ---[ end trace dbd25181ca0e64e1 ]---
[  440.949545]  xt_CT usbserial
[  440.962015] ------------[ cut here ]------------
[  440.967033]  usbnet
[  440.978539] WARNING: CPU: 3 PID: 66 at backports-2017-11-01/net/mac80211/driver-ops.h:17 drv_sta_state+0xd0/0x1c4 [mac80211]
[  440.982620]  ums_usbat
[  440.988180] mesh1:  Failed check-sdata-in-driver check, flags: 0x0
[... 2681 lines cut ...]
[  461.808757]  nf_conntrack libcrc32c iptable_mangle iptable_filter ipheth ip_tables hso ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 compat em_nbyte sch_codel sch_prio act_ipt em_meta sch_gred sch_dsmark sch_teql em_cmp em_text sch_pie sch_sfq sch_fq sch_multiq sch_red act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_htb sch_hfsc ledtrig_usbport ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables dummy ip_tunnel veth tun vfat fat ntfs nls_utf8 nls_iso8859_2 nls_iso8859_15 nls_iso8859_13 nls_iso8859_1 nls_cp852 nls_cp850 nls_cp437 nls_cp1250 vxlan udp_tunnel ip6_udp_tunnel uas usb_storage leds_gpio xhci_mtk
[  461.949460]  xhci_plat_hcd xhci_pci xhci_hcd sd_mod scsi_mod act_police cls_basic sch_tbf sch_ingress gpio_button_hotplug ext4 jbd2 mbcache usbcore nls_base usb_common mii crc32c_generic
[  461.982657] CPU: 3 PID: 66 Comm: kworker/3:1 Tainted: G        W       4.14.102 #0
[  461.997840] Workqueue: events_freezable ieee80211_alloc_hw_nm [mac80211]
[  462.011171] Stack : 805217e4 87cf5bc4 80590000 87d25100 80521878 87303414 00000009 00000011
[  462.027803]         00000000 800702ac 87d4357c 80594947 80590000 00000001 87cf5b88 53261687
[  462.044432]         00000000 00000000 00000000 0000c450 00000000 00002305 00000007 00000000
[  462.061059]         00000000 805a0000 000f39d0 00000000 00000000 00000000 805c0000 87303414
[  462.077687]         00000009 00000011 00000000 00000001 00000003 8029b7e8 0000000c 8060000c
[  462.094316]         ...
[  462.099173] Call Trace:
[  462.104051] [<800106c0>] show_stack+0x58/0x100
[  462.112908] [<8046290c>] dump_stack+0x9c/0xe0
[  462.121575] [<8002e948>] __warn+0xe0/0x114
[  462.129719] [<8002e9ac>] warn_slowpath_fmt+0x30/0x3c
[  462.139668] [<87303414>] drv_remove_interface+0x44/0x7c [mac80211]
[  462.152010] [<87317600>] ieee80211_add_virtual_monitor+0x988/0xa5c [mac80211]
[  462.166383] ---[ end trace dbd25181ca0e64fc ]---
[  462.378246] batman_adv: bat0: Removing interface: mesh1
[55224.822227] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[55224.833198] mt76x2e 0000:01:00.0: Build: 1
[55224.841381] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[55224.870477] mt76x2e 0000:01:00.0: Firmware running!
[55224.890717] ieee80211 phy0: Hardware restart was requested
```

mweinelt, 2019-02-28:
Also here is the output of logread which might be helpful in this case.

```
Thu Feb 28 13:01:27 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:01:33 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:01:39 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:01:45 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:01:52 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:01:58 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:02:04 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:02:10 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:02:16 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:02:22 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:02:28 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:02:34 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:02:40 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:02:46 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:02:52 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:02:58 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:03:04 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:03:10 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:03:16 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:03:22 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:03:28 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:03:34 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:03:40 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:03:46 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:03:52 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:03:58 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:04:04 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:04:10 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:04:16 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:04:22 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:04:28 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:04:34 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:04:40 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:04:46 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:04:52 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:04:58 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:05:04 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:05:10 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:05:16 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:05:22 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:05:28 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:05:34 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:05:40 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:05:46 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:05:52 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:05:58 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:06:04 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:06:10 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:06:16 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:06:22 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:06:28 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:06:34 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:06:40 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:06:46 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:06:52 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:06:58 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:07:04 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:07:10 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:07:16 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:07:22 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:07:28 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:07:34 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:07:40 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:07:46 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:07:52 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:07:58 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 13:08:04 2019 daemon.err hostapd: Failed to set beacon parameters
[... 855 lines cut ...]
Thu Feb 28 14:31:06 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 14:31:12 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 14:31:18 2019 daemon.err hostapd: Failed to set beacon parameters
[... cut ...]
Thu Feb 28 14:32:43 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 14:32:49 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 14:32:54 2019 authpriv.info dropbear[30539]: Child connection from 2001:67c:2ed8:1001:520c:3615:22b7:147f:56886
Thu Feb 28 14:32:55 2019 daemon.err hostapd: Failed to set beacon parameters
Thu Feb 28 14:32:55 2019 authpriv.notice dropbear[30539]: Pubkey auth succeeded for 'root' with key sha1!! 14:7b:8b:69:9b:b4:4e:a1:34:46:26:e3:a6:38:4a:32:de:04:1e:c1 from 2001:67c:2ed8:1001:520c:3615:22b7:147f:56886
```

nbd168, 2019-02-28:
Is that with 18.06 or master?

mweinelt, 2019-02-28:
OpenWrt 18.06 1be6ff6dc0ab53d2191c49c32cae7f6f9aadd14d
mt76 fa8340696f6135afc1895611080f628097570355

nbd168, 2019-02-28:
Could you please try this with OpenWrt master? The mac80211 version might make a difference here

nbd168, 2019-02-28:
I just pushed another commit that should help with mesh connection stability

mweinelt, 2019-02-28:
The router disappeared after flashing OpenWrt master, I'll have to recover it later tonight or tomorrow.

mweinelt, 2019-03-01:
Complete serial log with several stacktraces and the result that mesh1 is removed from bat0
https://gist.github.com/mweinelt/67754713ce6b879525fadede355c6567

openwrt https://github.com/openwrt/openwrt/commit/994428f3950c70bc92e5e6b68e784d8d660ca4ea
mt76 ead881b550e94fabdf48d35916be26ad576b00a9

nbd168, 2019-03-01:
I pushed some mac80211 fixes to openwrt and some more mt76 fixes to mt76.git. It should work much better now.

mweinelt, 2019-03-01:
Not quite yet. https://gist.github.com/mweinelt/e6beb6289b2f6162038fde5d61ee9a46

openwrt https://github.com/openwrt/openwrt/commit/c6caa7a27a38929f6d7e76795df6c3dbba7d7351
mt76 2c62d03d69f5e20f1129041382289b0865afabaf

nbd168, 2019-03-01:
Please show me the output of ifconfig and the contents of /etc/config/wireless.

nbd168, 2019-03-01:
Before the error occurs

mweinelt, 2019-03-01:
ifconfig
```
bat0      Link encap:Ethernet  HWaddr 90:8D:78:23:AF:1C  
          inet6 addr: fe80::928d:78ff:fe23:af1c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:635 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:19 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:42446 (41.4 KiB)  TX bytes:0 (0.0 B)

br-client Link encap:Ethernet  HWaddr 90:8D:78:23:AF:1C  
          inet6 addr: fd01:67c:2ed8:1001:928d:78ff:fe23:af1c/64 Scope:Global
          inet6 addr: fe80::928d:78ff:fe23:af1c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:11 errors:0 dropped:0 overruns:0 frame:0
          TX packets:28 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:868 (868.0 B)  TX bytes:3251 (3.1 KiB)

br-wan    Link encap:Ethernet  HWaddr 66:3A:05:3B:00:30  
          inet6 addr: fe80::643a:5ff:fe3b:30/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:4733 (4.6 KiB)

client0   Link encap:Ethernet  HWaddr 90:8D:78:23:AF:1E  
          inet6 addr: fe80::928d:78ff:fe23:af1e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:2216 (2.1 KiB)

client1   Link encap:Ethernet  HWaddr 92:8D:78:23:AF:1C  
          inet6 addr: fe80::908d:78ff:fe23:af1c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:27 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:2680 (2.6 KiB)

eth0      Link encap:Ethernet  HWaddr 3E:D7:4D:DD:48:10  
          inet6 addr: fe80::3cd7:4dff:fedd:4810/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:86 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:15004 (14.6 KiB)
          Interrupt:20 

eth0.1    Link encap:Ethernet  HWaddr 90:8D:78:23:AF:1C  
          inet6 addr: fe80::928d:78ff:fe23:af1c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:50 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:7254 (7.0 KiB)

eth0.2    Link encap:Ethernet  HWaddr 90:8D:78:23:AF:1F  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:4733 (4.6 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
[... 102 lines cut ...]
        option mesh_fwding '0'
        option mesh_id 'ffda-mesh-dom1'
        option mcast_rate '12000'
        option mode 'mesh'
        option disabled '0'
        option macaddr '96:8d:78:23:af:1c'

config wifi-iface 'client_radio0'
        option ifname 'client0'
        option network 'client'
        option disabled '0'
        option device 'radio0'
        option mode 'ap'
        option macaddr '90:8d:78:23:af:1e'
        option ssid 'darmstadt.freifunk.net'

config wifi-iface 'client_radio1'
        option ifname 'client1'
        option network 'client'
        option disabled '0'
        option device 'radio1'
        option mode 'ap'
        option macaddr '92:8d:78:23:af:1c'
        option ssid 'darmstadt.freifunk.net'
```

nbd168, 2019-03-01:
As I suspected, I couldn't reproduce your issue because it was triggered by your use of the macaddr option. I've pushed another change that should resolve this properly

mweinelt, 2019-03-01:
Yup, the stacktraces are gone. I'll be watching graphs again until something interesting happens (or not). Thanks so far!

mweinelt, 2019-03-01:
The wireless mesh connections, the node is has no wired connection, seem to be quite lossy now, but the router is otherwise stable. Due to the loss it fails to report metrics (see graph below) and with mtr I get roughly 40% loss over several minutes.

The node is very close (30cm) to the node it was meshing with since I attached the serial, I wouldn't think that this could influence the mesh connection in such a negative way.

[image]
[image]

nbd168, 2019-03-02:
Please try the latest mt76 version, I've pushed a fix for ED/CCA that might help

nbd168, 2019-03-03:
I just pushed more fixes to OpenWrt master, things should work much better now.

mweinelt, 2019-03-04:
Up 12 hours and everything looks pretty good. Packet loss is gone, metrics arrive properly, CPU isn't under pressure, clients and mesh connections work and are stable.

rotanid, 2019-03-04:
nice, let's hope this state persists, as your original report was about problems after ~3d

mweinelt, 2019-03-04:
Several restarts on both phy but no visible impact yet.

```
[  105.203738] batman_adv: bat0: MLD Querier appeared
[ 8852.174766] IPv6: ADDRCONF(NETDEV_UP): tmp.mesh0: link is not ready
[ 8855.494708] IPv6: ADDRCONF(NETDEV_UP): tmp.mesh1: link is not ready
[14943.773000] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[14943.783995] mt76x2e 0000:02:00.0: Build: 1
[14943.792211] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[14943.821001] mt76x2e 0000:02:00.0: Firmware running!
[14943.841086] ieee80211 phy1: Hardware restart was requested
[19651.891798] IPv6: ADDRCONF(NETDEV_UP): tmp.mesh0: link is not ready
[19655.231841] IPv6: ADDRCONF(NETDEV_UP): tmp.mesh1: link is not ready
[20910.087084] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[20910.099697] mt76x2e 0000:01:00.0: Build: 1
[20910.108713] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[20910.135202] mt76x2e 0000:01:00.0: Firmware running!
[20910.155333] ieee80211 phy0: Hardware restart was requested
[30451.654438] IPv6: ADDRCONF(NETDEV_UP): tmp.mesh0: link is not ready
[30454.946915] IPv6: ADDRCONF(NETDEV_UP): tmp.mesh1: link is not ready
[38317.740491] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
[38317.751487] mt76x2e 0000:02:00.0: Build: 1
[38317.759722] mt76x2e 0000:02:00.0: Build Time: 201507311614____
[38317.788315] mt76x2e 0000:02:00.0: Firmware running!
[38317.808431] ieee80211 phy1: Hardware restart was requested
[41251.437426] IPv6: ADDRCONF(NETDEV_UP): tmp.mesh0: link is not ready
[41254.817340] IPv6: ADDRCONF(NETDEV_UP): tmp.mesh1: link is not ready
[50716.950962] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
[50716.962274] mt76x2e 0000:01:00.0: Build: 1
[50716.970605] mt76x2e 0000:01:00.0: Build Time: 201507311614____
[50716.999245] mt76x2e 0000:01:00.0: Firmware running!
[50717.019820] ieee80211 phy0: Hardware restart was requested
[52051.150334] IPv6: ADDRCONF(NETDEV_UP): tmp.mesh0: link is not ready
[52059.712120] IPv6: ADDRCONF(NETDEV_UP): tmp.mesh1: link is not ready
```

mweinelt, 2019-03-06:
Yep, looks stable three days in. Thanks a ton!

nbd168, 2019-03-07:
I will look into it. I probably need to backport some mac80211 fixes for that.

nbd168, 2019-03-07:
Done

closed 2019-03-07 by nbd168