openwrt/mt76 #247: 5g is very slow related to beacon

dimatto, 2019-02-15:
mt76x2 on mir3g, everytime flashing a new image , 5g wifi is very slow, by restarting wifi, it back to normal, and after running sometime, it turns slow again. After capturing packets, I found that beacon with right tim bit set can't awake sleeping client, seems like beacon time drift problem, there's no this kind of problem after beacon drift patch and before adding usb code .

dimatto, 2019-02-23:
This problem is not related to time drift, but still related to beacon. I found another way to reproduce it. Just increase  beacon interval from luci, for example, increase from 100 to 200, or from 50 to 100, save&apply, tim can’t awake client any more, restart wifi will let it back to normal. Decreasing beacon interval has no problem. I think something with same effect happen when reflashing or running for some time, so cause same problem.

nbd168, 2019-02-23:
Please try the latest version from mt76.git

dimatto, 2019-02-24:
I tried latest version, the problem still remain.
I try to compile older commits, and find the problem is commit 5bca2de. This commit move beacon interval setting after beacon enable setting, by puting beacon interval setting before enable, the whole problem get fixed. It is not necessary to change MT_BEACON_TIME_CFG_SYNC_MODE to fix this problem.

dimatto, 2019-02-24:
Another problem, by reading original beacon timestamp work-around code, it caculate offset before 
TbttTickCount++, so the offset of adjust beacon is 62, but your code use tbtt_count after tbtt_count++, so the tbtt_count of adjust beacon is 63. So, every 64 beacon still defer 1us.

nbd168, 2019-02-24:
Thanks a lot for tracking this down. I've pushed fixes for both issues, please test the latest version.

dimatto, 2019-02-24:
I'm now testing my own code for long time running, basically your fix code is same as mine, so it's same as testing your code.
But I still has one question, if beacon is disabled for some reason, before it's  reenabled again, will mt76x02_bss_info_changed() called with BSS_CHANGED_BEACON_INT set ?  If not, tbtt_count must be reset to 0 when beacon is being disabled, or it will cause same problem.

nbd168, 2019-02-24:
When an interface is disabled/enabled, it will always issue a call with both BSS_CHANGED_BEACON_ENABLED and BSS_CHANGED_BEACON_INT set.
I think the only scenario where this could become an issue is if one AP interface is brought down and back up while another is left running.
Even then, I don't expect it to make any real difference in practice, but I pushed a fix for it anyway.

dimatto, 2019-02-24:
Yes, that’s my scenario, I always create some guest AP interfaces and disable them, only enable them when needed, when enabling  them, will it reset tbtt_count of other running interface ?
Another question, will beacon stop or disable for some reason while an interface still enabled ? Then tbtt_count need reset at that time.

nbd168, 2019-02-24:
We only need to reset the count when enabling beacons after the beacon timer has been stopped. That's what my commit does.

dimatto, 2019-02-24:
I see, I’ll test it

dimatto, 2019-02-25:
There’s no problem for one day, I think it’s ok.

nbd168, 2019-02-25:
Thanks for testing

closed 2019-02-25 by nbd168