openwrt/mt76 #248: mt7603e zbt-wg3526-32m ipv6 multi-cast packets noticeable drop

0x5a17ed, 2019-02-15:
I'm observing a problem with the open mt7603e driver in combination with IPv6 multi-cast packets on a zbt-wg3526-32m. It seems that there is a noticeable drop in multi-cast packets over the air. It also seems that uni-cast packets are just fine.

Running `ping fe80::9699:fe5c:626d:aa58%br-guest` to ping the given host just runs fine. All echo requests are answered properly.

Running `ping -6 ff02::1%br-guest | grep fe80::9699:fe5c:626d:aa58` at the same time yields periods of longer silence. Running wireshark on the other system shows that these echo requests never reach the host. This is problematic because it also affects router advertisement packets. This leads to flaky IPv6 connectivity.

Tested with the state of <https://github.com/openwrt/openwrt/commit/420376ab50b1>.

Unfortunately I don't know how to further isolate the root of the issue so tell me please, if I can be of any more help here.

0x5a17ed, 2019-02-24:
The noticeable drop was also observed with IPv4 broadcast packages. It seems that this is a generic problem with non unicast packages across IPv4 and IPv6. Adjusting the title.

lukasz1992, 2019-03-27:
@0x5a17ed Could you check it again with mt76 master branch?

closed 2019-06-27 by nbd168

0x5a17ed, 2019-06-27:
We're currently testing this again. So far it looks good. Sorry for the late response.