openwrt/mt76 #250: TP-Link Archer T2U - mt7610u kernel crash

tluciomiranda, 2019-02-19:
Hi,

Running OpenWrt SNAPSHOT r9320-9485ea7 on a Raspberry Pi 3B+ makes the system crash followed by reboot when connecting a T2U based on mt7610u chip. It does not matter if I connect before or after boot, as the system always crashes. Please see crashlog attached.

Regards

[crash.txt](https://github.com/openwrt/mt76/files/2882255/crash.txt)

LorenzoBianconi, 2019-02-19:
> Hi,
>
> Running OpenWrt SNAPSHOT r9320-9485ea7 on a Raspberry Pi 3B+ makes the
> system crash followed by reboot when connecting a T2U based on mt7610u
> chip. It does not matter if I connect before or [...]

Hi,

the issue should be already fixed here: https://github.com/nbd168/wireless

Regards,
Lorenzo

tluciomiranda, 2019-02-19:
Right, thank you for your response.
But, can you please enlighten me on how/when those fixes will arrive on main OpenWRT mt76x0u kmods ?

Regards

tluciomiranda, 2019-02-24:
After trying a bit to compile things, managed to compile a new image with updated mt76 drivers.
Now there's a different crash.
Please see attached log.

Regards
[crash.txt](https://github.com/openwrt/mt76/files/2898572/crash.txt)

LorenzoBianconi, 2019-02-25:
@tluciomiranda : are you running latest mac80211? The crash log remember me an old mac80211 issue:
a5ae326418b353301c6acc787a8988782e1762f6 ('mac80211: fix issue with possible txq NULL pointer')

closed 2019-03-10 by nbd168