openwrt/mt76 #252: The reason of my 2.4g lag

dimatto, 2019-02-26:
My 2.4g is mt7603e on mir3g. For long time it's laggy. I compiled older commits to find reason, and the reason is commit 9d009be. I can't figure out what's wrong with that commit, for now, I simply comment mt7603_false_cca_check() to wipe out  lag.

nbd168, 2019-02-26:
Thanks for tracking this down. Could you please show me a few snapshots of the contents of the "radio" debugfs file while it's lagging?

dimatto, 2019-02-26:
now I'm running code without mt7603_false_cca_check() and lag, I remember when it's lagging Sensitivity is -72 while my signal is -46 displayed at luci , not clear about ofdm and cck, if you need more accurate data, I'll try it tomorrow.

nbd168, 2019-02-26:
I’ve pushed a change to disable this by default. I found out that the vendor driver also does not enable this code

dimatto, 2019-02-27:
ok, it's fine.

closed 2019-02-27 by dimatto