openwrt/mt76 #253: 2.4 ghz mir3g error

9000000, 2019-02-27:
I tried a lot of different settings but it was still unstable, the connection was always interrupted or if there was a connection there was no internet, sometimes I saw the message: disconnected due to excessive missing ACKs. everything is only stable when I switch to legacy mode

nbd168, 2019-03-02:
What's the last version that you tried? Did you try fd315bd0f2761b277e09f9142eb0b8f9295f71a0 or newer?

9000000, 2019-03-02:
@nbd168  I always update the latest test versions of the homepage 
https://git.openwrt.org/?p=openwrt/openwrt.git;a=summary

fabsys, 2019-03-02:
Just update with the latest Snapshoot (https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/) the commit fd315bd is included.

root@OpenWrt:~# opkg list-installed|grep mt76
kmod-mt76-core - 4.14.103+2019-02-26-fd315bd0-1
kmod-mt7603 - 4.14.103+2019-02-26-fd315bd0-1
kmod-mt76x02-common - 4.14.103+2019-02-26-fd315bd0-1
kmod-mt76x2 - 4.14.103+2019-02-26-fd315bd0-1
kmod-mt76x2-common - 4.14.103+2019-02-26-fd315bd0-1

nbd168, 2019-03-03:
I just pushed more fixes to OpenWrt master, things should work much better now.

9000000, 2019-03-04:
I have updated to the latest update, the auto-disconnect error still occurs, my device always has more than 15 connections continuously

[log.txt](https://github.com/openwrt/mt76/files/2924810/log.txt)

nbd168, 2019-03-04:
What kind of client is this? Also, please show me your /etc/config/wireless

9000000, 2019-03-04:
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
	option country '00'
	option noscan '1'
	option channel '11'
	option legacy_rates '0'
	option txpower '22'
	option htmode 'HT40'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'Quick_Snow'
	option encryption 'psk2'
	option key '6666688888'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '36'
	option hwmode '11a'
	option path 'pci0000:00/0000:00:01.0/0000:02:00.0'
	option htmode 'VHT80'
	option country '00'
	option legacy_rates '0'
	option __toggle 'Enable'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'Quick_Snow_2'
	option encryption 'psk2'
	option key '6666688888'

nbd168, 2019-03-04:
Is it more stable if you disable HT40 on 2.4 GHz? Also, what kind of client is triggering these issues?

9000000, 2019-03-04:
Every connection to my router is a smartphone, a TV, an Android box, all of which are wireless connections, there are no links across the network.

9000000, 2019-03-04:
I tried it off but the result is still the same

Every time there is a new update I always try all the settings that do not exclude any settings, any errors I know right away

9000000, 2019-03-12:
@nbd168 please push the latest fixes to openwrt master

r0ck3r, 2019-03-12:
I think same for me, but how I knew, this issue is triggered by Windows 10 hosts with Realtek wireless cards. But sometime have this on Android devices too

9000000, 2019-03-24:
Thanks @nbd168 latest update very stable, I will check more

closed 2019-03-27 by nbd168