openwrt/mt76 #255: Big problem with 5g 80mhz channel width

dimatto, 2019-03-04:
On mir3g, a few channels can be used with 80mhz, such as channel 36，but displayed on luci is channel 40.  
On newifi d2, 80mhz channel width is totally unusable. If you choose channel 36,  sometimes it display channel 40 on luci. If you choose channel 40,  it display channel 36 .
All these cases, you can't see ssid on client scan list ,  seldom, you can see it but can't join.
20mhz and 40mhz have no problem. 
I compiled commit a year ago, problem still exist, but I remember there're no problem at that time.  So there must be something else together with mt76 driver cause this problem, such as mac80211,hostapd or else.

xavier83, 2019-03-05:
sounds like i'm facing similar issue with my 5ghz radio as well
https://github.com/openwrt/mt76/issues/199

dimatto, 2019-03-06:
@xavier83 On my newifi d2, for not appearing in client scan list, wrong password, can't join, I restore three mtd partition to factory original, ant it's fixed. These three partition are "u-boot", "u-boot-env", "factory". I notice in #199, someone mentioned modify mac address cause these problem, so I restore those partition to factory, then it's get fixed. Maybe the driver or openwrt or chip firmware need orignal factory data, but I flashed a pandorabox system which based on openwrt with kernel 3.14 and vendor original wireless driver, it has no problem at all without orignal factory partitions,I don't know why. If you use breed bootloader and non-original factory data, maybe it has problem with latest openwrt.
@WBINVD I haven't seen wrong channel displaying since I post this issue, I don't kow why, next time I see it, I'll see what's going on.

xavier83, 2019-03-06:
@dimatto any reference on how to restore specific partitions to factory?

dimatto, 2019-03-07:
@xavier83 Actually, I find out that it's "factory" partition that need to be restored, "u-boot" and "u-boot-env" has nothing to do with this. So, you need a system that has a writable "factory" partition, not offical openwrt, some pandorabox does. Or, you have to re-compile the dts file to make "factory" writable and use it to build a custom openwrt image.

dimatto, 2019-03-08:
It's actually an eeprom toleration problem address at #258

closed 2019-03-08 by dimatto

xavier83, 2019-03-23:
@dimatto my head is spinning now :face_with_head_bandage: lol