openwrt/mt76 #258: mt76 driver, flash data and mir3g 2.4g lag

dimatto, 2019-03-07:
Recently, I got a newifi d2 which had an unusable 80mhz 5g radio which works fine under vendor driver.
So I replace "factory" partition with another device's  flash data, all problems with 5g disappeared. But, 2.4g is very laggy, far more than mir3g. After modifying "factory" again, 2.4g is better than mir3g now, and it's almost equal to vendor driver on the same "factory" data. 
So I think maybe mir3g's 2.4g lag including flash data misread.  I don't know how mt76 driver read flash data, since every device has different location to place data, and every device may have its own data meaning, mt76 driver could misunderstand data from flash, and lead to many strange problem unrelated to driver itself,  so if mt76 driver can exclude error from flash data, it's good for identifying real bug.

lukasz1992, 2019-03-07:
Could you write what modifications did you on factory partition or provide flash dumps for both 2Ghz and 5Ghz radio?

nbd168, 2019-03-07:
Were you using a vendor driver with OpenWrt, or the vendor firmware to compare?

dimatto, 2019-03-07:
@nbd168  I use  pandorabox to compare, it’s actually an older version openwrt shipped with kernel 3.14 and vendor driver.

dimatto, 2019-03-07:
@lukasz1992 In fact, it only need change a few bytes to solve the 2.4g and 5g problem of my newifi d2. But I don’t know where’s these magic bytes of mir3g. I hope @nbd168 could remove the effect of these bytes fro mt76 driver.

redchenjs, 2019-03-07:
I have two newifi d2 using latest mt76 driver, and the 80MHz 5G wifi works very well, the only problem is ustable 2.4G wifi.

nbd168, 2019-03-07:
Can you show me a full boot log of the firmware with the vendor driver on an affected device?

dimatto, 2019-03-08:
@nbd168 attached file is boot log of pandorabox which 5g works fine, with this same "factory" data, mt76 driver does not work.
Latest mt76 driver has 5g speed no more than ~300Mbits/s on mir3g and newifi d2 while vendor driver has ~700Mbits/s sometimes reach 867Mbits/s.
From boot log, I knew that vendor driver read eeprom from "factory" partition which designated by dts file, I think so does mt76 driver. It seems that vendor driver can tolerate some eeprom that mt76 can't.
The modification of 2.4g "factory" is actually modification of eeprom read by driver, maybe it adjust tx power or rate power just like internet rumor said.

julianocss, 2019-03-08:
@nbd168

Did you make some change for mac80211 in Openwrt master yesterday? I cannot ser this change anymore.

Juliano

nbd168, 2019-03-10:
Which of the EEPROM files is the factory one and which one is the working one? Filenames only refer to device1 and device2.
Also, please show me the contents of the "otp" debugfs file from both radios on the affected device.

redchenjs, 2019-03-10:
@nbd168 Oh the EEPROM files I attatched are "factory" from my two devices, which have stable 5G wifi and laggy 2.4G wifi, I don't know how to modify them to make 2.4G wifi better.
@dimatto Could you please show us how to modify the "factory" data?

xavier83, 2019-03-10:
does restoring factory partition means flashing with the non-existent zbt-we1326-squashfs-**factory**.bin for me?

dimatto, 2019-03-11:
@xavier83 No, you need flash eeprom partition. It has nothing to do with a image which have factory in its name.

dimatto, 2019-03-11:
@redchenjs After reading source code of driver, I found that I actually change the tx power of my selected channel to make 2.4g better, so if you want to raise tx pwoer, you'd like to change it at luci not  eeprom.
But, obviously mt76 driver and vendor driver are different at eeprom offset definition and how to use them to config wifi, for example, EEPROM_G_TX_PWR_OFFSET and external PA. I don't know if this will lead to unnecessary raising tx power.
Maybe you can upload your 2.4g radio's otp file as a complement to your previous eeprom upload.

dimatto, 2019-03-11:
For 5g

[eeprom-and-otp-only-work-under-vendor-driver.zip](https://github.com/openwrt/mt76/files/2950665/eeprom-and-otp-only-work-under-vendor-driver.zip)

redchenjs, 2019-03-11:
My 2.4g radio's otp and eeprom files.
[mt7603e.zip](https://github.com/openwrt/mt76/files/2950677/mt7603e.zip)

nbd168, 2019-03-11:
@dimatto which vendor driver source are you using as reference? Got a link or at least a version number?

dimatto, 2019-03-11:
@nbd168  2.4g : https://github.com/OpenSNA/MT7603e
5g : https://github.com/mqmaker/mt76x2e
Maybe they are obsolete , I read them for reference.

nbd168, 2019-03-11:
@dimatto, one thing that's somewhat confusing about the vendor drivers is the fact that they contain large amounts of dead code for older chipsets.
For instance, the code using the EEPROM_G_TX_PWR_OFFSET definition you quoted above is not used on MT7603 at all.
I did a lot of code review to ensure that mt76 processes the EEPROM information in the same way as the vendor driver. The main exception is the fact that the vendor driver does not allow configuring the tx power in dBm, so I changed the code in mt76 to make that work.

dimatto, 2019-03-11:
@nbd168 I haven’t read 2.4g eeprom parse code very carefully, because they are long, maybe you’re right.
How about 5g ? Why that eeprom doesn’t work? I briefly read the code, and I found 
in mt76x2_phy_set_gain_val()
```
val = 0x1836 << 16;
	if (!mt76x2_has_ext_lna(dev) &&
	    dev->mt76.chandef.width >= NL80211_CHAN_WIDTH_40)
		val = 0x1e42 << 16;
```
But vendor source seems use 0x1836 only on 20mhz.
For unusable eeprom, if I set lnagain from 0x0 to 0x8f, it can work with mt76 driver, but mir3g’s lnagain is also 0x0, it can work with mt76, this device can’t, don’t know why.

redchenjs, 2019-03-13:
I changed my 2.4g wifi's tx-power from 20dbm to 29dbm, but after few minutes it became "no internet" status just like before, so maybe tx-power is not the main cause of this problem.

z1022, 2019-03-14:
Hi dimatto,

I also use newifi d2.  Is "factory" partition =  EEPROM ?
Which EEPROM you use in d2 ?

Thanks

redchenjs, 2019-03-15:
I disabled "WMM Mode" on 2.4g wifi and then the "no internet" problem was gone, now the 2.4g wifi works as stably as 5g wifi, although the bitrate is limited to 54 Mbit/s.

tkso1997, 2019-03-17:
in my kernel.log I get many "compare signature failed" could this point towards a corrupt eprom file or is it completely meaningless?

`
[    2.535461] Scanning device for bad blocks
[    2.684424] 10 fixed-partitions partitions found on MTD device MT7621-NAND
[    2.691280] Creating 10 MTD partitions on "MT7621-NAND":
[    2.696593] 0x000000000000-0x000000080000 : "Bootloader"
[    2.702994] 0x000000080000-0x0000000c0000 : "Config"
[    2.708970] 0x0000000c0000-0x000000100000 : "Bdata"
[    2.714835] 0x000000100000-0x000000140000 : "Factory"
[    2.720928] 0x000000140000-0x000000180000 : "crash"
[    2.726815] 0x000000180000-0x0000001c0000 : "crash_syslog"
[    2.733312] 0x0000001c0000-0x000000200000 : "reserved0"
[    2.739593] 0x000000200000-0x000000600000 : "kernel_stock"
[    2.746167] 0x000000600000-0x000000a00000 : "kernel"
[    2.752160] 0x000000a00000-0x000007f80000 : "ubi"
[    2.758845] [mtk_nand] probe successfully!
[    2.763086] compare signature failed ffc0
[    2.767235] compare signature failed ff80
[    2.771362] compare signature failed ff40
[    2.775486] compare signature failed ff00
[    2.779632] compare signature failed fec0
[    2.783759] compare signature failed fe80
[    2.787899] compare signature failed fe40
[    2.792026] compare signature failed fe00
[    2.796177] compare signature failed fdc0
[    2.800304] compare signature failed fd80
[    2.804428] compare signature failed fd40
[    2.808569] compare signature failed fd00
[    2.812695] compare signature failed fcc0
[    2.816841] compare signature failed fc80
[    2.820968] compare signature failed fc40
[    2.825092] compare signature failed fc00
[    2.829233] compare signature failed fbc0
[    2.833359] compare signature failed fb80
[    2.837505] compare signature failed fb40
[    2.841632] compare signature failed fb00
[    2.845773] compare signature failed fac0
[    2.849899] compare signature failed fa80
[    2.854023] compare signature failed fa40
[    2.858171] compare signature failed fa00
[    2.862299] compare signature failed f9c0
[    2.866439] compare signature failed f980
[    2.870565] compare signature failed f940
[    2.874689] compare signature failed f900
[    2.878835] compare signature failed f8c0
[    2.882962] compare signature failed f880
[    2.887103] compare signature failed f840
[    2.891229] compare signature failed f800
[    2.895212] load_fact_bbt failed
`

redchenjs, 2019-03-24:
On my device, the 2.4g wifi issue has been fixed by the latest version of OpenWrt.

closed 2019-03-27 by nbd168