openwrt/mt76 #259: 802.11s nodes not meshing

jsphpl, 2019-03-10:
Hi everyone,

i cannot make 802.11s (unencrypted) work. Neither on 2,4GHz nor 5GHz. Tested all modes (N @ 5GHz, N @ 2,4GHz, AC @ 5GHz), each with various channels and channel widths.

Devices are [TP-Link Archer C2 AC750](https://openwrt.org/toh/hwdata/tp-link/tp-link_archer_c2_ac750). Tested on both v1.0 and v1.1 (in all combinations: 1.1x1.1, 1.0x1.0, 1.1x1.0) with `OpenWrt SNAPSHOT, r9544-457de5a154` (mt76 @ 461f3b0, later 00ac79d).

802.11s works perfectly fine on the other radio at 2,4GHz.

On the MT7610E radio, the nodes either don't mesh at all, or only one node sees the other with a ridiculously low signal strength and tx rate. Devices are right next to each other. Possibly related to  #216?

Here's what it looks like in luci on both nodes:
[image]

This is the station that sees the other:
```
root@AP-3:~# iw dev wlan0 station dump
Station 00:0c:43:26:60:10 (on wlan0)
	inactive time:	152540 ms
	rx bytes:	308
	rx packets:	4
	tx bytes:	1839
	tx packets:	15
	tx retries:	192
	tx failed:	12
	rx drop misc:	0
	signal:  	-95 [-95] dBm
	signal avg:	-93 [-93] dBm
	Toffset:	433459524 us
	tx bitrate:	6.0 MBit/s
	mesh llid:	0
	mesh plid:	0
	mesh plink:	LISTEN
	mesh local PS mode:	UNKNOWN
	mesh peer PS mode:	UNKNOWN
	mesh non-peer PS mode:	ACTIVE
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	2
	beacon interval:100
	connected time:	535 seconds
```

Kernel output from the same device:
```
root@AP-3:~# dmesg
[    0.000000] Linux version 4.14.104 (jsphpl@joseph-pc) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r9544-457de5a154)) #0 Wed Mar 6 21:02:56 2019
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:3
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
[    0.000000] MIPS: machine is TP-Link Archer C2 v1
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] On node 0 totalpages: 16384
[... 224 lines cut ...]
	option hwmode '11a'
	option channel '36'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '11'
	option hwmode '11g'
	option path 'platform/10180000.wmac'
	option htmode 'HT40'
	option country '00'
	option legacy_rates '1'

config wifi-iface
	option encryption 'none'
	option device 'radio0'
	option mode 'mesh'
	option mesh_fwding '1'
	option mesh_rssi_threshold '0'
	option network 'lan wan wan6'
	option mesh_id 'mesh_backbone'
```

Am i missing something? What can i do to further debug things?

Thanks a lot!

nbd168, 2019-10-11:
Please try the latest version

zhoreeq, 2019-12-17:
I can confirm it doesn't work on a build from 24 October 2019.  OpenWrt SNAPSHOT r11323-49db2026e5 on Asus RT-AC51U, MediaTek MT7620A ver:2 eco:6

Mesh network configuration works on a router, the interface is created. My laptop can see the router with `iw dev mesh0 station dump`, but the router can't see laptop. Pings are not working either.

dangowrt, 2019-12-17:
@zhoreeq Did you test 2.4GHz as well? (because that's not mt76 then but rather the old rt2x00 driver and problems should be reported to linux-wireless)

zhoreeq, 2019-12-17:
@dangowrt yes, it was a test on 2.4. I don't have 5HGz in a laptop.

zhoreeq, 2019-12-19:
@dangowrt can you please clarify, should this issue be reported to linux-wireless? I'm a bit confused :)

dangowrt, 2019-12-19:
@zhoreeq So that Asus RT-AC51U got an MT7620A SoC, which contains Ralink RT6352 2.4GHz WiFi radio supported by the rt2x00 driver. Only the 5 GHz chip is supported by the mt76 driver which is developed in this repository. Hence, if you experience problems when using the 2.4GHz radio, please report them to the linux-wireless mailinglist http://vger.kernel.org/vger-lists.html#linux-wireless and CC to Stanislaw Gruszka <stf_xl@wp.pl>, the diver's maintainer.

jsphpl, 2019-12-29:
I've tested again on Archer C2 v1.0 and v1.1 and it is now working properly. Also seems to correctly report and adjust transmission power. Tested with openwrt @ `30301df` (mt67 @ `2a0edbb4`). Good job, thanks a lot!

closed 2019-12-29 by jsphpl