openwrt/mt76 #261: 2.4 Ghz (11g) on Mac Book Pro 2018 + constant connection interrupts

lumixen, 2019-03-18:
Testing environment:
```
Hardware: Xiaomi Mi Router 3G, MediaTek MT7621 ver:1 eco:3
Openwrt: OpenWrt 18.06-SNAPSHOT r7717-400601f / LuCI openwrt-18.06 branch (git-19.051.55698-76cf653) 
```

```
/etc/config/wireless
config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11g'
	option path 'pci0000:00/0000:00:00.0/0000:01:00.0'
	option channel 'auto'
	option htmode 'HT40'
	option country 'UA'
	option txpower '20'
	option legacy_rates '1'
```
I'm trying to connect my MBP 2018 to 2.4Ghz interface but experiencing constant connection interrupts.

Pings MBP -> Xiaomi 3G looks like this:
```
Request timeout for icmp_seq 645
Request timeout for icmp_seq 646
Request timeout for icmp_seq 647
Request timeout for icmp_seq 648
Request timeout for icmp_seq 649
Request timeout for icmp_seq 650
Request timeout for icmp_seq 651
Request timeout for icmp_seq 652
64 bytes from 192.168.34.1: icmp_seq=653 ttl=64 time=1.807 ms
64 bytes from 192.168.34.1: icmp_seq=654 ttl=64 time=1.255 ms
64 bytes from 192.168.34.1: icmp_seq=655 ttl=64 time=128.992 ms
64 bytes from 192.168.34.1: icmp_seq=656 ttl=64 time=45.887 ms
Request timeout for icmp_seq 657
Request timeout for icmp_seq 658
Request timeout for icmp_seq 659
64 bytes from 192.168.34.1: icmp_seq=657 ttl=64 time=3114.983 ms
64 bytes from 192.168.34.1: icmp_seq=658 ttl=64 time=2114.810 ms
64 bytes from 192.168.34.1: icmp_seq=659 ttl=64 time=1112.451 ms
64 bytes from 192.168.34.1: icmp_seq=660 ttl=64 time=110.039 ms
64 bytes from 192.168.34.1: icmp_seq=661 ttl=64 time=3.227 ms
64 bytes from 192.168.34.1: icmp_seq=662 ttl=64 time=3.575 ms
64 bytes from 192.168.34.1: icmp_seq=663 ttl=64 time=5.853 ms
64 bytes from 192.168.34.1: icmp_seq=664 ttl=64 time=1.926 m
```

I'd be glad to provide more logs from a router, but unfortunately, there is nothing specific logged there when interruption happens.

lukasz1992, 2019-03-20:
Maybe a stupid idea, but does it change anything when you replace /lib/firmware/mt7603_e2.bin with this one https://github.com/openwrt/mt76/raw/f2ba65f73f384919aa7644a685c79f8ae7338032/firmware/mt7603_e2.bin ? or that https://github.com/openwrt/mt76/raw/99273a3c8b74d0cf6b2ef7b2489731c055525b20/firmware/mt7603_e2.bin or that https://github.com/openwrt/mt76/raw/c395a030093350ff8b0833502c772eef3c1666df/firmware/mt7603_e2.bin

lumixen, 2019-03-20:
Is it safe to replace firmware without changing driver version?

lukasz1992, 2019-03-20:
Yes - worst thing that can happen is that other versions will not work.

Possibly this, #258, #253, #245, maybe #235 and #169 are the same issue.

redchenjs, 2019-03-21:
@lukasz1992 I tried 5 firmwares but interruptions still happen.

Firmware version I tried: 

- 20150624233050
- 20151020155902
- 20160107100755
- 20161027164355
- 20170925114906

[mt7603_e2_firmwares.zip](https://github.com/openwrt/mt76/files/2990676/mt7603_e2_firmwares.zip)

When I turn off HT mode the interruption will disappear.

lumixen, 2019-03-21:
I've tried HT20 mode and interrupts became less frequent. But, unfortunately, they still happen.

redchenjs, 2019-03-22:
@lumixen Only non-HT mode is stable, you need to use "Legacy" operating mode or disable "WMM Mode", which will totally disable HT features.

lumixen, 2019-03-22:
Well, disabling WMM == using `11g` with `52 MB\s` at most. And that doesn't sound to me as a good solution.

lumixen, 2019-03-22:
Also, this commit https://github.com/openwrt/mt76/commit/8167074dab20b4f434f882b3ceb737bc953c2f61 reverts firmware to the previous one, though I'm not sure what was were those reported regressions.

redchenjs, 2019-03-23:
#225

lumixen, 2019-03-23:
Hey @nbd168 !
I've managed to do some pcapping using another device in monitor mode: [MBP2018-X3G-STALE-CONNECTION.pcapng](https://drive.google.com/open?id=1kFFH55yafPmQxw-gh1_pDjhCNa93xU6w)
Resulting file is quite huge (~1G) cause I had to run `speedtest` in the background to force reproducing this issue.
During capturing interruptions happened (when ping command was reporting `Request timeout for icmp_seq`) several times at least:

- Around 13s from capturing start. It was quite a short outage, not more than 2 secs.
- Around 146s from capturing start. Then connection recovered around 190s from capturing start.

Hope, this pcap can help you understand what is happening.

nbd168, 2019-03-23:
@lumixen, thanks for that capture, it was very helpful! With it, I was able to create a test to reproduce this issue with my own MacBook. I've pushed a fix to OpenWrt master, I believe it should work now.

lumixen, 2019-03-23:
@nbd168 , thank you for the reactive fix!
 I've been running the latest build for 30 mins already and haven't seen this issue so far even under excessive traffic load.
I'm going to spend a few days more checking if this issue gone and let you know on the results!
Also, @redchenjs could you please test the latest fixes as well? Perhaps you ran into the same issue and fix works for you as well?

redchenjs, 2019-03-24:
@nbd168 @lumixen, thanks for all your work! These fixes are really effective!
I tried the latest version, and now the 2.4g wifi can run as expected. There was no interruption for a few hours!

nbd168, 2019-03-24:
Thanks a lot for testing and reporting your findings! Without that, it would have taken me much longer to fix these issues.

tkso1997, 2019-03-24:
**@all**, works for me as well! Thanks a lot **@nbd168** !!!

nbd168, 2019-03-24:
Nice, thanks for testing!

closed 2019-03-24 by nbd168

lumixen, 2019-03-26:
I haven't seen this issue so far after several days of testing, nice!