openwrt/mt76 #267: mt76 driver 2019-03-23 implicit declaration of function 'ieee80211_schedule_txq

tkmbe99, 2019-03-26:
Trying to compile latest changes for mt76 driver (2019-03-23) with OpenWRT (18.06.1, r7258-5eb055306f) on OpenSuse Leap 15.0
I have the following warning, which is treated as error by default:

/Devel/OpenWRT/OpenWRT-18.06.2-r1-w/build_dir/target-mipsel_24kc_musl/linux-ramips_mt76x8/mt76-2019-03-23-6fc50ea9/mac80211.c:624:4: error: implicit declaration of function 'ieee80211_schedule_txq'; did you mean 'ieee80211_check_tim'? [-Werror=implicit-function-declaration]
    ieee80211_schedule_txq(dev->hw, sta->txq[i]);
    ^~~~~~~~~~~~~~~~~~~~~~
    ieee80211_check_tim

This OpenWRT version with orig mt76 drivers compiles...
I've searched and not found this function.
Were was this function defined?
Do I have to update the whole OpenWRT repository to the newest version?

jow-, 2019-03-26:
https://github.com/openwrt/mt76/issues/265#issuecomment-476502672

closed 2019-03-26 by jow-

tkmbe99, 2019-03-26:
Broswing the resporitory with web ui shows me for file
 openwrt/package/kernel/mt76/Makefile
selecting branches master or openwrt-18.06 latest update (2019-03-23)
lede-17.01 has old version (2017-12-03)

I've cloned the repository with
git clone https://git.openwrt.org/openwrt/openwrt.git
git branch shows me
* (HEAD detached at v18.06.1)
  master
But my Makefile is the old one (2017-12-03)
So I guess other stuff too is old
So what do I wrong? I want to compile latest version...