openwrt/mt76 #269: 5GHz speed oscilations

lukasz1992, 2019-03-28:
I have made today some tests by iperf3 on a snapshot (on a11b6734). Both tests were done on unsecured networks.

I got up to 109 Mbps in 2.4GHz - over 70% of the theoretical maximum speed. 2GHz is also very stable. I consider this result as perfect. 

On 5GHz I am not able to exceed 250Mbps (with average about 210 Mbps). Also reported theoretical speed on STA often drops from 433 to 390 or 292.5 Mbps.

On MTK driver, I was able to gain up to 305Mbps real speed (which is about 70% of max theoretical speed).

So mt7612e maximum speed is about 70% of what I could expect.

There are some neighboor networks on 2.4GHz, but none on 5GHz. Changing the channel does not help. How could I check for possible reasons or maybe mt76x2 needs to be tweaked?

Tested on Intel 3160 1x1 radio, Win8.1. Device: R3G router. Distance from device during tests is about 1.5m.

nbd168, 2019-06-25:
Please try the latest version, it should work much better now

closed 2019-06-27 by nbd168

Openwrtfunboy, 2020-02-24:
Hi, @lukasz1992 
Have this problem now?
Have same platform as you.