openwrt/mt76 #270: luci shows 0 dBm noise

tkso1997, 2019-03-29:
luci shows 0 dBm noise, while "iwinfo wlan1 info" shows the noise. 
Not sure wether I should report it to the luci people or here?
Plus its just cosmetics.

```
root@OpenWrt:~# iwinfo wlan1 info
wlan1     ESSID: "dd-wrt"
          Access Point: 40:31:3C:D7:46:C4
          Mode: Master  Channel: 48 (5.240 GHz)
          Tx-Power: 20 dBm  Link Quality: 64/70
          Signal: -46 dBm  Noise: unknown
          Bit Rate: 432.7 MBit/s
          Encryption: mixed WPA/WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11nac
          Hardware: 14C3:7662 14C3:7662 [MediaTek MT76x2E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

dangowrt, 2019-03-29:
Ralink and MediaTek radios do not expose the noisefloor information to the driver/OS, so it says 'unknown' in iwinfo and should say the same in LuCI...

tkso1997, 2019-03-29:
Thanks for the clarification!

closed 2019-03-29 by tkso1997