openwrt/mt76 #271: saturation load starves other wifi connections (TP-Link WR841 v13)

differentblue, 2019-03-29:
When I saturate my wifi connection with a single bulk data transfer from my wifi connected laptop to an upstream-LAN Ethernet connected server, other connections from the same laptop seem to stall.   This stalling is **not** seen on other routers tested.

The router has  r9707-bee7ff7 installed, snapshot from last weekend, so has most of the recent @nbd168 changes.  I saturate using the following commands.
```
Load source command on laptop:    nc -v -v -n $serverIP 22123 </dev/zero
Load sink command on server:     nc -v -v -k -l -n -p 22123
```

While doing this saturation test I see the following from my laptop:
1. very slow to open up a web site in Chrome (page takes 2 minutes to begin showing, and then longer for pictures to fill in)
2. Google "speed test" shows very slow effective download WAN speed (even though the WAN is unused).

Using the factory TP Link firmware (same model router and same wifi channel) and doing the saturation, I can still browse the web and Google "speed test" shows full purchased WAN bandwidth available.   I also have no problems when saturating using OpenWRT on a TP Link -wr902ac-v1 on the same wifi channel.

Mushoz, 2019-03-29:
Do you experience the same issue with the 18.06.2 or 18.06.1 build? eg, is this a regression?

differentblue, 2019-03-30:
The mt76 wifi drivers for this device on 18.06.1 are pretty problematic so hard to tell.   I have not tried 18.06.2

araujorm, 2019-03-30:
I advise you to try the latest builds. Last week several problems were
solved and it's now much better. Please report if the problem persists with
the latest version.

differentblue, 2019-03-30:
I have updated the the latest snapshot build (6df5ab89cff8c2b2cd1a24a3e365c11b472497db).   Problem is still there.

nbd168, 2019-06-25:
Please try the latest version

closed 2019-08-28 by nbd168