openwrt/mt76 #274: Client Disconnect Issues on 5GHz MT76X2

agunzagun, 2019-04-08:
I'm experiencing client disconnection issues - namely on the 5GHz band - on the HooToo HT-ND001 (rebranded MTC WR1201). I posted in another thread but was recommended to make a new ticket.
 
Logread comes up
 
 ```
root@OpenWrt:~# logread | tail
Mon Apr  8 02:35:37 2019 daemon.info dnsmasq-dhcp[3138]: DHCPACK(br-lan) X.X.X.X xx:xx:xx:xx:xx:xx WirelessClient  
Mon Apr  8 02:35:38 2019 daemon.warn odhcpd[1202]: DHCPV6 CONFIRM IA_NA from xxxxxxxxxxxxxxxxxxxxxxxxxxxx on lan: ok xxxx:xxxx:xxxx::xxx/xxx  
Mon Apr  8 02:35:45 2019 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disconnected due to excessive missing ACKs  
Mon Apr  8 02:35:45 2019 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx  
Mon Apr  8 02:35:47 2019 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED xx:xx:xx:xx:xx:xx 1  
Mon Apr  8 02:35:47 2019 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated  
Mon Apr  8 02:35:47 2019 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED xx:xx:xx:xx:xx:xx 2  
Mon Apr  8 02:35:47 2019 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 1)  
Mon Apr  8 02:35:47 2019 daemon.notice hostapd: wlan0: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx  
Mon Apr  8 02:35:47 2019 daemon.info hostapd: wlan0: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
 ```
And more or less repeats as such. No outstanding issues (so far) on 2.4GHz.
 
The build used is based on the latest git as of this post. The profile is based on RAMIPS>MTC WR1201.
 
 ```
root@OpenWrt:~# lsmod | grep mt
cfg80211              220032  3 mt76x02_lib,mt76,mac80211
mac80211              426368  3 mt76x2e,mt76x02_lib,mt76
mmc_core               71744  2 mmc_block,mtk_sd
mt76                   24864  3 mt76x2e,mt76x2_common,mt76x02_lib
mt76x02_lib            36672  2 mt76x2e,mt76x2_common
mt76x2_common          10816  1 mt76x2e
mt76x2e                 9632  0
mtk_sd                 11504  0
usbcore               109824  5 ledtrig_usbport,xhci_plat_hcd,xhci_pci,xhci_mtk,xhci_hcd
xhci_hcd               76656  3 xhci_plat_hcd,xhci_pci,xhci_mtk
xhci_mtk                4160  0
 ```

If there is any more I can post to help with the issue, I'll do my best to assist.

julianocss, 2019-04-16:
Hello @nbd168

A feedback about your latest version on staiging dir, using mt7628 SoC (mt7603 + mt7612). Tplink Archer C50.

mt7603 (works very good).
mt7612 (doesn´t work, the latest driver does not even initialize).

Juliano

julianocss, 2019-04-19:
Hello @nbd168 

Tried to compile again from scratch, but MT76X2 radio does not initialize on my Archer C50, just MT7603.

Here my steps:

1) Created a new code on my github account
2) Copyed all the files from mt76 repositorie to it
3) Replaced all .c and .h files from your staiging dir.

https://github.com/julianocss/mt76_nbd168_staging

Am i doing something wrong?

Juliano

julianocss, 2019-04-29:
Hello @nbd168 

Output log from mt76 staging dir:

kern.info kernel: [   27.020720] mt76_wmac 10300000.wmac: ASIC revision: 76280001
kern.info kernel: [   28.051145] mt76_wmac 10300000.wmac: Firmware Version: 20151201
kern.info kernel: [   28.057241] mt76_wmac 10300000.wmac: Build Time: 20151201183641
kern.info kernel: [   28.077000] mt76_wmac 10300000.wmac: firmware init done
kern.debug kernel: [   28.254397] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
kern.err kernel: [   28.292286] mt76x2e 0000:01:00.0: card - bus=0x1, slot = 0x0 irq=4
kern.info kernel: [   28.298830] mt76x2e 0000:01:00.0: ASIC revision: 76120044
kern.info kernel: [   29.412283] mt76x2e 0000:01:00.0: ROM patch build: mt7662_patch_e3
kern.err kernel: [   29.418950] mt76x2e 0000:01:00.0: Invalid firmware
kern.warn kernel: [   29.423952] mt76x2e: probe of 0000:01:00.0 failed with error -2

---------------

Output log from latest mt76 commit on github:

kern.info kernel: [   29.868521] mt76_wmac 10300000.wmac: ASIC revision: 76280001
kern.info kernel: [   30.899469] mt76_wmac 10300000.wmac: Firmware Version: 20151201
kern.info kernel: [   30.905558] mt76_wmac 10300000.wmac: Build Time: 20151201183641
kern.info kernel: [   30.925379] mt76_wmac 10300000.wmac: firmware init done
kern.debug kernel: [   31.102773] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
kern.err kernel: [   31.140488] mt76x2e 0000:01:00.0: card - bus=0x1, slot = 0x0 irq=4
kern.info kernel: [   31.147057] mt76x2e 0000:01:00.0: ASIC revision: 76120044
kern.info kernel: [   32.257002] mt76x2e 0000:01:00.0: ROM patch build: 20141115060606a
kern.info kernel: [   32.268902] mt76x2e 0000:01:00.0: Firmware Version: 0.0.00
kern.info kernel: [   32.274475] mt76x2e 0000:01:00.0: Build: 1
kern.info kernel: [   32.278693] mt76x2e 0000:01:00.0: Build Time: 201507311614____
kern.info kernel: [   32.305378] mt76x2e 0000:01:00.0: Firmware running!

Am i doing something wrong?
If i can help with some other information, please tell me.

Juliano

nbd168, 2019-06-27:
Please try the latest version and let me know if this is still an issue

julianocss, 2019-07-12:
Hello @nbd168 

Running latest version for 7 days.
Everything looks good now.

Thank you very much.
Juliano

nbd168, 2019-07-12:
Thanks for testing

closed 2019-07-12 by nbd168