openwrt/mt76 #275: mt76x02 - problem with 5GHz

podtor, 2019-04-17:
I've bought a D-Link DWR-118 A1 and I have a huge problem with 5GHz. On original firmware, 5GHz works well. With snapshot (4.14.111-1) OpenWrt firmware only 2 meters range signal 5GHz. 2.4GHz work ok on OpenWrt and on the original firmware.

podtor, 2019-04-22:
On OpenWrt forum is a similar issue regarding ASUS RT-AC51U [mt76x0e - support for 5ghz](https://forum.openwrt.org/t/asus-rt-ac51u-support-for-5ghz/35041)

CHKDSK88, 2019-06-25:
Please try patch from https://github.com/openwrt/mt76/issues/286 .

ffrediani, 2019-06-25:
This router (Rev A1) uses the same radio as TP-Link C20 v4 and 5Ghz has serious issues to the point it doesn't work with snapshot versions so far. Seems this driver in mt76 hasn't been touched for a while, at least there was no apparent improvements.
There is another issue about this, please check: #227 and feedback there if your still persist.

ffrediani, 2019-07-10:
Have updated to r10348-577174cf60 and now it seems to be able to control the Transmit Power correctly. Also the Scan button that has never worked correctly seems now to work.

closed 2019-07-10 by nbd168

podtor, 2019-07-10:
Yes, I confirm. 5GHz works well.