openwrt/mt76 #276: How to install

Loganrose, 2019-04-21:
No matter what i try, i cant figure out how to install this?

gjdunga, 2019-04-21:
While using Kali Linux - 4.19.0-kali4-amd64 #1 SMP Debian 4.19.28-2kali1 (2019-03-18) x86_64

The Makefile has no label for install, clean or all.  It does have the following, but do not work! 

mt76-y 
mt76-usb-y
mt76x02-lib-y
mt76x02-usb-y 

Is there some secret sauce we are missing? Or is this a module that OpenWRT uses on a build, and it's not suitable for a normal Linux to build?

RavSS, 2019-04-22:
You can use DKMS. Here's a [dkms.conf file](https://aur.archlinux.org/cgit/aur.git/plain/dkms.conf?h=mt76-git) for mt76 which I use. You can create your own conf file if you want as well with your selected modules or even just create the *.ko files with a similar `make` command.

If you want to use that file, then replace the entire package version string with some number, which I'll refer to as $VER. Put the file inside the repository that is checked out to openwrt-18.06 (or the master branch if it compiles for your kernel). Once done, copy the directory over to `/usr/src/mt76-$VER` with the dash. Finally, do `dkms install -m mt76 -v $VER`.

I've tested this on both Arch Linux and Kali Linux, although I'm still unfortunately getting MCU message timeouts and disconnects on different systems with a MT7612E. Maybe I'm not doing something right? Hopefully it works better for you.

lukasz1992, 2019-04-25:
You can use necessary Makefile, Kconfig (and maybe other files) from kernel 5.1-rc6.

closed 2019-06-27 by nbd168

usama7628674, 2020-01-05:
@RavSS I've compiled it.Now how can i update old firware.bin files.I've comiled `.ko` files.