openwrt/mt76 #277: No AC enabled with A6210 card? (slow speeds)

maluhia, 2019-05-04:
My speeds are orders of magnitude slower on linux than on windows. I get ~150mb/s down on windows and and ~5mb/s down on linux, respectfully. A6210 uses MT7612U chipset. 

running 5.0.10-arch1-1-ARCH. Connecting to my 5GHz network.

>iw dev wlp0s20u3u1 station dump
```
Station fa:1d:0f:59:86:60 (on wlp0s20u3u1)
        inactive time:  4 ms
        rx bytes:       140750225
        rx packets:     135930
        tx bytes:       21824199
        tx packets:     58467
        tx retries:     8131
        tx failed:      395
        beacon loss:    17
        beacon rx:      11595
        rx drop misc:   1932
        signal:         -83 [0, -89] dBm
        signal avg:     -82 [0, -86] dBm
        beacon signal avg:      -83 dBm
        tx bitrate:     263.3 MBit/s VHT-MCS 6 80MHz VHT-NSS 1
        rx bitrate:     6.5 MBit/s VHT-MCS 0 VHT-NSS 1
        expected throughput:    51.360Mbps
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       long
        WMM/WME:        yes
        MFP:            no
        TDLS peer:      no
        DTIM period:    1
        beacon interval:100
        short slot time:yes
        connected time: 1579 seconds
```

>lsmod | grep mt76
```
mt76x2u                28672  0
mt76x2_common          20480  1 mt76x2u
mt76x02_usb            16384  1 mt76x2u
mt76_usb               36864  2 mt76x02_usb,mt76x2u
mt76x02_lib            73728  3 mt76x02_usb,mt76x2u,mt76x2_common
mt76                   57344  5 mt76_usb,mt76x02_lib,mt76x02_usb,mt76x2u,mt76x2_common
mac80211              950272  6 mt76,mt76_usb,ath9k,mt76x02_lib,mt76x2u,ath9k_hw
cfg80211              798720  7 mt76,ath9k_common,ath9k,mt76x02_lib,ath,mac80211,mt76x02_usb

```

>iw list
```
Wiphy phy1
        max # scan SSIDs: 4
        max scan IEs length: 2243 bytes
        max # sched scan SSIDs: 0
        max # match sets: 0
        max # scan plans: 1
        max scan plan interval: -1
        max scan plan iterations: 0
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports RSN-IBSS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
[... 401 lines cut ...]
        Device supports HT-IBSS.
        Device supports SAE with AUTHENTICATE command
        Device supports low priority scan.
        Device supports scan flush.
        Device supports AP scan.
        Device supports per-vif TX power setting
        P2P GO supports CT window setting
        Driver supports full state transitions for AP/GO clients
        Driver supports a userspace MPM
        Device supports active monitor (which will ACK incoming frames)
        Driver/device bandwidth changes during BSS lifetime (AP/GO mode)
        Device supports ACK timeout estimation.
        Device supports configuring vdev MAC-addr on create.
        Supported extended features:
                * [ RRM ]: RRM
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs

```

If you need more logs, I will happily supply them.

Thanks in advance.

LorenzoBianconi, 2019-05-06:
@maluhia : mt76x2u dongle is just receiving, right? Moreover signal strength seems pretty low to me

maluhia, 2019-05-06:
@LorenzoBianconi I can send and receive data with this dongle, as I did a speed test and got 11 Mbps down, 702 kbps up. Signal strength isnt the best, but I havent moved the dongle and get 10x the speed when I boot into windows and use their drivers, connecting the my wifi over ac band.

maluhia, 2019-06-04:
I ran another test right beside my router in Linux, and booted into windows and ran the same test (internet speed is 300/15):

> Linux: 167Mbps down and 5Mbps up
> Windows: 165Mbps down and 15Mbps up

I wonder why the upload is slow in Linux? No other devices were on the network.