openwrt/mt76 #278: 802.11r + 802.11w broken bss transition

dissent1, 2019-05-05:
There's an issue when FT (11r) and PMF (11w) are enabled at the same time. The issue can be reproduced as follows:
1. Connect to the 1st AP - OK
2. Move to the 2nd AP coverage area - FT to the 2nd AP is ok
3. Move back to 1st AP coverage area - the 1st AP log spams "Could not set STA to kernel driver" and fails to authenticate.

FT alone works ok and PMF alone work ok, but not concurrently.
This is a serious issue because wpa3 requires PMF to be enabled, so the choice becomes either or.

redchenjs, 2019-05-05:
Also happen to me.

dissent1, 2019-05-11:
Seems like it's not directly related to mt76, because it also happens on ath10k.

closed 2019-05-11 by dissent1

NerdyProjects, 2019-08-23:
I just discovered the same thing on a 18.06.4 archer c5 v1 (ath10k) where I try to get FT-over-DS going (802.1x & vlan setup) and have 802.11w enabled.
My Intel 7265 notebook reproduces the error as described in the issue, my galaxy s5 smartphone does not.

Any idea where to look further or where there is more communication about this issue? I found a cisco bug about the same topic, but no further discussions...

marekbedy, 2020-04-20:
I have the same problem on a 19.07.2 with Archer C7 v5

ktgeek, 2020-06-18:
Just adding to the issue.  Turning off 802.11w made it go away on an Omnia with raw OpenWrt and with archer c7v2 both on 19.07.3

enmaskarado, 2021-12-24:
I couldn't find the report in bugs.openwrt.org, so I created one. If it is duplicated, I have no problem if someone close my report.

https://bugs.openwrt.org/index.php?do=details&task_id=4201