openwrt/mt76 #279: Low-incorrect txpower on Mediatek 7628 (TP-Link 840 ver 6.0)

sanitariu, 2019-05-07:
It seems like txpower is totally incorrect on mediatek 7628 models.
I see -55dB when my laptop is 30 cm away from the router.
Same test with openwrt Tp-link 740 version 4 signal strenght in luci is -27dB. Version 740 power is limited to 18dB (63 mw) so seems like Mediatek (-55dB) are something like 14dB (30mw) which is wrong. Also any changes like country  or txpower does not change signal strength in Luci (still -55dB). Testing mediatek 7628 in 5 meters away make signal -80dB.
May be your driver does not set power at all or at least does not respect country regulatory ?
I am using latest OpenWrt SNAPSHOT, r9945-bc85640cdc.
Driver is stable and working good and 1 meter away but next room is not so good.

sanitariu, 2019-05-07:
Also tested version 6.1 and 6.2. Results are the same. Stable but no range.

root@telnet-router:~# cat /sys/kernel/debug/ieee80211/phy0/mt76/rate_txpower
       CCK:  0  0  0  0
      OFDM:  0  0  0  0  0  0  0  0
      STBC:  0  0  0  0  0  0  0  0  0  0
        HT:  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0
       VHT:  0  0  0  0  0  0  0  0  0  0

This seems wrong to me too. Like wrong eeprom reading ?
According to TL-WR840NV6.dts these are my partitions:

        memory@0 {
                device_type = "memory";
                reg = <0x0 0x2000000>;
        };

                partitions {
                        compatible = "fixed-partitions";
                        #address-cells = <1>;
                        #size-cells = <1>;

                        partition@0 {
                                label = "boot";
                                reg = <0x0 0x10000>;
                                read-only;
                        };

                        partition@10000 {
                                compatible = "tplink,firmware";
                                label = "firmware";
                                reg = <0x10000 0x3d0000>;
                        };

                        factory: partition@3f0000 {
                                label = "factory";
                                reg = <0x3f0000 0x10000>;
                                read-only;
                        };
                };

sanitariu, 2019-05-07:
tp-link 840 version 6 eeprom dump:

---------------------------------------
/lib/firmware# hexdump eeprom.bin
0000000 7628 0200 0c00 e143 0876 0000 0000 0000
0000010 ffff ffff ffff ffff ffff ffff ffff ffff
0000020 0000 0000 0020 0000 0c00 e143 2976 0c00
0000030 e143 2a76 3422 2000 ffff 0100 0000 0000
0000040 0000 0022 0000 0000 0030 0000 0000 0000
0000050 0000 9400 bc40 cac0 c024 c0c0 ca40 c024
0000060 c0c0 0000 0000 0000 0000 0000 0000 0000
0000070 0000 0000 0000 0000 0000 0000 0000 0000
*
00000a0 c2c4 c2c4 c0c2 c4c0 c4c4 c0c2 c0c0 0000
00000b0 0000 0000 0000 0000 0000 0000 0000 0000
00000c0 0000 0000 0000 0000 1a00 2a22 3531 3501
00000d0 4039 4d46 7f7f 007f 0000 0000 0000 0000
00000e0 1d11 1d11 351c 351c 351e 351e 1917 1917
00000f0 0002 0000 80b5 8888 000a 0000 0000 0000
0000100 ffff ffff ffff ffff ffff ffff ffff ffff
*
0000120 0000 0000 0000 0000 0000 0000 0000 0077
0000130 1d11 1d11 7f15 7f15 7f17 7f17 3b10 3b10
0000140 ffff ffff ffff ffff ffff ffff ffff ffff
*
0000200
------------------------

Somewhere here must be default txpower values ?

iscilyas, 2019-08-04:
seeing the same all-zero values for rate_txpower on xiaomi mir3p (mt7621 with dual mt7615)... it's odd that nobody commented on this issue yet

sanitariu, 2019-08-23:
I extraced factory bin file from same mediatek chip 7628 and from tp-link 840 (mediatek 7628) version 6.0.
First router TX power perfect also openwrt works like a charm
Second router tp-link 840 version 6.0 very LOW tx-power also works like a charm but 2 meters near router.
Attaching both bin files
If someone knows how to decode the file ? You can view via hexdump or other but i need to know which value is which.

[mediatekhigh-tx.txt](https://github.com/openwrt/mt76/files/3534382/mediatekhigh-tx.txt)
[mediateklow-tx.txt](https://github.com/openwrt/mt76/files/3534383/mediateklow-tx.txt)

sanitariu, 2019-08-23:
I extraced factory bin file from same mediatek chip 7628 and from tp-link 840 (mediatek 7628) version 6.0.
First router TX power perfect also openwrt works like a charm
Second router tp-link 840 version 6.0 very LOW tx-power also works like a charm but 2 meters near router.
Attaching both bin files
If someone knows how to decode the file ? You can view via hexdump or other but i need to know which value is which.

[mediatekhigh-tx.txt](https://github.com/openwrt/mt76/files/3534470/mediatekhigh-tx.txt)
[mediateklow-tx.txt](https://github.com/openwrt/mt76/files/3534471/mediateklow-tx.txt)

Anyone knows where in which line is country code or tx-power ?

rettichschnidi, 2019-08-24:
> Anyone knows where in which line is country code or tx-power ?

Answers to this you can find in the document [mt7628_eeprom_guideline_v2_00.pdf](https://www.widora.io/_media/mt7628_eeprom_guideline_v2_00.pdf).

sanitariu, 2020-01-09:
Still no fixes for tp-link 840 version 6,6.1,6.2 and 14 ?
The only problem is missing correct txpower.

sanitariu, 2020-06-03:
Correct partition is:
------------------
partition@10000 {
                            compatible = "tplink,firmware";
                            label = "firmware";
                            reg = <0x10000 0x3e0000>;
                    };

Also this fix in mt76x8.mk
---------------------------

define Device/tplink_tl-wr840n-v6
  $(Device/tplink)
  SOC := mt7628an
  IMAGE_SIZE := 3968k
  DEVICE_VENDOR := TP-Link
  DEVICE_MODEL := TL-WR840N
  DEVICE_VARIANT := v6
  TPLINK_FLASHLAYOUT := 4Mmtk
  TPLINK_HWID := 0x08400006
  TPLINK_HWREV := 0x1
  TPLINK_HWREVADD := 0x6
  TPLINK_HVERSION := 3
  SUPPORTED_DEVICES += tl-wr840n-v6
  IMAGE/tftp-recovery.bin := pad-extra 64k | $$(IMAGE/factory.bin)
endef
TARGET_DEVICES += tplink_tl-wr840n-v6

Now txpower and everything is okay. Router is usable.