openwrt/mt76 #280: 2.4 ghz bitrate low

9000000, 2019-05-23:
My modem usually has more than 20 devices connected, bandwidth above 2.4 ghz only about 86-94 Mb/s ht40 band is not enough to supply connected devices, hope you fix the bandwidth upgrade error to 300 Mb /S
Thanks

lukasz1992, 2019-05-24:
Could you dump the results of:
`iwinfo wlan0 station dump`

20 is a big number of devices in one network.

9000000, 2019-05-24:
I think xiaomi r3g can be loaded

root@OpenWrtFull:~# iwinfo                                                wlan0     ESSID: "Quick_Snow"
          Access Point: 50:64:2B:A9:70:85
          Mode: Master  Channel: 11 (2.462 GHz)
          Tx-Power: 26 dBm  Link Quality: 56/70
          Signal: -54 dBm  Noise: unknown
          Bit Rate: 81.2 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: 14C3:7603 14C3:7603 [MediaTek MT7603E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

lukasz1992, 2019-05-24:
`iwinfo wlan0 station dump`
not
`iwinfo wlan0`

9000000, 2019-05-24:
root@OpenWrtFull:~# iwinfo wlan0 station dump
No such wireless backend: wlan0
root@OpenWrtFull:~# iwinfo wlan0
Usage:
iwinfo <device> info
iwinfo <device> scan
iwinfo <device> txpowerlist
iwinfo <device> freqlist
iwinfo <device> assoclist
iwinfo <device> countrylist
iwinfo <device> htmodelist
iwinfo <backend> phyname <section>
root@OpenWrtFull:~#

lukasz1992, 2019-05-24:
@9000000 sorry, `iwinfo wlan0 assoclist` , I confused with iw

bond2bill, 2019-05-27:
Me too, i using MTK7621A D-Team Newifi D2, 2.4ghz slow.

root@OpenWrt:~# iwinfo wlan0 assoclist
C0:F8:DA:A0:1F:74  -48 dBm / unknown (SNR -48)  10 ms ago
        RX: 13.0 MBit/s, MCS 1, 20MHz                  66445 Pkts.
        TX: 21.7 MBit/s, MCS 2, 20MHz                 107868 Pkts.
        expected throughput: 11.4 MBit/s

9000000, 2019-05-29:
`root@OpenWrtFull:~# iwinfo wlan0 assoclist
9C:5C:F9:3B:97:AD  -75 dBm / unknown (SNR -75)  1750 ms ago
RX: 24.0 MBit/s                                26213 Pkts.
TX: 39.0 MBit/s, MCS 4, 20MHz                  43014 Pkts.
expected throughput: 25.1 MBit/s

34:CE:00:D3:C8:5C  -56 dBm / unknown (SNR -56)  210 ms ago
RX: 65.0 MBit/s, MCS 7, 20MHz                  10223 Pkts.
TX: 19.5 MBit/s, MCS 2, 20MHz                   2074 Pkts.
expected throughput: 15.9 MBit/s

E8:9E:B4:09:D5:39  -54 dBm / unknown (SNR -54)  8310 ms ago
RX: 11.0 MBit/s                                 4462 Pkts.
TX: 144.4 MBit/s, MCS 15, 20MHz                 4940 Pkts.
expected throughput: 114.3 MBit/s

3C:DC:BC:24:35:52  -79 dBm / unknown (SNR -79)  17410 ms ago
RX: 6.0 MBit/s                                   269 Pkts.
TX: 7.2 MBit/s, MCS 0, 20MHz                     182 Pkts.
expected throughput: 4.5 MBit/s

D8:C4:6A:D6:20:71  -51 dBm / unknown (SNR -51)  50 ms ago
RX: 144.4 MBit/s, MCS 15, 20MHz                 1462 Pkts.
TX: 144.4 MBit/s, MCS 15, 20MHz                 1168 Pkts.
expected throughput: 114.3 MBit/s

root@OpenWrtFull:~# iwinfo wlan0 assoclist                                9C:5C:F9:3B:97:AD  -75 dBm / unknown (SNR -75)  22150 ms ago
RX: 24.0 MBit/s                                26244 Pkts.
TX: 43.3 MBit/s, MCS 4, 20MHz                  43026 Pkts.
expected throughput: 29.7 MBit/s

34:CE:00:D3:C8:5C  -56 dBm / unknown (SNR -56)  2790 ms ago
RX: 65.0 MBit/s, MCS 7, 20MHz                  10426 Pkts.
TX: 19.5 MBit/s, MCS 2, 20MHz                   2115 Pkts.
expected throughput: 15.9 MBit/s

E8:9E:B4:09:D5:39  -54 dBm / unknown (SNR -54)  860 ms ago
RX: 130.0 MBit/s, MCS 15, 20MHz                 4491 Pkts.
TX: 144.4 MBit/s, MCS 15, 20MHz                 4971 Pkts.
expected throughput: 114.3 MBit/s

D8:C4:6A:D6:20:71  -51 dBm / unknown (SNR -51)  0 ms ago
RX: 144.4 MBit/s, MCS 15, 20MHz                 3430 Pkts.
TX: 130.0 MBit/s, MCS 14, 20MHz                 2665 Pkts.
expected throughput: 102.9 MBit/s

root@OpenWrtFull:~#`

9000000, 2019-06-03:
```
` OpenWrt SNAPSHOT, r10113-085f66d
 -----------------------------------------------------
root@OpenWrtFull:~# iwinfo wlan0 assoclist
E8:9E:B4:09:D5:39  -52 dBm / unknown (SNR -52)  70 ms ago
RX: 144.4 MBit/s, MCS 15, 20MHz             28666952 Pkts.
TX: 144.4 MBit/s, MCS 15, 20MHz             63204161 Pkts.
expected throughput: 114.3 MBit/s

34:CE:00:D3:C8:5C  -61 dBm / unknown (SNR -61)  510 ms ago
RX: 65.0 MBit/s, MCS 7, 20MHz                  26761 Pkts.
TX: 65.0 MBit/s, MCS 6, 20MHz                   6131 Pkts.
expected throughput: 48.0 MBit/s

9C:5C:F9:3B:97:AD  -68 dBm / unknown (SNR -68)  280 ms ago
RX: 24.0 MBit/s                                37073 Pkts.
TX: 86.7 MBit/s, MCS 12, 20MHz                 76129 Pkts.
expected throughput: 70.8 MBit/s

74:8D:08:DD:6F:F6  -64 dBm / unknown (SNR -64)  150 ms ago
RX: 24.0 MBit/s                                24958 Pkts.
TX: 57.8 MBit/s, MCS 5, 20MHz                  40738 Pkts.
expected throughput: 48.0 MBit/s

90:B0:ED:90:16:91  -65 dBm / unknown (SNR -65)  1810 ms ago
RX: 6.0 MBit/s                                 81740 Pkts.
TX: 144.4 MBit/s, MCS 15, 20MHz                72502 Pkts.
expected throughput: 98.3 MBit/s

D8:C4:6A:D6:20:71  -58 dBm / unknown (SNR -58)  80 ms ago
RX: 78.0 MBit/s, MCS 12, 20MHz                 32334 Pkts.
TX: 144.4 MBit/s, MCS 15, 20MHz                89167 Pkts.
expected throughput: 114.3 [MBit/s]()  `
```

9000000, 2019-06-03:
```
 OpenWrt SNAPSHOT, r10113-085f66d
 -----------------------------------------------------
root@OpenWrtFull:~# iwinfo wlan0 assoclist
E8:9E:B4:09:D5:39  -51 dBm / unknown (SNR -51)  30 ms ago
RX: 144.4 MBit/s, MCS 15, 20MHz             29290912 Pkts.
TX: 144.4 MBit/s, MCS 15, 20MHz             64531238 Pkts.
expected throughput: 114.3 MBit/s

34:CE:00:D3:C8:5C  -48 dBm / unknown (SNR -48)  600 ms ago
RX: 65.0 MBit/s, MCS 7, 20MHz                  29142 Pkts.
TX: 21.7 MBit/s, MCS 2, 20MHz                   6655 Pkts.
expected throughput: 18.2 MBit/s

74:8D:08:DD:6F:F6  -69 dBm / unknown (SNR -69)  110 ms ago
RX: 24.0 MBit/s                                44583 Pkts.
TX: 57.8 MBit/s, MCS 5, 20MHz                  53793 Pkts.
expected throughput: 48.0 MBit/s

90:B0:ED:90:16:91  -63 dBm / unknown (SNR -63)  5220 ms ago
RX: 24.0 MBit/s                               178842 Pkts.
TX: 144.4 MBit/s, MCS 15, 20MHz               159988 Pkts.
expected throughput: 114.3 MBit/s

D8:C4:6A:D6:20:71  -57 dBm / unknown (SNR -57)  10 ms ago
RX: 117.0 MBit/s, MCS 14, 20MHz                57116 Pkts.
TX: 130.0 MBit/s, MCS 15, 20MHz               128267 Pkts.
expected throughput: 102.9 MBit/s

F0:EE:10:37:4F:3E  -51 dBm / unknown (SNR -51)  510 ms ago
RX: 6.0 MBit/s                                 13825 Pkts.
TX: 72.2 MBit/s, MCS 7, 20MHz                  44482 Pkts.
expected throughput: 59.4 MBit/s

9C:5C:F9:3B:97:AD  -70 dBm / unknown (SNR -70)  11430 ms ago
RX: 24.0 MBit/s                                  279 Pkts.
TX: 39.0 MBit/s, MCS 4, 20MHz                    199 Pkts.
expected throughput: 31.9 MBit/s

DC:6D:CD:D8:B7:AA  -45 dBm / unknown (SNR -45)  46260 ms ago
RX: 6.0 MBit/s                                  5674 Pkts.
TX: 57.8 MBit/s, MCS 5, 20MHz                   3559 Pkts.
expected throughput: 48.0 MBit/s

78:FD:94:13:BA:DC  -66 dBm / unknown (SNR -66)  166420 ms ago
RX: 24.0 MBit/s                                  180 Pkts.
TX: 43.3 MBit/s, MCS 4, 20MHz                     70 Pkts.
expected throughput: 36.5
```

lynxis, 2019-06-03:
@9000000 please improve your bug report by editing the top post.
It is not clear **what's wrong?** How it should be? How is your test setup? What is your configuration? Which version do you use?

Overall Wifi is not a simple problem.
Usually 300mbit won't be archived on 2.4ghz, because 40Mhz is usually not possible.

9000000, 2019-06-03:
@lynxis all of my connected devices already support 40 Mhz band but I check that it doesn't 40mhz with no effect on 2.4ghz

lynxis, 2019-06-04:
@9000000 40 Mhz won't be used, as long there are other Accesspoint which uses the Spectrum. The specification forbids this. 40Mhz on 2.4Ghz is defacto non-existent, because you're not alone on the frequency.

lynxis, 2019-06-04:
@9000000 Please see my previous comment and update the missing informations.

lukasz1992, 2019-06-05:
@9000000 
I think in that many number of devices, we can not expect high bitrates; there are probably some delays (high pings). You could maybe check with another chip, but I don't see anything wrong in values from your dump.

@bond2bill
Can you check again during data transmission? Does it happen on other clients?

closed 2019-06-27 by nbd168