openwrt/mt76 #281: system crashed when high cpu load by  ksoftirqd/2

ybbid, 2019-05-30:
system crashed when high cpu load by ksoftirqd/2. Usually occurs once every 1 to 2 days
router:Newifi-D2(mt7621+mt7612+mt7603 rom:32m ram:512m)
arch:MediaTek MT7621 ver:1 eco:3
git-version:OpenWrt SNAPSHOT r10073-0293aa72d1 / LuCI Master (f138fc93) 
kernel version:4.14.120

blow is the kernel crash log:
Time: 1559144955.178036
Modules:        pppoe@8eaa4000+22b0     ppp_async@8ea9c000+1be0 snd_usb_audio@8ea60000+1d182    pppox@8ec75000+58a      ppp_generic@8ea48000+5b02       nf_conntrack_ipv6@8ea40000+1240 mt76x2e@8e934000+2991   mt76x2_common@8ee40000+2d68  mt76x02_lib@8ed20000+9db1       mt7603e@8ee60000+8e61   mt76@8ee58000+68e4      mac80211@8ec00000+70a26 lz4@8e929000+730        iptable_nat@8e92d000+310        ipt_REJECT@8e884000+410 ipt_MASQUERADE@8e9f1000+2f0     cfg80211@8ee80000+38ed0      xt_time@8f591000+730    xt_tcpudp@8f791000+750  xt_state@8ef0d000+310   xt_nat@8f783000+650     xt_multiport@8f7c2000+550       xt_mark@8f79d000+2f0    xt_mac@8ef0c000+2b0     xt_limit@8f790000+4f0   xt_conntrack@8f79c000+950    xt_comment@8ef0e000+230 xt_TCPMSS@8e87c000+b30  xt_REDIRECT@8e847000+310        xt_LOG@8f79f000+350     xt_FLOWOFFLOAD@8f7e5000+b40     ums_usbat@8e964000+20d0 ums_sddr55@8e956000+1790        ums_sddr09@8e810000+2410        ums_karma@8e808000+6f0       ums_jumpshot@8e960000+1010      ums_isd200@8f7bc000+1760        ums_freecom@8f7cb000+8d0        ums_datafab@8f780000+1370       ums_cypress@8f7e7000+a70        ums_alauda@8ee1c000+2270        uinput@8f792000+1d50    snd_usbmidi_lib@8ee30000+4213        slhc@8ef20000+149b      rfcomm@8ee28000+73fa    nf_reject_ipv4@8efcb000+883     nf_nat_redirect@8f6b2000+5fb    nf_nat_masquerade_ipv4@8efca000+67c     nf_conntrack_ipv4@8e962000+1150 nf_nat_ipv4@8e9ec000+d91     nf_nat@8ee18000+356b    nf_log_ipv4@8e9f2000+dd0        nf_flow_table_hw@8e9f5000+920   nf_flow_table@8ee0c000+365f     nf_defrag_ipv6@8ee08000+135e    nf_defrag_ipv4@8e95b000+436     nf_conntrack_rtcache@8e8e9000+af0       nf_conntrack@8e970000+f845   lz4_decompress@8e930000+34bf    lz4_compress@8e968000+5dc3      iptable_mangle@8e938000+3d0     iptable_filter@8e93c000+2d0     ip_tables@8e8f0000+294d hidp@8f788000+2e51      hci_uart@8e910000+30cf  crc_ccitt@8e8ea000+42b       compat@8e80a000+19b7    btusb@8e890000+6260     btintel@8e854000+182f   bnep@8e958000+256d      bluetooth@8e980000+46169        hid@8e940000+15156      evdev@8e924000+2970     input_core@8e918000+688b        ledtrig_usbport@8e8fe000+b00 nf_log_ipv6@8e90c000+1070       nf_log_common@8e8be000+acf      ip6table_mangle@8ef27000+4d0    ip6table_filter@8e809000+2d0    ip6_tables@8e8f4000+2841        ip6t_REJECT@8ff7d000+450        x_tables@8e8ec000+37cf  nf_reject_ipv6@8e82f000+a08  snd_rawmidi@8e8e0000+42ca       snd_seq_device@8e8dc000+8cf     snd_pcm_oss@8e8d0000+94b1       snd_pcm@8e8c0000+fafe   snd_timer@8e8b0000+456e snd_mixer_oss@8e888000+3439     snd_hwdep@8e81a000+121e snd_compress@8f7cc000+1d44   snd@8e8a0000+a27f       soundcore@8f706000+10fc ecdh_generic@8e898000+43ca      kpp@8f75d000+3bc        ecb@8ef90000+5b0        cmac@8e861000+930       uas@8f728000+2e50       usb_storage@8f710000+a4eb       leds_gpio@8f6d4000+bb0       xhci_plat_hcd@8f5fe000+15f0     xhci_pci@8f5f4000+c10   xhci_mtk@8f5fa000+12f0  xhci_hcd@8f6e0000+16c29 uhci_hcd@8f6d8000+4b60  ohci_platform@8f6cc000+1070     ohci_hcd@8f6c0000+5cff  ehci_platform@8f6b0000+1250     sd_mod@8f6a0000+7700 scsi_mod@8f680000+16e92 ehci_hcd@8f620000+8a17  gpio_button_hotplug@8f606000+1950       usbcore@8f640000+2109d  nls_base@8f5f8000+1550  usb_common@8ff26000+a17 zram@8f5b4000+2c70      zsmalloc@8f5ac000+2b26
<6>[    7.342686] hub 1-0:1.0: USB hub found
<6>[    7.346515] hub 1-0:1.0: 2 ports detected
<6>[    7.351106] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
<6>[    7.356419] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 2
<6>[    7.363911] xhci-mtk 1e1c0000.xhci: Host supports USB 3.0  SuperSpeed
<6>[    7.370520] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
<6>[    7.379491] hub 2-0:1.0: USB hub found
<6>[    7.383310] hub 2-0:1.0: 1 port detected
<6>[    7.396180] usbcore: registered new interface driver usb-storage
<6>[    7.403893] usbcore: registered new interface driver uas
<14>[    7.409517] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
<14>[    7.436929] init: - preinit -
<6>[    7.866683] usb 1-1: new high-speed USB device number 2 using xhci-mtk
<6>[    8.049080] hub 1-1:1.0: USB hub found
<6>[    8.053142] hub 1-1:1.0: 7 ports detected
<5>[    8.063837] random: procd: uninitialized urandom read (4 bytes read)
<6>[    8.427547] usb 1-1.1: new full-speed USB device number 3 using xhci-mtk
<6>[    8.472114] mtk_soc_eth 1e100000.ethernet: PPE started
<14>[   11.702021] mount_root: loading kmods from internal overlay
<14>[   11.730704] kmodloader: loading kernel modules from //etc/modules-boot.d/*
<14>[   11.738871] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
<5>[   12.347142] jffs2: notice: (501) jffs2_build_xattr_subsystem: complete building xattr subsystem, 21 of xdatum (3 unchecked, 18 orphan) and 61 of xref (18 dead, 1 orphan) found.
<14>[   12.363885] block: attempting to load /tmp/jffs_cfg/upper/etc/config/fstab
<14>[   12.388658] block: extroot: not configured
<5>[   12.542005] random: crng init done
<5>[   12.669239] jffs2: notice: (499) jffs2_build_xattr_subsystem: complete building xattr subsystem, 21 of xdatum (3 unchecked, 18 orphan) and 61 of xref (18 dead, 1 orphan) found.
<14>[   12.688943] mount_root: loading kmods from internal overlay
<14>[   12.738307] kmodloader: loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
<14>[   12.753897] kmodloader: done loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
<14>[   13.132411] block: attempting to load /tmp/jffs_cfg/upper/etc/config/fstab
<14>[   13.146450] block: extroot: not configured
<14>[   13.188785] mount_root: switching to jffs2 overlay
<4>[   13.222884] overlayfs: upper fs does not support tmpfile.
<12>[   13.246767] urandom-seed: Seeding with /etc/urandom.seed
<6>[   13.336080] mtk_soc_eth 1e100000.ethernet: 0x100 = 0x6060000c, 0x10c = 0x80818
<14>[   13.360430] procd: - early -
<14>[   13.363423] procd: - watchdog -
<14>[   14.046941] procd: - watchdog -
<14>[   14.050423] procd: - ubus -
<14>[   14.199627] procd: - init -
<14>[   14.792228] kmodloader: loading kernel modules from /etc/modules.d/*
<6>[   14.838405] ip6_tables: (C) 2000-2006 Netfilter Core Team
<6>[   14.862722] hidraw: raw HID events driver (C) Jiri Kosina
<6>[   14.886916] Bluetooth: Core ver 2.22
<6>[   14.890730] NET: Registered protocol family 31
<6>[   14.895180] Bluetooth: HCI device and connection manager initialized
<6>[   14.901570] Bluetooth: HCI socket layer initialized
<6>[   14.906438] Bluetooth: L2CAP socket layer initialized
<6>[   14.911644] Bluetooth: SCO socket layer initialized
<6>[   14.918240] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
<6>[   14.923578] Bluetooth: BNEP filters: protocol multicast
<6>[   14.928871] Bluetooth: BNEP socket layer initialized
<6>[   14.939241] usbcore: registered new interface driver btusb
<6>[   14.946289] Loading modules backported from Linux version v4.19.32-0-g3a2156c839c7
<6>[   14.953890] Backport generated by backports.git v4.19.32-1-0-g1c4f7569
<6>[   14.962969] Bluetooth: HCI UART driver ver 2.3
<6>[   14.967486] Bluetooth: HCI UART protocol H4 registered
<6>[   14.972629] Bluetooth: HCI UART protocol BCSP registered
<6>[   14.979697] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
<6>[   14.985676] Bluetooth: HIDP socket layer initialized
<6>[   14.993050] ip_tables: (C) 2000-2006 Netfilter Core Team
[... 105 lines cut ...]
<4>[ 7308.330048] ra    : 80468ad8 __do_softirq+0x128/0x2ec
<4>[ 7308.335079] Status: 11007c03      KERNEL EXL IE
<4>[ 7308.339268] Cause : 50800024 (ExcCode 09)
<4>[ 7308.343258] PrId  : 0001992f (MIPS 1004Kc)
<4>[ 7308.347332] Modules linked in: pppoe ppp_async snd_usb_audio pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 lz4 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda uinput snd_usbmidi_lib slhc rfcomm nf_reject_ipv4 nf_n [... cut ...]
<4>[ 7308.516341] [<80036da4>] tasklet_action+0xf4/0x1d0
<4>[ 7308.521124] [<80468ad8>] __do_softirq+0x128/0x2ec
<4>[ 7308.525820] [<80036eb8>] run_ksoftirqd+0x38/0x6c
<4>[ 7308.530444] [<80053858>] smpboot_thread_fn+0x1a8/0x1d8
<4>[ 7308.535581] [<80050098>] kthread+0x130/0x144
<4>[ 7308.539852] [<8000b0d8>] ret_from_kernel_thread+0x14/0x1c
<4>[ 7308.545240] Code: 0000000f  14800002  00000000 <000c000d> 8c43000c  0060f809  8c440010  0000000f  c2020000
<4>[ 7308.554969]
<4>[ 7308.556692] ---[ end trace 395a3d800055a403 ]---

=================

dissent1, 2019-05-31:
I have the same issue on 1 of my similar devices. Happens when txpower is set to around 20. Lowering txpower to around 10 doesn't produce the issue.
Maybe it is overheating? Try to place the device to a place with good ventilation.

ybbid, 2019-06-03:
It has been observed that this problem occurs when the signal of the terminal device connected to this router is low.

nbd168, 2019-06-27:
Fixed in the current version

pataha, 2019-07-27:
It still happen to me with latest snapshot. It’s rebooted while watching youtube after running ~2 days

pftbest, 2020-01-05:
Is this the same bug or something different?

```
<6>[747111.743694] mtk_soc_eth 1e100000.ethernet eth0: port 3 link up
<4>[752633.872801] Kernel bug detected[#1]:
<4>[752633.876739] CPU: 2 PID: 19 Comm: ksoftirqd/2 Not tainted 4.14.121 #0
<4>[752633.883169] task: 8fc78640 task.stack: 8fc82000
<4>[752633.887770] $ 0   : 00000000 00000001 8f3062f0 00000001
<4>[752633.893084] $ 4   : 00000000 00c50000 805f3c40 00000001
<4>[752633.898396] $ 8   : 00000000 000006f2 0000024d 001bca58
<4>[752633.903712] $12   : 06d9956d 00000040 3b9aca00 00000000
<4>[752633.909026] $16   : 8f3062f4 00000002 00000000 805861e0
<4>[752633.914339] $20   : 805dd258 805f3c40 805861e0 805dd2c0
<4>[752633.919654] $24   : 33c24cd6 0002ac84                  
<4>[752633.924969] $28   : 8fc82000 8fc83e28 00000008 80474b78
<4>[752633.930283] Hi    : 000000d4
<4>[752633.933253] Lo    : 00000035
<4>[752633.936249] epc   : 80036d94 tasklet_action+0xf4/0x1d0
<4>[752633.941462] ra    : 80474b78 __do_softirq+0x128/0x2ec
<4>[752633.946582] Status: 11007c03	KERNEL EXL IE 
<4>[752633.950856] Cause : 50800024 (ExcCode 09)
<4>[752633.954937] PrId  : 0001992f (MIPS 1004Kc)
<4>[752633.959103] Modules linked in: pppoe ppp_async pppox ppp_generic nf_conntrack_ipv6 mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat fuse ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsd nfs rpcsec_gss_krb5 auth_rpcgss oid_registry tun loop vfat fat lockd sunrpc grace dns_resolver
<6>[752634.002792] mt76x2e 0000:02:00.0: Firmware Version: 0.0.00
<4>[752634.030525]  dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic md5 hmac ecb des_generic cts cbc usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk xhci_hcd sd_mod scsi_mod gpio_button_hotplug ext4 mbcache jbd2 usbcore nls_base usb_common crc32c_generic
<4>[752634.030646] Process ksoftirqd/2 (pid: 19, threadinfo=8fc82000, task=8fc78640, tls=00000000)
<4>[752634.030650] Stack : 8fc78640 8fc78640 8057805c 00000001 00000001 80578058 00000007 00000040
<4>[752634.030682]         00000006 00000100 80578040 80474b78 00000001 00000200 00000000 00000040
<4>[752634.030709]         ffffff97 ffffffff 805f3c40 80518910 047bfc0b 80580000 8047ca40 04208040
<4>[752634.030738]         8fc21900 00000001 8058c040 00000002 80053850 80520000 805f0000 00000001
<4>[752634.030768]         8058f6e0 80036ea8 11007c03 001a731c a457989e 8fc21900
<6>[752634.036388] mt76x2e 0000:02:00.0: Build: 1
<4>[752634.063067]  8fc21900 800539f8
<4>[752634.063078]         ...
<4>[752634.063088] Call Trace:
<4>[752634.063127] [<80036d94>] tasklet_action+0xf4/0x1d0
<4>[752634.063154] [<80474b78>] __do_softirq+0x128/0x2ec
<6>[752634.071551] mt76x2e 0000:02:00.0: Build Time: 201507311614____
<4>[752634.079967] [<80036ea8>] run_ksoftirqd+0x38/0x6c
<4>[752634.079998] [<800539f8>] smpboot_thread_fn+0x1a8/0x1d8
<4>[752634.080038] [<800500a8>] kthread+0x130/0x144
<6>[752634.102465] mt76x2e 0000:02:00.0: Firmware running!
<4>[752634.105242] [<8000b0d8>] ret_from_kernel_thread+0x14/0x1c
<4>[752634.105256] Code: 0000000f 
<6>[752634.122587] ieee80211 phy1: Hardware restart was requested
<4>[752634.124425]  14800002  00000000 <000c000d> 8c43000c  0060f809  8c440010  0000000f  c2020000 
<4>[752634.181434] 
<4>[752634.183192] ---[ end trace 4441228ed08d34b8 ]---
```